geerlingguy/sbc-reviews #1: Automate benchmarking to the greatest extent possible with Ansible

geerlingguy, 2023-01-26:
TODO: Fill in some details here.

minecraftchest1, 2023-01-27:
Somewhat related. I am building a script that updates a table in the README (as was mentioned as a TODO in the README. For now, it is more of a template or example then final implementation (and not close to finished anyhow), but mighgt be useful down the road. Draft PR #14

geerlingguy, 2025-01-31:
I've been working on Ansible and Pyinfra automation scripts in parallel, to see what I like more for this.

Going to go forward with Pyinfra for now, but there are some things that feel a bit more hacky there, especially for multi-platform compatibility and getting output from shell commands.

geerlingguy, 2025-01-31:
Any follow-up work will go into new issues.

For now I'm just testing this across Arm, x86, and RISC-V.

closed 2025-01-31 by geerlingguy