geerlingguy/sbc-reviews #11: Banana Pi BPI-CM4

geerlingguy, 2023-01-26:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://wiki.banana-pi.org/Banana_Pi_BPI-CM4
  - Board purchased from: AliExpress
  - Board purchase date: 2023-01-05
  - Board specs (as tested): 4G RAM / 16G eMMC
  - Board price (as tested): $109.73

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

geerlingguy, 2023-01-26:
First fun issue... I went to boot up the board today, and searched all over the BPI-CM4 pages on the Wiki:

  - [Getting Started with CM4](https://wiki.banana-pi.org/Getting_Started_with_CM4)
  - [Banana Pi BPI-CM4 Linux Images](https://wiki.banana-pi.org/Banana_Pi_BPI-CM4#Linux)

...and there are no links to download any images, anywhere. After asking around on Twitter, [@PomstaZLesa found](https://twitter.com/PomstaZLesa/status/1618664438529200136) [this old Wiki page revision](https://wiki.banana-pi.org/index.php?title=Banana_Pi_BPI-CM4&oldid=13245#Linux) which does have some links (from 2022-06-20...) to downloads in Google Drive or Baidu Cloud.

Not super confidence-inspiring :/

kklem0, 2023-01-31:
> First fun issue... I went to boot up the board today, and searched all over the BPI-CM4 pages on the Wiki:
>
> * [Getting Started with CM4](https://wiki.banana-pi.org/Getting_Started_with_CM4)
> * [...]

It looks like they have updated your 2nd link 2 days ago.

geerlingguy, 2023-01-31:
Default username / login is `pi`/`bananapi` for the Debian image. Just posting it here since it took way too much reading through their Wiki to find that.

geerlingguy, 2023-01-31:
Tried compiling `top500-benchmark` and got:

```
    configure: WARNING: X11 not found; GL disabled
    configure: WARNING: compilation failed
    configure: error: ****  Incompatible Fortran and C Object File Types!  ****
    F77 Object File Type produced by "gfortran -fallow-argument-mismatch " is : : cannot open `' (No such file or directory).
     C  Object File Type produced by "gcc    -O2" is : : ELF 64-bit LSB relocatable, ARM aarch64, version 1 (SYSV), not stripped.
```

On the "Build MPI" step.

geerlingguy, 2023-01-31:
To get WiFi working, I did a `sudo modprobe 88x2cs` to enable the kernel module (see [this doc](https://wiki.banana-pi.org/Getting_Started_with_CM4#Enable_Wifi_and_BT)), then I installed Network Manager:

```
pi@bananapi:~$ sudo apt install -y network-manager
...
pi@bananapi:~$ nmcli
wlan0: disconnected
        "wlan0"
        wifi (rtl88x2cs), 36:B4:8F:83:24:6E, hw, mtu 1500

pi@bananapi:~$ sudo nmcli device wifi list
sudo: unable to resolve host bananapi: Name or service not known
IN-USE  SSID                   MODE   CHAN  RATE        SIGNAL  BARS  SECURITY 
        geerling...  Infra  36    540 Mbit/s  54      **    WPA2     
        ...
        --                     Infra  11    130 Mbit/s  0             WPA2

pi@bananapi:~$ sudo nmcli dev wifi connect geerling... password "my-password-here"
...
connected!
```

And also, all the details of the interface:

<details><summary>`iw list` details</summary>

```
pi@bananapi:~$ sudo iw list
sudo: unable to resolve host bananapi: Name or service not known
Wiphy phy0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
	Band 1:
		Capabilities: 0x1963
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
[... 116 lines cut ...]
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
	interface combinations are not supported
	Device supports scan flush.
	Supported extended features:
```
</details>

melroy89, 2023-05-16:
Nice you also have the BPI-CM4 module. I'm trying to get this module working in the BliKVM PCIe card, instead of the Raspberry Pi variant.. 

Related: https://github.com/ThomasVon2021/blikvm/issues/15

sgofferj, 2023-06-02:
I also got one of those together with an MCUZone POE/SSD baseboard. Everything works except the USB ports. There's a forum post reporting the same issue with a Waveshare Mini B baseboard. Not sure if it's an OS thing or they messed up the USB connection. I tried Debian, Ubuntu and Armbian to no avail.
@geerlingguy Have you ever gotten the USB working on a non-BPI baseboard?

sgofferj, 2023-09-04:
Somebody from the manufacturer provided a solution for the USB port issue: https://forum.banana-pi.org/t/cm4-with-mcuzone-board-no-usb-port-works/15552/7

billzez, 2023-09-22:
Quick update: we now support Banana Pi CM4, and we now have OMV and OpenWrt images.

https://www.axzez.com/software-downloads

melroy89, 2023-09-23:
> Quick update: we now support Banana Pi CM4, and we now have OMV and OpenWrt images.
> 
> https://www.axzez.com/software-downloads

WOW! Nice.. I wish [blikvm](https://github.com/ThomasVon2021/blikvm) would create images for the Banana Pi CM4 as well.