geerlingguy/sbc-reviews #2: Standardize the benchmarks I use across multiple boards

geerlingguy, 2023-01-26:
Some of the more standard benchmarks I've run include:

  - Geekbench 5 (for ARM / RISC-V, in preview)
  - disk-benchmark.sh (my script in my pi-cluster project, uses fio and iozone)
  - PiBenchmarks.com (community project by @TheRemote, uses fio, hdparm, dd, iozone)
  - Top500 / HPL Linpack (compiled on the board — this is nice to test long-term stability and thermals under heavy load)
  - Phoronix Test Suite (sometimes, for certain tests... it doesn't work OOTB on ARM64 for a lot of tests)
  - Power (testing power consumption over time during HPL and Geekbench, at least)

I've looked at some other tests, too, like @ThomasKaiser's [sbc-bench](https://github.com/ThomasKaiser/sbc-bench) scripts.

I'd like to have a nice standard suite of benchmarks on a basic level for the main things—like CPU, GPU (maybe), RAM, Disk, and Network). Then it would be nice to have a deeper set if I'm trying to get a better understanding of edge cases.

ThomasKaiser, 2023-01-26:
> I've looked at some other tests, too, like @ThomasKaiser's [sbc-bench](https://github.com/ThomasKaiser/sbc-bench) scripts.

Just to quickly clarify: `sbc-bench` doesn't invent new/own benchmarks but just fires up some already existing benchmarks and tools to measure RAM latency/bandwidth and real CPU clockspeeds in a controlled environment to generate insights instead of just random numbers. 

What are numbers worth if they're too low since too much background activitiy or throttling happened or CPU frequency capping (RPi with input voltage dropping below ~4.65V) or even CPU cores having been shut down due to overheating?

Which conclusion can be drawn about efficiency of a certain SoC or ARM/RISC-V core when we blindly trust into clockspeeds reported via sysfs when in reality they differ: with some Amlogic SoCs we still have [a mismatch of -40% comparing 'advertised' clockspeed vs. real](http://ix.io/4m8P) and `sbc-bench` reports that like this:

    Cpufreq OPP: 2016    Measured: 1197 (1197.338/1197.311/1197.121)    (-40.6%)
    Cpufreq OPP: 1752    Measured: 1197 (1197.392/1197.338/1197.284)    (-31.7%)
    Cpufreq OPP: 1536    Measured: 1197 (1197.365/1197.338/1197.202)    (-22.1%)
    Cpufreq OPP: 1416    Measured: 1197 (1197.338/1197.338/1197.202)    (-15.5%)
    Cpufreq OPP: 1200    Measured: 1197 (1197.256/1197.229/1197.229)
    Cpufreq OPP: 1000    Measured:  997    (997.431/997.290/997.266)
    Cpufreq OPP:  667    Measured:  664    (664.061/664.045/664.012)
    Cpufreq OPP:  500    Measured:  497    (497.358/497.346/497.346)
    Cpufreq OPP:  250    Measured:  246    (246.553/246.268/245.968)     (-1.6%)
    Cpufreq OPP:  100    Measured:   96       (96.994/96.592/96.208)     (-4.0%)

Same goes for Raspberries with inappropriate powering and here it's even worse: the main OS (ThreadX on the VC) silently drops ARM clockspeeds to 600 MHz while the Linux kernel running on the ARM core still happily reports to run at 1200/1500/1800 MHz. IMO stuff like this needs to be closely monitored so you're able to throw away results that are invalid (and since benchmarking server/storage systems is part of my day job I can confirm that throwing away results instead of relying on them is the 'norm' or what we mostly do since something went wrong we didn't know upfront so we need to adjust setup/methodology and repeat).

In that sense `sbc-bench` is more of an environment to execute the actual benchmark (see Geekbench and Phoronix 'piggyback' modes). 

As for the [few individual benchmarks chosen](https://github.com/ThomasKaiser/sbc-bench#which-tools-are-used-and-why) two are more load generators / stressors (cpuminer, stockfish) and there is actually just one that really generates numbers to compare and that's `7-zip` focusing on the stuff _I'm personally_ am interested in: integer+memory since me being a server guy. This limited focus is far away from generating a general performance recommendation since based on the use case people are using their SBC they need to pay attention to totally different areas (floating point, crypto acceleration, 2D/3D graphics acceleration, video acceleration, even I/O can be way more important than any of the other areas).

But while 7-zip's focus is pretty limited it has one **huge** advantage: it generates scores that are persistent accross OS and compiler releases (at least true for the 16.02 version that's shipped with Debian/Ubuntu since years). Most other benchmarks generate numbers that will change once you or your OS's maintainers update OS or compiler version. Sometimes it goes into the opposite direction and with more recent GCC the benchmark scores drop (cpuminer for example on Debian 11). 

As such that's the huge advantage of those 7-zip scores even if they represent performance metrics the average SBC user can not make lot use of: they represent 'hardware performance' way more precise in situations where software changed significantly and as such it makes sense to compare scores made in different years (**not** true for the majority of other passive benchmarking tools where numbers fluctuate over time 'by design').

This (ability to compare/validate results made in different years) is also the reason why `ramlat` and `tinymembench` are included since they often can explain why the actual benchmark numbers differ. For example RPi benchmark scores usually get better over the years. Even on the same RPi. And this is not since the hardware magically got faster but since RPi Trading Ltd. employees tweaked ThreadX with better DRAM initialisation/timings and once you `apt update`after a reboot memory access improved. 

It's important to know _why_ numbers differ and it's also important to realise when numbers made in different years are still valid or not (and that's something ignored in this SBC world almost everywhere - the basic rules of passive benchmarking simply do not apply since it's a true hobbyist's world)

TheRemote, 2023-01-26:
You're exactly right and it's actually worse than you're saying.  The results aren't invalidated by years.  They can be invalidated by days from how fast these apt packages can change.  They can be invalidated on the same day if they're running slightly different operating systems or firmware versions.

Having a lot of tests certainly helps get closer to the truth though.  Pi Benchmarks is kind of crowd sourcing that for the storage aspect to the entire community.  

On Pi Benchmarks we can actually go back and figure out *when* they changed gcc and the benchmarks started dropping for example.  That's because we have tens of thousands of tests to work with (probably a dozen or so per day on average for years across various boards).

That means it's not really a bug that these scores can change over time when we're crowdsourcing the benchmark.  It's a feature.  Now we can go back and look really carefully at the dates/scores/system configurations and figure out what changed and why.

You're right though that if you are taking individual benchmarks in a bubble and you are trying to compare them with one another that is a totally invalid/useless comparison.  Once we have hundreds or thousands of benchmarks though this turns from a weakness into a strength.  Now we have enough data to answer some of those more important questions:  what changed, when and why?

geerlingguy, 2023-01-26:
My main goal with _this_ repository is to document my benchmarking results for the times I test these boards for a particular purpose (comparison, a review, a new project)—and almost every time I do a comparison, I will re-benchmark using the latest OS/updates since that does make a huge difference a lot of times.

But the secondary purpose is to put up all my numbers (and any surprising notes, like the RK3588S on the Orange Pi being noticeably faster than the RK3588 on the Rock 5 B, both using the vendor Debian images), so others who benchmark or just have an interest in the boards can either confirm or deny the results, or give any feedback if desired (e.g. if some of my methodology is lacking... which it is, sometimes!).

So I always welcome any and all feedback!

TheRemote, 2023-01-26:
Beautiful system, that's a great way to investigate the root causes of some of these issues and differences as well.

I will sometimes do head-to-head testing between different boards and I always do fresh tests every time.  Like even though I've ran my same CPU benchmarks on the RPI 4 a dozen times I don't just reuse the old numbers each time.  I want to make sure they have all the apt updates and firmware updates available as they would be if a user bought the board and set it up for the first time.

I think it's a great methodology.  These are all great tools for us to help diagnose performance related issues.  Different tools are appropriate for different jobs as well.  I love having lots of tools available and that you shared / linked / documented a bunch of different ways to do it!

ThomasKaiser, 2023-01-27:
> the RK3588S on the Orange Pi being noticeably faster than the RK3588 on the Rock 5 B

Simple reason: Rockchip's Dynamic Memory Interface (DMC) is 'at fault'. Xunlong hasn't discovered this feature/mechanism yet (so DMC is inactive in their kernel config just as it was with Radxa last July). Radxa in the meantime uses a newer/different kernel branch where DMC is enabled, the governor defaults to `dmc_ondemand` and `upthreshold` defaults to `40`. 

This default results in a drop in idle consumption of about 600mW since DRAM is clocked almost all the time now at just 528 MHz instead of 2112 MHz as before! But the rather conservative `upthreshold=40` setting results in the memory controller not ramping up DRAM clocks quickly enough for Linux workloads and also... benchmarks.

Once you adjust this to `upthreshold=25` the idle consumption remains at this lower level but benchmarks and also real-world tasks as well perform almost identical as if you would set the DMC governor to `performance` which then behaves like Xunlong's kernel config but there accompanied by a 600mW higher idle consumption since DRAM is clocked at 2112 MHz all the time ([all DMC details](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md)). 

SoCs like RK3588/RK3588s are that complex that we have to deal with trade-offs at various locations. Though that's not entirely new since DMC and other such governors are there since 2015/RK3399.

The DMC governor's impact on RK3588 benchmarks was the main reason why `sbc-bench` when executed reports now each governor it detects directly to the tester (not only memory is affected but GPU and NPU too of course and with SoCs not available yet maybe something else):

[image]

And the results output lists all information about the DMC governor available (the below is from an [OPi 5 running with Armbian](http://ix.io/4m0V). The Armbian guys use Radxa's kernel branch for the OPi 5 so DMC is available/adjustable but of course don't give a sh*t about settings like `uptreshold`):

    SoC guess: Rockchip RK3588/RK3588s (35881000)
      DMC gov: dmc_ondemand (upthreshold: 40)
    DT compat: rockchip,rk3588s-orangepi-5
               rockchip,rk3588

The DRAM clockspeed transition table is also listed when available:

    Transitions since last boot (9012920ms ago):
    
    /sys/devices/platform/dmc/devfreq/dmc:
    
         From  :   To
               : 528000000106800000015600000002112000000   time(ms)
    * 528000000:         0         0         0        25   8557290
     1068000000:        20         0         0         3    113906
     1560000000:         0        14         0         0      8916
     2112000000:         6         9        14         0    329386
    Total transition : 91

Once you set `uptreshold=25` on Rock 5B it will most likely slightly outperform your OPi 5 for a reason called PVTM, another dynamic mechanism implemented by Rockchip in all their recent RK35xx line-up. The exact details you'll also find in my Rock 5B review.

I had the impression within the last months that Rockchip seems to put all the 'strong' silicon into RK3588 while the 'weak' silicon ends up in RK3588s SoCs where the A76 then will clock significantly lower than 2400 MHz. That's at least what I thought to observe after checking a bunch of `sbc-bench` results since my script of course also collects this information after discovering the PVTM mechanisms in addition to measuring real clockspeeds instead of trusting into `sysfs` values. 

But on the aforementioned OPi 5 the PVTM values show a rather 'strong' silicon variant (allowing the A76 cores to clock at ~2315 MHz):

           cpu cpu0: pvtm=1488
           cpu cpu0: pvtm-volt-sel=4
           cpu cpu4: pvtm=1721
           cpu cpu4: pvtm-volt-sel=5
           cpu cpu6: pvtm=1722
           cpu cpu6: pvtm-volt-sel=5

While [this OPi 5](http://ix.io/4ldh) shows weaker PVTM values (it's not only silicon variation but also temperature and Xunlong never thinking about appropriate cooling adds to the problem):

           cpu cpu0: pvtm=1401
           cpu cpu0: pvtm-volt-sel=0
           cpu cpu4: pvtm=1632
           cpu cpu4: pvtm-volt-sel=2
           cpu cpu6: pvtm=1640
           cpu cpu6: pvtm-volt-sel=2

I hope this painfully lenghty explanation backs my point of collecting all 'environmental data' known and available while benchmarking since _only those_ provide the answers for 'why do the numbers look different'.

ThomasKaiser, 2023-01-27:
> Pi Benchmarks is kind of crowd sourcing that for the storage aspect to the entire community.

@TheRemote please correct me if I'm wrong. As I understand it, PiBenchmarks.com is all and only about measuring _storage performance_ on different hosts (regardless of host type) and then presenting this as some sort of listings that vaguely differentiate between 'SBC' and 'PC'?

TheRemote, 2023-01-27:
Why would Orange Pi not implementing the DMC controller driver into the kernel make it *faster* than the Rock 5B?  Why does this technology exist then?  Shouldn't we want to just disable it for the Rock 5B if it makes the device perform poorly?  Or are you saying the default setting is just so bad that it's actually worse than not having it at all?

You don't think the disabling of the Specter mitigations / unprivileged eBPF is even a factor after that write-up?  Does sbc-bench test for this?  You should start testing for this.  That's known to have a huge performance impact as I'm sure you're well aware because I am well aware of it and you are way more in the weeds with this than I am.  People still do that all the time on Linux/Windows to get a big performance gain at the cost of security.

Why not test both?  The best thing for Jeff to do would be to update his kernel and retest so all of the security mitigations match.  Next it would probably be to change upthreshold to 25 and retest.

ThomasKaiser, 2023-01-27:
> Why would Orange Pi not implementing the DMC controller driver into the kernel make it faster than the Rock 5B?

When DMC is not enabled in kernel config DRAM is clocked at 2112 MHz by defauilt (or whatever highest value is defined in the boot BLOBs, with recent RK3328 BLOBs it's 1056 MHz and with RK3399 it depends on type of RAM: DDR for whatever reasons being clocked slightly higher than LPDDR). 

Once DMC is enabled the same could be achieved by setting the DMC governor to `performance` either manually or directly into kernel config. Then nothing would've changed.

But the default is `dmc_ondemand` with an `uptreshold` of 40 and this will severly limit all workloads and all benchmarks _that access RAM_. Stuff that runs entirely inside CPU caches (e.g. `sysbench's` CPU bench or `openssl`) isn't affected at all. And board vendors usually have no clue anyway what's happening (notable exceptions: RPi Trading Ltd. and Hardkernel).

As a result of my testing I came up first with `20` as a better replacement but after some more kernel updates it looks like `uptreshold=25` is the better compromise between low(est) idle consumption and high(est) peak performance. Workloads that involve 100% CPU utilization all the time won't see much of a difference when they're running long enough since the DMC mechanism will keep the DRAM clocked at 2112 MHz almost all the time. But this access pattern isn't that common even with benchmarks.

As for Spectre mitigations... isn't that stuff that requires at least 5.17 or something? RK3588 running with mainline kernel isn't an option today since most of the HW IP blocks aren't supported yet. And with an insecure 5.10 BSP kernel a few vulnerabilities more or less shouldn't matter anyway...

TheRemote, 2023-01-27:
Thanks for the explanation on DMC. I can absolutely see why this could happen from your explanation.  It does indeed sound like a matter of they implemented it so badly it was actually worse than nothing!

So for the specter mitigations this is actually user controllable to an extent.  I tried playing with this a little bit.  The specific setting that doesn't seem to match between the two (until you update the kernels).  It is:

`cat /proc/sys/kernel/unprivileged_bpf_disabled`

This will not exist on the original Rock 5B kernel.  It's too old.  It will exist on the Orange Pi 5.

You can actually toggle this until a reboot with:

`echo 1 > /proc/sys/kernel/unprivileged_bpf_disabled`

Before: 

```
root@rock-5b:/proc/sys/kernel# lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
...
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable: Unprivileged eBPF enabled
```

After:
```
root@rock-5b:/proc/sys/kernel# lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
...
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; CSV2, BHB
```

There is actually also a "2" setting in newer kernels but this one is too old to have it.  It wouldn't surprise me if that is the 1.17 version that is popping into your head.  In these old kernels it's either off or on.  In newer kernels there's a 2 which lets certain trusted system processes use it but nothing else.  

When I used to CPU mine years ago for Monero we would always disable the CPU security mitigations.  You could get a good 15-20% boost out of it sometimes for certain types of CPU tasks since it enables a bunch of predictive branch code that while is very fast has known security risks. 

It's interesting that this is a user-controllable one until reboot that doesn't match.  Theoretically you could test this on the old kernel to see if just toggling this specific setting would do it (or have some kind of impact).

TheRemote, 2023-01-27:
I checked out sbc-bench and I actually really like it.  I'll try shutting off the security mitigations and testing with your sbc-bench tool a little bit later here.

It would actually integrate really well with PiBenchmarks (at least some of the tests you've done here).  Like some of your tests could just be popped right into PiBenchmarks and submitted as an additional field with curl.

I think Pi Benchmarks shares a lot of your same philisophy.  Like it gathers a *ton* of data on each machine.  Lshw, inxi, udev, everything to capture all of this.  The script works almost exactly the same way as in it's a bash script that takes care of everything including the dependencies for it.

The only difference is you seem to do more processing of brands and stuff like that in the script and I do more of that in the backend.  The backend isn't released but there's nothing really any different than what you're doing in the bash script happening.  It's just parsing a whole bunch of brands / models / etc.

This was exactly how I was planning to expand Pi Benchmarks eventually into other type of tests but when I did it I wanted to do it the right way.  I wanted to do it the way you're doing here.  Crazy amounts of information so that we can go through and *see* these differences especially the weird ones.

If you aren't opposed to it I might pick your brain at some point to see about properly integrating some of this kind of testing into Pi Benchmarks (or I might just look at how you did it and credit you and your repository/work for sure).  You're definitely doing some good work here though in an area I don't really cover right now!

ThomasKaiser, 2023-02-03:
> My main goal with this repository is to document my benchmarking results for the times I test these boards

I decided to implement a `-j` switch for sbc-bench recently which should server two purposes:

  * provide as much performance relevant information as possible (e.g. this PVTM and DMC stuff on Rockchip SoCs)
  * adjust settings so that immediately following benchmark attempts more represent 'hardware performance' than software/settings as it's usually the case. This especially implies setting each and every available governor to `performance`!

I guess the help text will read then:

   tk@mac-tk sbc-bench % ./sbc-bench.sh -h
   
   Usage: sbc-bench.sh [-c] [-g] [-G] [-h] [-j] [-m] [-P] [-t $degree] [-T $degree] [-s]
   
   ############################################################################
   
    Use sbc-bench.sh for the following tasks:

    sbc-bench.sh invoked without arguments runs a standard benchmark
    sbc-bench.sh -c also executes cpuminer stress tester (NEON/SSE/AVX)
    sbc-bench.sh -g graphs 7-zip MIPS for every cpufreq OPP
    sbc-bench.sh -G Geekbench mode, ensures benchmark is properly monitored
    sbc-bench.sh -h displays this help text
    sbc-bench.sh -j Jeff Geerling mode. Don's use it if you're not him
    sbc-bench.sh -m [$seconds] provides CLI monitoring (5 sec default interval)
    sbc-bench.sh -P Phoronix mode, ensures benchmark is properly monitored
    sbc-bench.sh -t [$degree] runs thermal test waiting to cool down to this value
    sbc-bench.sh -T [$degree] runs thermal test heating up to this value
    sbc-bench.sh -s also executes stockfish stress tester (NEON/SSE/AVX/RAM)

More information and an early implementation follows soon!

geerlingguy, 2023-02-03:
@ThomasKaiser - Haha I'm flattered :D

ThomasKaiser, 2023-02-05:
Here we go. [This commit](https://github.com/ThomasKaiser/sbc-bench/commit/1cae41be2f4c3bfc9c17b321d498f97bf692247a) contains early functionality:

    root@rock-5b:/home/tk# sbc-bench.sh -j
    Starting to examine hardware/software for review purposes...
    
    Average load and/or CPU utilization too high (too much background activity). Waiting...
    
    Too busy for benchmarking: 19:06:58 up 0 min,  2 users,  load average: 1.00, 0.28, 0.10,  cpu: 15%
    Too busy for benchmarking: 19:07:03 up 0 min,  2 users,  load average: 1.00, 0.29, 0.10,  cpu: 13%
    Too busy for benchmarking: 19:07:08 up 0 min,  2 users,  load average: 0.92, 0.29, 0.10,  cpu: 4%
    Too busy for benchmarking: 19:07:13 up 0 min,  2 users,  load average: 0.84, 0.28, 0.10,  cpu: 0%
    Too busy for benchmarking: 19:07:18 up 0 min,  2 users,  load average: 0.78, 0.28, 0.10,  cpu: 0%
    Too busy for benchmarking: 19:07:23 up 1 min,  2 users,  load average: 0.71, 0.27, 0.10,  cpu: 0%
    Too busy for benchmarking: 19:07:28 up 1 min,  2 users,  load average: 0.66, 0.27, 0.10,  cpu: 0%
    Too busy for benchmarking: 19:07:33 up 1 min,  2 users,  load average: 0.60, 0.26, 0.10,  cpu: 0%
    
    sbc-bench v0.9.11
    
    Installing needed tools: Done.
    Checking cpufreq OPP. Done.
    Executing RAM latency tester. Done.
    Executing OpenSSL benchmark. Done.
    Executing cpuminer. 5 more minutes to wait. Done.
    Checking cpufreq OPP again. Done (9 minutes elapsed).
    
    It seems neither throttling occured nor too much background activity.
    
    Full results uploaded to http://ix.io/4nek
    
    
    
    
    
    # Radxa ROCK 5B
    
    Tested on Sun, 05 Feb 2023 19:16:40 +0100. Full summary: [http://ix.io/4nek](http://ix.io/4nek)
    
    ## General information:
    
        Rockchip RK3588/RK3588s (35880000), Kernel: aarch64, Userland: arm64
        
        CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                         cpufreq   min    max
         CPU    cluster  policy   speed  speed   core type
          0        0        0      408    1800   Cortex-A55 / r2p0
          1        0        0      408    1800   Cortex-A55 / r2p0
          2        0        0      408    1800   Cortex-A55 / r2p0
          3        0        0      408    1800   Cortex-A55 / r2p0
          4        1        4      408    2400   Cortex-A76 / r4p0
          5        1        4      408    2400   Cortex-A76 / r4p0
          6        2        6      408    2400   Cortex-A76 / r4p0
          7        2        6      408    2400   Cortex-A76 / r4p0
    
    ## Governors (tradeoff between performance and idle consumption):
    
    Original settings:
    
        cpufreq-policy0: ondemand (conservative ondemand userspace powersave performance schedutil)
        cpufreq-policy4: ondemand (conservative ondemand userspace powersave performance schedutil)
        cpufreq-policy6: ondemand (conservative ondemand userspace powersave performance schedutil)
        dmc: dmc_ondemand (dmc_ondemand userspace powersave performance simple_ondemand)
        fb000000.gpu: simple_ondemand (dmc_ondemand userspace powersave performance simple_ondemand)
        fdab0000.npu: userspace (dmc_ondemand userspace powersave performance simple_ondemand)
    
    Tuned settings:
    
        cpufreq-policy0: performance
        cpufreq-policy4: performance
        cpufreq-policy6: performance
        dmc: performance
        fb000000.gpu: performance
        fdab0000.npu: performance
    
    ## Clockspeeds:
    
    Before at 40.7°C:
    
        cpu0-cpu3 (Cortex-A55): OPP: 1800, Measured: 1824      (+1.3%)
        cpu4-cpu5 (Cortex-A76): OPP: 2400, Measured: 2336      (-2.7%)
        cpu6-cpu7 (Cortex-A76): OPP: 2400, Measured: 2337      (-2.6%)
    
    After at 68.4°C:
    
        cpu0-cpu3 (Cortex-A55): OPP: 1800, Measured: 1806 
        cpu4-cpu5 (Cortex-A76): OPP: 2400, Measured: 2305      (-4.0%)
        cpu6-cpu7 (Cortex-A76): OPP: 2400, Measured: 2306      (-3.9%)
    
    All known settings adjusted for performance. System now ready for benchmarking.
    Once finished stop with [ctrl]-[c] to get info about throttling, frequency cap
    and too high background activity all potentially invalidating benchmark scores.
    root@rock-5b:/home/tk# 

Makes use of colors in Terminal and is hopefully Github issues friendly:

[image]

And will soon be extended. I want to implement a mode where all of this info is quickly gathered, then all governors are switched to `performance` and then `sbc-bench -j` just runs showing every 60 seconds basic monitoring info and waits for `[ctrl]-[c]` to then give another overview of clockspeeds and whether throttling and/or frequency capping (the RPi undervoltage phenomena) has happened.

ThomasKaiser, 2023-02-05:
With [latest commit](https://github.com/ThomasKaiser/sbc-bench/commit/c63feead29de2c41b5ad5a63bdb1942f28a11dba) the observation of other benchmarks that were fired up works now.

As can be seen after `sbc-bench` had fired up the `cpuminer` stressor for 5 minutes temperatures and average load drop for a few minutes until I fired up a `stockfish` benchmark (most demanding I know of, freezes my Rock 5B reliably at 2112 MHz DRAM clock). Then load and temperatures rise again and when `sbc-bench` is terminated it prints throttling info as last step:

    ...
    ## Software versions:
    
      * Ubuntu 22.10 arm64
      * Armbian: Rock 5B, rockchip-rk3588, rockchip-rk3588, 22.11.0-trunk.0114, https://github.com/armbian/build
      * Compiler: /usr/bin/gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0 / aarch64-linux-gnu
      * OpenSSL 3.0.5, built on 5 Jul 2022 (Library: OpenSSL 3.0.5 5 Jul 2022)
      * Kernel 5.10.72-rockchip-rk3588 / CONFIG_HZ=300
    
    All known settings adjusted for performance. System now ready for benchmarking.
    Once finished stop with [ctrl]-[c] to get info about throttling, frequency cap
    and too high background activity all potentially invalidating benchmark scores.
    
    Time       big.LITTLE   load %cpu %sys %usr %nice %io %irq   Temp   DC(V)
    21:42:00: 2400/1800MHz  5.91  41%   0%  41%   0%   0%   0%  58.2°C   5.20
    21:43:00: 2400/1800MHz  2.17   0%   0%   0%   0%   0%   0%  50.8°C   5.15
    21:44:01: 2400/1800MHz  0.83   0%   0%   0%   0%   0%   0%  47.2°C   5.16
    21:45:01: 2400/1800MHz  0.30   0%   0%   0%   0%   0%   0%  44.4°C   5.26
    21:46:01: 2400/1800MHz  0.11   0%   0%   0%   0%   0%   0%  41.6°C   5.21
    21:47:01: 2400/1800MHz  0.04   0%   0%   0%   0%   0%   0%  39.8°C   5.35
    21:48:01: 2400/1800MHz  3.91  63%   0%  63%   0%   0%   0%  57.3°C   5.11
    21:49:01: 2400/1800MHz  6.50  99%   0%  99%   0%   0%   0%  65.6°C   5.05
    21:50:01: 2400/1800MHz  7.45  99%   0%  99%   0%   0%   0%  71.2°C   5.02
    21:51:02: 2400/1800MHz  7.80  99%   0%  99%   0%   0%   0%  76.7°C   5.04
    21:52:02: 2400/1800MHz  7.93  99%   0%  99%   0%   0%   0%  80.4°C   5.01
    21:53:02: 2400/1800MHz  7.97  99%   0%  99%   0%   0%   0%  81.3°C   5.03
    21:54:02: 2208/1608MHz  8.83  99%   0%  99%   0%   0%   0%  81.3°C   5.09
    21:55:02: 2208/1608MHz  8.42  99%   0%  99%   0%   0%   0%  82.2°C   5.04
    ^C
    
    Cleaning up. Done.
    Checking cpufreq OPP again. Done.
    
    Clockspeeds now at 72.1°C:
    
        cpu0-cpu3 (Cortex-A55): OPP: 1800, Measured: 1794 
        cpu4-cpu5 (Cortex-A76): OPP: 2400, Measured: 2288      (-4.7%)
        cpu6-cpu7 (Cortex-A76): OPP: 2400, Measured: 2288      (-4.7%)
    
    ATTENTION: Throttling might have occured on CPUs 0-3 (Cortex-A55).
    
    ATTENTION: Throttling might have occured on CPUs 4-5 (Cortex-A76).
    
    ATTENTION: Throttling might have occured on CPUs 6-7 (Cortex-A76).
    
    
    Throttling statistics (time spent on each cpufreq OPP) for CPUs 0-3 (Cortex-A55):
    
    1800 MHz:  727.38 sec
    1608 MHz:  100.01 sec
    1416 MHz:       0 sec
    1200 MHz:       0 sec
    1008 MHz:       0 sec
     816 MHz:       0 sec
     600 MHz:       0 sec
     408 MHz:    0.30 sec
    
    Throttling statistics (time spent on each cpufreq OPP) for CPUs 4-5 (Cortex-A76):
    
    2400 MHz:  729.18 sec
    2208 MHz:   98.22 sec
    2016 MHz:       0 sec
    1800 MHz:       0 sec
    1608 MHz:       0 sec
    1416 MHz:       0 sec
    1200 MHz:       0 sec
    1008 MHz:       0 sec
     816 MHz:       0 sec
     600 MHz:       0 sec
     408 MHz:    0.29 sec
    
    Throttling statistics (time spent on each cpufreq OPP) for CPUs 6-7 (Cortex-A76):
    
    2400 MHz:  734.29 sec
    2208 MHz:   93.13 sec
    2016 MHz:       0 sec
    1800 MHz:       0 sec
    1608 MHz:       0 sec
    1416 MHz:       0 sec
    1200 MHz:       0 sec
    1008 MHz:       0 sec
     816 MHz:       0 sec
     600 MHz:       0 sec
     408 MHz:    0.28 sec
    root@rock-5b:/home/tk#

ThomasKaiser, 2023-02-06:
To elaborate on why the collected info matters and why adjusting **all** governors is important... let's start with the latter first.

On today's CPUs (even MCUs) there's conflicting goals to meet: max performance and min consumption and obviously you don't get both at the same time. There's stuff like DVFS (dynamic voltage frequency scaling --> ramping up chip's supply voltages only when needed) or 'race to idle' (e.g. sending everything storage to deep sleep to just wake the storage components every now and then to quickly write stuff and then send them to deep sleep again) and many more.

This stuff needs to be configured and SoC vendors ship with settings that meet _their_ customer's criteria to provide a high performing device with also an excellent battery life (low overall consumption). These settings obviously are designed around Android use cases since Linux doesn't exist as a market for the SoCs that can be found on today's SBC (the exceptions are TI, Renesas, NXP, maybe RPi/Broadcom and a few others). But what works well for a TV box or a tablet where someone watches movies, surfs the Internet or plays a game from time to time is almost never a good fit for a general purpose Linux system.

That's where settings matter: are those SoC vendor's defaults made for Android use cases a proper fit for benchmarking? Especially if the goal is 'benchmarking _the hardware_' and not 'benchmarking _the SoC vendor's compromises to achieve low power consumption with Android_'. Unfortunately this is what you're really benchmarking if you stay with the defaults.

The following is all Geekbench 5.5 (a software that IMO has many flaws) running on the same Rock 5B with the same software stack:

  * [Plain defaults](https://browser.geekbench.com/v5/cpu/20336539)
  * [Only cpufreq governors set to `performance`, `sbc-bench -G` mode](https://browser.geekbench.com/v5/cpu/20337379)
  * [All governors set to `performance`, `sbc-bench -j` mode](https://browser.geekbench.com/v5/cpu/20337000)

This is [plain defaults vs. all governors set to `performance`](https://browser.geekbench.com/v5/cpu/compare/20336539?baseline=20337000) and this [only cpufreq vs. all governors set to `performance`](https://browser.geekbench.com/v5/cpu/compare/20337330?baseline=20337000). Some individual benchmarks see a drop in performance of almost 40% and this is all just due to the board maker shipping with software/settings the SoC maker designed for a different target audience. At least now that RK3588(s) devices run with a vendor's BSP kernel. 

With 'latest and greatest' mainline kernel it might get even worse since at the SoC vendor hardware and software _experts_ worked together while with mainline it's pure luck which individual out there upstreams which portion of these adjustable mechanisms like DMC.

BTW: those governors at defaults do not only affect CPU benchmarks but storage and even network as well, see again this [Rock 5B example](https://github.com/radxa/kernel/commit/55f540ce97a3d19330abea8a0afc0052ab2644ef#commitcomment-79484235) and there the Phoronix comparison at the bottom (the SQLite test is almost pure random I/O). The proposed `ondemand/io_is_busy` tweaks are a good fit as Linux defaults (that's why I added them to Armbian long time ago) but for storage and network benchmarks all governors need to be set to `performance` since otherwise in a lot of cases just random numbers are generated.

@TheRemote maybe I missed it but I haven't found any governor settings when looking through the info collected by your script. Default governors do not affect storage benchmarks that much on x86 and with Raspberry Pi OS but on other SBCs based on the kernel version it can make a difference like night and day.

TheRemote, 2023-02-06:
@ThomasKaiser inxi captures all of this information (and the information I talked to in my other posts which was fixed in the kernel by Radxa):

```
CPU:
  Info: model: N/A variant: cortex-a72 bits: 64 type: MCP arch: ARMv8 family: 8 model-id: 0
    stepping: 3
  Topology: cpus: 1x cores: 4 smt: N/A cache: L1: 320 KiB desc: d-4x32 KiB! i-4x48 KiB
    L2: 1024 KiB desc: 1x1024 KiB
  Speed (MHz): avg: 1500 min/max: 600/1500 scaling: driver: cpufreq-dt governor: ondemand cores:
    1: 1500 2: 1500 3: 1500 4: 1500 bogomips: 432
  Features: asimd cpuid crc32 evtstrm fp
  Vulnerabilities:
  Type: itlb_multihit status: Not affected
  Type: l1tf status: Not affected
  Type: mds status: Not affected
  Type: meltdown status: Not affected
  Type: mmio_stale_data status: Not affected
  Type: retbleed status: Not affected
  Type: spec_store_bypass status: Vulnerable
  Type: spectre_v1 mitigation: __user pointer sanitization
  Type: spectre_v2 status: Vulnerable
  Type: srbds status: Not affected
  Type: tsx_async_abort status: Not affected
```

inxi really is very good.  I have a lot of other tests that show this as well that aren't shown by default but they don't need to be since inxi encapsulates all of it and that entire report is shown.

ThomasKaiser, 2023-02-07:
> inxi captures all of this information

@TheRemote unfortunately it does not. It just displays the `cpufreq` governor for `cpu0` and nothing else while with RK3588's BSP kernel we have 5 more governors in action all contributing to a diverse performance picture. 

And my point was and still is that it's not only about _reporting_ governors but _adjusting them_ prior to benchmarking. At least if the goal is benchmarking _the hardware_ and not just _SoC vendor settings made for Android use cases_.

As explained in my last comment: [this here](https://browser.geekbench.com/v5/cpu/compare/20337330?baseline=20337000) is two times Geekbench on an otherwise identical system. The only difference is DMC settings (even cpufreq governors were already set to `performance`). The difference this makes is massive.

Memory clock is important and ignoring this just introduces a random number generator into 'measurements'. That's why when adding the Stockfish load generator to sbc-bench a while ago I took my time to check influence of DRAM clock on RK3588 on a variety of benchmarks (benchmarking the benchmark): https://github.com/ThomasKaiser/sbc-bench/issues/55#issuecomment-1309375720

And wrt storage benchmarks... as mentioned in the last comment already: not setting at least all cpufreq governors to `performance` prior to benchmarking invalidates results at least with random I/O measurements: https://github.com/radxa/kernel/commit/55f540ce97a3d19330abea8a0afc0052ab2644ef#commitcomment-79484235

ThomasKaiser, 2023-02-07:
@geerlingguy I think after latest commit the `sbc-bench -j` functionality is now worth a look even if quick memory bandwidth measurements are still not implemented yet.

Basic idea (for now – later Ansible integration is a different topic) is to execute a benchmark or a set of benchmarks on a freshly booted SBC, save the results, then run `sbc-bench -j` as root in another Terminal, wait the 10 minutes for info collection, adjusting settings and throttling/stability tests until it beeps three times and sbc-bench switches into background monitoring:

    All known settings adjusted for performance. System now ready for benchmarking.
    Once finished stop with [ctrl]-[c] to get info about throttling, frequency cap
    and too high background activity all potentially invalidating benchmark scores.
    
    Time       big.LITTLE   load %cpu %sys %usr %nice %io %irq   Temp   DC(V)
    11:10:27: 2400/1800MHz  6.50  38%   0%  38%   0%   0%   0%  55.5°C   5.22
    11:11:28: 2400/1800MHz  2.26   0%   0%   0%   0%   0%   0%  48.1°C   5.30
    ...

Now the aforementioned benchmarks are executed again (this time generating higher/real scores) while sbc-bench shows potential background and swapping activity in real time (the `%sys` and `%io` columns as well as overall `%cpu`) and when stopped with `[ctrl]-[c]` it measures clockspeeds again and reports throttling (also frequency capping on underpowered RPis) both based on measurements, kernel statistics if the kernel supports it (`CONFIG_CPU_FREQ_STAT=y`) and on VideoCore platforms by asking `vcgencmd get_throttled`.

The amount of collected info now as follows (see also the link with [full info](http://ix.io/4nmJ)):

    root@rock-5b:/home/tk# sbc-bench.sh -j
    Starting to examine hardware/software for review purposes...
    
    sbc-bench v0.9.12
    
    Installing needed tools: Done.
    Checking cpufreq OPP. Done.
    Executing RAM latency tester. Done.
    Executing OpenSSL benchmark. Done.
    Executing cpuminer. 5 more minutes to wait. Done.
    Checking cpufreq OPP again. Done (10 minutes elapsed).
    
    It seems neither throttling occured nor too much background activity.
    
    Full results uploaded to http://ix.io/4nmJ
    
    
    
    
    
    # Radxa ROCK 5B
    
    Tested on Tue, 07 Feb 2023 11:10:27 +0100. Full info: [http://ix.io/4nmJ](http://ix.io/4nmJ)
    
    ## General information:
    
        Rockchip RK3588/RK3588s (35880000), Kernel: aarch64, Userland: arm64
        
        CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                         cpufreq   min    max
         CPU    cluster  policy   speed  speed   core type
          0        0        0      408    1800   Cortex-A55 / r2p0
          1        0        0      408    1800   Cortex-A55 / r2p0
          2        0        0      408    1800   Cortex-A55 / r2p0
          3        0        0      408    1800   Cortex-A55 / r2p0
          4        1        4      408    2400   Cortex-A76 / r4p0
          5        1        4      408    2400   Cortex-A76 / r4p0
          6        2        6      408    2400   Cortex-A76 / r4p0
          7        2        6      408    2400   Cortex-A76 / r4p0
    
    ## Governors (tradeoff between performance and idle consumption):
    
    Original settings:
    
        cpufreq-policy0: ondemand / 1008 MHz (conservative ondemand userspace powersave performance schedutil)
        cpufreq-policy4: ondemand / 408 MHz (conservative ondemand userspace powersave performance schedutil)
        cpufreq-policy6: ondemand / 408 MHz (conservative ondemand userspace powersave performance schedutil)
        dmc: dmc_ondemand / 528 MHz (dmc_ondemand userspace powersave performance simple_ondemand)
        fb000000.gpu: simple_ondemand / 300 MHz (dmc_ondemand userspace powersave performance simple_ondemand)
        fdab0000.npu: userspace / 1000 MHz (dmc_ondemand userspace powersave performance simple_ondemand)
    
    Tuned settings:
    
[... 43 lines cut ...]
    
    All known settings adjusted for performance. System now ready for benchmarking.
    Once finished stop with [ctrl]-[c] to get info about throttling, frequency cap
    and too high background activity all potentially invalidating benchmark scores.
    
    Time       big.LITTLE   load %cpu %sys %usr %nice %io %irq   Temp   DC(V)
    11:10:27: 2400/1800MHz  6.50  38%   0%  38%   0%   0%   0%  55.5°C   5.22
    11:11:28: 2400/1800MHz  2.26   0%   0%   0%   0%   0%   0%  48.1°C   5.30
    11:12:28: 2400/1800MHz  0.83   0%   0%   0%   0%   0%   0%  44.4°C   5.32
    11:13:28: 2400/1800MHz  0.30   0%   0%   0%   0%   0%   0%  42.5°C   5.35
    11:14:28: 2400/1800MHz  0.11   0%   0%   0%   0%   0%   0%  40.7°C   5.18
    11:15:28: 2400/1800MHz  0.04   0%   0%   0%   0%   0%   0%  38.8°C   5.25
    11:16:28: 2400/1800MHz  0.01   0%   0%   0%   0%   0%   0%  37.9°C   5.15
    ^C
    
    Cleaning up. Done.
    Checking cpufreq OPP again. Done.
    
    Clockspeeds now at 38.8°C:
    
        cpu0-cpu3 (Cortex-A55): OPP: 1800, Measured: 1826      (+1.4%)
        cpu4-cpu5 (Cortex-A76): OPP: 2400, Measured: 2339      (-2.5%)
        cpu6-cpu7 (Cortex-A76): OPP: 2400, Measured: 2339      (-2.5%)
    
    root@rock-5b:/home/tk#

ThomasKaiser, 2023-02-07:
> Power (testing power consumption over time during HPL and Geekbench, at least)

I added support for NetIO switchable powermeters to sbc-bench a while ago. In case you have something from here https://www.gridconnect.com/collections/netio (the 'PowerCable 101S' is already sufficient) then you could already enjoy precise power monitoring.

Those NetIO thingies measure power in two different ways at the same time ('actual' as well as cumulative consumption) and with fluctuating loads like e.g. Geekbench with its way too short execution times only cumulative mode is able to provide meaningful numbers. Some more info to this mode [here](https://github.com/ThomasKaiser/Check_MK/blob/master/agents/plugins/netio.cfg#L49-L89) but I don't know whether it makes more sense...

TheRemote, 2023-02-07:
@ThomasKaiser Oh, well that was only one test that captures it.  I also do lscpu and cat /proc/cpuinfo:

```
james@pop-os:~$ lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         43 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  48
  On-line CPU(s) list:   0-47
Vendor ID:               AuthenticAMD
  Model name:            AMD Ryzen Threadripper 3960X 24-Core Processor
    CPU family:          23
    Model:               49
    Thread(s) per core:  2
    Core(s) per socket:  24
    Socket(s):           1
    Stepping:            0
    Frequency boost:     enabled
    CPU max MHz:         3800.0000
    CPU min MHz:         2200.0000
    BogoMIPS:            7600.29
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb 
                         rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 mo
                         vbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt t
                         ce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibpb stibp vmmcall fsgsbase bmi1 avx2 s
                         mep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_loc
                         al clzero irperf xsaveerptr rdpru wbnoinvd amd_ppin arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists paus
                         efilter pfthreshold avic v_vmsave_vmload vgif v_spec_ctrl umip rdpid overflow_recov succor smca sev sev_es
Virtualization features: 
  Virtualization:        AMD-V
Caches (sum of all):     
  L1d:                   768 KiB (24 instances)
  L1i:                   768 KiB (24 instances)
  L2:                    12 MiB (24 instances)
  L3:                    128 MiB (8 instances)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-47
Vulnerabilities:         
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Mitigation; untrained return thunk; SMT enabled with STIBP protection
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; Retpolines, IBPB conditional, STIBP always-on, RSB filling, PBRSB-eIBRS Not affected
  Srbds:                 Not affected
  Tsx async abort:       Not affected
```

Then there is also cat /proc/cpuinfo:

```
james@pop-os:~$ cat /proc/cpuinfo
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 49
model name	: AMD Ryzen Threadripper 3960X 24-Core Processor
stepping	: 0
microcode	: 0x8301055
cpu MHz		: 2200.000
cache size	: 512 KB
physical id	: 0
siblings	: 48
core id		: 0
cpu cores	: 24
[... 1311 lines cut ...]
cache size	: 512 KB
physical id	: 0
siblings	: 48
core id		: 30
cpu cores	: 24
apicid		: 61
initial apicid	: 61
fpu		: yes
fpu_exception	: yes
cpuid level	: 16
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr rdpru wbnoinvd amd_ppin arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif v_spec_ctrl umip rdpid overflow_recov succor smca sev sev_es
bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass retbleed
bogomips	: 7600.29
TLB size	: 3072 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]

```

Those are just off the top of my head.  Some of my  other tests likely gather this information.

I don't see any changes that need to be made to pibenchmarks.com whatsoever here.  Thanks for the suggestions though!

ThomasKaiser, 2023-02-07:
> I don't see any changes that need to be made to pibenchmarks.com whatsoever here.

So let's stop here.

What I'm currently doing is preparing an `sbc-bench` mode for Jeff and hopefully other reviewers that can be used with whatever benchmark to provide two things:

  * sane settings that are a prerequisit for running any benchmark in a reasonable way
  * proper monitoring to get later an idea why numbers are this or that way

Nothing of this has any special relationship with Radxa, Rockchip or the Rock 5B. The Rock 5B is just here to illustrate why 'sane settings' matter since I went through all the details months ago.

And the point I tried to make (obviously to no avail if you think nothing needs to be changed with pibenchmarks.com) is that setting **all** performance related governors always to `performance` prior to benchmarking is a prerequisit for benchmarking 'hardware performance' and not hardware + settings.

This example (editing out `ondemand/io_is_busy`) was there to illustrate that:

https://openbenchmarking.org/result/2208036-NE-2208032NE57

    performance ......... 502.96 |=======================================
    schedutil ........... 543.28 |=============================================
    ondemand ............ 549.89 |==============================================

Cpufreq governors matter even with storage benchmarks and on platforms with dynamic DRAM clocking these settings of course matter as well. They are the reason for benchmark score differences of up to 200%, same with real-world tasks.

That's the whole point: adjust settings first since otherwise benchmark numbers are partially meaningless as long as the claim is 'benchmarking _hardware_'.

BTW: if your point of posting irrelevant `/proc/cpuinfo` contents here was to demonstrate walls of text being annoying... I know that very well and that's why the lenghty `sbc-bench` output is not meant to be thrown at consumers but on reviewers to get an idea what's really happening on the device in question while the reviewer executes whatever benchmark he likes.

geerlingguy, 2023-02-07:
@ThomasKaiser - One other thing to keep in mind: my main goal (unlike many other reviewers who might get deeper into the tuning weeds, ala CNX or ServeTheHome) is to give people a picture of the performance they could expect if they go buy the board, flash an OS to it, and start using it.

I do always strive to have a better understanding of the performance and raw potential—and I think there's a lot of progress towards that just in this thread—but I also need to keep things at a higher level in the videos, at least, and one of my goals is to compare like for like. If you buy a Pi, flash Pi OS, and run XYZ on it, will that run faster or slower than the same thing on a Rock 5 B with Radxa's official OS and all its defaults?

At a base level, that doesn't incorporate any settings like vulnerability mitigations being enabled, frequency scaling, memory speed tweaks, etc., but if run enough times with a small enough deviation in results, it gives one picture of the performance an end user will expect with a default experience.

There are two goals here—one is to get that 'default experience' score, which is useful to the majority who will never learn enough to be trusted overclocking or tweaking kernel settings, and the other is to get the 'raw performance' score which is better to expose bugs in the OS images, shady software hacks in a BSP, that sort of thing. Both are valuable to me :)

geerlingguy, 2023-02-07:
(And one thing I really _really_ hate about video—I can never change the video once posted. So I try to link back to a blog post, at least, where I can add notes and update links if I need to.

ThomasKaiser, 2023-02-07:
> There are two goals here ... 'default experience' score ... 'raw performance' score. Both are valuable to me :)

Exactly. And my efforts now are all about making this comparison as easy as possible while trying to provide the answer to 'why do numbers differ' at the same time.

You mentioned/asked why 'the RK3588S on the Orange Pi being noticeably faster than the RK3588 on the Rock 5 B' above, I answered with 'easy explanation' followed by DMC, PVTM and whatever. Based on the research done on RK3588 within the last 6 months and the stuff currently still floating around in my head.

Then I thought: why not compiling my 'style of research' (`find /sys -name "*governor*`and so on) into another sbc-bench mode so this primitive methodology is available to everyone and even in an automated fashion.

And that's all this `sbc-bench -j` approach is about. Tuning the available settings for 'raw performance' while also providing hints or maybe even insights why numbers differ between devices. Just execute your set of benchmarks with defaults, then run `sbc-bench -j` to gather information and tune settings and rerun the benchmarks to compare what has changed in which way.

BTW: since having watched your 5B review in the meantime I added `/sys/module/pcie_aspm/parameters/policy` to governor reporting since your experience with crappy network bandwidth in one direction smells like PCIe ASPM again set to `powersave`.

Now big kudos to you: the way you review/benchmark publicly in the open / here on Github makes the whole process almost scientific but at least _bidirectional_. Problems/bottlenecks identified along the journey of getting in touch with a new device could result in a feedback loop to board/SoC vendor who then almost always adjust defaults if able to understand why it matters.

At least that's the mode I prefer and for example with Radxa it worked rather well: they reacted directly to issues brought up: [increasing MMC clock after me complaining](https://github.com/radxa/kernel/commit/dc46fcef4ac28b044f88320318c28b0a7132fa53) and [some of the other stuff listed here](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#suggestions-to-radxa).

For me benchmarking/reviewing something is always the process of improving settings, as such my 'default experience' while testing should be way worse compared to those final consumers enjoying better settings once the SBC reaches their hands since improving 'default experience' is always part of the journey.

But maybe I'm too biased due to the way benchmarking works as part of my day job. I never got hired to 'report numbers' but only to improve them. And ofc it isn't about numbers but challenges like 'is the server fit for another 50 employees'. The numbers just help identifying the bottlenecks. And looking back 2 decades it was never hardware that had to be replaced but always software/settings.

RadxaYuntian, 2023-02-14:
TK thanks for your investigation. I have applied the suggested DMC tuning here: https://github.com/radxa/kernel/commit/6e4e9e0a7f6ed85ec099f34e95ed687b4893b031. The fix is applied to the tree on `rk3588s.dtsi` so Armbian folks can benefit on this for their OPi image as well.

As we are currently supporting RK3399 with the same kernel as RK3588, do you think we can apply some DMC tuning for that family as well?

ThomasKaiser, 2023-02-14:
@RadxaYuntian let's continue here to not further hijack this issue: https://forum.radxa.com/t/probable-dmc-tunings-with-5-10-bsp-kernel-on-rk3399/14719

geerlingguy, 2023-09-07:
For the time being, I've tacked on some phoronix tests (which are easily accessible to anyone who's tried PTS before, and don't take too long to complete, and added on a `tinymembench` directive in the 'Memory' section.

Still haven't settled on some simple benchmarks for GPU.

ThomasKaiser, 2024-03-01:
Just a few notes/thoughts on the infamous Geekbench especially wrt RISC-V.

These are Geekbench **6** results for Sophon SG2042 (64 C910 cores): https://browser.geekbench.com/search?utf8=✓&q=rv64imafdcv

[image]

Single-core results are all pretty much the same but multi-core the four results at the bottom look even more weird than the rest since being between 521 and 760 (or in other words: _3.5 to 5 times_ the single-core score). What do these insanely low scores at the bottom all have in common? Made with Geekbench 6.2.0 and 6.2.1 while the 'better' scores are all made with 6.2.2 so obviously the Geekbench guy slowly improves his RISC-V coding skills.

And even these 'better' multi-core scores are a bad joke since the 'best' is at 1458 or only _10 times_ the single-core score. This is still a 64-core SoC and with other benchmarks the single-core to multi-core ratio is for example 1:25 ([Geekbench **5**](https://browser.geekbench.com/search?k=v5_cpu&q=rv64imafdcv&utf8=✓)) or even 1:37 ([7-zip](https://github.com/ThomasKaiser/sbc-bench/blob/master/results/4wYE.txt#L517))

geerlingguy, 2024-11-25:
For GPU testing, I think I may add on [GravityMark](https://gravitymark.tellusim.com) at 720p

CoenWarmer, 2025-01-07:
Very interested to see what Linux 6.3 will bring to Orange Pi 5 with Vulcan support.

svanimisetti, 2025-02-17:
@geerlingguy - #46 lists a GPU benchmark. Did you use the M.2 JMT adapter (as shown in the recent [YouTube video](https://youtu.be/AyR7iCS7gNI)?). It would be great if you could share any details on how to get the Milk-V Mars to talk to an NVIDIA RTX 3050 or AMD Radeon RX 6600. I can try to run my benchmarks and post it to #45.

geerlingguy, 2025-02-17:
@svanimisetti - For the Mars, it was just using the internal GPU (or might've been using llvmpipe, using the CPU as a GPU). I have only experimented with dGPUs on their larger boards.