geerlingguy/sbc-reviews #3: Radxa Rock 5 model B

geerlingguy, 2023-01-26:
TODO.

geerlingguy, 2023-01-26:
I'm perplexed as to how the Orange Pi 5 (with the RK3588S) is benchmarking faster than the RK3588 in the Rock 5 model B, at least according to both my testing _and_ [Christopher Barnatt](https://twitter.com/ChrisBarnatt/status/1618417249412591620).

samuk, 2023-01-26:
I'd be interested how it goes with the Armbian kernel/ ISO https://forum.armbian.com/topic/24625-mainline-kernel/#comment-153904

TheRemote, 2023-01-27:
Curious.  I tried this really quickly on my Orange Pi 5 vs. Rock 5B:

```
orangepi@orangepi5:~$ sysbench --threads=8 cpu run
sysbench 1.0.20 (using system LuaJIT 2.1.0-beta3)

Running the test with following options:
Number of threads: 8
Initializing random number generator from current time

Prime numbers limit: 10000

Initializing worker threads...

Threads started!

CPU speed:
    events per second: 13758.30

General statistics:
    total time:                          10.0007s
    total number of events:              137621

Latency (ms):
         min:                                    0.38
         avg:                                    0.58
         max:                                   28.04
         95th percentile:                        1.10
         sum:                                79836.93

Threads fairness:
    events (avg/stddev):           17202.6250/6752.63
    execution time (avg/stddev):   9.9796/0.01

orangepi@orangepi5:~$ uname -a
Linux orangepi5 5.10.110-rockchip-rk3588 #1.1.0 SMP Fri Jan 6 15:58:17 CST 2023 aarch64 GNU/Linux

orangepi@orangepi5:~$ cat /etc/os-release
PRETTY_NAME="Orange Pi 1.1.0 Bullseye"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

And the Rock 5B:

```
rock@rock-5b:~$ sysbench --threads=8 cpu run
sysbench 1.0.20 (using system LuaJIT 2.1.0-beta3)

Running the test with following options:
Number of threads: 8
Initializing random number generator from current time

Prime numbers limit: 10000

Initializing worker threads...

Threads started!

CPU speed:
    events per second: 14803.50

General statistics:
    total time:                          10.0006s
    total number of events:              148066

Latency (ms):
         min:                                    0.36
         avg:                                    0.54
         max:                                   11.03
         95th percentile:                        1.10
         sum:                                79954.50

Threads fairness:
    events (avg/stddev):           18508.2500/9344.12
    execution time (avg/stddev):   9.9943/0.00

rock@rock-5b:~$ uname -a
Linux rock-5b 5.10.66-27-rockchip-gea60d388902d #rockchip SMP Mon Oct 24 08:25:47 UTC 2022 aarch64 GNU/Linux

rock@rock-5b:~$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

Great catch.  I'd imagine that you're on to something.  I know there are a lot of revisions of Rock 5B already whereas for the Orange Pi 5 we should have relatively the same board.  I wonder if some of this can be explained by board revisions / differences on the Rock 5B.

Which version of the board do you have?  I have V1.42.2022.08.29.  There are already several different versions of the Rock 5B out there:

- V1.3(Developer Edition)
- V1.41(Pre Production Edition)
- V1.42

They don't make it clear what the changes are to these at all but [they all have their own spec sheets / schematics / etc. on the Radxa site](https://wiki.radxa.com/Rock5/hardware).

TheRemote, 2023-01-27:
I actually have a follow-up on this for you to try.  I was able to update the Radxa Rock 5B kernel.  You and I were both running the stock kernel 5.10.66-27.

The Radxa updates are broken in the stock kernel.  If you try to do sudo apt update it will tell you you need to add the apt repository key.  It's quite broken in other words.  Here's how we fix it with a one-liner:

`wget -O - apt.radxa.com/bullseye-stable/public.key | sudo apt-key add -`

Now you can do:

`sudo apt update && sudo apt upgrade`

It will tell you that a few packages will be downgraded.  Let it.  This took me all the way up to kernel 5.10.110:

```
rock@rock-5b:~$ uname -a
Linux rock-5b 5.10.110-35-rockchip-g98c1daa32982 #rockchip SMP Thu Jan 12 03:36:12 UTC 2023 aarch64 GNU/Linux
```

Now our kernel is 5.10.110 just like the Orange Pi 5's.  They are (very nearly) identical versions.

That's not the interesting part though.  Take a look at this:

```
rock@rock-5b:~$ uname -a 
Linux rock-5b 5.10.110-35-rockchip-g98c1daa32982 #rockchip SMP Thu Jan 12 03:36:12 UTC 2023 aarch64 GNU/Linux
rock@rock-5b:~$ lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
...
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable: Unprivileged eBPF enabled
```

The security settings now match!

Can you try updating your kernel after fixing the apt key for Debian and retest and see if the scores are more reasonable?

geerlingguy, 2023-01-28:
> The security settings now match! It's for the worse as far as vulnerabilities go since they're both now vulnerable (yay) but it's very good for benchmarking purposes that they're now the same.

Heh... that's not a great endorsement, that when you update your system, you end up with more vulnerabilities :P

But I will try.

RadxaYuntian, 2023-01-28:
I did not recall we touched the mitigation configuration, so this might be a side effect of updating our kernel to a newer Rockchip SDK release. In any case I'll add them to my kernel config override.

TheRemote, 2023-01-28:
Absolutely!  I would have tried against the Geekbench but I didn't realize it needs a license.  It's pretty cheap at least but it's hard to want to pay for someone else's benchmark when I am developing one because it makes me feel like "ugh, I should just add whatever features I want to pay someone else to mine!".

It turns out you can toggle this setting too.  On the Orange Pi 5:

```
root@orangepi5:/home/orangepi# sudo lscpu
Architecture:                    aarch64
...
Vulnerability Spectre v2:        Vulnerable: Unprivileged eBPF enabled
```

Now the magic to shut it off:

`echo 1 > /proc/sys/kernel/unprivileged_bpf_disabled`

```
root@orangepi5:/home/orangepi# sudo lscpu
Architecture:                    aarch64
...
Vulnerability Spectre v2:        Mitigation; CSV2, BHB
```

And now we have security mitigations enabled again (but only until a reboot).  Interestingly these still don't quite match toggling the setting.  The only way I got them to completely match was updating the Rock 5Bs.

[Here's an interesting reference](https://www.phoronix.com/news/Linux-6.1-Disable-ARM64-BHI-BHB).  They actually added a new kernel flag to disable the BPF mitigations in the official kernel because according to this article it can impact benchmarks by up to 20%.  This is specifically BHB (branch history buffer).

This might not be anything you guys did at Radxa.  I'd imagine it's upstream.  Maybe the reason the settings match after updating is simply due to upstream kernel changes in the RockChip kernel perhaps?

samuk, 2023-01-28:
Is anyone intersted in comparing with the 6.xx kernel? https://github.com/armbian/build/releases/tag/23.02.0-trunk.0191

RadxaYuntian, 2023-01-29:
@TheRemote: The security config has been enabled: https://github.com/radxa-repo/bsp/commit/8584bc2648fd6a59a64f6fb8f416197b9bd92e67 This config set `unprivileged_bpf_disabled` to 2, so admin can either change it to 0 to enable unprivileged eBPF, or to 1 to disable it until next reboot. You can check the linked SUSE KB article for some additional details.

@samuk: We won't officially support Armbian since ultimately Armbian cannot provide the same hardware support as the Rockchip SDK (which is a good thing because Rockchip SDK comes with its own compromises), and we don't have enough resources to support 2 systems. We do submit PRs to Armbian because some of our customers want to use that one, but they usually understand the limitations, and are capable for developing their own solutions.

TheRemote, 2023-01-29:
@RadxaYuntian That's fantastic, I'm familiar with the 2 setting for privileged users and that's great that is available.  Using it in trusted/secure places would be my preference as well.  It's definitely a more precise and elegant solution to give you options to still take advantage of that performance when safe / appropriate.

Thank you for such a fast response and taking care of this!

samuk, 2023-01-29:
@RadxaYuntian understood. My particular interest is the upcoming CM5 modules. Would you able to provide $free hardware to the Armbian devs to assist their work?

RadxaYuntian, 2023-01-30:
We have provided free hardware samples to known projects for quite awhile. However, that also depends on if they are interested in it. Last time when I offered Zero 2 to 2 Armbian developers only one person took it.

rpardini, 2023-01-30:
> Last time when I offered Zero 2 to 2 Armbian developers only one person took it.

True. I took it a few months ago, and is it's a great/fantastic board, and Armbian support for it is in now what I consider in good shape, with all firmware and patches included, and fully working mainline u-boot 22.10 and kernel `6.2-rc5`; thanks to Neil Armstrong & co. I just wish it was... _actually released_ so it can be mainlined...

rpardini, 2023-01-30:
Ref the rk3588(s) boards: Armbian has a dedicated `legacy`/`vendor` kernel with the 2.6.x/5.10.x stuff from rk/Radxa, plus extra patches/drivers/overlays etc. The OPi5 has joined that party and the Khadas Edge2 is heading there too. Adding the Radxa variants (CM5, A/B, etc) shouldn't be too hard. 

We've also an `edge` kernel carrying some of `sre`'s mainline patches for the rk3588 -- but that's not really usable _yet_, but we're ready for when it is.

geerlingguy, 2023-09-05:
FYI I've run a few more tests (phoronix test suite ones, added to the OP), running Armbian's latest release for the Orange Pi because their download folder in Google Drive is empty, and running the latest Debian 11 build (35, I think?) for the Rock 5 B, and the OP5 is still faster in a couple benchmarks, but others are neck in neck.

lukeomatik, 2023-10-22:
Hi. Is it wise to assume sort of same results for the Orange Pi 5 Plus? It is a RK3588 (non S) with 2 Independent 2.5gbps ports (they did a dirty job on the usb3.0 ports instead, ive read the schematics)

uri-tech, 2025-01-31:
Hi, @geerlingguy  can you compare the "Radxa ROCK 5B+"? the 32GB side by site to the "Radxa ROCK 5C" 32GB will be great to see

kurtmorris, 2025-08-28:
I'd be curious to see a follow up using the mainline kernel, [Collabora](https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md) has been working hard!

nsapa, 2026-10-09:
Running vanilla Debian is now possible with [edk2-rk3588](https://github.com/edk2-porting/edk2-rk3588).
One strange issue: if you connect the USB-serial adaptor before the power supply, the nvme go away a few minutes after booting.
```
[itoperation@xavia:~] $ fastfetch
        _,met$$$$$gg.          itoperation@xavia
     ,g$$$$$$$$$$$$$$$P.       -----------------
   ,g$$P""       """Y$$.".     OS: Debian GNU/Linux 13 (trixie) aarch64
  ,$$P'              `$$$.     Host: ROCK 5 Model B (0)
',$$P       ,ggs.     `$$b:    Kernel: Linux 6.12.111+deb13-arm64
`d$$'     ,$P"'   .    $$$     Uptime: 5 days, 5 hours, 31 mins
 $$P      d$'     ,    $$P     Packages: 456 (dpkg)
 $$:      $$.   -    ,d$$'     Shell: bash 5.2.37
 $$;      Y$b._   _,d$P'       Terminal: /dev/pts/0
 Y$$.    `.`"Y$$$$P"'          CPU: rk3588 (8) @ 2.40 GHz
 `$$b      "-.__               GPU: Rockchip rk3588-mali [Integrated]
  `Y$$b                        Memory: 417.58 MiB / 15.16 GiB (3%)
   `Y$$.                       Swap: 0 B / 12.30 GiB (0%)
     `$$b.                     Disk (/): 1.93 GiB / 220.61 GiB (1%) - ext4
       `Y$$b.                  Local IP (enP4p101s0): 10.132.2.22/25
         `"Y$b._               Locale: en_CA.UTF-8
             `""""

```