geerlingguy/sbc-reviews #4: Raspberry Pi 4 model B

geerlingguy, 2023-01-26:


geerlingguy, 2023-03-02:
Benchmark of SD card on PiBenchmarks.com: https://pibenchmarks.com/benchmark/67793/

arekm, 2024-01-31:
I wonder about one thing. You seem to never experienced rpi 4 wifi instability (causing lockup (only wifi lockup) requiring hard reset to get wifi working again). It's daily (or rather hourly) experience here on *every* rpi 4 board (but only have 4GB, rev 1.5 boards; official PSU). Leaving iperf -t0 running is enough to trigger.

k-koeberl, 2024-03-04:
May be this question is wrong here but some of you may have seen the same effect as myself.
I am using Sandisk Ultra Fit from 32GB up to 512GB for a long time now on my RPI3 and RPI4 boards and they work fine. Lately I have ordered "new" 256GB Ultra Fit's and they are unusable slow.

This are the parameters of the "old" ones that are working well
SanDisk Ultra Fit 256GB (made in China)
=======================
Bus 002 Device 008: ID 0781:5583 SanDisk Corp. Ultra Fit
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0781 SanDisk Corp.
  idProduct          0x5583 Ultra Fit
  bcdDevice            1.00
  iManufacturer           1 SanDisk
  iProduct                2 Ultra Fit
  iSerial                 3 0401a026bcbb31d7dc3a43a4fd6a281e69edd0bfafede1b68eed300273ea9b981c960000000000000000000090719afaff806e18835581077c271bb5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
[... 89 lines cut ...]
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   2
      Lowest fully-functional device speed is High Speed (480Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled

Under Windows both are working at the same speed and without any problem. I also tried to get further information
from the vendor but the only comment was that the USB flash drives are only certified for Windows.
Has anyone seen similar effects ? Are there any parameters that could make the new Sandisk Ultra Fits work under Linux ?

L4wnmower, 2024-12-22:
Hi Jeff,
If you could manage a bit of extra testing time, it would be interesting to amend above results for the Pi4B & Pi400, using latest Bookworm with the NUMA patches. Same applies to Pi5 and Pi500 (although these may have the performance bump included?)
Thanks. 
Merry Xmas and A Wonderful New Year

geerlingguy, 2024-12-22:
@L4wnmower - I have re-tested both Pi 5 and 500 (and CM5), at least on many of the CPU-related tests, and added those results inline with their posts (for comparison to the original software tests). Will try to do Pi 4 at some point too.

ankumm, 2025-01-12:
I just bought one of these sticks. I copied an existing Dietpi installation to the stick - and on a RP4 it works, but only extremly slowly.
It was booting the whole night until the containers on it started ;-).
Dietpi runs with 100% IO on it. So everything extreeemly slow.
I couldnt find anything on the net speaking about this stick than this here.

I could do test for you, but i don't have any clue where to go