geerlingguy/sbc-reviews #5: Orange Pi 5

geerlingguy, 2023-01-26:
TODO.

geerlingguy, 2023-09-05:
I tried downloading Orange Pi OS today and all the Google Drive folders for Arch and OH (whatever that is?) were empty :(

[image]

So sticking with Armbian for my testing.

dkarter, 2023-12-28:
Hi Jeff,

First, huge thank you for all your work and for making technology more accessible and easy to understand!

I'm wondering if you've ever done/planning to do a review of the Orange Pi 5 Plus - the specs seem VERY compelling when compared even with Raspberry Pi 5. Particularly the 32GB RAM, built-in NVME and 2x 2.5Gbps ports, NPU and GPU. Too good to be true? :)

http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus-32GB.html

Cheers!

geerlingguy, 2023-12-28:
@dkarter - I would, but it seems impossible to get my hands on one... It seems to be even harder to get than a Raspberry Pi 5 lately.

leon332157, 2024-01-10:
It appears that the OH version is the orangepi os based on OpenHarmony.

amrutprabhu, 2024-08-18:
Hey @geerlingguy ,

I was doing some research about the Orange Pi 5 and its new MAX version. 

I see in the geekbench 6 scores you mentioned above, I see that it only shows the CPU to be 1.8 GHz, while the specs of the Orange Pi 5 have the A76 @2.4 GHz and an A55 @1.8 GHz. 

is it that GeekBench only uses the A55 to do the multicore test?

geerlingguy, 2024-08-18:
@amrutprabhu - Geekbench is always a bit funky with CPU + frequency detection. Sometimes it thinks there's just 1 core with 4 threads for a 4 core Arm SoC, sometimes it just reports 1/1, sometimes it reports a lower frequency or higher... I wouldn't put much stock into the Geekbench results.

It's better to run something like [`sbc-bench`](https://github.com/ThomasKaiser/sbc-bench) to see actual frequencies on all cores. Honestly I was also testing this board with Armbian as the Orange Pi OS version was hard to find when I was testing... so that could have something to do with it too.

amrutprabhu, 2024-08-22:
> @amrutprabhu - Geekbench is always a bit funky with CPU + frequency detection. Sometimes it thinks there's just 1 core with 4 threads for a 4 core Arm SoC, sometimes it just reports 1/1, sometimes [...]

ohh.. thanks for this. I haven't explored this tool yet. 
Geekbench kind of made it simple to measure scores across devices.