geerlingguy/sbc-reviews #15: Radxa Rock 3 Compute Module (CM3)

geerlingguy, 2023-02-22:
[image]

## Basic information

  - Board URL (official): https://wiki.radxa.com/Rock3/CM
  - Board purchased from: Allnet.China
  - Board purchase date: January 23, 2023
  - Board specs (as tested): No WiFi / 1GB RAM / no eMMC
  - Board price (as tested): $30 (+$12 shipping)

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

geerlingguy, 2023-02-22:
It looks like you need to manually modify the dtbs included with the Debian image before it will work correctly on the CM4 IO Board from Raspberry Pi: https://github.com/radxa-build/radxa-cm3-io/releases/tag/20221101-0118

geerlingguy, 2023-02-22:
Even with that... I put the thing on my CM4 IO Board, and nothing. The green ACT LED is very dim. The red PWR LED is brightly lit. The Ethernet port's LEDs are both on. No coil whine, and the RK3566 doesn't get hot.

[image]

I just found there are some newer releases. Going to try one of those... https://github.com/radxa-build/radxa-cm3-io/releases/tag/20230201-0059

geerlingguy, 2023-02-22:
Looks like someone else has similar issues here: https://forum.radxa.com/t/cm3-v1-3-waveshare-cm4-io-base-a-wont-boot/10324/7

I tried shorting GLOBAL_EN to GND (https://wiki.radxa.com/Rock3/CM3/raspcm4io/getting_started#Power_on), but that didn't do anything either. My board revision is V1.31A — it seems that shouldn't be required. Ah well... I'll wait to see if anyone has any ideas on Twitter or Mastodon and come back to this later.

I've already sunk about 6 hours of my life into the two CM3 revisions I have (V1.1 with eMMC, and V1.31A with no eMMC), not sure how much more time I'll spend here.

(And no, I don't think buying the Radxa CM3 IO board is the answer—my goal is to see how well these things are drop-in replacements for the CM4, since there are _hundreds_ of carrier boards that assume CM4 compatibility.)

RadxaYuntian, 2023-02-23:
I'd say we are in a pretty awkward timing as far as CM3 support goes right now. On one hand we have received a few reports about CM3 boot issue on Raspberry Pi CM4 IO, on the other hand our own IO board sample is broken and we are waiting for delivery.

While I can't help our kernel developers with their lack of hardware, I have created a [separate image](https://github.com/radxa-build/radxa-cm3-rpi-cm4-io/releases/) for use with Raspberry Pi IO board a while ago. I was working on another RK3566 based products for a few weeks, so there are some improvements to this platform since the last build, and I just initiated a new one. However, our current development and testing focus is not on CM3, so it will take a while before we officially release it.

I think in the short term you can give this new image a go. If it doesn't work (which I'm afraid is likely), I'll make sure once we get the RPi CM4 IO, there will be an boot-able image in that repository.

bastelfreak, 2023-02-23:
@RadxaYuntian are there any plans for upstreaming your patches so mainline linux will work? That would make a lot of stuff easier.

RadxaYuntian, 2023-02-23:
I'm not sure which patches you are talking about, but you can ask @RadxaNaoki about our up-streaming efforts.

RadxaNaoki, 2023-02-23:
patches for CM3 were already submitted
https://git.kernel.org/pub/scm/linux/kernel/git/mmind/linux-rockchip.git/commit/?h=v6.3-armsoc/dts64&id=096ebfb74b19f2d4bdcbc33ae02e857ff4b3e0a0

geerlingguy, 2023-02-23:
@RadxaYuntian Thank you so much for your help—I will give that a try later! I'm guessing the CM5 is getting some more focus (as it should—I think it is the more compelling alternative to the CM4 with it's way faster cores!)... I would love to take a look at one once they're available.

geerlingguy, 2023-03-08:
It seems like @TheRemote had a similar story as mine above... no ability to boot it on the CM4 IO board :(

[Radxa Rock 3 Compute Module Review](https://jamesachambers.com/radxa-rock-3-compute-module-review/)

Jamos1988, 2023-04-20:
Anybody tried the Armbian yet? From https://wiki.radxa.com/Rock3/downloads I get redirected to the Armbian website, there I can download the CLI or XFCE version. No luck boothing them. Only one I can get tot work is the Debian one. Hoped to get Armbian working too as I have projects that are more compatible there.

darioteixeira, 2023-04-24:
@Jamos1988 Could you explicit what you mean by the "Only one I can get tot work is the Debian one"? There's a Debian image from Armbian and one from the Radxa website -- which one do you mean?

geerlingguy, 2023-05-31:
See also makerbymistake's CM3 (eMMC) review: https://dphacks.com/2023/05/30/radxa-rock-cm3-review-all-you-need-to-know/

RadxaYuntian, 2023-06-01:
I just fixed [U-Boot support](https://github.com/radxa-repo/bsp/blob/main/u-boot/latest/0031-rk356x/0201-Add-Radxa-CM3-RPI-CM4-IO-dummy-support.patch) for CM3 on Raspberry Pi CM4 IO 2 days ago. We are working on upgrading our rk356x to use Rockchip 5.10 kernel, and will release a new image soon.

darioteixeira, 2023-06-01:
@RadxaYuntian Thank you for the update. Will these fixes allow one to use any variety of the Rock CM3 (with/without eMMC) on any board that expects a Raspberry Pi CM4?

RadxaYuntian, 2023-06-01:
For eMMC less CM3 you will need to install the bootloader on the on-board SPI, otherwise the SoC will not boot from microSD.

RadxaYuntian, 2023-06-01:
The code is not released yet and is still being worked on (I'm looking at the USB support on Raspi CM4 IO again now), but you can use [the last 2 commands](https://radxa-repo.github.io/bsp/firmware/management.html#setupsh-maskrom_autoupdate_spinor) to update SPI bootloader once you build it yourself.

RadxaYuntian, 2023-06-07:
I cannot answer that question unless I have the board in hand to test it.

darioteixeira, 2023-06-07:
> I cannot answer that question unless I have the board in hand to test it.

Ah, since Turing Pi has announced that they have a collaboration with Radxa to develop the RK1 (a compute module for the Turing Pi 2), I assumed that you guys already had access to a Turing Pi 2 board!

Saleh-At, 2023-07-27:
How to build Armbian os on cm3 io?!
Do you have any tutorials?!
Because I have a qtquick project that needs opengl.

xEBIx, 2023-09-06:
I do have a turing2 Board and 2 Radxa cm3 with raspi-cm4-io cards. To build u-boot is done also. 
Previous I switched on power to the cm3 and got a glowing and 2 lit LEDs on the CM3/raspi-card, they did not change brightness or anything. Now I did start one in Maskrom mode and could get the first of the last two commands from here throug. The last command wont work because I dont get an mtd device. I am writing SPI Flash with the Radxa cm3-io Board. As I didnt know how to do that with the turing board, nor if it is even possible. 
After flashing the CM3 is starting all 3 LED come on bud the lower one on the raspi-card is blinking twice. 
Also I cannot get any output via a serial console, I have a cable connected to the pins for that node. 
Any advice how to get further? How I can get output? Is the last command nessesary, if so when should that mtd device come up?

xEBIx, 2023-09-06:
I've got the Board. But thank you for the Information that it wont work on the turing at all.

RadxaYuntian, 2023-09-07:
@xEBIx ,

> The last command wont work because I dont get an mtd device.

[setup.sh update_spinor [mtd_device]](https://radxa-repo.github.io/bsp/firmware/management.html#setupsh-update_spinor-mtd_device) is intended to run within the system of CM3. You won't see the MTD device when using maskrom connected with OTG cable. It does basically the same thing as [setup.sh maskrom_autoupdate_spinor](https://radxa-repo.github.io/bsp/firmware/management.html#setupsh-maskrom_autoupdate_spinor) but one runs from another computer, while one runs from the system itself. So I won't worry about it as you got the maskrom one working.

> After flashing the CM3 is starting all 3 LED come on but the lower one on the raspi-card is blinking twice.

It's the heartbeat LED trigger. That means you have loaded Linux kernel.

> Also I cannot get any output via a serial console, I have a cable connected to the pins for that node.

Make sure your serial cable is set to baud 1.5M with no flow control. Not all cables support this baud, so check [our recommendation](https://docs.radxa.com/general-tutorial/serial) (urgh this is Chinese only. I'll create an issue to update it).

Another thing is that the build system you are using is not released for CM3+CM4 IO yet, although we are currently [iterate on it](https://github.com/radxa-build/radxa-cm3-rpi-cm4-io/releases) and should see it officially released soon. In the mean time if you want to do self build make sure you specify radxa-cm3-rpi-cm4-io as your product (i.e. `./bsp u-boot latest radxa-cm3-rpi-cm4-io`, `./rbuild -c latest radxa-cm3-rpi-cm4-io`). This will allow you to use the correct U-Boot and device tree.

@darioteixeira ,

> the reply I got was that due to a hardware mess-up the CM3 can only be flashed on Radxa's own IO board.

I'll need to double check the status on this. The engineer @nascs who is working on this is not in the office today.

xEBIx, 2023-09-07:
@RadxaYuntian 
Thank you for that detailed reply.
Shall I use the rpi-cm4-io variant for the cm4 sodimm adapter card, or the sodimm variant? I supposed the cm4-io is for the raspberry cm4 Io board, the equivalent to the cm3-io board. Right now I use the sodimm variant.

The serial cable is the one u are selling at radxa, and it is set for 1.5k baud yet. Also the led is flashing just twice and stays off after that. The kernel heartbeat would flash until the kernel shuts down, as I see on other SBC hardware.

RadxaYuntian, 2023-09-08:
Urgh if you are using CM4 SODIMM adapter then it is untested. In theory you should use rpi-cm4-io variant but we can't guarantee its compatibility. Do you have an instruction of your CM4 SODIMM so I can see if there is any extra steps required to run a normal Raspberry Pi CM4?

@darioteixeira , we just checked and confirmed that you can flash CM3 on Raspberry Pi CM4 IO, as long as you follow our instruction: https://wiki.radxa.com/Rock3/installusb-install-radxa-cm3-rpi-cm4-io

xEBIx, 2023-09-08:
@RadxaYuntian 
Thanks for your reply.
Here is the installation instruction Page for a Raspberry CM4 card on that adapter. 
https://docs.turingpi.com/docs/raspberry-pi-cm4-install-os-using-turing-pi2-cluster

I will try the rpi4-io variant next.

xEBIx, 2023-09-09:
@RadxaYuntian 
I tried the rspi-cm4-io variant of the Bootloader image, the lower one on the Raspi-adpapter card and the LED on the cm3 are blinking with a rather slow frequency. But again, nothing on Serial consoles.

RadxaYuntian, 2023-09-11:
If it works it works : ) 

> Shall I use the rpi-cm4-io variant for the cm4 sodimm adapter card, or the sodimm variant? I supposed the cm4-io is for the raspberry cm4 Io board, the equivalent to the cm3-io board. Right now I use the sodimm variant.

I missed this one but the SODIMM variant is for a different product [CM3S](https://wiki.radxa.com/Rock3/CM). That system won't work correctly on regular CM3.

darioteixeira, 2023-09-30:
@RadxaYuntian Can you confirm if it's possible to boot a Rock CM3 without eMMC from an SD card on a board compatible with the Raspberry Pi CM4 IO board? Someone on the Radxa forums suggested that the eMMC-less Rock CM3 only works with Radxa's IO board -- is this true or not? Thanks in advance!

RadxaYuntian, 2023-10-09:
You need to flash SPI with our latest bootloader first, then it can boot from microSD without eMMC on a RPI CM4 IO board.

I'd suggest you send further support questions to our forum, and ping me over there if you got no response, to keep this issue on track.

darioteixeira, 2023-10-09:
Thank you for your reply @RadxaYuntian, and I will indeed ask further questions on Radxa's forum. In the meantime -- and for the sake of others looking for guidance in this thread -- I have to note that though you say that "You need to flash SPI with our latest bootloader first", the instructions on Radxa's wiki are useless. First because the description of the hardware doesn't even match the actual hardware (my Radxa CM3 doesn't have a "SPI Flash disable button"), and second because the instructions assume you have a Radxa CM3 IO board.

Someone in the Radxa forum mentioned that at least some versions of the Radxa CM3 are not actually compatible with RPi CM4 due to a hardware mistake. If this is the case, Radxa should just admit their mistake and be upfront regarding compatibility instead of obfuscating the situation. Otherwise they are just wasting everybody's time.

darioteixeira, 2023-10-09:
Regarding the issue of Radxa CM3 compatibility, the information I got from the Radxa forum is that there is indeed a hardware fuck-up that prevents the eMMC-less CM3 from running on any board that is not Radxa's own IO board.

For goodness sake, @RadxaYuntian, could you guys please update Radxa's website to convey this information? I have been wasting my time for months trying to get the damn thing to work. The feeling I get is that you're just obfuscating for fear of losing face!

RadxaYuntian, 2023-10-10:
I own to my mistakes all the time, but the only issue I'm aware of that blocks eMMC-less CM3 from booting on RPi CM4 IO is missing CD line, which can be workaround by aforementioned `flash SPI with our latest bootloader first`.

If you know something I don't, just ping me on the forum so I can have the context, and expose me as a big liar.

darioteixeira, 2023-10-10:
@RadxaYuntian You keep talking about "flash SPI", but the production versions of the Radxa CM3 -- the ones you are actually selling -- don't have this. Just take a look at the photo in this issue's first comment and compare it with the one on Radxa's website. Can you see the difference? This is what I mean by obfuscation and not owning up to your mistakes!

RadxaYuntian, 2023-10-10:
Please refer to the forum thread.

darkikim, 2024-12-19:
Hi, did anyone manage to run a CM3 without eMMC on the turing pi 2 ?

I can boot the CM3 without problem on the associated board (CM3io) but even after updating the SPI Bootloader via `rsetup` i can't manage to boot an sd card on the turing pi ...