geerlingguy/sbc-reviews #16: Raspberry Pi 3 model B+

geerlingguy, 2023-02-27:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.raspberrypi.com/products/raspberry-pi-3-model-b-plus/
  - Board purchased from: Micro Center
  - Board purchase date: May 28, 2018
  - Board specs (as tested): WiFi, 1GB RAM
  - Board price (as tested): $29.99

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

geerlingguy, 2023-02-28:
To get Geekbench to complete, I had to increase the swap space from 100M to 1024M (fyi).

ThomasKaiser, 2023-03-02:
> To get Geekbench to complete, I had to increase the swap space from 100M to 1024M

100M sounds like the standard emergency swap on SD card, right?

Which simply means that Geekbench numbers have to go into the trash and not into a review when the claim is benchmarking _the device_ and not just generating numbers w/o meaning?

BTW: with Geekbench 6 everything got worse (sample standard deviation seems to be higher than before and swapping occurs even on systems with 2 GB RAM now). Geekbench + SBC with less than 4GB RAM -> numbers w/o meaning since you're partially not 'benchmarking' the SBC in question but just the random I/O performance of the device the swap device sits on.

Geekbench 6 numbers on a device with 1GB RAM with swap on an ordinary SD card compared to swap on an A1 card compared to [ZRAM](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Review_of_RPi_Zero_2_W.md#sd-card-endurance) will vary greatly. Same with Geekbench 5.5 but to a lower extent.

ThomasKaiser, 2023-03-02:
BTW: `geekbench_armv7` (gone with GB6) might run w/o swapping...

geerlingguy, 2023-03-02:
> Which simply means that Geekbench numbers have to go into the trash and not into a review when the claim is benchmarking the device and not just generating numbers w/o meaning?

The numbers are not invalid—the single core benchmarks required no swap, and only one of the multi-core tests seemed to dip into swap. The top line numbers have some inconsistency but the fact the benchmark doesn't run under defaults is an interesting point to make. There are other programs I've tried running on Pi 3 B+ and CM3+ that I couldn't run stably due to lack of RAM, so it just illustrates the point.

But that's why I link to the result, too—so people can compare it to other Pi 3 B+ runs on the Geekbench site, and also see the individual test results.

ThomasKaiser, 2023-03-02:
> There are other programs I've tried running on Pi 3 B+ and CM3+ that I couldn't run stably due to lack of RAM, so it just illustrates the point.

That's mostly due to running a 64-bit userland on these thingies where everything needs almost twice as much RAM as such you run way more likely into swapping. :)

> so people can compare it to other Pi 3 B+ runs on the Geekbench site, and also see the individual test results.

People can't compare since you can't search for RPi 3B+ on the Geekbench site. All you can do is to search for either "Raspberry" or "BCM2385" (since for whatever stupid reason most RPi from 2B on report this wrong SoC in `/proc/cpuinfo` unless it's RPi 4 **w/o** `arm_64bit=1` since only then the correct SoC name is reported). Searching for "BCM2837B0" or "BCM2710" returns 0 hits while "BCM2711" at least returns RPi 4B running a 32-bit kernel. 

Then you need to crawl through endless pages and search for "1400" since that's the only easy way to spot RPi 3B**+**.

And then you end up at this for example: https://browser.geekbench.com/v5/cpu/compare/18249926?baseline=20776700

Same device, same software (64-bit kernel and 64-bit userland), maybe different `gpu_mem` settings since amount of memory differs but this other board which has more memory available than yours shows lower scores. Why?

And there are plently of 3B+ listings which multi-cores below 200 or even 150.

Geekbench doesn't record real clockspeeds (but on RPi 3B**+** the 1400 MHz will be turned silently into 1200 MHz with even moderate loads unless you set `temp_soft_limit=80` in `config.txt` and reboot which ofc nobody knows), Geekbench doesn't report throttling, it doesn't check/report frequency capping (the RPi being silently downclocked to 600 caused by undervolting) and it doesn't report swapping. It just collects numbers be it somewhat representing HW performance or being pure garbage and throws it into a database you can't even search in a reasonable way :)

BTW: that's what the `-r` and `-R` sbc-bench modes have been designed to deal with (the latter w/o running own benchmarks, just collecting some info and then switching into monitoring mode and reporting afterwards was has really happened while in another terminal this and that was tested to tell then if and why results are invalid. :)

platima, 2023-11-26:
Very confused that this says BCM2835 but aarch64... I thought it was ARMv6 (32-bit)?

geerlingguy, 2023-11-26:
@platima - The chip is 64-bit capable. It can also run 32-bit stuff too, it's one of the older designs that can do both :)

platima, 2024-01-05:
> To get Geekbench to complete, I had to increase the swap space from 100M to 1024M (fyi).

Yeah on a 2GB board I had to kill lightdm to get it to finish without oom reaping the geekbench process. I honestly think we need to run it with swap disabled though to ensure results are more consistent, which might mean sticking to v5?