geerlingguy/sbc-reviews #19: Ampere Altra Developer Platform

geerlingguy, 2023-04-17:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

(This thing is... kinda more than a 'board' — but I still want data somewhere, and this is as good a place as any!)

  - Board URL (official): https://amperecomputing.com/systems/altra/kraken-comhpc-WS
  - Board purchased from: (delivered by Ampere)
  - Board purchase date: April 3, 2023
  - Board specs (as tested): 96-core / 64 GB RAM / 128 GB SSD
  - Board price (as tested): Approx $5000

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

geerlingguy, 2023-04-17:
Also tested `pts/phpbench` via `phoronix-test-suite benchmark phpbench`: https://openbenchmarking.org/result/2304174-NE-AMPEREALT14 (498149)

geerlingguy, 2023-04-17:
Tried to install Windows 11 and it's been quite the saga. Haven't been able to get an ISO that boots yet:

  - Tried https://uupdump.net, multiple configurations and using canary, dev, and beta releases, best I could get is it booting to the windows logo with an Exception appearing on screen and locking up
  - Tried https://uup.rg-adguard.net, same thing
  - Tried downloading Windows Developer Kit 2023 recovery image (had to dig up my serial number), copying that on top of a Recovery Media usb stick, that didn't work (it wouldn't boot at all... let me select the drive in BIOS but would immediately kick back to BIOS)
  - Tried installing Parallels Desktop for Mac and having it install Windows 11 — it did, creating an ISO that looked like a similar layout to the ones from UUPDump, and it actually booted for like 3 seconds longer, and instead of an exception printed to screen, I got the sad face blue screen of death
  - I've also tried more incantations with UUPDump, and older versions, like the one linked from the [UTM Docs](https://docs.getutm.app/guides/windows/#downloads), but those didn't work either.

geerlingguy, 2023-04-17:
On my Mac, I downloaded a copy of the VMDK file kindly provided by [DB Tech](https://dbte.ch/). Then I installed `qemu` with `brew install qemu`, and ran:

```
qemu-img convert -O raw Windows11_InsiderPreview_Client_ARM64_en-us_25324.VHDX Windows11_InsiderPreview_Client_ARM64_en-us_25324.raw
```

Now I formatted my Corsair USB stick (128 GB) as ExFAT and am copying over the `.raw` file (it's like 70 GB!).

geerlingguy, 2023-04-17:
It sounds like according to some, scaling beyond 32 or 64 cores might break with Geekbench (possibly due to some internal configuration), so another way to game that system is to lock an instance of Geekbench 5 to each core individually, e.g.:

```
for ((i=0; i<60; ++i)); do numactl -C $i ./geekbench5 --cpu & >> "gkbch"$i 2>&1; done
```

The 30,000 score does seem low even accounting for overhead losses, compared to an 800 or so single core.

geerlingguy, 2023-04-17:
Drat, the Windows boot results in a sad face blue screen of death with [`ACPI_BIOS_ERROR`](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xa5--acpi-bios-error).

geerlingguy, 2023-04-18:
To get an Nvidia Quadro RTX 8000 working, I ran `sudo apt install libglvnd-dev pkg-config`, then installed Nvidia's latest aarch64 driver for the Quadro RTX 8000 (running the `.run` file with `sudo`).

I can get full resolution video output through DisplayPort on the card, and Ubuntu (and `neofetch`) sees the card, alongside `lshw`. But:

```
$ nvidia-smi
No devices were found

$ sudo dmesg | tail
[  276.229559] NVRM: GPU 000d:01:00.0: RmInitAdapter failed! (0x24:0x65:1423)
[  276.229665] NVRM: GPU 000d:01:00.0: rm_init_adapter failed, device minor number 0
[  280.597471] NVRM: GPU 000d:01:00.0: RmInitAdapter failed! (0x24:0x65:1423)
[  280.597541] NVRM: GPU 000d:01:00.0: rm_init_adapter failed, device minor number 0
[  466.107317] audit: type=1107 audit(1681788653.426:93): pid=1779 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.login1.Manager" member="SessionNew" name=":1.12" mask="receive" pid=7171 label="snap.firefox.firefox" peer_pid=1858 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  473.623874] NVRM: GPU 000d:01:00.0: RmInitAdapter failed! (0x24:0x65:1423)
[  473.623963] NVRM: GPU 000d:01:00.0: rm_init_adapter failed, device minor number 0
[  477.995897] NVRM: GPU 000d:01:00.0: RmInitAdapter failed! (0x24:0x65:1423)
[  477.995979] NVRM: GPU 000d:01:00.0: rm_init_adapter failed, device minor number 0
```

In the Nvidia dev forums, I found [this post](https://forums.developer.nvidia.com/t/nvidia-smi-outputs-no-devices-were-detected-in-ubuntu-20-04-rtx3060-driver-515/219908/2) which states:

> The nvidia driver doesn’t like your board.
> ...
> This is a known and recurring bug in the nvidia driver, the known working versions list is based on reports from other users.

So... looks like it's off to the races to find a working driver!

geerlingguy, 2023-04-18:
Er... apparently I still had the VGA plugged into my monitor, so the DisplayPort output wasn't actually working. D'oh! Well, we'll see what others say. I might have better luck compiling Linux 6.2 with AMD's latest driver support and my RX 6700 XT. We'll see!

istori1, 2023-04-24:
I have been enabling aarach64 support of NVIDIA on Flathub and some apps. When you get the NVIDIA GPU running there are couple of apps you can run.

https://github.com/flathub/org.freedesktop.Platform.GL.nvidia/pull/148
https://github.com/flathub/org.freedesktop.Sdk.Extension.llvm15/pull/10
https://github.com/flathub/org.kde.kdenlive/pull/207
https://github.com/flathub/org.jellyfin.JellyfinServer

https://github.com/LizardByte/Sunshine/releases/download/v0.19.1/sunshine_aarch64.flatpak

ThomasKaiser, 2023-04-25:
> The 30,000 score does seem low even accounting for overhead losses, compared to an 800 or so single core.

Here someone tests an 80 core Ampere Altra machine, obviously with more reliable power monitoring and generating higher GB5 scores than your machine (882/444285): https://youtu.be/m6-juFXR9c0?t=560

All those 80 core benchmarks show a higher single-threaded score than your machine but multi differs drastically (33000 - 46500): https://browser.geekbench.com/search?page=1&q=MP32-AR1-SW-HZ-001

Why do these scores differ? Obviously nobody knows and nobody cares (at least all scores exceeding 40000 have 256 GB RAM in common)

ThomasKaiser, 2023-04-25:
BTW: could you please share output from this command:

    cat /sys/devices/system/cpu/cpu*/topology/physical_package_id | sort | uniq -c

I'm curious since a dual socket Ampere Altra Max system showed different package IDs for `cpu0` - `cpu8` but then identical package ID for all the remaining 247 cores. Wondering whether here it's the same (affects `sbc-bench -j` mode that is designed to help explaining why Geekbench scores differ)

geerlingguy, 2023-04-25:
> Why do these scores differ? Obviously nobody knows and nobody cares (at least all scores exceeding 40000 have 256 GB RAM in common)

Note that I was told by the folks at Ampere I should fill all six RAM slots in the machine for better scores, as the NUMA layout on the Ampere requires all channels to be full for best performance. Whether that affects Geekbench scores in particular remains to be seen, but judging by STH's testing on Epyc and Ampere, it seems that's a reasonable outcome.

I'll soon be ordering two more sticks of RAM to match what's already installed. Right now I have the Ampere workstation back in a box because I'm trying to wrap up another project that's taking up the majority of the space in my little workshop :( — I hope to get it back out in mid May.

geerlingguy, 2023-04-25:
This chip is the M96-28, which is a 96-core part running at 2.8 GHz. I will try to get the machine up and running this week if I get a chance, and I'll give you more info. But might not be able to until may again.

geerlingguy, 2023-04-27:
It looks like their EDK2 fork [just got a fix](https://github.com/ADLINK/edk2-platforms/commit/34807c173aa8aa55fce694a9e9d83a4ae2d76a0f) for the Windows BSOD I was running into (apparently it should've already worked on the 32/64/80 core Altra parts, but it needed fixing on Altra Max (96/128 core parts). Waiting for a new version of the firmware to be available and I'll test it out again!

joespeed, 2023-04-28:
> (apparently it should've already worked on the 32/64/80 core Altra parts, but it needed fixing on Altra Max (96/128 core parts). Waiting for a new version of the firmware to be available and I'll test it out again!

that's what I get for thinking myself clever for having 96 core swapped into your PC🙄

Bohanke, 2023-04-28:
Just send you the new release for edk2 that will enable you to run Windows 11 on the new Ampere Altra Max generation

geerlingguy, 2023-04-29:
I have Windows 11 Pro for Arm Insider Preview running now. A few stats:

  - I had to upgrade to a prerelease version of Adlink's EDK II firmware (version 2.04.100.10), which was sent directly but I believe will be available widely soon (if not already?)
  - I was given an Arm driver for the ASPEED graphics controller onboard, and... it worked perfectly, giving me clean 1080p output over the built-in VGA (before the driver, I could only get 1024x768 output)
    - Would be _really_ cool for any other graphics vendor to support any driver under Windows on Arm
  - Built-in Ethernet doesn't work under Windows (no driver is present for it), so I had to plug in one of my USB-to-Gigabit adapters (I used [this TRENDNET adapter](https://amzn.to/3LE1AFT)). It was identified immediately and I can get full 1 Gbps speeds.
  - Cinebench R23 for X86 emulated in Windows 11: 452 single, 5647 multi (again... noting this score is running under x86 emulation!)
  - Geekbench 6 for Windows: Did not install correctly (binary never got installed), opened [this issue](http://support.primatelabs.com/discussions/geekbench/82502-geekbench-6-doesnt-install-correctly-under-windows-on-arm-on-ampere?unresolve=true)
  - Geekbench 5 for Windows: 744 single / 21389 multi ([result here](https://browser.geekbench.com/v5/cpu/21128438))
  - Crysis runs, but only at maybe 1-2 fps since there is no hardware acceleration whatsoever (that was on low settings)

geerlingguy, 2023-04-29:
Yes, it can play Crysis. Depending on your definition of the word 'play' ;)

[image]

farcop, 2023-05-14:
I missed in the article an attempt to install esxi-arm on this platform and, for example, deploying k8s. Seems like it must have.

rbapat-ampere, 2023-05-16:
A few tips and tricks to boost HPL performance.
1.  Change HPL.dat input values to P = 8  and Q = 12 respectively.

2.  Change HPL.dat Ns value to maximize the RAM utilization.

3. Use the Ampere-Oracle BLIS Math libraries. Instructions on how to install/use them are as follows.
- git clone https://github.com/flame/blis.git MyBlisDir
- change to  MyBlisDir
- #Switch to the new ampere branch 
- git checkout ampere
- ./QuickStart.sh altramax
- #Ensure that the test bench contains Ampere Oracle Blis exported to PATH and LD_LIBRARY_PATH appropriately.
- source ./blis_build_altramax.sh
- source blis_setenv.sh
- export LD_LIBRARY_PATH=<Install_Path>/MyBlisDir/lib/altramax

4. Use the attached Makefile to build the benchmark.
[Make.zip](https://github.com/geerlingguy/sbc-reviews/files/11487931/Make.zip)
Note : I had to zip the makefile since I could not attach it directly.

5. Edit lines 70, 95,96 & 97 to match your HPL and Ampere-Oracle BLIS installation paths

6. Build the binary : make arch=Altramax_oracleblis -j

7. Run benchmark as usual.

geerlingguy, 2023-06-11:
I'm re-testing my system a bit using the above guidelines from @rbapat-ampere — thanks! See https://github.com/geerlingguy/top500-benchmark/issues/10

rbapat-ampere, 2023-06-12:
> I'm re-testing my system a bit using the above guidelines from @rbapat-ampere — thanks! See [geerlingguy/top500-benchmark#10](https://github.com/geerlingguy/top500-benchmark/issues/10)

Thanks @geerlingguy . Also, if you'd like a a more detailed guide including performance expectations (ie scores that we are observing on our end) refer : https://github.com/AmpereComputing/HPL-on-Ampere-Altra

geerlingguy, 2023-06-12:
Interesting findings after upgrading to 96 GB of RAM (6 x 16 GB ECC DDR4 sticks)

  - Power consumption is about +30W across the board
  - Geekbench 5.4 multicore score goes from [30121](https://browser.geekbench.com/v5/cpu/compare/21070727) with 64G to [35818](https://browser.geekbench.com/v5/cpu/compare/21323770) with 96G
  - Linpack result goes from 400 Gflops at 200W with 64G to 600 Gflops at 235W with 96G ([source](https://github.com/geerlingguy/top500-benchmark/issues/10#issuecomment-1586355411))

geerlingguy, 2023-06-12:
The Geekbench 6.1.0 for Windows aarch64 preview seems to work, the result is submitted, but then the console window it's running in disappears without giving a claim key. But here is the result of one of my test runs: https://browser.geekbench.com/v6/cpu/1567571 (1008 single core / 10481 multi core).

Watching Windows' built-in CPU core monitor, it seems less than half of the available 96 cores are actually in use, so the multi core score is wildly inaccurate. See: http://support.primatelabs.com/discussions/geekbench/82502-geekbench-6-doesnt-install-correctly-under-windows-on-arm-on-ampere

geerlingguy, 2023-06-12:
As an additional data point, with 96 GB of RAM, the measly 5647 Cinebench R23 score under Windows 11 for Arm bumps to a slightly-less-measly 6858 - that's a 20% speedup! Just imagine how well Cinebench would run if it weren't only optimized for X86 :)

geerlingguy, 2023-06-12:
Following the recommended installation for NVIDIA drivers for Linux, I am now seeing output through my 3080 Ti, and can have some more fun.

`nvidia-smi` output:

```
jgeerling@ampere-altra:~$ nvidia-smi
Mon Jun 12 09:55:16 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.116.04   Driver Version: 525.116.04   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 0000000D:01:00.0  On |                  N/A |
| 48%   53C    P0   111W / 400W |    315MiB / 12288MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      3631      G   /usr/lib/xorg/Xorg                 96MiB |
|    0   N/A  N/A      3805      G   /usr/bin/gnome-shell              128MiB |
|    0   N/A  N/A      5079      G   ...7/usr/lib/firefox/firefox       87MiB |
+-----------------------------------------------------------------------------+
```

glmark2 score:

```
jgeerling@ampere-altra:~$ glmark2
=======================================================
    glmark2 2021.02
=======================================================
    OpenGL Information
    GL_VENDOR:     NVIDIA Corporation
    GL_RENDERER:   NVIDIA GeForce RTX 3080 Ti/PCIe
    GL_VERSION:    4.6.0 NVIDIA 525.116.04
=======================================================
[build] use-vbo=false: FPS: 6367 FrameTime: 0.157 ms
[build] use-vbo=true: FPS: 12194 FrameTime: 0.082 ms
[texture] texture-filter=nearest: FPS: 12209 FrameTime: 0.082 ms
[texture] texture-filter=linear: FPS: 12207 FrameTime: 0.082 ms
[texture] texture-filter=mipmap: FPS: 12224 FrameTime: 0.082 ms
[shading] shading=gouraud: FPS: 12028 FrameTime: 0.083 ms
[shading] shading=blinn-phong-inf: FPS: 11940 FrameTime: 0.084 ms
[shading] shading=phong: FPS: 11976 FrameTime: 0.084 ms
[shading] shading=cel: FPS: 11990 FrameTime: 0.083 ms
[bump] bump-render=high-poly: FPS: 11369 FrameTime: 0.088 ms
[bump] bump-render=normals: FPS: 12373 FrameTime: 0.081 ms
[bump] bump-render=height: FPS: 12112 FrameTime: 0.083 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 12015 FrameTime: 0.083 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 11416 FrameTime: 0.088 ms
[pulsar] light=false:quads=5:texture=false: FPS: 12023 FrameTime: 0.083 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 5273 FrameTime: 0.190 ms
[desktop] effect=shadow:windows=4: FPS: 6034 FrameTime: 0.166 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 1432 FrameTime: 0.698 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 1760 FrameTime: 0.568 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 1996 FrameTime: 0.501 ms
[ideas] speed=duration: FPS: 8453 FrameTime: 0.118 ms
[jellyfish] <default>: FPS: 11187 FrameTime: 0.089 ms
[terrain] <default>: FPS: 1677 FrameTime: 0.596 ms
[shadow] <default>: FPS: 10011 FrameTime: 0.100 ms
[refract] <default>: FPS: 7168 FrameTime: 0.140 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 12174 FrameTime: 0.082 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 12092 FrameTime: 0.083 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 12086 FrameTime: 0.083 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 12119 FrameTime: 0.083 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 12069 FrameTime: 0.083 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 12090 FrameTime: 0.083 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 12016 FrameTime: 0.083 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 11917 FrameTime: 0.084 ms
=======================================================
                                  glmark2 Score: 9878 
=======================================================
```

Super Tux Kart is giving 112 fps on max settings at 1440x900, and I'm going to test the Doom 3 install they have in their repo next.

geerlingguy, 2023-06-12:
The Doom 3 demo gave a constant 60fps, and I couldn't seem to get it unlocked. I was able to install `openarena` as well, and that gave 1000 fps continuous when I set max framerate to 0 but it started getting jittery (I don't know if I could go beyond that), and if I set 900 or below, it would just lock that frame rate all day.

I'd say the 3080 Ti is complete overkill for 10-20+ year old games :D

geerlingguy, 2023-06-12:
I was trying to get Stable Diffusion running, but ran into some issues with CUDA + NVIDIA driver versions. I might attempt again with a completely clean install, but NVIDIA's support makes it all quite 'fun', as finding compatible driver / CUDA version pairings can be a challenge, and it doesn't help that most of the stable diffusion guides want to you to install their own mix of packages which may or may not break your working CUDA install, lol.

geerlingguy, 2023-06-15:
Using the optimized blis library from Oracle, I was able to get 985 Gflops at 270W, for a new efficiency of 3.64 Gflops/W

ThomasKaiser, 2023-06-28:
> This chip is the M96-28, which is a 96-core part running at 2.8 GHz

Have you tried to verify the '2.8 GHz' (e.g. with `sbc-bench -j`)? 

The reason why I'm asking (link might not work with Safari for whatever reason) is Willy's cpufreq measurements here at the very bottom: [https://www.cnx-software.com/2023/04/18/adlink-ampere-altra-dev-kit-features-atx-motherboard-with-32-to-80-core-arm-com-hpc-cpu-module/#comment-608818](https://www.cnx-software.com/2023/04/18/adlink-ampere-altra-dev-kit-features-atx-motherboard-with-32-to-80-core-arm-com-hpc-cpu-module/#comment-608818)

geerlingguy, 2023-07-03:
New video today: https://www.youtube.com/watch?v=ydGdHjIncbk

And @ThomasKaiser - I'll try to get a run on it later today.

AlexLandherr, 2023-07-03:
Hi Jeff,

To run my benchmark `prime_threads` just follow the README. I've coded everything in C++ and it's standard library.
My dev system is an Ubuntu laptop using an i7-12700H. My precise OS version is `Ubuntu 22.04.2 LTS` 64-bit.

So far on ARM I've only tried on a RPi4B without issue. One thing that might be a problem is that one LD flag is `-lpthread`, though I could be wrong as I am only about 9 months into C++.

Repo:
[prime_threads](https://github.com/AlexLandherr/prime_benchmark)

geerlingguy, 2023-07-03:
@ThomasKaiser - Posting results here: https://github.com/ThomasKaiser/sbc-bench/issues/72

geerlingguy, 2023-07-03:
@AlexLandherr - I ran the benchmark and noticed only one core was getting any load:

```
jgeerling@adlink-ampere:~/prime_benchmark$ ./prime_benchmark
Start prime_benchmark between 1 and 100000000? [Y/n]: Y
Starting prime benchmark...
Iteration 1 of 10 Runtime (ns): 98088310584
Iteration 2 of 10 Runtime (ns): 98097435960
Iteration 3 of 10 Runtime (ns): 98094310171
Iteration 4 of 10 Runtime (ns): 98087703900
Iteration 5 of 10 Runtime (ns): 98089309445
...
```

I think it might be faster if it ran on all cores :D

AlexLandherr, 2023-07-03:
> @AlexLandherr - I ran the benchmark and noticed only one core was getting any load:
>
>
>
> ```
>
> jgeerling@adlink-ampere:~/prime_benchmark$ ./prime_benchmark
>
> Start prime_benchmark between 1 and [...]

Check the corrected link, I messed up the URLs (oops...)

ThomasKaiser, 2023-07-04:
Talking about reinventing the wheel...

> I ran the benchmark and noticed only one core was getting any load

Same with Bruce's primes benchmark he introduced [over there](https://github.com/geerlingguy/sbc-reviews/issues/10#issuecomment-1423536813) a while ago: https://hoult.org/primes.txt

> I think it might be faster if it ran on all cores

I wonder which use case / workload such a multi-threaded 'benchmark' could represent since the whole working set fits inside CPU caches but usually server machines like these Ampere boxes run tasks that depend (heavily) on RAM access...

geerlingguy, 2023-07-05:
@AlexLandherr - results:

```
jgeerling@adlink-ampere:~/prime_threads$ ./prime_threads
System supports 96 threads.
    ****    
Search limit for all options are (both are inclusive limits):
Lower search limit: 1
Upper search limit: 100000000

Select benchmark mode from list below:
0. Single-threaded benchmark.
1. Multi-threaded benchmark (96 threads).
2. Exit program.
Enter one of the listed values: 1

Numerator (also upper search limit): 100000000
Denominator (also how many threads): 96
Quotient: 1041666
Remainder: 64
  ****  
Starting multithreaded prime benchmark...

Started at: 2023-07-05 02:10:34 UTC
Iteration 1 of 10 Runtime (ns): 1650924933
Iteration 2 of 10 Runtime (ns): 1381674931
Iteration 3 of 10 Runtime (ns): 1381530659
Iteration 4 of 10 Runtime (ns): 1381874490
Iteration 5 of 10 Runtime (ns): 1382733573
Iteration 6 of 10 Runtime (ns): 1380218386
Iteration 7 of 10 Runtime (ns): 1392295779
Iteration 8 of 10 Runtime (ns): 1380862075
Iteration 9 of 10 Runtime (ns): 1381572084
Iteration 10 of 10 Runtime (ns): 1381265241
Prime multithreaded benchmark is done!

Stopped at: 2023-07-05 02:10:48 UTC

**** Results ****
Program mode is: Multithreaded
Thread count: 96
Search started at: 2023-07-05 02:10:34 UTC
Search ended at: 2023-07-05 02:10:48 UTC
Program ran for total of (DD:HH:MM:SS.SSSSSSSSS): 00:00:00:14.095112951

Program ran for: 14095112951 ns

Average time to find all primes between 1 and 100000000 was (DD:HH:MM:SS.SSSSSSSSS):
00:00:00:01.409495215

Average search time: 1409495215 ns
Number of primes found is: 5761455
```

geerlingguy, 2023-07-05:
@ThomasKaiser - I compiled Bruce's [primes.txt](https://hoult.org/primes.txt) and got:

```
jgeerling@adlink-ampere:~$ gcc -O3 primes.c -o primes
jgeerling@adlink-ampere:~$ ./primes
Starting run
3713160 primes found in 5641 ms
-480 bytes of code in countPrimes()
```

That seems slow to me, though—but the Ampere doesn't necessarily have the fastest single core performance, either...

ThomasKaiser, 2023-07-05:
>  I compiled @brucehoult's primes.txt ... That seems slow to me

Why? The Ampere's N1 cores are [A76 siblings](https://en.wikichip.org/wiki/arm_holdings/microarchitectures/neoverse_n1) (see also the [core IDs](https://github.com/ThomasKaiser/sbc-bench/blob/f2404ae6f21561eb604f21c25be30b2a78ce1342/sbc-bench.sh#L426-L427)) and for an A76 at 2.8 GHz with these cache sizes this looks reasonable.

And we shouldn't forget that the 2.8 GHz are still questionable, see [Willy's investigations](https://www.ipi.wiki/community/forum/topic/100157/cpu-frequency-capped-to-23-ghz-instead-of-26-ghz) _really measuring_ clockspeeds since unfortunately we can't trust the values the cpufreq driver spits out in general :(

AlexLandherr, 2023-07-05:
> @AlexLandherr - results:
>
> ```
> jgeerling@adlink-ampere:~/prime_threads$ ./prime_threads
> System supports 96 threads.
>     ****
> Search limit for all options are (both are inclusive limits):
> Lower [...]

Thanks very much Jeff! Here's my result for the max power settings on my i7-12700H:
```**** Results ****
Program mode is: Multithreaded
Thread count: 20
Search started at: 2023-07-05 08:33:13 UTC
Search ended at: 2023-07-05 08:34:11 UTC
Program ran for total of (DD:HH:MM:SS.SSSSSSSSS): 00:00:00:58.569630496

Program ran for: 58569630496 ns

Average time to find all primes between 1 and 100000000 was (DD:HH:MM:SS.SSSSSSSSS):
00:00:00:05.856952223

Average search time: 5856952223 ns
Number of primes found is: 5761455

**** Runtime for each iteration ****
Iteration 1 of 10 Runtime (ns): 5806562542
Iteration 2 of 10 Runtime (ns): 5679193186
Iteration 3 of 10 Runtime (ns): 5739044034
Iteration 4 of 10 Runtime (ns): 5868012868
Iteration 5 of 10 Runtime (ns): 5759130597
Iteration 6 of 10 Runtime (ns): 5850852134
Iteration 7 of 10 Runtime (ns): 6006729002
Iteration 8 of 10 Runtime (ns): 5994649762
Iteration 9 of 10 Runtime (ns): 5895252589
Iteration 10 of 10 Runtime (ns): 5970095523

Program started at: 2023-07-05 08:33:04 UTC
Program ended at: 2023-07-05 08:34:11 UTC
```

Is it okay with you for me to link to your results in my README?

geerlingguy, 2023-07-06:
> Is it okay with you for me to link to your results in my README?

Sure!

Bohanke, 2023-07-10:
Was a little bored in the weekend and since you started this on Windows 11, had to try it on Ubuntu with a graphics card installed. 

Most of the time it will not start up, when it does its a miracle, got it working with Nvidia RTX6000 card. 
Online instructions are terrible had to combine several directions to even get box86/64 correctly compiled and installed
Borrowed some instructions from RP4, but build flags are a little different
This is the steam.deb from official homepage

We did not run any games yet because when it aborts it takes many starts to get it back up. We will spend some time to document it. 
[image]

A lot of complaints though asking for i386 libraries that are of course not available for arm (armhf)

[image]
[image]

geerlingguy, 2023-07-10:
@Bohanke - I was having similar issues with the i386 packages.

Bohanke, 2023-07-15:
Rock solid 

Start with fulll Ubuntu Server 20.04.5 install 
 

$ sudo apt update
$ sudo apt upgrade

$ sudo apt install tasksel
$ sudo tasksel install ubuntu-desktop
$ sudo systemctl set-default graphical.target
$ sudo reboot

$ sudo nano /etc/default/grub
Add “pcie_aspm=off” into the line of GRUB_CMDLINE_LINUX=”” and save the change.
Execute the command to update the grub and then reboot:
$ sudo update-grub

$ sudo touch /etc/modprobe.d/blacklist-nouveau.conf
$ sudo nano /etc/modprobe.d/blacklist-nouveau.conf

add : 

blacklist nouveau
options nouveau modeset=0

save file

$ sudo update-initramfs -u

W: Possible missing firmware /lib/firmware/ast_dp501_fw.bin for module ast
ignore it !

$ sudo reboot

install some tools for Nvidia graphics and maybe CUUDA later ??
$ sudo apt-get install g++ freeglut3-dev build-essential libx11-dev libxmu-dev libxi-dev libglu1-mesa-dev libfreeimage-dev libglfw3-dev

Goto Nvidia homepage to download the install package for your particular card !!!
In this example the card is RTX6000

Copy the file into home directory, make sure its executable 

run the installer 
$ sudo ./NVIDIA-Linux-aarch64-535.54.03.run

(do not do this in SSH shell, it has to be done local !!!!)

$ sudo reboot

You now have Ubuntu desktop with Nvidia drivers installed and booting up the Nvidia display
============================================================================================

BOX86 + SteAM and BOX64

sudo dpkg --add-architecture armhf

==== Do not compile right now because standard GCC and gcc-arm-linux-gnueabihf need for box86 are not clear to me !!!

Use @Itai-Nelken's apt repository to install precompiled box86 debs, updated weekly.

sudo wget https://itai-nelken.github.io/weekly-box86-debs/debian/box86.list -O /etc/apt/sources.list.d/box86.list
wget -qO- https://itai-nelken.github.io/weekly-box86-debs/debian/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box86-debs-archive-keyring.gpg
sudo apt update && sudo apt install box86:armhf -y

Still use the script file to install steam but we need to clone git BOX86 sources for that 	

$ git clone https://github.com/ptitSeb/box86

$ cd box86

$ ./install_steam.sh

 
$ sudo wget https://ryanfortner.github.io/box64-debs/box64.list -O /etc/apt/sources.list.d/box64.list
$ wget -qO- https://ryanfortner.github.io/box64-debs/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box64-debs-archive-keyring.gpg
$ sudo apt update && sudo apt install box64-generic-arm -y

maybe needed ??
$ sudo systemctl restart systemd-binfmt

$ sudo reboot

Bohanke, 2023-07-15:
We, me and my son :)

made a target list for games to try out first. 
they are listed as Platinum where is comes to compatibility

Here is a lot of compatibility info :  https://www.protondb.com/

Platinum most compatible and most popular 
=============================================
Castle Crashers		    	255MB !!
Human: Fall Flat	    	        500MB !!
Untitled Goose Game	    	800MB
Amazing Cultivation Sim.    	1GB			xxx
MudRunner		    	1GB
Cuphead		            	2GB
Inside			    	3GB   
Deep Rock Galactic          	3GB			xxx
Pit People		    	3GB
Farming Simulator 15	    	3GB			xxx
Risk of Rain 2		    	4GB
Grim Dawn	            	5GB			xxx
What Remains of Edith Finch 	5GB
Alien Swarm: Reactive Drop  	6GB
ICEY			    	6GB
Rocket League	            	7GB
Outer Wilds		    	8GB			xxx
Mordheim: City of the Damned	8GB
Dishonored		    	9GB

Little Nightmares	   			10GB
Little Nightmares II	   			10GB
DRAGON BALL XENOVERSE	   			10GB 	xxx
Bloodstained: Ritual of the Night	  	10GB	xxx
Ori and the Blind Forest: Def Edition	  	11GB
STEINS;GATE				  	13GB	xxx
Satisfactory		    		  	15GB	xxx
Atelier Ryza: Ever Darkness & Hideout	  	15GB    xxx
AI＊Shoujo/AI＊少女			  	15GB    xxx
A Plague Tale: Innocence    		  	20GB    xxx
鬼谷八荒 Tale of Immortal    		  	25GB    xxx
Sekiro: Shadows Die Twice - GOTY Edition  	25GB
Ryse: Son of Rome			  	26GB
Control Ultimate Edition		  	26GB
Aliens: Fireteam Elite			  	30GB
Wolfenstein II: The New Colossus	  	42GB
Wolfenstein: The New Order		  	44GB
The Witcher 3: Wild Hunt    		  	50GB
NieR:Automata		    		  	50GB
ACE COMBAT 7: SKIES UNKNOWN		  	50GB
SCARLET NEXUS				  	50GB
Doom			       		  	55GB

xxx means maybe small issues

geerlingguy, 2023-08-06:
Still trying to replicate the 1.2+ Tflop results with the 96-core CPU on my Dev Workstation, but still only getting just under 1 Tflop: https://github.com/AmpereComputing/HPL-on-Ampere-Altra/issues/10

ThomasKaiser, 2023-08-07:
> still only getting just under 1 Tflop

Have you tried to monitor benchmark execution this time? So do you really know your CPU cores clock at a certain clockspeed and neither throttling nor swapping occured? Linpack FAQ clearly states:

> If the problem size you pick is too large, swapping will occur, and the performance will drop

`sbc-bench -R` is _designed_ for just this: checking whether the execution of any other benchmarks does what it should. Executed in another terminal window prior to benchmarking and being stopped by `[ctrl]-[c]` afterwards. Real clockspeeds (measured) will be reported prior/after benchmarking and whether throttling and/or swapping occured.

geerlingguy, 2023-09-11:
@ThomasKaiser - Yes, we've been monitoring that, at this point I went to the extent of buying new Samsung RAM to match the sticks in Ampere's test system, and we've seen that the Samsung x4 RAM sticks are producing 20-30% faster RAM benchmarks than the Transcend x8 sticks (specs otherwise identical, DDR4-3200 ECC CL22).

I just re-tested in tinymembench and the Samsung RAM is getting:

```
 standard memcpy                                      :  12107.0 MB/s
 standard memset                                      :  44746.4 MB/s
```

Whereas the Transcend RAM was getting:

```
 standard memcpy                                      :   9894.0 MB/s
 standard memset                                      :  44745.2 MB/s
```

For more details, see https://github.com/AmpereComputing/HPL-on-Ampere-Altra/issues/10#issuecomment-1711717393

Re-running Top500 HPL benchmark now.

geerlingguy, 2023-09-11:
New result is 1.188 Tflops at 296W, see: https://github.com/geerlingguy/top500-benchmark/issues/10

geerlingguy, 2023-09-15:
I've also just upgraded the system to some new Samsung 64GB RAM sticks (for a total of 384 GB of RAM, sheesh), and here's the tinymembench result:

<details>
<summary>Click to show tinymembench results</summary>

```
TODO
```
</details>

zjlywjh001, 2023-10-30:
Hi, I'm currently test my AADK with Q80-30, but strangely my score is very low: https://browser.geekbench.com/v5/cpu/21893639.
I use 4-channel 32GB 3200 DRAM, total 128GB. But seen a very huge differs from others'. Does anybody know why?

ThomasKaiser, 2023-10-30:
> I'm currently test my AADK with Q80-30, but strangely my score is very low

Your CPU cores only clock at 2.3 GHz, see the `"processor_frequency"` node: https://browser.geekbench.com/v5/cpu/21893639.gb5 (you need an user account to see the `.gb5` and `.gb6` raw data files)

While Jeff's testing with his 2.8 GHz SKU the cores really clocked with those 2.8 GHz: https://browser.geekbench.com/v5/cpu/21323770.gb5

I would always recommend running Geekbench from inside `sbc-bench -G` since my tool not only measures clockspeeds but also monitors for e.g. swap (which for obvious reasons can also ruin benchmark scores).

zjlywjh001, 2023-10-31:
> Your CPU cores only clock at 2.3 GHz, see the `"processor_frequency"` node: https://browser.geekbench.com/v5/cpu/21893639.gb5 (you need an user account to see the `.gb5` and `.gb6` raw data files) [...]

I still don't know why my CPU only runs at  2.3GHz frequency. In system it reports `cpuinfo_max_freq` is 3000000 and even if the `scaling_cur_freq` is set to 3000000, the cpuinfo_cur_freq still only 2300000 .