geerlingguy/sbc-reviews #21: Raspberry Pi 5 model B

geerlingguy, 2023-09-28:
[image]

> **NOTE**: The board used for this testing was a pre-release revision. The final production Pi 5 model B and any future revisions may have slight differences, especially as the Raspberry Pi OS image is updated.

## Basic information

  - Board URL (official): https://www.raspberrypi.com/products/raspberry-pi-5-model-b/
  - Board purchased from: (pre-production version delivered to Alpha testing team)
  - Board purchase date: N/A
  - Board specs (as tested): 8GB
  - Board price (as tested): $80

## Linux/system information

```
# output of `neofetch`
       _,met$$$$$gg.          pi@pi5 
    ,g$$$$$$$$$$$$$$$P.       ------ 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 12 (bookworm) aarch64 
 ,$$P'              `$$$.     Host: Raspberry Pi 5 Model B Rev 1.0 
',$$P       ,ggs.     `$$b:   Kernel: 6.1.32-v8+ 
`d$$'     ,$P"'   .    $$$    Uptime: 4 mins 
 $$P      d$'     ,    $$P    Packages: 1536 (dpkg) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.15 
 $$;      Y$b._   _,d$P'      Resolution: 1920x1080 
 Y$$.    `.`"Y$$$$P"'         Terminal: /dev/pts/0 
 `$$b      "-.__              CPU: BCM2835 (4) @ 2.400GHz 
  `Y$$                        Memory: 416MiB / 7942MiB 
   `Y$$.
     `$$b.                                            
       `Y$$b.                                         
          `"Y$b._
              `"""

# output of `uname -a`
Linux pi5 6.1.32-v8+ #1 SMP PREEMPT Sat Aug  5 07:03:33 BST 2023 aarch64 GNU/Linux
```

## Benchmark results

### CPU

  - Geekbench 6: (748 single / 1507 multi - https://browser.geekbench.com/v6/cpu/2808487)
  - 30.249 Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): 1.8 W
  - Maximum simulated power draw (`stress-ng --matrix 0`): 9.7 W
  - During Geekbench multicore benchmark: 7.9 W
  - During `top500` HPL benchmark: 11 W (2.75 Gflops/W)

### Disk

#### SanDisk Extreme 128GB microSD

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | 91.5 MB/s |
| iozone 1M random read | 90.09 MB/s |
| iozone 1M random write | 62.28 MB/s |
| iozone 4K random read | 9.86 MB/s |
| iozone 4K random write | 4.07 MB/s |

#### Kioxia XG8 1TB NVMe at PCIe Gen 2

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | 454 MB/s |
| iozone 1M random read | 425.57 MB/s |
[... 204 lines cut ...]
pi@pi5:~ $ dmesg | grep ptp
[    4.692928] pps pps0: new PPS source ptp0
[    4.692979] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
pi@pi5:~ $ ls /dev/ptp*
/dev/ptp0
pi@pi5:~ $ cat /sys/class/ptp/ptp0/clock_name 
gem-ptp-timer
pi@pi5:~ $ ethtool -T eth0
Time stamping parameters for eth0:
Capabilities:
  hardware-transmit
  software-transmit
  hardware-receive
  software-receive
  software-system-clock
  hardware-raw-clock
PTP Hardware Clock: 0
Hardware Transmit Timestamp Modes:
  off
  on
  onestep-sync
Hardware Receive Filter Modes:
  none
  all
```

geerlingguy, 2023-09-28:
Note: Most benchmarks were run with 16k page size; Geekbench 6 would not work with 16k so I rebooted on 4k page size kernel8.

geerlingguy, 2023-09-28:
sbc-bench results: https://github.com/ThomasKaiser/sbc-bench/issues/77

ThomasKaiser, 2023-09-28:
Can you please post output from `lsusb` and `lspci`?

geerlingguy, 2023-09-28:
`lsusb`:

```
pi@pi5:~ $ sudo lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@pi5:~ $ sudo lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
```

`lspci`:

```
pi@pi5:~ $ sudo lspci -vvvv
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 40
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x4, ASPM L1, Exit Latency L1 <2us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x4
			TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible-
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd+
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS+
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported, DRS-
			 DownstreamComp: Link Up - Present
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
[... 55 lines cut ...]
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable+ Count=61 Masked-
		Vector table: BAR=0 offset=00000000
		PBA: BAR=0 offset=00002000
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: rp1
```

geerlingguy, 2023-09-28:
Note that the above `lspci` was run when I had nothing attached to the 'ext' port—it's currently booting off a microSD card in the official case, and working with the external connector with the Pi inside the case is a bit tricky :)

rgov, 2023-09-28:
The new UART header is presumably to make it easier to attach the Pi Debug Probe. Do we get GPIO 14/15 as a second UART, then?

ThomasKaiser, 2023-09-28:
Thank you. I'm just trying to figure out how the whole PCIe setup looks like...

Does the 'external' PCIe lane originate from the SoC or from RP1? When you force PCIe to Gen3 does this affect only the external 'slot' or also interconnection with RP1? Which USB controller is inside the thing?

The NIC has the RPi vendor ID (`1de4:0001`) but what about USB? Maybe some USB peripherals need to be connected for the controller showing up on the PCIe bus?

will127534, 2023-09-28:
They got their own PCIe Vender ID! : D
Also interesting to see that most interface is packed in a single device and named as ethernet device. I wonder how is the CSI camera sending back the data through it.

geerlingguy, 2023-09-28:
> The new UART header is presumably to make it easier to attach the Pi Debug Probe. Do we get GPIO 14/15 as a second UART, then? Can we use the UART header for something other than the serial console?

@rgov you still get the separate UART via RP1 on GPIO 14/15, that one is configured as always via config.txt — the UART header is direct into the BCM2711 SoC.

> Does the 'external' PCIe lane originate from the SoC or from RP1? When you force PCIe to Gen3 does this affect only the external 'slot' or also interconnection with RP1? Which USB controller is inside the thing?

@ThomasKaiser ext PCIe from SoC (so when you set `dtparam=pciex1_gen=3` that only affects the external PCIe lane, not RP1).

> The NIC has the RPi vendor ID (1de4:0001) but what about USB? Maybe some USB peripherals need to be connected for the controller showing up on the PCIe bus?

With a 1TB SSD (SanDisk Extreme) attached:

```
pi@pi5:~ $ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0781:5588 SanDisk Corp. Extreme Pro
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@pi5:~ $ sudo lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
pi@pi5:~ $ lspci
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

Not sure what magic they're pulling, it all seems to route through RP1 on those x4 lanes.

DamonHD, 2023-09-28:
Thanks for the idle power value: important for running off-grid, though still higher than my current 3B at about 1W...

https://www.earth.org.uk/note-on-Raspberry-Pi-3-setup.html

cleverca22, 2023-09-28:
https://gist.github.com/cleverca22/97e46998b5fbd4e2f6772127c0cd034a
device tree in here, where you can see all of the fun bits in the rp1, including an adc!

ThomasKaiser, 2023-09-28:
> https://gist.github.com/cleverca22/97e46998b5fbd4e2f6772127c0cd034a

Thanks for this. So in the RP1 Ethernet is Cadence MACB/GEM IP and USB the usual Synopsys DesignWare 3 stuff while the SoC itself also seems to contain a RGMII capable GMAC (if RPi Trading Ltd. comes up with a new CM5 pinout they could maybe use this other Gigabit Ethernet as well)

cleverca22, 2023-09-28:
ah, good catch, i missed that

lukts30, 2023-09-28:
Looks like it is still limited to **USB OTG 2.0** like the Pi4 [\[1\]](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/#comment-1594029).
Rock 3/4 have USB OTG 3.0 but Rock 5 only has USB OTG 2.0 like all the raspberrypi's. 

I need a SBC with USB OTG 3.0 for a fast [piSCSI USB Drive](https://matt.olan.me/post/making-a-piscsi-usb-drive-part-2/).

ThomasKaiser, 2023-09-28:
@geerlingguy have you tried to power the RPi 5 with something else than their new 5V/5A power brick?

Curious whether it supports higher profiles with supply voltages exceeding 5V or whether this limitation is a hard one? Also curious whether `debugfs` is enabled and something shows up below `/sys/kernel/debug/usb/` that hints on which USB PD chip they use. On Rock 5B for example you get detailed negotiation status/process when doing this:

    cat /sys/kernel/debug/usb/fusb302-4-0022 /sys/kernel/debug/usb/tcpm-4-0022 | sort

Also curious whether USB-C details show up when doing something like `grep "" /sys/class/typec/port0/* 2>/dev/null`?

And in general providing `dmesg` output would be great since answering a bunch of questions without us annoying you :)

hexwab, 2023-09-28:
> See video: [Rasbperry Pi 5: EVERYTHING you need to know](https://www.youtube.com/watch?v=nBtOEmUqASQ)

Graph at 9:25 looks suspect: you appear to be using a cumulative plot. So for example the blue "108.9" point is actually at y position 108.9+54.9=163.8. Screenshot: https://sphere.chronosempire.org.uk/~HEx/shots/latency.png

Here's what I'd expect it to look like: https://sphere.chronosempire.org.uk/~HEx/pi5.svg

Code:
```#!/usr/bin/python3
import matplotlib.pyplot as plt
import numpy as np
from matplotlib import ticker

size = range(0,17)
boards = {
    'Pi 4': [0.0,0.0,0.0,0.0,0.0,0.0,4.7,7.2,10.3,11.9,22.7,80.9,108.9,129.4,139.8,145.1,156.5],
    'Pi 5': [0.0,0.0,0.0,0.0,0.0,0.0,0.0,1.1,1.6,1.9,8.3,12.2,54.9,86.8,103.0,112.9,118.7]
}

plt.rcParams["figure.figsize"] = (12,8)
fig, ax = plt.subplots()
ax.xaxis.set_major_formatter(ticker.FuncFormatter(lambda n,pos: "%d"%(2**n)))
ax.grid()
for q in boards.keys():
    ax.plot(size, boards[q], marker='.', markeredgecolor='black',label=q)
    ax.fill_between(size, boards[q])
    for xy in zip(size,boards[q]):
  	    ax.annotate('%.1f' % xy[1], xy=xy, textcoords='data')
ax.legend(loc='upper left')
ax.set_title('Memory latency, single random read')
ax.set_xlabel('Size (KiB)')
ax.set_ylabel('Latency (ns)')
plt.savefig("pi5.svg")```

jack-ma, 2023-09-28:
> Looks like it is still limited to **USB OTG 2.0** like the Pi4 [[1]](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/#comment-1594029). Rock 3/4 have USB OTG 3.0 but Rock 5 only has USB OTG 2.0 like all the raspberrypi's.
> 

@lukts30 

ROCK 5A and ROCK 5B both has USB OTG 3.0, ROCK 5B OTG is at USB C. ROCK 5A OTG is at the upper double layer USB 3.0. Please check the Tech Spec section: https://radxa.com/rock5a/

AndrewRentschler, 2023-09-28:
Now lets just hope there is plenty of supply!

geerlingguy, 2023-09-28:
> Graph at 9:25 looks suspect: you appear to be using a cumulative plot. So for example the blue "108.9" point is actually at y position 108.9+54.9=163.8.

@hexwab - Oh my! You're correct. Chalk that up to 'was up too late benchmarking' :O

geerlingguy, 2023-09-28:
Full `dmesg` output is below the fold here:

<details>
<summary>Click to show full `dmesg` output</summary>

```
pi@pi5:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.47-v8_16k+ (pi@pi5) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Thu Sep 21 21:07:03 CDT 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001a000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s176872 r8192 d44312 u229376
[    0.000000] pcpu-alloc: s176872 r8192 d44312 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521984
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=B8:27:EB:AE:74:3B vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=22c2ce57-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 9, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbffc000-0x00000000ffffc000] (64MB)
[    0.000000] Memory: 7915024K/8384512K available (12288K kernel code, 2156K rwdata, 4160K rodata, 4416K init, 1211K bss, 141808K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39682 entries in 39 pages
[    0.000000] ftrace: allocated 39 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000149] Console: colour dummy device 80x25
[... 480 lines cut ...]
[    4.340161] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    4.340166] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    4.340480] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    4.631212] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.631221] Bluetooth: BNEP filters: protocol multicast
[    4.631230] Bluetooth: BNEP socket layer initialized
[    4.637365] Bluetooth: MGMT ver 1.22
[    4.643890] NET: Registered PF_ALG protocol family
[    5.147720] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    5.147729] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[... cut ...]
[    9.247310] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
[    9.247348] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   10.406380] EXT4-fs (sda1): mounted filesystem with ordered data mode. Quota mode: none.
[   11.049112] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   14.132840] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   14.139623] Bridge firewalling registered
[   14.290571] Initializing XFRM netlink socket
```
</details>

FCLC, 2023-09-28:
Following up on https://mast.hpc.social/deck/@geerlingguy@mastodon.social/111143860188995259

2 questions: 

CPU: What ISA extensions are supported by the chip? the A76 should be an Aarch64 chip that implements baseline Arm v8.2 with select extensions such as the Dot product extensions from V8.4 the speculative load/store memory extensions from V8.5 amongst others (lscpu -e should do it? at worse `cat /proc/cpuinfo >> ~/cpuinfo_dump.txt`)

GPU/Multimedia ASIC: What formats/profile does the HEVC decoder support? is it 8bit 420 only? does it implement 10bit (required for "proper" HDR)? 
Easiest way to check with be vainfo, but that assumes that VAAPI is implemented and that it's installed

Headcrabed, 2023-09-28:
Hello, can you test glmark2 and glmark2-es2 score and do a simple comparison between rpi4b?

hexwab, 2023-09-28:
> @hexwab - Oh my! You're correct. Chalk that up to 'was up too late benchmarking' :O
> 
> I'm updating the graph and wording in my blog post.

Cheers. While I'm nitpicking: have you considered SVG graphs (rather than JPEG of all things)?

ThomasKaiser, 2023-09-28:
> What ISA extensions are supported by the chip?

Either grep for "Flags" [here](http://ix.io/4HDw) or see [there](https://github.com/ThomasKaiser/sbc-bench/blob/df866063a93ca5246cea148373d2fd49a35fd687/sbc-bench.sh#L6344): `fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp`

lukasMega, 2023-09-28:
Jeff, do you plan to focus more on "tweaking" in order to achieve the lowest possible idle power consumption?
BTW, thanks for your great report and video about RPi 5!

DamonHD, 2023-09-28:
I would certainly like to know that set of tweaks for my application...  B^>

FCLC, 2023-09-28:
> Either grep for "Flags" [here](http://ix.io/4HDw) or see [there](https://github.com/ThomasKaiser/sbc-bench/blob/df866063a93ca5246cea148373d2fd49a35fd687/sbc-bench.sh#L6344): `fp asimd evtstrm aes [...]

Gotcha, so looks like we get
neon (ASIMD),
FP16 (FPHP + asimdhp or "Advanced SIMD half precision), 
FP64 FMA( asimdrdm or Rounding Double Multiply Accumulate/Subtract)
SIMD dot products (asimddp)

ThomasKaiser, 2023-09-28:
> device tree in here, where you can see all of the fun bits in the rp1, including an adc!

BTW: [sbc-bench output](http://ix.io/4HDw) collected by Jeff shows `lm-sensors` output:

    rp1_adc-isa-0000
    in1:           1.47 V  
    in2:           2.53 V  
    in3:           1.41 V  
    in4:           1.38 V  
    temp1:        +55.5 C

Also accessible via `sysfs`. There it's `hwmon1`:

    /sys/class/hwmon/hwmon1/in1_input:1475
    /sys/class/hwmon/hwmon1/in1_raw:1824
    /sys/class/hwmon/hwmon1/in2_input:2539
    /sys/class/hwmon/hwmon1/in2_raw:3149
    /sys/class/hwmon/hwmon1/in3_input:1413
    /sys/class/hwmon/hwmon1/in3_raw:1544
    /sys/class/hwmon/hwmon1/in4_input:1385
    /sys/class/hwmon/hwmon1/in4_raw:1522
    /sys/class/hwmon/hwmon1/name:rp1_adc
    /sys/class/hwmon/hwmon1/temp1_input:51985
    /sys/class/hwmon/hwmon1/temp1_raw:823
    /sys/class/hwmon/hwmon1/uevent:OF_NAME=adc
    /sys/class/hwmon/hwmon1/uevent:OF_FULLNAME=/axi/pcie@120000/rp1/adc@c8000
    /sys/class/hwmon/hwmon1/uevent:OF_COMPATIBLE_0=raspberrypi,rp1-adc
    /sys/class/hwmon/hwmon1/uevent:OF_COMPATIBLE_N=1

geerlingguy, 2023-09-28:
> Hello, can you test glmark2 and glmark2-es2 score and do a simple comparison between rpi4b?

@Headcrabed - mentioned in my video (linked earlier), GLMark2 scores:

  - 117 fullscreen
  - 905 windowed

theofficialgman, 2023-09-28:
hey @geerlingguy . First off great video.
I have tried leaving a youtube comment though all my comments get deleted (this has been an ongoing thing for years now, gotta love youtube/google).

I am a developer/maintainer at pi-apps https://github.com/Botspot/pi-apps and just want to say that your experience with installing Steam is not expected or shown to occur by any of our own testing on Bookworm on pi4 (Bookworm beta has been available through invite only on the rpi forum for a month or two now).

If you still have that install I would be interested in seeing the Steam install logs from `~/pi-apps/logs/`

My initial guess as to what caused the issue would be that the rpi foundation gave you an image to test and did not update the apt repository with new packages until later (likely today as I saw many updates). Installing the armhf mesa (and dependencies) which the box86/steam scripts do could have inadvertently removed critical arm64 OS packages if the repository had only older versions of armhf packages compared to the locally installed arm64 packages.

theofficialgman, 2023-09-28:
> Note: Most benchmarks were run with 16k page size; Geekbench 6 would not work with 16k so I rebooted on 4k page size kernel8.

@geerlingguy wait wait wait... this is news to me. On the pi5 they intend on using the 16K pagesize kernel by default? Why? This is actually REALLY important information. Please send back the output of
```bash
getconf PAGE_SIZE
```
in each of the modes you are testing

CC: @botspot

ThomasKaiser, 2023-09-28:
> Geekbench 6: (748 single / 1507 multi - https://browser.geekbench.com/v6/cpu/2808487)

For anyone interested in how important memory access is, this is the comparison with another 5B (from Radxa) testing only the four A76 cores of Rockchip RK3588: https://browser.geekbench.com/v6/cpu/compare/2808487?baseline=335677 (`sbc-bench` in Geekbench-Mode by default also tests each CPU cluster individually)

So this is two times four A76 cores (RPi 5B running at 2.4 GHz, Rock 5B at only ~2.3 GHz) in comparison:

[image]

(full results output: [http://ix.io/4pnb](http://ix.io/4pnb))

wormley, 2023-09-28:
Do we know if it supports non-PD USB-C at full power? I have a bunch of Pi 4b just hanging off a 5.1V Meanwell with USB-C cables and no negotiation.

dwillmore, 2023-09-29:
> Do we know if it supports non-PD USB-C at full power? I have a bunch of Pi 4b just hanging off a 5.1V Meanwell with USB-C cables and no negotiation.

I, too, am interested in the differences in performance between a 5V/3A and the Trading Companies 5.1V/5A supply.  Supposedly the board does something different when it recognizes the supply, but it's unclear to me is that is just in passing through more power to the USB ports or if the CPU will use more of it--also depending on cooling capacity, one assumes.

geerlingguy, 2023-09-29:
@dwillmore - There's a flag you can set to override the functionality, which already exists for the Pi 4, `usb_max_current_enable` — setting this to `usb_max_current_enable=1` in `/boot/config.txt` will remove the current limiting for a PSU that negotiates less than 5A, or for powering the Pi via GPIO pins.

There may also be a mechanism for throttling the CPU (either automatically or via some other flag) while preserving the USB max current (1.6A), otherwise it's best to find a suitable 5V/5A PSU.

Also, regarding **overclocking**, I haven't done much testing yet, but did push `arm_freq=2600` and even `2800`; 2600 was stable though temps were up about 3-5°C. 2800 was slightly unstable under load, but the system never crashed.

Instead of `over_voltage`, Pi engineers are recommending using `over_voltage_delta` (e.g. `over_voltage_delta=50000` for an increase of 50 mV), and have explained the reasoning thusly:

> `over_voltage_delta` adds the offset AFTER [DVFS](https://semiengineering.com/knowledge_centers/low-power/techniques/dynamic-voltage-and-frequency-scaling/) has run whereas `over_voltage` essentially defeats the voltage scaling.

geerlingguy, 2023-09-29:
> Edit 2: Since Jeff right now seems to be the only one who ran GB 6 let's look at [the few GB5 scores](https://browser.geekbench.com/search?k=v5_cpu&q=Raspberry+Pi+5+Model+B+Rev+1.0). Multi-threaded to single-threaded ratio not as bad as with Jeff's GB6 result but still hinting at something seriously wrong since MT scores are just 2.5 higher than ST.

@ThomasKaiser - A couple notes: Geekbench 6.2 doesn't run with 16K page size, so I had to switch to the `kernel8` 4K page size to get it to run; I have opened an issue with Primate Labs for it: [Can't run Geekbench on systems with 16k page size](http://support.primatelabs.com/discussions/geekbench/83404-cant-run-geekbench-on-systems-with-16k-page-size).

Geekbench 6 in general is a strange beast on Arm (thus it's still in Preview officially ;). I think there are still a few bugs to be ironed out, though the single core Geekbench 6 numbers between my Pi 4 and Rock 5 B are identical: https://browser.geekbench.com/user/446940

geerlingguy, 2023-09-29:
@theofficialgman - Yes, currently they are trialling 16k page size in the Bookworm release for Pi 5, but switching is as simple as:

```
[pi5]
kernel=kernel8.img
```

The alpha testers have been tracking a few issues with software running 32-bit binaries that have issues... the default (between 4k and 16k) isn't set in stone yet, so now would be a good time to run it up the flagpole if there are more implications :)

StuartIanNaylor, 2023-09-29:
> > Geekbench 6: (748 single / 1507 multi - https://browser.geekbench.com/v6/cpu/2808487)
> 
> Pretty obvious that there's something seriously wrong if a quad-core CPU scores only twice as much 'multi-threaded' compared to 'single-threaded'. Looks like some internal bottleneck with multithreaded tasks?
> 
> For anyone interested in how important memory access is, this is the comparison with another 5B (from Radxa) testing only the four A76 cores of Rockchip RK3588: https://browser.geekbench.com/v6/cpu/compare/2808487?baseline=2818821 (`sbc-bench -G` in Geekbench-Mode by default also tests each CPU cluster individually)
> 
> So this is two times four A76 cores (RPi 5B running at 2.4 GHz, Rock 5B at only ~2.3 GHz) in comparison:
> 
> [image]
> 
> (full results output: http://ix.io/4HGm)
> 
> Edit: retested with Geekbench 6.2 to match Jeff's version.
> 
> Edit 2: Since Jeff right now seems to be the only one who ran GB 6 let's look at [the few GB5 scores](https://browser.geekbench.com/search?k=v5_cpu&q=Raspberry+Pi+5+Model+B+Rev+1.0). Multi-threaded to single-threaded ratio not as bad as with Jeff's GB6 result but still hinting at something seriously wrong since MT scores are just 2.5 higher than ST.

Some of the multi-thread benches work in cycles and wait for all threads to finnish before issuing another. In real like with the likes of whisper.cpp or llama.ccp running on all cores is either scaling doesn't work well or that it causes queueing for the small cores to finnish as the RK3588 is faster using 4 threads and slows on 8 with the previous.
So guess Geekbench maybe similar also under linux core 0-3 is little, 4-7 is big and often wondered if Geekbench just picks the 1st core in some tests which may be 0 and little.
Geekbench is a bit pants and https://github.com/ThomasKaiser/sbc-bench is likely a bit better, dunno if it is optimised for newer Arm v8.2 though.

Headcrabed, 2023-09-29:
> > Hello, can you test glmark2 and glmark2-es2 score and do a simple comparison between rpi4b?
> 
> @Headcrabed - mentioned in my video (linked earlier), GLMark2 scores:
> 
>     * 117 fullscreen
> 
>     * 905 windowed

Thank you, and could you test clpeak through rusticl? So we could compare its compute performance between other sbcs.

ThomasKaiser, 2023-09-29:
> under linux core 0-3 is little, 4-7 is big

Not with Amlogic AD311D2, there it's the other way around.

> and often wondered if Geekbench just picks the 1st core in some tests which may be 0 and little.

As written before: when using `sbc-bench`'s Geekbench mode (`-G`) the clusters will also be tested individually. The little cores were all killed prior to running Geekbench to get an apples to apples comparison. Again: this is RK3588 castrated to meet BCM2712 capabilities: two times quad-core A76, no little cores involved, exact same software, huge multi-threaded difference: https://browser.geekbench.com/v6/cpu/compare/2808487?baseline=2818821

> Geekbench 6 in general is a strange beast on Arm

Sure but still... how is it possible that one quad-core A76 (BCM2712) performs so poorly compared to another (RK3588 w/o little cores)?

mariobalanica, 2023-09-29:
> Sure but still... how is it possible that one quad-core A76 (BCM2712) performs so poorly compared to another (RK3588 w/o little cores)?

Single LPDDR4X chip (dual-channel) on RPi vs two (quad-channel) on RK3588 boards.

RAM bandwidth has a great effect on multi-core performance.

ThomasKaiser, 2023-09-29:
> device tree in here

BTW: according to the commented official [bcm2712.dtsi](https://github.com/raspberrypi/linux/blob/2d3d2030c0564790a76cf94fa0ca1913873381af/arch/arm/boot/dts/bcm2712.dtsi#L935-L1092) there's three PCIe 3.0 controllers inside BCM2712 providing six Gen3 lanes in total (all downgraded to Gen2 speeds by default for obvious reasons like idle consumption)

ayufan, 2023-09-29:
@geerlingguy I'm interested in what is exposed as part of `v4l2_m2m` (if anything at all). Could you run the https://github.com/ayufan/camera-streamer/blob/main/tools/dump_cameras.sh?

theofficialgman, 2023-09-29:
> @theofficialgman - Yes, currently they are trialling 16k page size in the Bookworm release for Pi 5, but switching is as simple as:
>
> ```
> [pi5]
> kernel=kernel8.img
> ```
>
> The alpha testers have been [...]

Great thanks. CC: @ptitseb  looks like you will be wanting to implement 16K pagesize in box86 after all.

ptitSeb, 2023-09-29:
> Great thanks. CC: @ptitSeb looks like you will be wanting to implement 16K pagesize in box86 after all.

Oh, thanks for the info. I'll add that in my todo.

theofficialgman, 2023-09-29:
@geerlingguy what does the armhf 32bit image use as it's kernel? An armv6 kernel with 4k pages? An armv7 kernel with 4k pages? Armv8 with 4 or 16k pages?

There is a lot of Linux software that is pagesize specific (look at asahi Linux's list) and most of these things were only fixed for arm64 (and only in arch, not necessarily in the debian bookworm packages)
<https://github.com/AsahiLinux/docs/wiki/Broken-Software>

ThomasKaiser, 2023-09-29:
> Single LPDDR4X chip (dual-channel) on RPi vs two (quad-channel) on RK3588 boards.

This explains it partially, at least for Geekbench while 7-zip's internal benchmark for example is not affected since here memory latency is important and not so much bandwidth.

This is still crippled RK3588 (little cores killed) with big cores slightly lower clocked than the A76 in BCM2712 and DRAM clocked the first time at 2112 MHz and then only at 528 MHz:

| Device / DRAM clock | GB6 multi | GB6 single | 7-zip multi | 7-zip single | memcpy | memset |
| ----- | ------: | ------: | ------: | ------: | ------: | -----: | -----: |
| Rock 5B / 2112 MHz | 2644 | 861 | 10898 | 3146 | 10830 | 29220 |
| Rock 5B / 528 MHz | 1830 | 730 | 10030 | 2570 | 3520 | 8400 |
| RPi 5B | 1507 | 748 | 10950 | 3160 | 5260 | 11520 |

So while Rock 5B with DRAM clocked at just 528 MHz shows significantly lower memcpy/memset scores it still easily outperforms RPi 5B multi-threaded in GB6 which hints at something else going on inside BCM2712 too.

And this is 528 MHz vs. 2112 MHz in direct comparison to see which of the individual Geekbench tests is affected by memory bandwidth and which isn't (that much): https://browser.geekbench.com/v6/cpu/compare/2826821?baseline=2818821

fanoush, 2023-09-29:
@geerlingguy can you possibly add photo of the other side of the board below the you you have there? Maybe even x-ray photo would be nice :-)

There are 7 pairs of traces going  to the PCI-E FPC connector, that may be too much for 1x pci-e? However two pairs  go to another layer, maybe these are SD slot data pins after all - is the sd slot on the other side? However I guess there would be another pair of capacitors if there were two pcie lanes going that way.

erdnussIO, 2023-09-29:
@geerlingguy Thank you very much for the detailed launch-day coverage!

Does the PTP support of the phy in the RP1 include an user-accessible PPS output?

geerlingguy, 2023-09-29:
@fanoush - I don't have x-rays yet, but here's the bottom:

[image]

@ayufan - `dump_cameras.sh` output below:

<details>
<summary>dump_cameras.sh output - click to expand</summary>

```
pi@pi5:~/Downloads $ bash ./dump_cameras.sh 
++ cat /sys/firmware/devicetree/base/model
++ tr -d '\0'
+ echo 'Raspberry Pi 5 Model B Rev 1.0'
Raspberry Pi 5 Model B Rev 1.0
+ uname -a
Linux pi5 6.1.47-v8_16k+ #1 SMP PREEMPT Thu Sep 21 21:07:03 CDT 2023 aarch64 GNU/Linux
+ v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.
v4l2-ctl | pispbe (platform:1000880000.pisp_be):
v4l2-ctl | 	/dev/video20
v4l2-ctl | 	/dev/video21
v4l2-ctl | 	/dev/video22
v4l2-ctl | 	/dev/video23
v4l2-ctl | 	/dev/video24
v4l2-ctl | 	/dev/video25
v4l2-ctl | 	/dev/video26
v4l2-ctl | 	/dev/video27
v4l2-ctl | 	/dev/video28
v4l2-ctl | 	/dev/video29
v4l2-ctl | 	/dev/video30
v4l2-ctl | 	/dev/video31
v4l2-ctl | 	/dev/video32
v4l2-ctl | 	/dev/video33
v4l2-ctl | 	/dev/video34
v4l2-ctl | 	/dev/video35
v4l2-ctl | 	/dev/video36
v4l2-ctl | 	/dev/video37
v4l2-ctl | 	/dev/media0
v4l2-ctl | 	/dev/media2
v4l2-ctl | 
v4l2-ctl | rpivid (platform:rpivid):
v4l2-ctl | 	/dev/video19
v4l2-ctl | 	/dev/media1
v4l2-ctl | 
+ libcamera-hello --list-cameras
libcamera | No cameras available!
====================================
DEVICE: /dev/video19
====================================
+ v4l2-ctl -d /dev/video19 --info --list-formats-ext --list-fields --list-formats-out --list-fields-out --list-ctrls
/dev/video19 | Driver Info:
/dev/video19 | 	Driver name      : rpivid
/dev/video19 | 	Card type        : rpivid
/dev/video19 | 	Bus info         : platform:rpivid
/dev/video19 | 	Driver version   : 6.1.47
/dev/video19 | 	Capabilities     : 0x84204000
/dev/video19 | 		Video Memory-to-Memory Multiplanar
/dev/video19 | 		Streaming
/dev/video19 | 		Extended Pix Format
/dev/video19 | 		Device Capabilities
/dev/video19 | 	Device Caps      : 0x04204000
/dev/video19 | 		Video Memory-to-Memory Multiplanar
/dev/video19 | 		Streaming
/dev/video19 | 		Extended Pix Format
/dev/video19 | Media Driver Info:
/dev/video19 | 	Driver name      : rpivid
/dev/video19 | 	Model            : rpivid
/dev/video19 | 	Serial           : 
/dev/video19 | 	Bus info         : platform:rpivid
[... 1768 lines cut ...]
/dev/video37 | Media Driver Info:
/dev/video37 | 	Driver name      : pispbe
/dev/video37 | 	Model            : pispbe
/dev/video37 | 	Serial           : 
/dev/video37 | 	Bus info         : platform:1000880000.pisp_be
/dev/video37 | 	Media version    : 6.1.47
/dev/video37 | 	Hardware revision: 0x02252700 (35989248)
/dev/video37 | 	Driver version   : 6.1.47
/dev/video37 | Interface Info:
/dev/video37 | 	ID               : 0x0300003d
/dev/video37 | 	Type             : V4L Video
/dev/video37 | Entity Info:
/dev/video37 | 	ID               : 0x0000003b (59)
/dev/video37 | 	Name             : pispbe-config
/dev/video37 | 	Function         : V4L2 I/O
/dev/video37 | 	Pad 0x0100003c   : 0: Source
/dev/video37 | 	  Link 0x0200003f: to remote pad 0x100000a of entity 'pispbe' (Video Pixel Formatter): Data, Enabled, Immutable
/dev/video37 | ioctl: VIDIOC_ENUM_FMT
/dev/video37 | 	Type: Video Capture
/dev/video37 | 
/dev/video37 | ioctl: VIDIOC_ENUM_FMT
/dev/video37 | 	Type: Video Output
/dev/video37 | 
```
</details>

geerlingguy, 2023-09-29:
> what does the armhf 32bit image use as it's kernel? An armv6 kernel with 4k pages? An armv7 kernel with 4k pages? Armv8 with 4 or 16k pages?

AFAICT, Pi 5 doesn't yet have a 32-bit Arm image (not sure if there will be one?). All of us have been testing on 64-bit image. Good question to ask one of the Pi engineers.

@erdnussIO - I believe the PHY *might* have a pin for PPS_OUT, but it is not routed anywhere accessible on the Pi 5 (at least not at a quick glance). Maybe there's a test pin somewhere on the bottom it can be accessed?

On the CM4, there was a separate header where PPS_IN/PPS_OUT was exposed. I don't see any way for it to route to GPIO so we'd need to figure out the pin on the BCM chip and solder to it.

erdnussIO, 2023-09-29:
> > what does the armhf 32bit image use as it's kernel? An armv6 kernel with 4k pages? An armv7 kernel with 4k pages? Armv8 with 4 or 16k pages?
> 
> AFAICT, Pi 5 doesn't yet have a 32-bit Arm image (not sure if there will be one?). All of us have been testing on 64-bit image. Good question to ask one of the Pi engineers.
> 
> @erdnussIO - I believe the PHY _might_ have a pin for PPS_OUT, but it is not routed anywhere accessible on the Pi 5 (at least not at a quick glance). Maybe there's a test pin somewhere (TP64?) on the bottom it can be accessed?
> 
> On the CM4, there was a separate header where PPS_IN/PPS_OUT was exposed. I don't see any way for it to route to GPIO so we'd need to figure out the pin on the BCM chip and solder to it.

Thank you so much for your investigation. 
That would be a bit of a pity - I would have hoped for something like that pin to be routed through the RP1 as an AF or so (which, I have to admit, would probably be quite unconventional). So much for an audio HAT synchronized to/clocked by PTP via the PPS output (having AES67 in mind here)

will127534, 2023-09-29:
> /dev/video21 | 	[40]: 'PC1B' (PiSP Bayer Comp 1, compressed)
> /dev/video21 | 	[41]: 'PC1R' (PiSP Bayer Comp 1, compressed)
> /dev/video21 | 	[42]: 'PC1G' (PiSP Bayer Comp 1, compressed)
> /dev/video21 | [...]

This looks very interesting.... do RPI have their special compressed Bayer format?

Also thank you for all the work and follow up you did, this thread has been a amazing source of information!

ayufan, 2023-09-29:
@will127534 I believe this is for supporting more advanced and higher-pixel camera sensors.

theofficialgman, 2023-09-29:
@geerlingguy I have compiled vulkancapsviewer and glcapsviewer for you to run if you can. this will allow you to upload the OpenGL/GLES/Vulkan capabilities of the Pi5 to the respective online databases

You will need a few libraries installed to run the binaries. I will list them in debian control automated format for each after each zip
[glcapsviewer.zip](https://github.com/geerlingguy/sbc-reviews/files/12771751/glcapsviewer.zip)
```
Depends=libc6 (>= 2.34), libgcc-s1 (>= 3.0), libglew2.2 (>= 2.2.0-4+b1), libglfw3 (>= 3.0), libglx0, libopengl0, libqt5core5a (>= 5.14.1), libqt5gui5 (>= 5.0.2) | libqt5gui5-gles (>= 5.0.2), libqt5network5 (>= 5.0.2), libqt5widgets5 (>= 5.4.0), libstdc++6 (>= 11), libx11-6
```
[vulkanCapsViewer.zip](https://github.com/geerlingguy/sbc-reviews/files/12771752/vulkanCapsViewer.zip)
```
Depends=libc6 (>= 2.34), libgcc-s1 (>= 3.0), libqt5core5a (>= 5.15.1), libqt5gui5 (>= 5.0.2) | libqt5gui5-gles (>= 5.0.2), libqt5network5 (>= 5.0.2), libqt5widgets5 (>= 5.14.1), libstdc++6 (>= 11), libvulkan1 (>= 1.2.131.2), libwayland-client0 (>= 1.0.2)
```
In glcapsviewer make sure to upload both the OpenGL Default and OpenGLES 3.0 Context results

For anyone curious, these dependencies were autogenerated using `dpkg-shlibdeps` which is what many debian packages use to generate their dependencies. So I just did this in the folder that contained the binary:
```bash
# create dummy debian control file
mkdir debian
touch debian/control
# generate dependency list in debian control format and print to stdout
dpkg-shlibdeps --ignore-missing-info -O vulkanCapsViewer
```

geerlingguy, 2023-09-29:
@theofficialgman - I will try to run that later (have a kids' event to attend tonight, was painting most of the day!). @will127534 and @ayufan — when I visited Pi HQ in May, I noticed there was a *lot* of buzz around ISP in the new chip, they were excited to talk about camera capabilities (and some ideas using RP1 for low power stuff, like detecting simple light/dark levels via camera even if BCM2712 is powered off).

When engineers talk with some excitement, it often indicates they are proud of some things that they know you won't understand. And I understand little of the guts of the thing!

will127534, 2023-09-30:
> Looks like it is still limited to **USB OTG 2.0** like the Pi4 [[1]](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/#comment-1594029). Rock 3/4 have USB OTG 3.0 but ~Rock 5~ only has [...]

I wonder if we can switch that DW3 USB3 IP to be in device mode and we can then have the USB OTG 3.0 on the USB-A connector.

FCLC, 2023-09-30:
Any luck on VAINFO? if it does have HEVC 10bit decode, could make for a nice little HDR transcode capable jellyfish server

geerlingguy, 2023-09-30:
@theofficialgman - See:

  - [OpenGL Report for V3D 7.1](https://opengl.gpuinfo.org/displayreport.php?id=10501)
  - [Vulkan Report for V3D 7.1.7](https://vulkan.gpuinfo.org/displayreport.php?id=24529)

FCLC, 2023-09-30:
> @theofficialgman - See:
> 
> * [OpenGL Report for V3D 7.1](https://opengl.gpuinfo.org/displayreport.php?id=10501)
> * [Vulkan Report for V3D 7.1.7](https://vulkan.gpuinfo.org/displayreport.php?id=24529)
> 
> @FCLC - `vainfo` returns:
> 
> ```
> pi@pi5:~/Downloads/vktest $ sudo vainfo
> error: XDG_RUNTIME_DIR is invalid or not set in the environment.
> libva info: VA-API version 1.17.0
> libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
> vaInitialize failed with error code -1 (unknown libva error),exit
> ```

You shouldn't need sudo here, but it shouldn't hurt. Could be that they haven't enabled vaapi and that they've only got v4l2 up and running right now

[127 more replies]