geerlingguy/sbc-reviews #22: Milk-V Mars CM

geerlingguy, 2023-10-27:
[image]

## Basic information

  - Board URL (official): https://milkv.io/mars-cm
  - Board purchased from: [ARACE](https://arace.tech/products/milk-v-mars-cm)
  - Board purchase date: 2023-09-05
  - Board specs (as tested): 4 GB RAM / 16 GB eMMC
  - Board price (as tested): $54 (+$12 shipping)

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2023-10-27:
Just plopped this board on my Compute Module 4 IO Board, and it seems to do some sort of boot process, arriving at a flashing green LED (two flashes, then 1 second break, repeating).

The network LEDs light up too, but I don't see a new device on the network. No HDMI output, and I'm not 100% sure if the GPIO pinout is the same as Pi CM4, so I don't want to plug in UART yet...

I tried accessing the 'Get Started' docs page listed on the side of the box ([http://milkv.io/docs/mars-cm](http://milkv.io/docs/mars-cm)), but that page is currently giving a 404 Page Not Found.

I've opened up an issue on the MilkV Community Forum: [Mars CM - Getting Started guide missing?](https://community.milkv.io/t/mars-cm-getting-started-guide-missing/845).

It would be great if this JH7110-based board could run as a drop-in replacement of the Compute Module 4... we'll see!

geerlingguy, 2023-10-27:
Attaching to UART using the same pins as Mars ([see Serial Console docs](https://milkv.io/docs/mars/getting-satrted/setup)), I was able to monitor boot output...

```
U-Boot SPL 2021.10 (Oct 08 2023 - 17:13:43 +0800)
LPDDR4: 4G version: g8ad50857.
Trying to boot from SPI

OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|___./_____|
        | |
        |_|

Platform Name             : StarFive VisionFive V2
Platform Features         : medeleg
Platform HART Count       : 5
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 4000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : pm-reset
Platform Shutdown Device  : pm-reset
Platform Suspend Device   : ---
Firmware Base             : 0x40000000
Firmware Size             : 392 KB
Firmware RW Offset        : 0x40000
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 1
Domain0 HARTs             : 0*,1*,2*,3*,4*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000040000000-0x000000004003ffff M: (R,X) S/U: ()
Domain0 Region02          : 0x0000000040040000-0x000000004007ffff M: (R,W) S/U: ()
Domain0 Region03          : 0x0000000000000000-0xffffffffffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x0000000040200000
Domain0 Next Arg1         : 0x0000000042200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 1
Boot HART Domain          : root
Boot HART Priv Version    : v1.11
Boot HART Base ISA        : rv64imafdcbx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 4096
Boot HART PMP Address Bits: 34
Boot HART MHPM Count      : 2
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2021.10 (Oct 08 2023 - 17:13:43 +0800)

CPU:   rv64imacu_zba_zbb
Model: StarFive VisionFive V2
DRAM:  4 GiB
cdns3_bind: unsupported dr_mode
MMC:   sdio0@16010000: 0, sdio1@16020000: 1
Loading Environment from SPIFlash... SF: Detected gd25lq128 with page size 256 Bytes, erase size 4 KiB, total 16 MiB
*** Warning - bad CRC, using default environment

StarFive EEPROM format v2
...
```

It looks like it does automatically boot into Debian, and it says an eth0 link is up... (this is with no microSD card attached). Trying to find it on my network now.

geerlingguy, 2023-10-27:
I can't figure out the login/password. I tried `user`/`starfive`, `user`/`user`, and `riscv`/`starfive` so far, to no avail.

geerlingguy, 2023-10-27:
SSH password authentication was disabled, so to enable it (without messing with vi/nano over UART), I ran the oneliners:

```
sed -i '/PasswordAuthentication yes/s/^#//g' /etc/ssh/sshd_config
sed -i '/PermitRootLogin prohibit-password/s/^#//g' /etc/ssh/sshd_config
sed -i '/PermitRootLogin prohibit-password/s/PermitRootLogin prohibit-password/PermitRootLogin yes/g' /etc/ssh/sshd_config
```

And restarted SSH with:

```
systemctl restart sshd
```

Now I can login via SSH!

geerlingguy, 2023-10-27:
I'm in, but apt is incredibly slow (like in the single-digit-kilobytes per second range). Not sure why. Waiting for an install to complete so I can fetch some more info about the system. An `apt upgrade` might take all night at this range!

geerlingguy, 2023-10-27:
The disk partition was not expanded initially, so I realized when I ran out of disk space (only about 4 GB is allocated by default), I had to run through the process documented on StarFive's website for [extending the partition on the eMMC storage](https://doc-en.rvspace.org/VisionFive2/Quick_Start_Guide/VisionFive2_QSG/extend_partition.html).

Once I did that, I wound up with about 11 GB of usable disk space.

geerlingguy, 2023-10-28:
I'm unable to install Ansible via `pip` on the board because `pip` can't be upgraded to the latest version (easily at least), and thus `cryptography` libraries aren't getting installed.

So I'll run my Top500 benchmark through Ansible on my Mac, over SSH instead.

christiancelona, 2023-10-28:
From the image provided by MilkV, following the instructions previously reported:
[https://github.com/milkv-mars/mars-buildroot-sdk/releases/tag/20230723](https://github.com/milkv-mars/mars-buildroot-sdk/releases/tag/20230723)

`sudo apt install python3-pip`

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  build-essential bzip2 dpkg dpkg-dev fakeroot javascript-common
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libbz2-1.0 libdpkg-perl libexpat1 libexpat1-dev libfakeroot
  libfile-fcntllock-perl libjs-jquery libjs-sphinxdoc libjs-underscore
  liblzma5 libperl5.36 libpython3-dev libpython3.11-dev patch perl perl-base
  perl-modules-5.36 python3-dev python3-distutils python3-lib2to3
  python3-pkg-resources python3-setuptools python3-wheel python3.11-dev
  xz-utils zlib1g zlib1g-dev
Suggested packages:
  bzip2-doc debsig-verify debian-keyring apache2 | lighttpd | httpd git bzr ed
  diffutils-doc perl-doc libterm-readline-gnu-perl
  | libterm-readline-perl-perl libtap-harness-archive-perl
  python-setuptools-doc
The following NEW packages will be installed:
  build-essential bzip2 dpkg-dev fakeroot javascript-common
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libdpkg-perl libexpat1-dev libfakeroot libfile-fcntllock-perl libjs-jquery
  libjs-sphinxdoc libjs-underscore libpython3-dev libpython3.11-dev patch
  python3-dev python3-distutils python3-lib2to3 python3-pip python3-setuptools
  python3-wheel python3.11-dev xz-utils zlib1g-dev
The following packages will be upgraded:
  dpkg libbz2-1.0 libexpat1 liblzma5 libperl5.36 perl perl-base
  perl-modules-5.36 python3-pkg-resources zlib1g
10 upgraded, 27 newly installed, 0 to remove and 825 not upgraded.
Need to get 30.1 MB of archives.
After this operation, 146 MB of additional disk space will be used.
Do you want to continue? [Y/n] 

`sudo apt install ansible`
    
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  ansible-core ieee-data libselinux1 libyaml-0-2 python-babel-localedata
  python3-anyio python3-argcomplete python3-babel python3-cffi-backend
  python3-click python3-colorama python3-cryptography python3-dnspython
  python3-h11 python3-h2 python3-hpack python3-httpcore python3-httplib2
  python3-httpx python3-hyperframe python3-jinja2 python3-jmespath
  python3-kerberos python3-libcloud python3-lockfile python3-markdown-it
  python3-markupsafe python3-mdurl python3-netaddr python3-ntlm-auth
  python3-packaging python3-passlib python3-pygments python3-pyparsing
  python3-requests-ntlm python3-requests-toolbelt python3-resolvelib
  python3-rfc3986 python3-rich python3-selinux python3-simplejson
  python3-sniffio python3-tz python3-winrm python3-xmltodict python3-yaml
Suggested packages:
  cowsay sshpass python-cryptography-doc python3-cryptography-vectors
  python3-trio python3-aioquic python-jinja2-doc python-lockfile-doc ipython3
  python-netaddr-docs python-pygments-doc ttf-bitstream-vera
  python-pyparsing-doc
The following NEW packages will be installed:
  ansible ansible-core ieee-data libyaml-0-2 python-babel-localedata
  python3-anyio python3-argcomplete python3-babel python3-cffi-backend
  python3-click python3-colorama python3-cryptography python3-dnspython
  python3-h11 python3-h2 python3-hpack python3-httpcore python3-httplib2
  python3-httpx python3-hyperframe python3-jinja2 python3-jmespath
  python3-kerberos python3-libcloud python3-lockfile python3-markdown-it
  python3-markupsafe python3-mdurl python3-netaddr python3-ntlm-auth
  python3-packaging python3-passlib python3-pygments python3-pyparsing
  python3-requests-ntlm python3-requests-toolbelt python3-resolvelib
  python3-rfc3986 python3-rich python3-selinux python3-simplejson
  python3-sniffio python3-tz python3-winrm python3-xmltodict python3-yaml
The following packages will be upgraded:
  libselinux1
1 upgraded, 46 newly installed, 0 to remove and 824 not upgraded.
Need to get 30.3 MB of archives.
After this operation, 343 MB of additional disk space will be used.
Do you want to continue? [Y/n]

christiancelona, 2023-10-28:
> I'm unable to install Ansible via `pip` on the board because `pip` can't be upgraded to the latest version (easily at least), and thus `cryptography` libraries aren't getting installed.

We use different images
[https://github.com/milkv-mars/mars-buildroot-sdk/releases/tag/20230723](https://github.com/milkv-mars/mars-buildroot-sdk/releases/tag/20230723)

geerlingguy, 2023-10-28:
To monitor the SoC temps, I installed `lm-sensors` and monitored with `watch`:

```
# apt install -y lm-sensors
# watch sensors
Every 2.0s: sensors                                                              starfive: Sat Oct 28 03:22:46 2023

120e0000.tmon-isa-0000
Adapter: ISA adapter
temp1:        +52.6 C
```

(That was with a heatsink only. SoC throttle limit seems to be 85°C.)

geerlingguy, 2023-10-28:
It's funny, about 10 minutes after I boot the thing, I get a system message that it will suspend. Not sure why... and it doesn't _actually_ suspend:

```
Broadcast message from user@starfive (Sat 2023-10-28 03:38:48 UTC):

The system will suspend now!
```

geerlingguy, 2023-10-28:
I tried using `nmtui` to configure a wifi connection (there were already details for a test connection in there), but after that it just output errors about trying to connect, like:

```
[ 6178.841606] [dhd] [wlan0] dhd_pri_stop : tx queue stopped
[ 6178.847110] [dhd] [wlan0] dhd_stop : Enter
[ 6178.851239] [dhd] dhd_stop: ######### called for ifidx=0 #########
[ 6178.877674] [dhd] dhd_stop: making dhdpub up FALSE
[ 6178.882480] [dhd] dhd_tcpack_suppress_set 377: already set to 0
[ 6178.888442] [dhd] [wlan0] dhd_stop : Exit
[ 6178.893123] [dhd] [wlan0] dhd_set_mac_address : macaddr = ae:34:61:e0:da:c8
[ 6178.900123] [dhd] CFG80211-ERROR) wl_cfg80211_macaddr_sync_reqd : no macthing if type
[ 6178.908044] [dhd] dhd_set_mac_addr_handler: interface info not available/down 
[ 6179.014288] [dhd] [wlan0] dhd_open : Enter
[ 6179.018445] [dhd] dhd_open: ######### called for ifidx=0 #########
[ 6179.047426] [dhd] CFG80211-ERROR) wl_cfg80211_generate_mac_addr : wl_cfg80211_generate_mac_addr:generated new MAC=36:85:ba:e6:7b:50 
[ 6179.065959] [dhd] CFG80211-ERROR) __wl_update_wiphybands : error reading txbf_bfe_cap (-23)
[ 6179.075253] [dhd] CFG80211-ERROR) __wl_update_wiphybands : error reading txbf_bfr_cap (-23)
[ 6179.101182] [dhd] CFG80211-ERROR) init_roam_cache : roamscan_mode iovar failed. -23
[ 6179.108860] [dhd] CFG80211-ERROR) wl_cfg80211_up : Failed to enable RCC.
[ 6179.116226] [dhd] [wlan0] dhd_open : Exit ret=0
[ 6179.120789] [dhd] [wlan0] dhd_pri_open : tx queue started
[ 6179.127869] [dhd] [wlan0] _dhd_set_mac_address : MACID ae:34:61:e0:da:c8 is overwritten
[ 6179.237736] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 184, bssidx: 0
[ 6181.683969] [dhd] CFG80211-ERROR) wl_set_auth_type : set auth failed (-2)
[ 6181.690788] [dhd] CFG80211-ERROR) wl_config_assoc_security : Invalid auth type
[ 6181.698024] [dhd] CFG80211-ERROR) wl_cfg80211_connect : config assoc security failed
[ 6181.705772] [dhd] CFG80211-ERROR) wl_cfg80211_connect : connect error (-2)
```

eshattow, 2023-10-28:
With regard to gnome not functioning after apt upgrade of the vendor provided OS via Debian unstable package repo, `gnome-shell` (and gdm3...) count among the packages that are failing to build at the moment on the Debian unstable repo (`task-gnome-desktop` is uninstallable). For a while now package building machines as part of Debian infrastructure ran into a Linux kernel regression causing builds to randomly fail with riscv64 on real hardware. This regression was identified and reverted in stable kernel series 6.5.8 which was only just recently packaged for Debian unstable and may/may not yet be rolled out to actually be running on the build machines. You can verify this with a Debian QuIck Baker image in a virtual machine for example https://people.debian.org/~gio/dqib/ to get a baseline riscv64 Debian unstable system, and then following along of https://buildd.debian.org/status/architecture.php?a=riscv64&suite=sid for an overview of what packages are stuck in the re-build churn.

geerlingguy, 2023-10-28:
A number of the Phoronix Test Suite tests failed, for example, trying to do the mp3 encode test:

```
UNAME_MACHINE = riscv64
UNAME_RELEASE = 5.15.0
UNAME_SYSTEM  = Linux
UNAME_VERSION = #9 SMP Sun Oct 8 17:14:43 CST 2023
configure: error: cannot guess build type; you must specify one
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

And then for things like the linux kernel build, it ran out of space. For now I'll leave those tests out, 16 GB isn't that much space to play with :)

geerlingguy, 2023-10-28:
I still can't get WiFi to connect.

```
# ip link
...
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state DORMANT mode DORMANT group default qlen 1000
    link/ether ee:49:09:26:18:b5 brd ff:ff:ff:ff:ff:ff permaddr b8:2d:28:25:5d:0e
```

`nmtui` upon activating a WiFi link just sits `Connecting...` while error messages like those earlier are printed in dmesg.

That was with my 5 GHz network. Trying two different 2.4 GHz networks (one is on an older router that works better with ancient or weird WiFi hardware), I could get nothing from either of those too (same errors).

So maybe the WiFi driver's a bit flaky in the OS image I have installed?

geerlingguy, 2023-10-29:
Checking on the [JH7110 Linux kernel Upstreaming Status](https://rvspace.org/en/project/JH7110_Upstream_Plan), it looks like most interfaces are upstream now, with the rest on their way. Quite nice! And happening more quickly than many of the Arm SoCs these days...

geerlingguy, 2023-10-29:
Testing PCIe:

```
root@starfive:~# sudo lspci -vvvv
00:00.0 PCI bridge: PLDA XpressRich-AXI Ref Design (rev 02) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 32 bytes
	Interrupt: pin A routed to IRQ 0
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 30000000-300fffff [size=1M] [32-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [80] Express (v2) Root Port (Slot+), MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0
			ExtTag+ RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
			ClockPM- Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1
			TrErr- Train- SlotClk- DLActive- BWMgmt+ ABWMgmt-
		SltCap:	AttnBtn- PwrCtrl- MRL- AttnInd- PwrInd- HotPlug- Surprise-
			Slot #0, PowerLimit 0W; Interlock- NoCompl-
		SltCtl:	Enable: AttnBtn- PwrFlt- MRL- PresDet- CmdCplt- HPIrq- LinkChg-
			Control: AttnInd Unknown, PwrInd Unknown, Power- Interlock-
		SltSta:	Status: AttnBtn- PowerFlt- MRL- CmdCplt- PresDet- Interlock-
			Changed: MRL- PresDet- LinkState-
		RootCap: CRSVisible-
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna- CRSVisible-
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr+ BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn- NFERptEn- FERptEn-
		RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
			 FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
		ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0000
[... 64 lines cut ...]
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [360 v1] Data Link Feature <?>
	Capabilities: [380 v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [400 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
			  PortCommonModeRestoreTime=60us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [470 v1] Lane Margining at the Receiver <?>
	Kernel driver in use: nvme
```

```
root@starfive:~# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
...
nvme0n1      259:0    0  1.9T  0 disk 
|-nvme0n1p1  259:1    0  256M  0 part 
`-nvme0n1p2  259:2    0  1.9T  0 part 
```

geerlingguy, 2023-10-29:
I [formatted the 2TB NVMe SSD](https://www.jeffgeerling.com/blog/2021/htgwa-partition-format-and-mount-large-disk-linux-parted), then ran my disk benchmarks, and got 150 MB/sec for fio, and 250 MB/sec for 1M sequential read/write via iozone. Not amazing, but a lot better than built-in eMMC!

christiancelona, 2023-10-30:
There are several [commits regarding RISC-V](https://kernelnewbies.org/Linux_6.6#RISC-V) and many more on StarFive’s JH7110 SoC:

- [add bindings for the PLL clock generator on the JH7110 RISC-V SoC](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=bd348ca24d81cca2a27f8ffa12adc8f30f184275);
- [add documentation to describe StarFive System Controller Registers](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c81f7845b2ce7a2ea1beb2ac4621b5d568d2b644);
- [add PLL clock inputs from PLL clock generator](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=2110add84bc6e21a1bf55f2c9d1fc14d408ce2e0);
- [add driver for the StarFive JH7110 PLL clock controller and they work by reading and setting syscon registers](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=616bc1dea1ac8909dfcd6d32802df6fe50eddde8);
- [modify PLL clocks source to be got from DTS or the fixed factor clocks](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=a013e9818734ade52944e73f64242633c6cd4128);
- [add stg_syscon/sys_syscon/aon_syscon/PLL nodes for JH7110 SoC](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3fcbcfc496f0cc08f9dc004a92915ce1cfb7ea95);
- [add PLL clocks input from PLL clocks driver in SYSCRG node](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3e6670a28b009cc381b40ee26a6f41509aca46eb);
- [add Starfive JH7110 SoC PCIe 2.0 PHY driver support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fd097f48eea95c9776a8846edfe9e328e9eaf7e2);
- [add StarFive JH7110 SoC USB 2.0 PHY dt-binding](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=2689c9c4ab6006342fce9dd88a0d63a24cf9c05f);
- [add StarFive JH7110 SoC PCIe 2.0 PHY dt-binding](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=69d41115b83905d77474846cbcea91b84bbb7175);
- [add Starfive JH7110 SoC USB 2.0 PHY driver support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=16d3a71c20cf2e7cb5f0fef669fe8dc7c2cfd87d);
- [add Starfive JH7110 SoC PCIe 2.0 PHY driver support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fd097f48eea95c9776a8846edfe9e328e9eaf7e2).

geerlingguy, 2023-10-30:
@christiancelona - Do you know why the buildroot image for these devices is still on kernel 5.15.x?

christiancelona, 2023-10-30:
I would have preferred not to write it to you :)
I believe that there is little point in engaging in benchmarks of this SoC, except in the case in which you want to measure the benefits brought by good software support, which has so far been absent.

I'm waiting for the new integrated SMB kernel that might make my SBC suitable for my purposes. Up until now I have always used Raspberry Pi but the latest one is absolutely not in my purchase plans and not because of the cost. For those who wanted to use the Milk-V Mars SBC as a low-power desktop, the biggest obstacle to using the new kernels was certainly the display controller driver for Verisilicon DC8200 (for me the [Chinese ideograms](https://www.verisilicon.com/cn/Home) are a big obstacle)and an HMDI driver, [now patched](https://patchwork.kernel.org/project/linux-riscv/cover/20231025103957.3776-1-keith.zhao@starfivetech.com/)
and check again the [documentation](https://rvspace.org/en/project/JH7110_Upstream_Plan) at the line HDMI/ DC8200.

At the moment I am not able to compile a new kernel and what follows, I lack the time and maybe even some skill, I am waiting for the new images from Armbian or DietPi and on November 1st or 2nd a Debian mouse should be ready to try.

But perhaps you have clearer ideas about the propagation time of new kernels in Linux distributions.

christiancelona, 2023-10-30:
Update:
`Skipping acquire of configured file 'main/binary-riscv64/Packages' as repository 'http://deb.debian.org/debian bookworm-backports InRelease' doesn't support architecture 'riscv64'`

christiancelona, 2023-10-31:
Milk-V Mars SBC benchmark: http://ix.io/4KoX
StarFive JH7110 | 1500 MHz | 5.15 | Armbian 23.11.0-trunk.243 lunar riscv64/riscv | 4000 | 1174 | 24900 | 870 | 790 | - |

geerlingguy, 2023-10-31:
@christiancelona - Interesting, it looks like every score besides the memcpy/memset scores was slower in your run (compared to [the one I ran on the StarFive kernel](http://ix.io/4K7E)).

christiancelona, 2023-10-31:
I wondered if it could depend on the software and I redid the tests with a Raspberry Pi 4B and the same OS version.

[RPi 4 Model B Rev 1.4](https://www.raspberrypi.com/products/raspberry-pi-4-model-b/), sbc-bench v0.9.49 result: [http://ix.io/4Kpg](http://ix.io/4Kpg)
BCM2711 Rev B0 | 1800 MHz | 6.1 | Armbian 23.11.0-trunk.243 jammy arm64 | 5580 | 1704 | 36240 | 2400 | 2940 | - |

ThomasKaiser, 2023-10-31:
@christiancelona with 6.5 it doesn't look better (memory latency and CPU clockspeeds the same but scores for whatever reason slightly lower):

| Kernel | Clockspeed | Distro | 7-zip multi | 7-zip single | AES | memcpy | memset |
| ----- | :--------: | :----: | ----: | ------: | ------: | -----: | -----: |
| [5.15](http://ix.io/4KoX) | 1500 MHz | Lunar riscv64 | 4000 | 1174 | 24900 | 870 | 790 |
| [6.5](http://ix.io/4DbJ) | 1500 MHz | Lunar riscv64 | 3910 | 1114 | 24100 | 870 | 840 |

christiancelona, 2023-10-31:
Thank you. I installed the StarFive image, they are different for SD or whatever. Indeed, the difference in performance is also noticeable in use, not only from the benchmarks, the system is more responsive.

I have no idea what changes between kernels, other than the version, at the moment the image provided by the SoC manufacturer is the best, despite being older.

rhinot, 2023-10-31:
Hey Folks - How can I help contribute here?
I have a Milk-V Mars. What do you recommend as an image, and have either of you figured out how to setup and run this thing headless?

geerlingguy, 2023-11-01:
@rhinot - See the [3rd comment above](https://github.com/geerlingguy/sbc-reviews/issues/22#issuecomment-1783550433); you'll need to get console access using UART if you want to set things up headless. You might also be able to flash a newer image to microSD if you bought a 'lite' version without eMMC baked in.

I'll be posting more about the Mars CM tomorrow too, on my blog and on YouTube. I haven't personally tested Mars (sans CM) though.

tumetuoppi, 2023-11-01:
I was able to get an image over HDMI with the eMMC version. First boot there was no image, unplugged power and plugged the power back on -> Linux loading texts showed up on the monitor and eventually it booted in to Gnome. Subsequent boots the HDMI has been working fine.

geerlingguy, 2023-11-01:
@tumetuoppi - That's good—I wonder if it could be that my older HP monitor is just not recognized for some reason? I have booted and rebooted at least 20-30x now, both reboots and unplug/replugs (sometimes waiting a while between boots).

I also don't know the process for re-flashing the eMMC, so I wonder if maybe the slightly newer OS image for the Mars available for download from the Debian page could help?

taspenwall, 2023-11-02:
Has anyone been able to get the buildroot image working?  it just hangs while booting for me.

carbonfix, 2023-11-03:
@geerlingguy The process for re-flashing the eMMC image has been updated to the document:

[https://milkv.io/docs/mars/compute-module/boot](https://milkv.io/docs/mars/compute-module/boot)

rhinot, 2023-11-06:
In case it helps...
[Milk-V Mars](https://milkv.io/docs/mars/overview), sbc-bench v0.9.50 result: http://ix.io/4KNT
StarFive JH7110 | 1500 MHz | 6.1.59 | Debian GNU/Linux trixie/sid riscv64 | 4090 | 1191 | 24600 | 865 | 828 | - |

Similar to @christiancelona's Armbian Lunar numbers on 5.15 & 6.5, although slightly higher.
Big differences: 8GB version, counterfeit SD, DietPi stock build from Nov 3

platima, 2023-11-06:
> Attaching to UART using the same pins as Mars ([see Serial Console docs](https://milkv.io/docs/mars/getting-satrted/setup)), I was able to monitor boot output...
>
> ```
> U-Boot SPL 2021.10 (Oct 08 2023 [...]

Had this same issue last week testing the non-CM board. Corrupt SBC config on SPI flash seemingly resulted in it looking for kernel on eMMC as default, but their image wanted SD card. I flashed both, booted, then changed /boot/extlinux/extlinux.cfg to use mmcblk0 so no SD card required, and it works a treat.

Next week I'm going to try to flash OpenSBI again with correct config, see if that does it. Released eMMC image for now anyway; https://plati.ma/milk-v-mars/ (should have seen this post and enabled password-full SSH, dang it) 

Cheers for your RV contributions mate!

mr-c, 2023-11-08:
> Update: `Skipping acquire of configured file 'main/binary-riscv64/Packages' as repository 'http://deb.debian.org/debian bookworm-backports InRelease' doesn't support architecture 'riscv64'`

@christiancelona Switch to `trixie` (the testing version of the next stable Debian release, where `riscv64` is a first class architecture) from `bookwarm` in your `/etc/apt/sources.list`

platima, 2023-11-08:
> Switch to `trixie` (the testing version of the next stable Debian release, where `riscv64` is a first class architecture) from `bookwarm` in your `/etc/apt/sources.list`

OH this repo is live? I am so damn excited; was looking for it the other day when messing with a Mars SBC. Thank you kindly!

mr-c, 2023-11-08:
> > Switch to `trixie` (the testing version of the next stable Debian release, where `riscv64` is a first class architecture) from `bookwarm` in your `/etc/apt/sources.list`
> 
> OH this repo is live? I am so damn excited; was looking for it the other day when messing with a Mars SBC. Thank you kindly!

Whoops, I should have tested that. Use `sid` instead, I guess the riscv64 packages haven't migrated to Trixie yet.

platima, 2023-11-08:
> Whoops, I should have tested that. Use `sid` instead, I guess the riscv64 packages haven't migrated to Trixie yet.

You set me up, just to let me down 💔

eshattow, 2023-11-08:
> Whoops, I should have tested that. Use `sid` instead, I guess the riscv64 packages haven't migrated to Trixie yet.

/etc/apt/sources.list.d/debian.sources:
```
Types: deb deb-src
URIs: http://deb.debian.org/debian
Suites: unstable
Components: main contrib non-free non-free-firmware
Signed-By: /usr/share/keyrings/debian-archive-keyring.gpg
```

/etc/apt/sources.list/debian-experimental.sources:
```
Types: deb deb-src
URIs: http://deb.debian.org/debian
Suites: experimental
Components: main contrib non-free non-free-firmware
Signed-By: /usr/share/keyrings/debian-archive-keyring.gpg
```

There is a PCB revision needed to fix some problems and so the next batch of Mars CM is delayed until later this month (November 2023). Until I get a Mars CM in-hand I am following along with Debian riscv64 progress by a running qemu instance and the DQIB image [as mentioned above](https://github.com/geerlingguy/sbc-reviews/issues/22#issuecomment-1783729076).

platima, 2023-11-09:
@eshattow oh super interesting to know! Can I ask what problems are being fixed if you have any insight?

Ripper to see rv64 in unstable now, did not recall seeing that previously. Looks like it's got kernel 6.5 in there too; https://packages.debian.org/unstable/linux-image-6.5.0-4-riscv64

I've got a Mars SBC (not CM) I am tempted to test with.

eshattow, 2023-11-09:
@platima I too am curious. That was the summary of a reply from the distributor customer support person. 🤷

platima, 2023-11-09:
> From "E Shattow" ***@***.***>
> To "geerlingguy/sbc-reviews" ***@***.***>
> Cc "platima" ***@***.***>; "Mention"
> Date 9/11/2023 4:40:41 PM
> Subject Re: [geerlingguy/sbc-reviews] Milk-V Mars CM (Issue #22)

oh, Crush? Hah

rhinot, 2023-11-11:
@platima I installed the [linux-image-riscv64](https://packages.debian.org/unstable/linux-image-riscv64) which installs 6.5.10-1 and a few other utils.

Install goes fine, but haven't figured how to get the system to boot that kernel (no grub). Happy to try any ideas you have, so you don't have to switch over for nothing.

platima, 2023-11-17:
Hey if it's all setup right, should just be a matter of modifying 
/boot/extlinux/extlinux.conf from memory! You can even have both kernel 
options in there just in case.

Eg see how the Ox64 image has two extlinux entries depending on the 
platform: 
https://github.com/openbouffalo/buildroot_bouffalo/blob/main/board/pine64/ox64/rootfs-overlay/boot/extlinux/extlinux.conf

Just make sure the kernels are both present and have unique names.

From "rhinot" ***@***.***>

Date 18/11/2023 6:32:57 AM

rhinot, 2023-11-18:
Thanks, @platima!

That was helpful, as I tried loading the 6.5 kernel, it failed, and this allowed me to fall back to 6.1.
For other's reference, I also found [this extlinux.conf sample](https://github.com/jetsonhacks/bootFromUSB/blob/main/sample-extlinux.conf) clarifying.

Do you know where I can find what failed when the system tried to boot into my 6.5 kernel? 
I tried journalctl, but there is nothing even with the timestamps of the attempted boot with that kernel.

platima, 2023-11-18:
Very welcome.

And damn, that's a tough one. Not too sure TBH, I'd just watch the debug 
output via serial in PuTTy, and scroll back up to read it hah.

Date 18/11/2023 2:14:26 PM

eshattow, 2023-12-30:
[image]
[image]
Flatbed scan of rev 1.1

eshattow, 2024-01-18:
Fun fact you can use the U-Boot 'dhcp' command to set up the network PHY ahead of loading a kernel. Otherwise MOTORCOMM_PHY in addition to defconfig is needed for mainline Linux kernel, and if you want SD Card function that will be I2C_DESIGNWARE_PLATFORM && MFD_AXP20X_I2C && REGULATOR_AXP20X. No patches needed for SDCard and it seems "good enough" to use the in-tree DTB: starfive/jh7110-starfive-visionfive-2-v1.3b.dtb (although it doesn't exactly match the Mars CM hardware).

With some manual intervention the official debian-installer can be run successfully in EFI mode.

eshattow, 2024-01-24:
### Linux Kernel, EFI stub, and DTB

Linux Kernel sources:
    Linux 6.8 release candidate 1 Download: [linux-6.8-rc1.tar.gz](https://git.kernel.org/torvalds/t/linux-6.8-rc1.tar.gz)

Patchwork series:
    [Refactoring Microchip PCIe driver and add StarFive PCIe](https://patchwork.kernel.org/project/linux-riscv/list/?series=815039&state=%2A&archive=both) Download: [Refactoring-Microchip-PCIe-driver-and-add-StarFive-PCIe.patch](https://patchwork.kernel.org/series/815039/mbox/)
    [riscv: dts: starfive: add Milkv Mars board device tree](https://patchwork.kernel.org/project/linux-riscv/list/?series=806238&state=%2A&archive=both) Download: [riscv-dts-starfive-add-Milkv-Mars-board-device-tree.patch](https://patchwork.kernel.org/series/806238/mbox/)

Attached: 
    Adapt patch 22/22 from series 815039 to add the VisionFive2 PCIe dts updates to the Milk-V Mars dts Download: [add-vf2-pcie-to-milkv-mars-dts.txt](https://github.com/geerlingguy/sbc-reviews/files/14031782/add-vf2-pcie-to-milkv-mars-dts.txt)

```
tar zxvf linux-6.8-rc1.tar.gz
cd linux-6.8-rc1
patch -p1 < ../Refactoring-Microchip-PCIe-driver-and-add-StarFive-PCIe.patch
patch -p1 < ../riscv-dts-starfive-add-Milkv-Mars-board-device-tree.patch
patch -p1 < ../add-vf2-pcie-to-milkv-mars-dts.txt
make defconfig

# Milk-V Mars CM Lite PCIe PHY and proposed PCIe PLDA support from series 815039
scripts/config --enable PHY_STARFIVE_JH7110_PCIE --enable PCIE_STARFIVE_HOST

# networking PHY and eMMC/SD attach to an AXP15060 (as modules)
scripts/config --module I2C_DESIGNWARE_PLATFORM --module MFD_AXP20X_I2C --module REGULATOR_AXP20X --module MOTORCOMM_PHY

# networking PHY and eMMC/SD attach to an AXP15060 (compiled-in)
scripts/config --enable I2C --enable I2C_DESIGNWARE_PLATFORM --enable MFD_AXP20X_I2C --enable REGULATOR_AXP20X --enable MOTORCOMM_PHY
scripts/config --enable DWMAC_STARFIVE --enable STMMAC_PLATFORM  --enable STMMAC_ETH
scripts/config --enable CLK_STARFIVE_JH7110_AON --enable CLK_STARFIVE_JH7110_STG --enable CLK_STARFIVE_JH7110_ISP --enable CLK_STARFIVE_JH7110_VOUT
scripts/config --enable EFIVAR_FS --enable CONFIGFS_FS --enable NLS_ISO8859_1

make olddefconfig
make -j4 bindeb-pkg LOCALVERSION=-cmlite KDEB_PKGVERSION=1
cd ..
mkdir linux-image-6.8.0-rc1-cmlite_1_riscv64
dpkg -x linux-image-6.8.0-rc1-cmlite_1_riscv64.deb linux-image-6.8.0-rc1-cmlite_1_riscv64
gunzip -c linux-image-6.8.0-rc1-cmlite_1_riscv64/boot/vmlinuz-6.8.0-rc1-cmlite > vmlinux-6.8.0-rc1-cmlite.efi
cp linux-image-6.8.0-rc1-cmlite_1_riscv64/usr/lib/linux-image-6.8.0-rc1-cmlite/starfive/jh7110-milkv-mars.dtb jh7110-milkv-mars-cmlite.dtb
7z a linux-image-6.8.0-rc1-cmlite_1_riscv64.zip linux-image-6.8.0-rc1-cmlite_1_riscv64.deb vmlinux-6.8.0-rc1-cmlite.efi jh7110-milkv-mars-cmlite.dtb
```

See attached: [linux-image-6.8.0-rc1-cmlite_1_riscv64.zip](https://github.com/geerlingguy/sbc-reviews/files/14033793/linux-image-6.8.0-rc1-cmlite_1_riscv64.zip)

### U-Boot cleanup

Connect uart with [tio](https://github.com/tio/tio) and interrupt the u-boot bootcmd countdown at power-on. If u-boot complains about the following commands being invalid and you're using Gnome Terminal with tio then try changing the Wide/Narrow Unicode compatibility setting in Gnome Terminal settings prior to paste. Firstly all of the Milk-V Mars CM u-boot environment variable complexity is removed:

`for i in boot_a_script boot_devs boot_efi_binary boot_efi_bootmgr boot_extlinux boot_net_usb_start boot_pci_enum boot_prefixes boot_script_dhcp boot_scripts boot_syslinux_conf boot_targets bootargs bootcmd bootcmd_dhcp bootcmd_distro bootcmd_mmc0 bootdelay bootdev bootdir bootenv bootenv_mmc bootenv_nvme bootenv_sdk bootfile bootm_size bootmode bootpart chip_vision chipa_gmac_set chipa_set chipa_set_linux chipa_set_uboot cma_1g cma_2g cma_4g cma_8g cma_ddr1g_set cma_ddr2g_set cma_ddr4g_set cma_ddr8g_set cma_node cma_resize cma_start cpu_max_vol cpu_speed_1250_set cpu_speed_1500_set cpu_vol_1020_set cpu_vol_1040_set cpu_vol_1060_set cpu_vol_set devnum distro_boot_env distro_bootcmd distro_bootenv_mmc distro_bootenv_nvme distro_mmc_test_and_boot efi_dtb_prefixes emmc_devnum eth0addr eth1addr ethaddr ethact; do env delete ${i}; done`

`for i in fdt_addr_r fdt_high fdt_loaddtb fdt_sizecheck fdtaddr fdtcontroladdr fdtfile fdtoverlay_addr_r initrd_high ipaddr kernel_addr_r kernel_comp_addr_r kernel_comp_size load_distro_uenv load_efi_dtb load_sdk_uenv loadaddr memory_addr memory_size mmc_boot mmc_devnum_l mmc_test_and_boot netmask nvme_boot nvme_devnum_l nvme_init partitions preboot pxefile_addr_r ramdisk_addr_r rootpart scan_dev_for_boot scan_dev_for_boot_part scan_dev_for_efi scan_dev_for_extlinux scan_dev_for_scripts scan_sf_for_scripts script_offset_f script_size_f scriptaddr sd_devnum sdev_blk sdk_boot_env serial# set_fdt_distro sf_boot type_guid_gpt_loader1 type_guid_gpt_loader2 type_guid_gpt_system uboot_fdt_addr usb_boot ver visionfive2_mem_set; do env delete ${i}; done`

`printenv`

Those are split over a few commands because there's a maximum line length for u-boot input. There should now only be $baudrate environment variable remaining. Commit and reset to trigger a restore of some missing basic variables:

`saveenv ; reset`

Set a few essential load address variables according to suggestions at [u-boot Environment Variable documentation](https://docs.u-boot.org/en/latest/usage/environment.html) then commit and reset:

```
setenv loadaddr 0x042000000   # start of DDR 0x040000000 + 32MiB
[... cut ...]
dhcp $kernel_addr_r $tftp_ip:/cmlite/vmlinux-6.8.0-rc1-cmlite.efi
dhcp $fdt_addr_r $tftp_ip:/cmlite/jh7110-milkv-mars-cmlite.dtb
dhcp $ramdisk_addr_r $tftp_ip:/cmlite/initrd.gz
setenv bootargs 'initrd='$ramdisk_addr_r',0x'$filesize
bootefi $kernel_addr_r $fdt_addr_r

Warnings from the installer about missing modules are to be expected since there is a mismatch between the kernel we have loaded and the modules from debian-installer initial ramdisk; choose the option to continue.

to be continued

moroznah, 2024-02-11:
I'm just a Gentoo user with no knowledge in embedded whatsoever trying to get something sane onto Mars CM. @eshattow instructions working nicely. 
For some reason a very early eMMC board that i have refuses to take SD card in any way so this was done directly on eMMC. Some caveats along the way:

Serial is flaky, tried multiple USB ones and FT232RL sort of worked. 
Tio for firmware updates did not work for me, minicom way:
loady
sf probe
ctrl+A S, ymodem, transfer SPL
crc32 $loadaddr $filesize
sf update $loadaddr 0x0 $filesize
ctrl+A S, ymodem, transfer U-BOOT
crc32 $loadaddr $filesize
sf update $loadaddr 0x100000 $filesize

Debian netinst hanged a few times at package installs with CPU soft lockups during untar. Power cycle / repeat.
During install manually created 100MB EFI and whatever ext4 parts, installed grub.
Anyone tried rEFInd on RISC-V?

Postinst to actually boot, since dtb is absent in EFI, used tftp again :
pci enum; nvme scan; mmc rescan
fatload mmc 0:1 ${kernel_addr_r} efi/debian/grubriscv64.efi
setenv tftp_ip 1.2.3.4
dhcp $fdt_addr_r $tftp_ip:/cmlite/jh7110-milkv-mars-cmlite.dtb
bootefi $kernel_addr_r $fdt_addr_r
Install kernel from .deb, copy dtb file to EFI part.

@eshattow 
Please comment how to setup u-boot envs permanently to default into grub without fatloading every time, extlinux.conf is missing and not getting what to do from RTFM.
Thanks!

platima, 2024-02-11:
> Please comment how to setup u-boot envs permanently to default into grub

Hi moroznah. Thanks for the comment but this is not the forum for such discussion - here we talk about the performance of the SBC in question and/or benchmark results, if anything. I think ideally though, there should be minimal discussion as these 'Issues' are just a way to collaboratively store result data.

Your best bet may be to reach out to @eshattow directly, else perhaps the uboot forums, Reddit or have a look at the starfive_tech repos.

I know I broke this guideline myself, but that's a lesson learned. Cheers!

eshattow, 2024-02-11:
@moroznah Thanks! Further discussion is welcome on the Matrix group chat https://matrix.to/#/#milkv-mars:matrix.org part of the [Milk-V Matrix Space](https://matrix.to/#/#milk-v:matrix.org) linked at their [website](https://milkv.io/).

platima, 2024-02-11:
> @moroznah Thanks! Further discussion is welcome on the Matrix group chat https://matrix.to/#/#milkv-mars:matrix.org part of the [Milk-V Matrix Space](https://matrix.to/#/#milk-v:matrix.org) linked at their [website](https://milkv.io/).

Thanks E! Have a great week everyone

eshattow, 2024-02-23:
> > JH7110 CPU MaskROM has a method of accepting XMODEM data over UART to run as SPL directly so don't worry about bricking your Mars CM Lite. The question then is what SPL programs can be used in this way to enact a recovery?
> > 
> > * Source code [was promised for the official recovery.bin tool from StarFive](https://github.com/starfive-tech/Tools/issues/2) in a comment from Michael Zhu dated January of 2023 but no source code was published in at least a year since then. Otherwise than no source code being available the tool is working for JH7110 on Mars CM.
> 
> I haven't used that in quite a while with my jh7110 stuff. @xypron shared this with me and I've used it the couple times I have changed the firmware since: https://github.com/xypron/JH71xx-tools

That is a serial transfer tool co-dependent on (not a replacement for) the StarFive recovery.bin SPL. It says right in the README.md `-r, --recovery <filename>      : Bootloader recovery firmware.`

xypron, 2024-02-23:
@eshattow  I guess https://github.com/starfive-tech/bootloader_recovery is what you are looking for. It produces jh7100_recovery_boot.bin.

eshattow, 2024-02-23:
> @eshattow I guess https://github.com/starfive-tech/bootloader_recovery is what you are looking for. It produces jh7100_recovery_boot.bin.
> 
> For the VisionFive 2 I found it much easier to set the DIP switches to boot from an SD card, flash the SPI flash from U-Boot and then reset the DIP switches.

Is jh7100_recovery_boot.bin valid for JH7110? There is no JH7100 on the MilkV Mars CM.

shanduur, 2024-04-08:
Hi! Have anyone tried running upstream kernel? Looks like [most of the patches are already there for CPU](https://rvspace.org/en/project/JH7110_Upstream_Plan).

rhinot, 2024-04-11:
@shanduur Just tried booting the upstream 6.7.9 kernel on a working DietPi stock install (which currently is running a patched 6.1.81). 

System didn't boot. Nothing even in the kernel logs that I can share to help debug.

eshattow, 2024-08-09:
Milk-V Mars CM Lite on DFRobot IoT Mini Router carrier, in DFRobot IoT Mini Router metal case:
[image]
[image]

Notes: Rubber plug included to cover SMA mounting hole if unused. Same hole I think could be used to mount some kind of connector to route to the GPIO pins as desired, maybe a 3.5mm audio jack for UART?

[16 more replies]