geerlingguy/sbc-reviews #25: Turing RK1

geerlingguy, 2023-11-02:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://turingpi.com/product/turing-rk1/
  - Board purchased from: (Review sample, but I also pre-ordered four 8GB modules)
  - Board purchase date: 2023-08-15 (pre-order placed)
  - Board specs (as tested): 16 GB RAM / 64 GB eMMC
  - Board price (as tested): $169

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2023-11-02:
First step for bringup was getting the boards working. I plugged them into my original Turing Pi 2 prototype board... which is a bit worse for the wear :)

I couldn't get them to boot, and that old board didn't have PWM fan headers either, so I popped them into my kickstarter "v2.4" board, and they booted and had fans running full blast—but I was told I should upgrade to the 2.0.0 (currently RC1) firmware to get full functionality.

So I initially tried updating over the BMC Web UI, and found that later versions only ship an .img file, not an `.swu` file, so you have to do one of the more 'advanced' BMC update methods.

I initially tried using `PhoenixSuit` on my Windows PC, but kept running into all kinds of permissions issues (would not like to work in an ecosystem that requires disabling so many security features, heh...), then was told I can just flash the .img to microSD, boot off it, and get it upgraded that way.

So it was as easy as:

  1. Flash the 2.0.0-RC1 .img from [here](https://github.com/turing-machines/BMC-Firmware/actions/runs/6721854692) to a microSD card with Etcher (if you're reading this later, go get the latest release...)
  2. Insert the microSD card into the Turing Pi 2's rear microSD card slot.
  3. Power on the Turing Pi 2 while observing UART through the BMC_UART3 header. (green to RX, white to TX, black to GND on my Adafruit adapter)
  4. When prompted, type `CONFIRM` over UART and press enter, the firmware will upgrade.

After removing the microSD card and rebooting the board, it shows:

```
 _____ _   _ ____  ___ _   _  ____ 
|_   _| | | |  _ \|_ _| \ | |/ ___|
  | | | | | | |_) || ||  \| | |  _ 
  | | | |_| |  _ < | || |\  | |_| |
  |_|  \___/|_| \_\___|_| \_|\____|

Welcome to Turing Pi

turingpi login: 2023-11-02T22:04:00.464Z INFO  [bmcd] Turing Pi 2 BMC Daemon v2.0.0
```

Yay!

geerlingguy, 2023-11-02:
I also attempted to install the RK1 into a Jetson development board. The [product page states](https://turingpi.com/product/turing-rk1/#block_cc95cb27816e01a8107c402b3216ff6c):

> The Turing RK1 compute module is compatible with Nvidia Jetson pin layout. This means you can plug it into any Nvidia Jetson carrier board.

Unfortunately, the tiny retention pins on the slot on my carrier board seemed to smash into two tiny components on either side of RK1 making it close to impossible to reliably insert the board (without bending some of the metal tabs...).

The Jetson Nano has two tiny holes in the PCB in those areas to accept the little underside retention tabs, the RK1 does not. I'll upload a picture of the problem if I can remember :)

So back to Turing Pi 2 we go!

geerlingguy, 2023-11-02:
It looks like @Joshua-Riek maintains a bunch of images for Rockchip RK3588 devices, and supports RK1 in that repo: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.29

I've downloaded `ubuntu-22.04.3-preinstalled-server-arm64-turing-rk1.img.xz` and will try flashing it to a node via the BMC web UI... fingers crossed!

geerlingguy, 2023-11-03:
After all that, I got:

```
2023-11-02T22:30:28.888Z INFO  [bmcd::app::firmware_runner] started writing to node 1
2023-11-02T22:35:26.221Z INFO  [bmcd::persistency::app_persistency] commiting persistency to disk
2023-11-02T23:31:21.412Z INFO  [bmcd::app::firmware_runner] Verifying checksum of written data to node 1
2023-11-02T23:46:51.843Z INFO  [bmcd::app::firmware_runner] Flashing node 1 successful, restarting device...
2023-11-02T23:46:52.269Z INFO  [bmcd::api::streaming_data_service] worker done. took 1h 16m 26s 669ms 421us 43ns (#1779206829)
thread 'main' panicked at 'source slice length (0) does not match destination slice length (16384)', src/utils/ring_buf.rs:22:33
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

And the web UI showed: `Upgrade failed: [object Object]`

geerlingguy, 2023-11-03:
Trying with an updated version from https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6740498946

geerlingguy, 2023-11-03:
...it looks like the RK1 is booted, but it's not giving me any SSH access. Default user/pass should be `ubuntu` (both).

Joshua-Riek, 2023-11-03:
It might need a moment for cloud-init to do its thing. But the default credentials are ubuntu and ubuntu. You can connect to the node through UART on the BMC if ssh does not want to work.

### Connect to the Turing RK1 through BMC 

On the BMC use microcom to connect 
```
microcom -s 115200 /dev/ttySX
```

Slot number to devnode:
| Slot   | Devnode      |
| ------ | ------------ |
| Slot 1 | `/dev/ttyS2` |
| Slot 2 | `/dev/ttyS1` |
| Slot 3 | `/dev/ttyS4` |
| Slot 4 | `/dev/ttyS5` |

alexellis, 2023-11-03:
That's a nice flashing experience, better than I was expecting having worked with the CM4. So you insert an SD card with an image, then use the BMC over HTTP to flash it to a node's eMMC? Is the serial console over USB a must - or would the HTTP BMC work on its own?

alexellis, 2023-11-03:
@Joshua-Riek Your images look like great community resource. [Those build times look really quite long](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6740498946/job/18323665253). Are you using QEMU for them, or how do you build them? [This is why I ask](https://actuated.dev/blog/native-arm64-for-github-actions).

Joshua-Riek, 2023-11-03:
I cross-compile the bootloader and kernel, but then the rootfs is built with QEMU. I have an army of RK3588 dev boards that would be perfect to set up as GitHub runners to speed things up, but lack the space right now to do so. I need to optimize the build process and figure out how best to proceed. I never thought I would need this much computational power when the project started, but it should only grow from here.

geerlingguy, 2023-11-03:
All my benchmarks a are a few percentage points faster than Rock 5 B, which is interesting.

ThomasKaiser, 2023-11-03:
> All my benchmarks are a few percentage points faster than Rock 5 B, which is interesting.

According to the submitted sbc-bench result the DMC memory governor was set to `performance` (no idea whether that's @Joshua-Riek's intention or caused by running `sbc-bench -r` prior which adjust the DMC memory governor to 'always top speed'). So if you ran Geekbench after `sbc-bench -r` w/o any reboot in between Geekbench was executed with DRAM clocked all the time at 2112 MHz.

When you benchmarked your Rock 5B most probably the software shipped back then defaulted to `dmc_ondemand` memory governor adjusting memory clock dynamically between 528 MHz and 2112 MHz based on CPU utilization. Actual benchmark scores then did depend on the `up_treshold` value, see at the very bottom of:

https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#important-insights-and-suggested-optimisations

The difference between clocking DRAM at either 528 MHz or 2112 MHz on RK3588 depends on the benchmark in question. Some (like the infamous `sysbench` or older benchmarks in general where the whole data fits inside modern CPU's caches) are not affected at all but for example with Geekbench 6 it looks like this (the individual tests are the interesting ones since they show which test depends on memory latency/bandwidth and which rather don't):

https://browser.geekbench.com/v6/cpu/compare/3337274?baseline=3336736

Joshua-Riek, 2023-11-03:
There is a system script to set the CPU / GPU governor to `performance` on boot, to address some performance and micro stutter issues for desktop users. But if the DMC memory governor is set to `performance` by default that is not intentional.

ThomasKaiser, 2023-11-04:
> There is a system script to set the CPU / GPU governor to `performance` on boot

That's important to know for @geerlingguy since measuring idle performance is negatively affected by that. 

With Jeff's use cases for the RK1 in mind I would switch back to `ondemand` if I were him and use `/etc/sysfs.d/tk-optimize-rk3588.conf` from the [aforementioned link](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#important-insights-and-suggested-optimisations) to get both low idle consumption as well as high overall performance since with RK's defaults (no `io_is_busy` and especially `up_threshold = 40`) the CPU clockspeeds won't ramp up quickly enough for mixed workloads (and also a lot of benchmarks).

alexellis, 2023-11-05:
@geerlingguy re this comment:

https://github.com/geerlingguy/sbc-reviews/issues/25#issuecomment-1791634464

I flashed the same firmware from the link you gave to a microSD card, inserted it on the rear, booted and watched the serial console. It just boots normally without the prompt you saw. Did you do anything else like short any headers?

geerlingguy, 2024-01-15:
It looks like the modules are complete! See: https://twitter.com/turingpi/status/1746937693110403303

I checked on my order from August 15 (for 4x 8 GB models plus heatsinks) and so far it still says 'Processing'.

AnEvilPenguin, 2024-01-24:
> It looks like the modules are complete! See: https://twitter.com/turingpi/status/1746937693110403303
> 
> I checked on my order from August 15 (for 4x 8 GB models plus heatsinks) and so far it still says 'Processing'.

Mine turned up last night. Just flashing them now!

geerlingguy, 2024-01-26:
@AnEvilPenguin - how goes testing? My shipment should arrive today! Hoping to re-test everything on a production unit and see how it goes.

AnEvilPenguin, 2024-01-26:
Installation and flashing went smoothly (though I had to reboot each module myself - docs imply that they should handle it themselves).

Heatsinks are nice and quiet (especially compared to some of the Rpi PoE modules I've tried!). Detected all of my nvme drives with no issues (I've had issues there in the past).

Done a basic k3s setup and just need to figure out where I want to take it next.
I've not really pushed them that hard yet, but so far they feel pretty snappy compared to my Rpi 4bs and Jetson Nano.

I can say that I'm pretty pleased with them! Overall a good experience so far! I look forward to hearing your thoughts as well!

nmaas87, 2024-01-30:
Got two 16 GB modules yesterday, looks like they decreased eMMC from 64 GB to 32 GB.

daniel-kukiela, 2024-02-03:
> Got two 16 GB modules yesterday, looks like they decreased eMMC from 64 GB to 32 GB.

All RK1 modules contain a 32GB flash. I checked the photo from the first post just to make sure about this specific unit, and the flash chip is a 32GB model. :)

- Daniel (Turing Machines)

daniel-kukiela, 2024-02-03:
@geerlingguy Could we correct the flash size in your post? The flash size is 32GB. There are 2 places that mention 64GB - the basic information section and the benchmark results section. Thank you! :)

- Daniel (Turing Machines)

geerlingguy, 2024-02-04:
@daniel-kukiela - Done! I hope to re-test with my new production copies this month!

daniel-kukiela, 2024-02-04:
Thank you!

As for the disk speeds, for anyone wondering, the maximum speed should be roughly twice what the benchmark results show right now - there was a software bug that caused the flash chip to not be configured correctly, but recent OS versions have this fixed.

sidick, 2024-02-05:
I got shipping notice for my couple of 32gb ram ones today

geerlingguy, 2024-02-22:
Following up with some testing of the 32GB module (since the 16GB ones I've been testing are earlier revision, and I would like to base my final review on the 32GB ones): https://github.com/geerlingguy/sbc-reviews/issues/38

closed 2024-02-22 by geerlingguy

daniel-kukiela, 2024-02-22:
Do you still consider updating the disk benchmark results here (or removing them)? The firmware version you used there was a pre-release and was not meant to be used for this purpose. The issue with the storage speed has been fixed with the first firmware release and these numbers here might suggest lower performance of the storage in 16GB modules. :)

nyanmisaka, 2024-02-22:
@geerlingguy Just in case you are interested in the H264/H265 **hardware** decoding/encoding performance on RK3588.
Try this [ffmpeg-rockchip](https://github.com/nyanmisaka/ffmpeg-rockchip), which has RKMPP & RKRGA support, and it also has a [Wiki](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki) page. Based on it, Jellyfin [recently added](https://www.cnx-software.com/2024/02/01/jellyfin-rockchip-rk3588-mpp-hardware-acceleration/) hardware transcoding support for RK3588.

geerlingguy, 2024-02-22:
@daniel-kukiela - Do you have a good guide for upgrading the firmware on the 16GB boards? Or is it just `apt upgrade`? I'll re-test tomorrow.

daniel-kukiela, 2024-02-22:
@Joshua-Riek I believe `apt upgrade` should work in this case, right?

Joshua-Riek, 2024-02-22:
Yes, that is correct.

daniel-kukiela, 2024-02-22:
Thank you @Joshua-Riek .
So, yes @geerlingguy, `apt upgrade`. Thank you!

geerlingguy, 2024-02-23:
@daniel-kukiela - Updated the results for eMMC, looks like it is quite improved, thanks :)