geerlingguy/sbc-reviews #26: Orange Pi Compute Module 4

geerlingguy, 2023-11-03:
[image]

## Basic information

  - Board URL (official): http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-CM4-1.html
  - Board purchased from: Amazon
  - Board purchase date: 2023-11-01
  - Board specs (as tested): 4 GB RAM / 32 GB eMMC
  - Board price (as tested): $43.99

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2023-11-03:
Supposedly this board is compatible with CM4 IO boards. So far I have tested the PiTray mini and am getting no boot.

geerlingguy, 2023-11-03:
Daaaang, the wiki has about 6 miles of instructions for flashing the eMMC, and all the screenshots are of RockChip's tool in Chinese: http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_CM4

I'm going to try [How to burn a Linux image to eMMC using the dd command](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_CM4#How_to_burn_a_Linux_image_to_eMMC_using_the_dd_command)...

geerlingguy, 2023-11-03:
Interestingly, you can just flash Orange Pi OS (I have 1.0.0 Bookworm) straight to a microSD card and the device will boot off it, even if the device has eMMC. Looks a lot like Armbian.

I wish the 'boot off microSD even though I have eMMC' trick worked on Raspberry Pi CM4 :P

Default user/pass is `orangepi`.

geerlingguy, 2023-11-03:
Running `sudo apt update` you wind up grabbing sources off `http://repo.huaweicloud.com/debian` (just an interesting observation...).

geerlingguy, 2023-11-03:
I copied the .img file from my Mac over to the microSD card:

```
scp Orangepicm4_1.0.0_debian_bookworm_server_linux5.10.160.img orangepi@10.0.100.247:~/
```

Then I erased the eMMC and wrote the image to it:

```
sudo dd bs=1M if=/dev/zero of=/dev/mmcblk0 count=1000 status=progress
sudo sync
sudo dd bs=1M if=Orangepicm4_1.0.0_debian_bookworm_server_linux5.10.160.img of=/dev/mmcblk0 status=progress
sudo sync
```

TheOnion7, 2023-11-04:
I'm .currently doing the exact same process than you . Very happy to see that you worked on that today!
I'm a bit disappointed, I thought it would be as easy as a RaspPi... but it's definitely not. I bought the exact same model than you; OrangePI CM4 Control Module with the CM4 board, for a total of 133$ CAD. But as of now, can't boot on anything I flash except the weird Android OS that it's obviously made for a touch screen. I bought that device hoping to install OpenVoiceOS on it since my Raspberry PI 4 was so slow and that new module has very great hardware.... but nothing impressive yet.. 

Thanks for you work Jeff.

geerlingguy, 2023-11-05:
Today when I tried popping the Orange Pi CM4 into my little 3D printed CM4 board carrier tray, I noticed the PCB is way too thick—it's the thickest of all the clones, maybe this is an 8-layer PCB?

UnKnoWn-Consortium, 2023-11-07:
@TheOnion7 Given the inferior core mix in the RK3566 (4x Cortex-A55) I dun think it will be faster than a Pi 4B (4x Cortex-A72), not to mention software maturity and optimization of the latter. If performance is your priority you may have better luck with the Orange Pi 5 series. Its RK3588 is an octa-core part (4x Cortex-A76 + 4x Cortex-A55) with a newer P-core architecture plus a 6-TOPS NPU.

TheOnion7, 2023-11-07:
> @TheOnion7 Given the inferior core mix in the RK3566 (4x Cortex-A55) I dun think it will be faster than a Pi 4B (4x Cortex-A72), not to mention software maturity and optimization of the latter. If [...]

Thank You for your replay and all that information. 
I would gladly try it, but for now, I'm very disappointed. It's been several days I'm trying to simply boot from an SD Card on any existing possible OS and... no chance yet. 

I tried probably anything I could try , I'm about to return it and see for another alternative for me. I have a RaspPi 4B, but it's not enough for what I'm trying to do. I'm currently looking at simple mini Intel PCs. Will see. haha

Thanks for the info

RicardoCst, 2023-12-22:
Hi @geerlingguy ,

Can I ask if you have tried the Orange Pi Compute Module 4 in the Radxa Taco?

RicardoCst, 2024-01-04:
@geerlingguy 

I tested it with the Radxa Taco and it works, no more buying expensive boards from Radxa...

toastmanAu, 2024-01-09:
> Running `sudo apt update` you wind up grabbing sources off `http://repo.huaweicloud.com/debian` (just an interesting observation...).

This is somewhat related to your "looks like Armbian" observation. Their image builder is an Armbian fork complete with the exact same default config file for building images. Apt mirror can be set to "China" for Great Firewall of China friendly apt sources. Armbian uses two Chinese universities for their mirrors and Opi use a Chinese commercial cloud storage service. I use both image builders and if you pass "" to the apt mirror it uses your default sources. My resultant images update from ports.ubuntu.com by default instead of Huawei (Ubuntu images ofc).

AlekseyMamontov, 2024-01-23:
Good afternoon, someone managed to assemble a core for this board higher -> 6., because on the manufacturer’s website there is only 5.10 and it’s somehow crooked.

In the main kernel there is not even .dts (of this board) for assembling DT.

AceAsket, 2024-01-29:
> @geerlingguy
> 
> I tested it with the Radxa Taco and it works, no more buying expensive boards from Radxa...

For radxa taco not working nvme

jpiccari, 2024-02-05:
> Good afternoon, someone managed to assemble a core for this board higher -> 6., because on the manufacturer’s website there is only 5.10 and it’s somehow crooked.
> 
> In the main kernel there is not even .dts (of this board) for assembling DT.

I'm kinda working on it as a side project. It is actually very very similar to the Radxa CM3 IO which seems to have great support in current linux kernels. I was able to boot and run for a few seconds with an OS built for the radxa board but the opi cm4 quickly overheats and continuously reboots. I'm working on comparing the .dts on the stock opi cm4 to the .dts for the radxa board. I'll keep you posted but its a nights and weekends (when I feel like it) sort of thing.

AlekseyMamontov, 2024-02-05:
I launched it on kernel 6.8, everything works without problems (you can take the device tree from RK3566 OrangePI 3b and remove unnecessary ones), except for SPI3 for some reason PINCTRL does not switch the pins responsible for SPI to alternative functions (probably problems in the rockchip driver, there is chaos in the code )) )

[image]

AlekseyMamontov, 2024-02-05:
[image]

jpiccari, 2024-02-05:
What is the filename for that DTS? I couldn't find it when browsing the source files :(

AlekseyMamontov, 2024-02-05:
> What is the filename for that DTS? I couldn't find it when browsing the source files :(
https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-6.6-rk35xx/arch/arm64/boot/dts/rockchip/rk3566-orangepi-3b.dts

AlekseyMamontov, 2024-02-05:
Taking into account the old device tree
https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-5.10-rk35xx/arch/arm64/boot/dts/rockchip/rk3566-orangepi-cm4.dts

AlekseyMamontov, 2024-02-05:
> > I launched it on kernel 6.8, everything works without problems (you can take the device tree from RK3566 OrangePI 3b and remove unnecessary ones), except for SPI3 for some reason PINCTRL does not switch the pins responsible for SPI to alternative functions (probably problems in the rockchip driver, there is chaos in the code )) )
> > ![Знімок екрана з 2024-02-05 11-34-48](https://private-user-images.githubusercontent.com/35818127/302243044-272fd673-e922-48dd-b271-d7433a42d027.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcxMjY2NzgsIm5iZiI6MTcwNzEyNjM3OCwicGF0aCI6Ii8zNTgxODEyNy8zMDIyNDMwNDQtMjcyZmQ2NzMtZTkyMi00OGRkLWIyNzEtZDc0MzNhNDJkMDI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA1VDA5NDYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmNWIyOGI4ZTJiNjFkYjE3ZDM3NjFmYThkMjBiMzNjNDM3NDQwZTViZmRhNjMzNzhmMWM2MTA5MjU4MzUzMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.V6boMNlMkdlYMHiAGoTJC7rN22M1vbK1_ynmDUbsQSg)
> 
> Nmve worked?

This was not necessary, I did not check.

It’s clear that Rockchip are assholes and you probably shouldn’t buy devices based on their chips; they stopped supporting their chips a couple of years ago with the Linux kernel.
https://github.com/rockchip-linux/kernel

AlekseyMamontov, 2024-02-05:
Everything is built on the basis
two files.

#include "rk3566.dtsi"
#include "rk356x.dtsi"
[rk3566-orangepi-cm4.zip](https://github.com/geerlingguy/sbc-reviews/files/14171018/rk3566-orangepi-cm4.zip)

AceAsket, 2024-02-06:
> > > I launched it on kernel 6.8, everything works without problems (you can take the device tree from RK3566 OrangePI 3b and remove unnecessary ones), except for SPI3 for some reason PINCTRL does not switch the pins responsible for SPI to alternative functions (probably problems in the rockchip driver, there is chaos in the code )) )
> > > ![Знімок екрана з 2024-02-05 11-34-48](https://private-user-images.githubusercontent.com/35818127/302243044-272fd673-e922-48dd-b271-d7433a42d027.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcxMjY2NzgsIm5iZiI6MTcwNzEyNjM3OCwicGF0aCI6Ii8zNTgxODEyNy8zMDIyNDMwNDQtMjcyZmQ2NzMtZTkyMi00OGRkLWIyNzEtZDc0MzNhNDJkMDI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA1VDA5NDYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmNWIyOGI4ZTJiNjFkYjE3ZDM3NjFmYThkMjBiMzNjNDM3NDQwZTViZmRhNjMzNzhmMWM2MTA5MjU4MzUzMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.V6boMNlMkdlYMHiAGoTJC7rN22M1vbK1_ynmDUbsQSg)
> > 
> > 
> > Nmve worked?
> 
> This was not necessary, I did not check.
> 
> It’s clear that Rockchip are assholes and it’s probably not worth buying devices based on their chips, they stopped supporting their chips in the Linux kernel a couple of years ago. https://github.com/rockchip-linux/kernel

I tested nvme on 5.10 and it works on the native board for cm4 from orange pi, but not on taco

AlekseyMamontov, 2024-02-06:
> I tested nvme on 5.10 and it works on the native board for cm4 from orange pi, but not on taco

Check if this is enabled in the device tree for your device.
Also note that on Github the rockchip branch is different from the main one...
In general, advice to everyone who wants to buy motherboards with a rockchip chip - don’t do it, unless of course you want to waste a lot of time looking for bugs, etc.

AlekseyMamontov, 2024-02-06:
http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_CM4#Linux_SDK.E2.80.94.E2.80.94orangepi-build_instructions

**The branch mentioned here is not the same thing as the branch of the orangepi-build source code, please do not confuse it. This branch is mainly used to distinguish different kernel source code versions.

Currently, the Linux5.10 bsp kernel provided by RK is defined as the legacy branch. If the mainline kernel is supported in the future, a current branch will be added.**

It looks like Rockchip has leaked its RK3566 chips to board makers on the cheap.
They leaked these boards to users))
As a result, you bought new boards with a chip that has been discontinued and is supported by user enthusiasm.
The technical description of the chip is minimally available.

[Rockchip_RK3566_Datasheet_V1.1.pdf](https://github.com/geerlingguy/sbc-reviews/files/14177335/Rockchip_RK3566_Datasheet_V1.1.pdf)

jpiccari, 2024-02-09:
> Everything is built on the basis two files.
> 
> #include "rk3566.dtsi" #include "rk356x.dtsi" [rk3566-orangepi-cm4.zip](https://github.com/geerlingguy/sbc-reviews/files/14171018/rk3566-orangepi-cm4.zip)

I tried it out and it works but drops a lot of support for things like eMMC, SD cards, wifi/bt, etc. I also started working on a dts that works with mainline linux and it works pretty good based on my testing. I've never written dts before so I'm sure there are basic mistakes being made but most things I need are working. The major exception is onboard wifi/bt. The sdio initialize for the wifi chip fails most of the time and randomly works other times. Even when initialization works the drives/firmware don't load properly. I think there is some kind of clock issue going on but I'm not really sure.
[rk3566-orangepi-cm4.dts.txt](https://github.com/geerlingguy/sbc-reviews/files/14221716/rk3566-orangepi-cm4.dts.txt)

AlekseyMamontov, 2024-02-09:
[.config.zip](https://github.com/geerlingguy/sbc-reviews/files/14224790/default.config.zip)
Don’t forget to compile and edit DT without a configured .config - meaningless work.
Because in the tree you can specify any devices, but if there are no drivers in the assembled kernel.
None of this makes sense.))

AlekseyMamontov, 2024-02-09:
wireless_wlan: wireless-wlan {
		compatible = "wlan-platdata";
		rockchip,grf = <&grf>;
		wifi_chip_type = "ap6256";
		WIFI,host_wake_irq = <&gpio0 RK_PD6 GPIO_ACTIVE_HIGH>;
		status = "okay";
	};

	wireless_bluetooth: wireless-bluetooth {
		compatible = "bluetooth-platdata";
		clocks = <&rk809 1>;
		clock-names = "ext_clock";
		uart_rts_gpios = <&gpio2 RK_PB5 GPIO_ACTIVE_LOW>;
		pinctrl-names = "default", "rts_gpio";
		pinctrl-0 = <&uart1m0_rtsn>;
		pinctrl-1 = <&uart1_gpios>;
		BT,reset_gpio    = <&gpio2 RK_PB7 GPIO_ACTIVE_HIGH>;
		BT,wake_gpio     = <&gpio2 RK_PC1 GPIO_ACTIVE_HIGH>;
		BT,wake_host_irq = <&gpio2 RK_PC0 GPIO_ACTIVE_HIGH>;
		status = "okay";
	};

jpiccari, 2024-02-10:
Maybe this is nothing but I noticed on the orangepi provided image the mmc2/sdio (used by wifi/bt module) frequency does not match the requested frequency.

```
orangepi@orange:~$ sudo cat /sys/kernel/debug/mmc2/ios
clock:          150000000 Hz
actual clock:   148500000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    6 (sd uhs SDR104)
signal voltage: 1 (1.80 V)
driver type:    0 (driver type B)
```

Notice the 1.5Mhz difference in actual and advertised clock frequencies. This seemed a bit sus so I tried lowering the max clock frequency in my modified dt against mainline linux and it not gets detected super reliably. However, I noticed now the actual clock frequency is down to about 50Mhz.

```
root@OpenWrt:/# cat /sys/kernel/debug/mmc2/ios
clock:          145000000 Hz
actual clock:   50000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    6 (sd uhs SDR104)
signal voltage: 1 (1.80 V)
driver type:    0 (driver type B)
```

Still can't get wifi/bt to work though. I have the drivers (I think) compiled and I've tried symlinking the firmware to the right location but it just fails to load and returns this no matter what I do. Maybe I'm using the wrong drivers or set the wrong device compatibility? Hard to say but on the orangepi image it is using some android driver which I find a little strange and it advertises as bcm43456 so it seems like that isn't exactly the issue. Maybe next I'll try copying over the firmware from the orangepi image but I'd really like to be able to find a source for it.

```
root@OpenWrt:/# dmesg | grep -iE 'mmc2|sdio|brcm'
[    0.832791] mmc_host mmc2: card is non-removable.
[    0.847037] mmc_host mmc2: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
[    0.975579] mmc_host mmc2: Bus speed (slot 0) = 50000000Hz (slot req 145000000Hz, actual 50000000HZ div = 0)
[    0.995438] mmc2: new ultra high speed SDR104 SDIO card at address 0001
[   12.641734] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[   13.924217] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   14.934260] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

```

AlekseyMamontov, 2024-02-10:
> Maybe this is nothing but I noticed on the orangepi provided image the mmc2/sdio (used by wifi/bt module) frequency does not match the requested frequency.
> 
> ```
You have chosen the path of experimentation))

If you assemble on the board itself

In principle the steps are simple
.
1. Place the .config file in the folder with the kernel sources
2. Run make menuconfig.
Connect additional drivers that you need.
3. Then in the folder arch/arm64/boot/dts
edit the device tree file of your board.
4. run build make -J4
5. Then definitely
make modules_install (when building the kernel for the first time or when making changes to the kernel itself)
6. Then make install
Check the /boot folder
7. Do not forget that you have a U-boot bootloader.
And copy the image of the new kernel Image to the /boot folder from the arch/arm64/boot/ folder
B check the .dtb file in the /boot folder corresponds to what was collected in the arch/arm64/boot/dts folder
then.
Reboot))

**And now you can experiment with assembling the device tree**.

В общем, вы можете изменять параметры, добавлять и удалять устройства из дерева устройств.
А потом запустив просто make -j4 можно его пересобрать (при собранном ядре пересобираются только обновленные файлы, то есть файл дерева устройств).
А потом из папки /arch/arm64/boot/dts/ можно скопировать в папку /boot и перезагрузиться и наблюдать за своими экспериментами))

А если что-то изменить в драйверах (в исходниках), то обновляешь файл образа в папке загрузки

jpiccari, 2024-02-13:
Thanks for the tip but I already know how to compile the kernel of device. :) My goal is slightly different, I don't care to rely on questionable vendor-provided binaries for my bootloader, kernel, and userland. Instead my focus is on replace all code on the board with patches from the official sources. I have that 90% working at this point. I have some more testing to do (specifically nvme ssds and see if I can't get wifi working) but hope to get patches into the official versions of u-boot and linux kernel so that we can have support for all OSes without needing to rely on OrangePi to provide distros.

I'll send an update once I get there.

AlekseyMamontov, 2024-02-13:
Wifi 

wireless_wlan: wireless-wlan {
совместимый = "wlan-platdata";
rockchip,grf = <&grf>;
wifi_chip_type = "ap6256";
WIFI,host_wake_irq = <&gpio0 RK_PD6 GPIO_ACTIVE_HIGH>;
статус = «ОК»;
};

Sfinx, 2024-03-18:
Anybody seeing this too ?

http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=147478

Looks like h/w issue that just ignored by orange

jpiccari, 2024-03-20:
> Anybody seeing this too ?
> 
> http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=147478
> 
> Looks like h/w issue that just ignored by orange

I see something similar. When uart console is connected and power is cycled by usb-c port, the device does not boot and the power light stays sort of half lit. Not sure if related but felt reminiscent. Good luck.

JL2010, 2024-05-07:
@jpiccari, not sure if this helps your efforts, but I did find this device tree `rk3566-orangepi-cm4.dts`: https://github.com/orangepi-xunlong/linux-orangepi/blame/orange-pi-5.10-rk35xx/arch/arm64/boot/dts/rockchip/rk3566-orangepi-cm4.dts

Probably better to start with that rather than trying to modify the orangepi-3b one? [which was mentioned here](https://github.com/geerlingguy/sbc-reviews/issues/26#issuecomment-1926573929)

It's in Orange Pi's [orange-pi-5.10-rk35xx](https://github.com/orangepi-xunlong/linux-orangepi/blame/orange-pi-5.10-rk35xx/arch/arm64/boot/dts/rockchip/rk3566-orangepi-cm4.dts) branch for kernel 5.10 only, don't know why it's not available in any other 5.x or 6.x branch. I'm slowly trying to unravel the world of building (and not just using) embedded linux.

I'm going to take that device tree and see if I can figure out how to build Armbian for the OPi CM4, I think this commit could serve as a guide: https://github.com/armbian/build/commit/05a72b8954e932089d4867b55da0d353dc4ba1ac

AlekseyMamontov, 2024-05-07:
Rockchip, Nvidia (Jetson Nano) are assholes, they release a product, and then a year later they refuse to support their products with old kernels maximum 5.10 no higher...

In short, you shouldn't buy their products... unless you want to spend a lot of time looking for problems...

jpiccari, 2024-05-09:
I’ve seen all the branches. I have everything working correctly (or as correctly as possible) except WiFi/bluetooth and with a dts that could be integrated into the Linux kernel (unlike the hacky ones that have been shared that rely on non-standard vendor drivers). I’ve tested it with the latest kernel releases and it seems mostly stable (although with some odd sdio clock issue). 

you will never get WiFi to work without correct drivers and firmware which I’ve not been able to find (even when copying firmware from the orange pi images). The dts in their repo is obfuscating some driver magic and it is not obvious or time efficient for me to jump into what a correct dts should look like.

if you want I can see if I can get the dts upstreamed into mainline Linux as-is and maybe get some help refining a couple things. But overall it seems more valuable to just go buy a rpi cm4 😬 digikey has them at MSRP last I checked.

AlekseyMamontov, 2024-05-09:
Until your board appears in this list
https://github.com/torvalds/linux/tree/master/arch/arm64/boot/dts/rockchip
it's not worth buying

the Armbian team seems to be trying hard and writing in their newsletters
Most popular Rokchip SoC is getting vendor kernel upgrade to 6.1.y with a focus on delivering a fully functional desktop experience.

luckylinux, 2025-05-11:
Question if anybody tried something similar 😃.

I got quite a few CM4 4GB Lite on the cheap.

But then of course I discovered that the IO/Base Boards are quite expensive 🤣.

The cheapest by far is the Orange Pi CM4 Base Board on Aliexpress:
https://www.aliexpress.com/item/1005005988611020.html

Followed by an official Raspberry Pi CM4 to 4 Adapter Board:
https://www.aliexpress.com/item/1005007408324374.html

Did anybody try to put a Raspberry Pi CM4 Lite into the Orange Pi CM4 Base Board ? Did you manage to get the M.2 Slot (NVMe specifically) working ?

The Orange Pi Website states `upporting access to Orange Pi CM4/Raspberry Pi CM4 core board`:
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-CM4-2.html

But I wonder if it's really 1:1 Mapping and all Features (of both Boards) being supported.

Thanks in advance !

masterlog80, 2025-09-13:
Hello,
I got a couple of those boards for dollars and are pretty nice once comparing perf/W: 1-2W on idle and 3-4W on boot/load (lower than a Raspberry PI). I think the main issue is that images are based on old kernel `5.10.160-rockchip-rk356x` and some modules (e.g. **ceph**, which I planned to use) are missing. Unfortunately there is no Armbian support, so no prebuilt newer images. 
As @AlekseyMamontov had successfully [upgraded]((https://github.com/geerlingguy/sbc-reviews/issues/26#issuecomment-1926573929)) the kernel I followed official instructions [1](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_CM4#Get_the_source_code_of_Linux_sdk) and [2](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_CM4#Compile_the_Linux_kernel) reaching [this](https://github.com/user-attachments/assets/7264c1b5-c462-4579-b942-ddb1dd52f48f) UI.
However, although using *current settings* or manually add the **ceph** module in the [current](https://github.com/user-attachments/assets/73163a18-7dff-435c-bc30-fae184c89548) branch, installing the compiled kernel, and reboot, the board didn't come up. I was just wondering if someone can share the `.deb` files for newer u-boot and Kernel, or know a place to get those.

Regards,

luckylinux, 2025-09-13:
@masterlog80: how is that answering my Question, which was about installing a Raspberry Pi CM4 into a Orange Pi CM4 Base Board ? The Hardware will be mostly of the Raspberry Pi (minus possibly some USB3 Controller etc from the Orange Pi CM4 Base Board).

The Images you linked seem to be a Port of Raspberry Pi OS to the Orange Pi 3B/CM4 instead.

masterlog80, 2025-09-13:
> [@masterlog80](https://github.com/masterlog80): how is that answering my Question, which was about installing a Raspberry Pi CM4 into a Orange Pi CM4 Base Board ? The Hardware will be mostly of the [...]

oh, sorry @luckylinux I have misunderstood your point.... :-(