geerlingguy/sbc-reviews #27: Luckfox Core3566

geerlingguy, 2023-11-10:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.luckfox.com/Core3566-Series
  - Board purchased from: [Waveshare](https://www.waveshare.com/product/luckfox/boards-modules/core3566-1/core3566.htm?sku=24844)
  - Board purchase date: 2023-11-01
  - Board specs (as tested): 2GB Lite, No WiFi
  - Board price (as tested): $23.99 ($6 shipping)

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2023-11-10:
So far I went to the LUCKFOX WIKI, and on the [Downloads](https://wiki.luckfox.com/Core3566/Datasheets) page, I clicked on the Firmware link (which goes to [this Google Drive](https://drive.google.com/drive/folders/1q8zXpt3WzSfGyjJrTL8ObEmExVmXp-Ag). I went into Image, then downloaded `2023-09-13-debian-arm64-HDMI-lite.img` and flashed it to a microSD card with Etcher.

I popped that into my PiTray mini, and got no indication it was going to boot (no Ethernet lights, no onboard red or green LED). I then popped it over to my Compute Module 4 IO Board and the red D1 power LED comes on... but not the green ACT LED. Nor is there any LED blinking for the Ethernet port.

I am wondering if either (a) I have dead hardware, or (b) I need a specialized image to use it with the CM4 IO Board?

Their Chinese wiki leads to a password-protected folder [on Baidu](https://pan.baidu.com/share/init?surl=4u6m-oBmCDpC-wy-5u0kAQ&pwd=cn4z), and seemingly the same or a similar set of images.

geerlingguy, 2023-11-10:
Also exploring elsewhere, it seems like their Linux SDK is using Kernel 4.19 maybe? A bit old, most of the other RK3566 boards use 5.10 (which is old-ish, but not super old).

I am trying the non-`-lite` HDMI image now to see if it fares any better.

It looks like their Wiki uses [SDDiskTool](https://wiki.luckfox.com/Core3566/Core3566-quick-start#4-tf-card-image-burning), and I'm not sure but is that Windows-only? It seems like you wouldn't *need* that to flash an image, but I did notice a little checkbox in their screenshot for "SD Boot". Which I wonder if that indicates SDDiskTool does something funky to set up the boot partitions properly?

NeilHanlon, 2023-11-10:
possibly a similar experience -- https://github.com/willpuckett/smote/blob/6f9e6339e61c5da8b65a63038e332f2684dc22da/posts/sbc.md?plain=1#L142-L150

Akkiesoft, 2023-11-11:
Hi. I also have a Core3566(4GB Lite , No Wi-Fi).

SDDiskTool was needed to write the image to MicroSD.
It seems to write the image to the MicroSD card while it does some tricks, but I haven't analyzed the details yet.

I tested with Debian/Ubuntu images made by  Radxa for ROCK3 Model C, It can boot but MAC address didn't match luckfox's official image. I guess the DeviceTree is partially different.

geerlingguy, 2023-11-11:
Also some efforts to get UEFI working on this little board: https://github.com/jaredmcneill/quartz64_uefi/pull/56

me-peng, 2023-11-14:
The img provided by Luckfox is not actually in address order, but rather uses a file similar to a compressed package, which contains the BIN file (boot file) required for downloading the system and the system that needs to be burned to EMMC. The original operation method provided by RK is also the same

If you need to burn in a Linux environment, you can use the tools provided by RK to burn (this requires downloading the SDK provided by Luckfox and decompressing it in the tools Linux directory)

The PWR LED of Luckfox 3566 outputs an OD signal through the core voltage, without outputting a high level. The ACT LED is configured inside the core, and it cannot be output unless it is started to the core, which is also an OD signal

You can see if the 3.3V current required by the floor is too high, which makes the floor SD card unusable (the 3.3V output current of the core board is very low, and if the 3.3V current needs to be output from the core, it will cause the SD card to be unusable (the CM4 floor uses the 3.3V current of the core), which has been fed back to them. They said that the new board has been repaired)

abeisgoat, 2023-11-23:
For anyone looking into this - I believe I got close to getting this board to boot from an SD which was written from Linux. My steps were...

1. [Follow the steps to download the ancient Ubuntu VM](https://wiki.luckfox.com/Core3566/Core3566-SDK/) and build it.
2. [Use the RockChip wiki info about writing each img at the correct location](https://opensource.rock-chips.com/wiki_Boot_option#Boot_from_SD.2FTF_Card) including the idlboader, u-boot, etc.

Sadly I could not complete this loop as the VM would not build the img files without erroring, but I believe this is close to the process which would be involved.

sorphin, 2023-12-01:
> Sadly I could not complete this loop as the VM would not build the img files without erroring, but I believe this is close to the process which would be involved.

I actually was able to build everything in the VM (not under virtualbox... under UTM, but anyway.. took a DAY).. I have one of these boards too and also saw in one of the amazon reviews someone claimed they grabbed the Orange PI Debian 12 image and threw it right on.. I'd sure like to know how.. heh. I've spent the night tonight digging into this and if i can get the orange pi image broken apart, might try flashing each partition... (I have the EMMC version).. alas none of the rockpi tools THEY included in the SDK (rkdevtool, rkimagemaker, afptool) will touch the orange pi image to try and unpack it, which means orange pi didn't do it this way (which is confirmed more or less by orange pi's github repo that shows how to build their images (but i haven't dug into that real far yet)..

sorphin, 2023-12-01:
> I currently have the 2023-09-13 debian arm64 HDMI lite 006.img on the eMMC. After editing “/etc/apt/sources.list“ I had success in upgrading from Buster to Bullseye.
> 
> Welcome to Debian GNU/Linux 11 (bullseye) (GNU/Linux 4.19.232 aarch64) Linux linaro-alip 4.19.232 #1 SMP Wed Sep 13 10:33:16 CST 2023 aarch64

While that updated the distro, it's still the ancient 4.19 kernel... and doesn't solve the issue of switching to say ubuntu or arch or other OSes like the OP CM4 has available.. It seems that Luckfox has worked "hard' to "secure" this board to how THEY want it to run, but I don't play that way :-)

sorphin, 2023-12-01:
I actually just ate my words.. I just flashed the Orange Pi CM4 image....and it booted successfully....

steps used:
upgrade_tool_v2.3_mac/upgrade_tool DB images/MiniLoaderAll.bin 
rkdeveloptool/rkdeveloptool wl 0 images/Orangepicm4_1.0.2_debian_bullseye_server_linux5.10.160/Orangepicm4_1.0.2_debian_bullseye_server_linux5.10.160.img

resulted in:
Welcome to Orange Pi 1.0.2 Bullseye with Linux 5.10.160-rockchip-rk356x

Al1745, 2023-12-02:
Well that's interesting. 
I was able to flash the orangepi image (lite or desktop) to either a micro sd card or the on board emmc and boot up no problems.
Wifi connectivity was gaind by using a USB wifi dongle.

sorphin, 2023-12-02:
Well, the theory isn't reality... the dtb file from the core board panics the orange pi kernel..  guess that's no surprise..  since i have the source files, i should be able to recompile the device tree for 5.1.0, but now i have to set up a build env for that.. (smartly, orangepi uses docker instead of a massive VM, but still)

sorphin, 2023-12-04:
And.. an update.. After arguing (more like changing the code) with the DTBs and the induced random crashes (I hate device tree stuff).. I have the Core3566 booting not only under Orange Pi Debian and Ubuntu, but the green led is a blinking AND networking is working.

Proof:

`root@orangepicm4:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.94  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::162c:3f45:5220:76f4  prefixlen 64  scopeid 0x20<link>
        ether 00:00:a4:05:d0:bf  txqueuelen 1000  (Ethernet)
        RX packets 90  bytes 8770 (8.7 KB)
        RX errors 0  dropped 14  overruns 0  frame 0
        TX packets 62  bytes 5833 (5.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 41 
`

`root@orangepicm4:~# cat /boot/orangepiEnv.txt
verbosity=4
bootlogo=false
extraargs=cma=128M
overlay_prefix=rk356x
#fdtfile=rockchip/rk3566-orangepi-cm4-new.dtb
fdtfile=rockchip/core3566-hdmi-lp4x-v1-linux-new.dtb
bootargs="earlycon=uart8250,mmio32,0xfe660000 console=ttyFIQ0"
rootdev=UUID="b4cc0980-05cf-4e57-9c71-711060ab194d"
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
`

abeisgoat, 2023-12-11:
> I actually just ate my words.. I just flashed the Orange Pi CM4 image....and it booted successfully....
> 
> steps used: rkdeveloptool/rkdeveloptool db MiniLoaderAll.bin Downloading bootloader succeeded.
> 
> rkdeveloptool/rkdeveloptool wl 0x0 Orangepicm4_1.0.2_debian_bookworm_server_linux5.10.160/Orangepicm4_1.0.2_debian_bookworm_server_linux5.10.160.img

I managed to just download the Orange Pi CM4 Arch image and flash it to an SD using normal Linux img writing utilities and it booted successfully, but as you say - missing ethernet, etc.

sorphin, 2023-12-12:
> I managed to just download the Orange Pi CM4 Arch image and flash it to an SD using normal Linux img writing utilities and it booted successfully, but as you say - missing ethernet, etc.

Right. This was to the emmc. Im still working out some kinks (sometimes it crashes/panics as loading the kernel on boot) and once it gets here, I'll be seeing if the tc358743 in the blikvm will work on it with the device tree I have. I guess once I have the issues sorted out in the coreg* dtb file, I'll put it in my repo and link it. Ethernet pulls 900+ Mbps both directions without issues.

costastf, 2024-02-11:
@sorphin any update on this? Would love to get some help for a working solution.

schmooster, 2024-02-13:
I also have a conundrum trying to get this same board to run a 720x720 square DSI[?] touchscreen - would be very nice to have parity with the rpi-CM4 images I am using (which work in the context). Similar experience with an Ubuntu rk3566 img which had no networking & I halted.

leromarinvit, 2024-04-30:
Hi @sorphin, would you mind sharing your dts? I have a few of these boards lying around and would love to put them to use with a non-ancient kernel, and having something that mostly works as a starting point (even if it's not perfect) would be great.

dan-and, 2024-06-06:
Hey @geerlingguy , 
There are updated images for the Core3566 on your mentioned baidu link from April 2024. What caught my eyes are special NPU versions.

btbxbob, 2024-07-23:
successfully boot armbian 24.04 on core 3566.
methods and the dts/dtb files are uploaded here:
https://forums.luckfox.com/viewtopic.php?t=913