geerlingguy/sbc-reviews #33: SBC Name Here

platima, 2024-01-05:
[image]
 
## Basic information
 
  - Board URL (official): http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-2W.html
  - Board purchased from: https://www.aliexpress.com/item/1005006024572428.html
  - Board purchase date: 2023-11-04
  - Board specs (as tested): 2GB RAM + Expansion Board, 64GB microSD Card (w/ heatsink)
  - Board price (as tested): $37.60 AUD + $13.37 AUD shipping, inc any taxes
 
## Linux/system information
 
```
# output of `neofetch`
        #####           root@orangepizero2w
       #######          -------------------
       ##O#O##          OS: Orange Pi 1.0.0 Bookworm aarch64
       #######          Host: OrangePi Zero2 W
     ###########        Kernel: 6.1.31-sun50iw9
    #############       Uptime: 2 hours, 27 mins
   ###############      Packages: 1360 (dpkg)
   ################     Shell: bash 5.2.15
  #################     Resolution: 1280x720
#####################   CPU: (4) @ 1.512GHz
#####################   Memory: 155MiB / 1986MiB
  #################
 
#output of `uname -a`
Linux orangepizero2w 6.1.31-sun50iw9 #1.0.0 SMP Thu Sep  7 17:39:46 CST 2023 aarch64 GNU/Linux
```
 
## Benchmark results
 
### CPU
 
  - Geekbench 6: 155 single / 435 multi (https://browser.geekbench.com/v6/cpu/4265031
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)
 
### Power
 
  - Idle power draw (at wall): 1.22 W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: 2.20 W
  - During `top500` HPL benchmark: TODO W
 
### Disk
 
#### microSD (Lexar V30 U3 A1 633X 64GB)
 
| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | 26.6 MB/s |
| iozone 1M random read | 23.52 MB/s |
| iozone 1M random write | 22.52 MB/s |
| iozone 4K random read | 10.00 MB/s |
| iozone 4K random write | 4.58 MB/s |
 
 
### Network
 
`iperf3` results:
 
  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down
 
(Be sure to test all interfaces, noting any that are non-functional.)
 
## GPU
 
[... 134 lines cut ...]
     32768 :    0.0 ns          /     0.0 ns
     65536 :    4.3 ns          /     7.2 ns
    131072 :    6.5 ns          /    10.3 ns
    262144 :    7.7 ns          /    11.6 ns
    524288 :    8.3 ns          /    12.2 ns
   1048576 :    9.7 ns          /    13.8 ns
   2097152 :   94.2 ns          /   143.5 ns
   4194304 :  136.6 ns          /   181.5 ns
   8388608 :  157.9 ns          /   193.5 ns
  16777216 :  168.3 ns          /   197.6 ns
  33554432 :  173.3 ns          /   199.2 ns
  67108864 :  175.8 ns          /   199.9 ns
```
</details>
 
## Phoronix Test Suite
 
Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):
 
  - pts/encode-mp3: 48.19 sec
  - pts/x264 4K: 0.78 fps
  - pts/x264 1080p: 3.38 fps
  - pts/phpbench: 86536

_(https://openbenchmarking.org/result/2401052-NE-TEST6962944)_

leon332157, 2024-01-06:
Hello, thanks for the review! I have the OrangePi Zero 3 board 4GB, which has the same SOC has this board. I'd love to see how this board compared to the Zero 3. Perhaps I can help with the review process as well. How are you monitoring the power consumption during this review?

platima, 2024-01-06:
Hey that would be great! I use this little thing https://amzn.to/3vwyDG6 (afilliate link) and you can see how I connect and monitor it in the video https://youtu.be/4sRz4CKpIdk (being published in 11 hours).

I am about to upgrade to a newer model that supports PD though, like https://www.aliexpress.us/item/1005005863861438.html I think.

leon332157, 2024-01-06:
Great, I'll be sure to check out the video.

leon332157, 2024-01-06:
@platima I watched your video, and you mentioned that there is no GPU acceleration out-of-the box. According to [this](https://www.reddit.com/r/OrangePI/comments/17qisxw/orange_pi_zero_2w_gpu_3d_accelerationgpu/) reddit post, it is apparenly very easy to enable Panfrost with mesa. I have tested this on my Zero 3 and seems to work pretty well. 

[image]
[image]

In fact, I'm writing this message on the HW-accelerated chromium and I am happy with the performance. Hopefully this helps!

reticulatingsplines, 2024-01-06:
Hey @platima, if you're in the market for decent one of these usb passthrough meters with PD support, then I wanted to vouch for the TC66 from Ruideng, AKA "RIDEN", at least that's the name they seem to use when marketing stuff in the US (where I'm at). I've been using one for a while and it's been extremely useful. It's reversible (power can be flowing from the female port to the male port or vice-versa) and it can serve as a PD protocol trigger as well (use with caution folks!). I'm guessing that Fnirsi one does both of those things just fine too though.

The main reason I'd opt for the TC66 is that another youtuber, TheHWCave, has two videos uploaded where they extensively test and essentially reverse-engineering the thing, detailing exactly how to hook it up and what modes to use to get the most accurate logging possible. They've done a really great service, given that a lot of these little USB meters seem to come with several buttons and extra ports on the side, but no actual useful documentation about how they're to be used. It's also encouraging that Ruideng's youtube account has left replies in the comments with additional info (importantly, that powering the device through microUSB port on the side will give the most accurate readings of the usb-c power path, as the device won't be drawing from it to power itself).

Again, I don't doubt that the Fnirsi meter is a solid piece of tech as well, they have definitely proven themselves in the "cheap but surprisingly accurate testing gadgets" market and I've heard good things about the FNB58 (which actually comes in a nice enclosure rather than the DIY-style plastic+standoffs that all the rest of these things seem to ship in). Really, it's TheHWCave's feature-length documentary on the the TC66 that makes it the top choice for me, especially given that I've learned these things are pretty easy to use incorrectly, despite seeming to just be plug-n-play.

-----

TC66 on aliexpress: https://www.aliexpress.us/item/2251832781988598.html
You can also find it on Amazon, at least in the US. I'd try to buy it from the company's own Amazon store, which on the US version of Amazon is called "RD". There seem to be a lot of other accounts selling these things and I imagine it'd be pretty easy to make an identical knockoff with a garbage BOM.

TheHWCave's videos:
"Ruideng TC66 / TC66C USB Type-C Tester - Way too good for just USB!"
https://www.youtube.com/watch?v=rOlhibDUJgs

"Everything you always wanted to know about USB Type-C & the TC66/C (but were afraid to ask)"
https://www.youtube.com/watch?v=L70gtJ9hzfQ

reticulatingsplines, 2024-01-06:
Oh, I also forgot maybe the most important part, @TheHWCave also has a repository with their own short-and-sweet PDF documentation and a python script you can use to interface with the meter and log data from it into a CSV, which is a great alternative to using the manufacturer's software (which is useless to me given I don't use windows!)

https://github.com/TheHWcave/TC66

platima, 2024-01-06:
> According to [this](https://www.reddit.com/r/OrangePI/comments/17qisxw/orange_pi_zero_2w_gpu_3d_accelerationgpu/) reddit post, it is apparenly very easy to enable Panfrost with mesa

OH that is excellent, I will definitely have to try this out - thank you kindly! And yeah I was mostly referencing page 31 of the doco, the RaspiOS port doco, and the fact there was epic sheering. Could make for some good "Part 2 Updates" content though! I have no idea why it's not enabled out of the box =/

Again, many thanks!

platima, 2024-01-06:
> I wanted to vouch for the TC66 from Ruideng, AKA "RIDEN"

Excellent, thank you kindly mate! So we stop spamming poor Jeff, I just started https://www.reddit.com/r/AskElectronics/comments/190e1ky/usb_current_meters_w_pd_support/

richardhendricks, 2024-01-10:
Getting two of these soon, one of the 1.5GB and one of the 4GB versions.  Can you time how long it takes to from cold boot to idle prompt? Thanks!

platima, 2024-01-10:
Yo. Measured it [in this video](https://youtu.be/4sRz4CKpIdk?t=1343) (timestamped link), but long story short first boot was about ~2 mins while it sets itself up, then after that boot was 40s to desktop. Would be about 25s to CLI I reckon.

WarHawk8080, 2024-01-19:
Read the temps on the dies
`paste <(cat /sys/class/thermal/thermal_zone*/type) <(cat /sys/class/thermal/thermal_zone*/temp) | column -s $'\t' -t | sed 's/\(.\)..$/.\1°C/'`

el-quique, 2024-01-22:
Hello. Does anyone have a tutorial for connecting a 3.5 LCD RPi. I am missing the configuration and connection of the LCD_RST and LCD_RC pins to the Orange Pi Zero 2W. THX

platima, 2024-05-17:
> Anyone tested the usb speed on the opi ?

Not really the place for it mate, this isn't a forum for the product! I'd recommend a post over at http://www.orangepi.org/orangepibbsen/. Cheers

platima, 2024-05-17:
> Hello. Does anyone have a tutorial for connecting a 3.5 LCD RPi. I am missing the configuration and connection of the LCD_RST and LCD_RC pins to the Orange Pi Zero 2W. THX http://www.lcdwiki.com/3.5inch_RPi_Display

Hey I'd recommend posting over at http://www.orangepi.org/orangepibbsen/. This isn't a support group / discussion forum sorry! Cheers

platima, 2024-05-17:
> Read the temps on the dies
> 
> ```
> paste <(cat /sys/class/thermal/thermal_zone*/type) <(cat /sys/class/thermal/thermal_zone*/temp) | column -s $'\t' -t | sed 's/\(.\)..$/.\1°C/`
> ```
> 
> OOF....mine is running a bit warm!
> 
> ```
> root@orangepizero2w:/sys/class/thermal/thermal_zone0# paste <(cat /sys/class/thermal/thermal_zone*/type) <(cat /sys/class/thermal/thermal_zone*/temp) | column -s $'\t' -t | sed 's/\(.\)..$/.\1°C/'
> cpu_thermal_zone  62.2°C
> ddr_thermal_zone  63.4°C
> gpu_thermal_zone  62.7°C
> ve_thermal_zone   61.5°C
> ```

Not bad for no heatsink though; I'd say anything up to ~75C is fine. Good info though, thank you!