geerlingguy/sbc-reviews #35: Orange Pi Zero 3

leon332157, 2024-01-06:
[image]

## Basic information

  - Board URL (official): http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-3.html
  - Board purchased from: [Amazon US](https://www.amazon.com/gp/product/B0CBTKLNVB/)
  - Board purchase date: 12/28/2023
  - Board specs (as tested): 4GB + Expansion Board, 64GB PNY Elite-X MicroSD, no heatsink
  - Board price (as tested): $38 without tax

## Linux/system information
Official Ubuntu Image
```
orangepi@orangepizero3:~$ neofetch
            .-/+oossssoo+/-.               orangepi@orangepizero3
        `:+ssssssssssssssssss+:`           ----------------------
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.3 LTS aarch64
    .ossssssssssssssssssdMMMNysssso.       Host: OrangePi Zero3
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 6.1.31-sun50iw9
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 14 mins
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 1531 (dpkg)
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 5.1.16
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Terminal: /dev/pts/0
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   CPU: (4) @ 1.512GHz
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   Memory: 451MiB / 3930MiB
+sssshhhyNMMNyssssssssssssyNMMMysssssss+
.ssssssssdMMMNhsssssssssshNMMMdssssssss.
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/
  +sssssssssdmydMMMMMMMMddddyssssssss+
   /ssssssssssshdmNNNNmyNMMMMhssssss/
    .ossssssssssssssssssdMMMNysssso.
      -+sssssssssssssssssyyyssss+-
        `:+ssssssssssssssssss+:`
            .-/+oossssoo+/-.

Linux orangepizero3 6.1.31-sun50iw9 #1.0.2 SMP Thu Sep 28 13:55:18 CST 2023 aarch64 aarch64 aarch64 GNU/Linux
```

## Benchmark results

### CPU

  - Geekbench 6: (155 single/ 457 multi -(https://browser.geekbench.com/v6/cpu/4279497))
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): ~1 W
  - Maximum simulated power draw (`stress-ng --matrix 0`): ~2 W
  - During Geekbench multicore benchmark: ~2W
  - During `top500` HPL benchmark: TODO W

### Disk

#### PNY Elite-X 64GB micro SD

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

### Network

`iperf3` results:

[... 120 lines cut ...]
   4194304 :  135.3 ns          /   178.2 ns
   8388608 :  155.1 ns          /   189.3 ns
  16777216 :  164.3 ns          /   193.2 ns
  33554432 :  169.3 ns          /   196.0 ns
  67108864 :  172.0 ns          /   197.9 ns
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

JuniorJPDJ, 2024-03-04:
Can I please ask you to do iperf3 test? I'm interested in ethernet speed.

leon332157, 2024-03-05:
> Can I please ask you to do iperf3 test? I'm interested in ethernet speed.

Sorry, I no longer have access to the device, but I would imagine it's similar to other h618 devices.