geerlingguy/sbc-reviews #39: Lichee Console 4A

geerlingguy, 2024-03-22:
[image]

## Basic information

  - Board URL (official): [TODO](https://sipeed.com/licheepi4a)
  - Board purchased from: Review sample provided by Sipeed
  - Board purchase date: January, 2024
  - Board specs (as tested): 16 GB RAM / 128 GB eMMC / TH1520 SoC (4x C910, clocked at 1.5 GHz)
  - Board price (as tested): $429

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

  - TODO: Haven't determined standardized benchmark yet. See [Issue #2](https://github.com/geerlingguy/sbc-reviews/issues/2).

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-03-22:
Reading through the [Lichee Console 4A Wiki](https://wiki.sipeed.com/hardware/en/lichee/th1520/lcon4a/lcon4a.html), there are a few important notes:

  - The AliExpress listing shows the SoC running at 1.8 GHz, but the Wiki says "For the sake of Console battery life and heat generation, the current upper limit of the CPU frequency of the default image is limited to 1.5GHz." (under the 'Battery Life' section).
  - The Wiki says "It can be powered by 12V DC 3.5mm interface or 5V USB-C port." — however many reviews mentioned they couldn't get USB-C charging to work.
  - The GPU on the SoC is the "IMG™ B Series BXM-4-64" — I know there was some work towards making the open source driver available in Linux, not sure how far along that is.

geerlingguy, 2024-03-22:
Charging is interesting—it seems like the circuit is somewhat simple, loading up the battery at the beginning and exponentially decaying over time. The unit got quite hot right after I plugged it in, which is a little alarming as I tend to trust these devkit-style devices less when it comes to thermals...

[image]

geerlingguy, 2024-03-23:
As with Milk-V Mars CM (see #22), some of the Phoronix test suites are failing:

```
Phoronix Test Suite v10.8.4

    Installed:     pts/timed-audio-encode-1.0.1
    To Install:    pts/encode-mp3-1.7.4

    Determining File Requirements ..................................................................................
    Searching Download Caches ......................................................................................

    1 Test To Install
        5MB Of Disk Space Is Needed
        5 Seconds Estimated Install Time

    pts/encode-mp3-1.7.4:
        Test Installation 1 of 1
        1 File Needed [1.45 MB / 1 Minute]
        File Found: lame-3.100.tar.gz                                                                       [1.45MB]
        Approximate Install Size: 5 MB
        Estimated Install Time: 5 Seconds
        Installing Test @ 02:15:47
            The installer exited with a non-zero exit status.
            ERROR: cannot guess build type; you must specify one
            LOG: /var/lib/phoronix-test-suite/installed-tests/pts/encode-mp3-1.7.4/install-failed.log
```

Not sure if the test suite has a risc-v option available; it works on x86 and arm64 at least.

geerlingguy, 2024-03-24:
In terms of games:

  - `supertuxkart`: it's competent at `supertuxkart`, getting 30-40 fps at 1024x768 resolution. For some reason I was unable to switch resolutions or play full screen—any time I tried, it quit out of the game. I had to play with no blur or vsync, but it was perfectly playable on the keyboard since it's a keyboard-only game.
  - `dsda-doom`: Classic Doom but with `freedoom` resources, if I try OpenGL rendering, it crashes, but software rendering works a treat. It worked in all resolutions, and plays well—excepting the atrocious RedPoint control; it was excruciating trying to use that as a mouse. Better to plug in an external mouse if you want to frag some enemies.

Zepan, 2024-03-25:
If you want run some software with OpenGL, please try switch first:  sudo switch-gl gl4es
USB-C charging: It not support PD fast chraging, only can be charging with normal 5V2A (this feature is added temporarily, so it is not perfect)
Sometimes (rarely) keyboard not responding, you can try press Pow+Z to restart keyboard.
Temporarily, do not apt upgrade, it will upgrade the kernel and case boot error(upstream is debugging the sleep mode, buggy now...)

geerlingguy, 2024-03-25:
I can confirm if I plug in USB-C power, it seems to be charging the battery off there, pulling around 2.4A at 5V with the Raspberry Pi USB-C charger. I noticed it doesn't seem to charge while the power on the device is off. It lit up the little blue LED and pulled down 2.4A momentarily, but then stopped. It only pulled any power through USB-C while powered on.

The other charger seemed to be able to charge even while the device is powered off.

geerlingguy, 2024-03-25:
With the 12v 2A AC adapter, it puts out 28W while the device is running and charging the battery (from about 70%), and then around 20W while the device is powered off. So it looks like it'll run off the wall wart while charging the battery, adding on that extra 8W or so.

@Zepan - for USB-C, is it supposed to charge the battery while powered off? I know a few other people said "USB-C charging doesn't work", but maybe it works while powered on, but not powered off?

geerlingguy, 2024-03-25:
[image]

I tried installing a SKhynix 2242 256 GB NVMe SSD in the expansion slot on the bottom, but unfortunately (a) it was very difficult to install, as it is fairly deep in the case, and the WiFi antenna cable interfered with the install, and (b) I couldn't get it seated perfectly, as there was a screw pillar inside just barely touching the side of the SSD as I inserted it.

I booted it up anyway, since I could get it about 90% of the way in the slot, but it didn't enumerate on the PCIe bus nor show up in `lsblk`:

```
sipeed@lpi4a12d6:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0      179:0    0 116.5G  0 disk 
├─mmcblk0p1  179:1    0     2M  0 part 
├─mmcblk0p2  179:2    0   500M  0 part /boot
└─mmcblk0p3  179:3    0   116G  0 part /
mmcblk0boot0 179:8    0     4M  1 disk 
mmcblk0boot1 179:16   0     4M  1 disk 
sipeed@lpi4a12d6:~$ lspci
sipeed@lpi4a12d6:~$ sudo lspci -vvvv
```

I then tried to take off the bottom, to see if I could make it fit better without the bottom cover in place, but four of the six bottom cover screws were stripped already. I hadn't attempted to use them before, so this was how they came from the factory—one was completely rounded out, and three were rounded out enough I couldn't get any purchase from any of my precision drivers (I was just barely twisting them, with no pressure, and the driver would just spin).

[image]

I've heard other reviewers have trouble with the soft screw heads too ([example](https://www.cnx-software.com/2024/01/10/lichee-console-4a-portable-risc-v-development-terminal-review-unboxing-teardown-hands-on/)), but mine were pre-stripped out of the box :(

I'll have to see if I can find a screw extractor bit small enough for these tiny screws, or mangle them up with my Dremel maybe :/

geerlingguy, 2024-03-25:
My screw extractor set has bits far too large for these tiny screws, so I bought this [Moody 6-piece set](https://amzn.to/3TQerst) on Amazon, and I'll see if I can get enough purchase to pull out the screws. I'll have to dig around to find replacement screws—these guys are quite small!

geerlingguy, 2024-03-26:
I was testing Bluetooth... and it didn't seem to initialize the Bluetooth adapter. I get an error from `blueman-manager`:

> Connection to BlueZ failed
>
> Bluez daemon is not running, blueman-manager cannot continue.
> This probably means that there were no Bluetooth adapters detected or Bluetooth daemon was not started.

`hcitool dev` returns no Devices, and `hciconfig -a` returns nothing.

I started the bluetooth service (`sudo systemctl start bluetooth`), and that seemed to go okay, with no warnings in `dmesg`, but when I opened Bluetooth Adapters or Bluetooth Manager they'd just briefly start to draw the window then crash out.

geerlingguy, 2024-03-29:
New video: https://www.youtube.com/watch?v=8qDGV6LTOnk

I also got the tiny screw extraction set, so hopefully I can get inside today...