geerlingguy/sbc-reviews #40: Radxa CM5

geerlingguy, 2024-04-02:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://radxa.com/products/computer-on-module/cm/cm5
  - Board purchased from: [ARACE](https://arace.tech/products/radxa-cm5-radxa-cm5-lite?variant=42816188350644)
  - Board purchase date: April 2, 2024
  - Board specs (as tested): CM5 (RK3588S2) / 8GB RAM / 64GB eMMC
  - Board price (as tested): $88

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-04-24:
Nope, still no notification of shipment.

geerlingguy, 2024-05-16:
Received it today, popped it on my Raspberry Pi Compute Module 4 IO Board, and powered it up.

There's a blue LED on the CM5 right next to the board label (CM5 V2.21), and the Ethernet LEDs are lit, with the activity LED blinking, so it's definitely doing _something_. No HDMI output at least on HDMI0.

geerlingguy, 2024-05-16:
I'm downloading [radxa-cm5-io_debian_bullseye-test_kde_b1.img.xz](https://github.com/radxa-build/radxa-cm5-io/releases/download/b1/radxa-cm5-io_debian_bullseye_kde_b1.img.xz) from the [docs / download page](https://docs.radxa.com/en/compute-module/cm5/getting-started/download), and flashing it to a 32GB microSD card with Etcher.

Qvy-png, 2024-05-16:
Is there any jumper on the IO board that could cause a boot issue ? Perhaps it is just like the Milkv mars CM, where it requires a serial connection

geerlingguy, 2024-05-16:
After flashing the microSD card, it is booting. Still no HDMI out of HDMI0 or HDMI1, but I can log in over SSH with `rock`/`rock`, and will begin some testing.

geerlingguy, 2024-05-16:
Geekbench 6 power usage graph:

[image]

geerlingguy, 2024-05-16:
Regarding HDMI output, I've opened [CM5 - HDMI output on Pi CM4 IO Board](https://forum.radxa.com/t/cm5-hdmi-output-on-pi-cm4-io-board/20995) on the Radxa community forum.

geerlingguy, 2024-05-24:
The Radxa CM5 gets a quick mention in [today's video on the LattePanda Mu](https://www.youtube.com/watch?v=GKGtRrElu30).

myaliasisrithm, 2024-06-23:
Have you tested rhe CM5 to see if it works in the Super6c? I'd be willing to upgrade from the pi CM4 modules but it seems iffy that they're a clean drop-on replacement.

AdmShipper, 2024-08-08:
I'd also like to see a test of the radxa cm5 on the Super6c.

pterodyne, 2024-09-09:
correct me if I'm wrong, but dont you need builds that say "with cm 4 io" to use cm4 carriers? My cm5 radxa works fine with a cm4 nano and the cm4io builds.  And yes, I realize this an old thread, but I don't see that solution listed to the no HDMI issue.  These worked for me both Joshua Riek Ubuntu with cm4 io, and the Radxa builds...

michaelarmstrong, 2024-10-02:
Can someone verify if this works in normal CM4 carriers? I have a Super6C and intend on using this. I don't care about display etc.

pterodyne, 2024-10-03:
The two that I have both do (16 and 32GB variants). Just make sure the image you use says "cm4-io" or similar.  The CM5-io versions are only for the official carrier board.  Keep in mind there is a 3rd connector on the Radxa CM5 so you need to make sure there is nothing physically in the way on the carrier.  I had one CM4 board that had some surface mounted components in the location that the 3rd connector was, and it wouldn't fit.  Also if your carrier can't do a SD for each module then you'll need to use a different carrier to install the OS to the EMMC.  Just things to think about there is a tool rkdeveloper (from memory, not exact name) that allows you to write the image.  I was using Joshua Riek's Ubuntu from an SD card in the nano carrier B board, and then used the utility that comes with that build to write the image to the EMMC.

gulassi, 2024-10-05:
> Can someone verify if this works in normal CM4 carriers? I have a Super6C and intend on using this. I don't care about display etc.

I have been running three of these (a 4GB lite model) in Super6C without any major issues. The heatsinks provided with the official case are not 100% compatible, but can be made to work. Meaning that they don't physically conflict with any part of the module, but the heatpads need to be doubled up to get contact on the CPU

michaelarmstrong, 2025-01-04:
Just ordered a 32GB/256GB model to test in my Super6C deployment, will report back on current qwerks but will be hopefully buying more.

nekomatic, 2025-02-24:
Does the **testing complete** mean it works with the Super6C board? I'm about to push the buy button ;)

michaelarmstrong, 2025-02-24:
No idea but I can tell you I installed it in my Super6C and it’s working great. Temps are good too. Be sure to flash the CM4 IO board image. The CM5 one wouldn’t work

michaelarmstrong, 2025-02-24:
That’s the one!

HazMatt69, 2025-03-17:
Would this work on the compute blade?

braghome, 2025-06-07:
Can you add Radxa CM5 test with Sipeed NanoCluster, if at all possible

alpha754293, 2025-09-09:
If you add the columns for RAM (GB), Storage (GB), Price ($, as tested), and then compute $/(GFLOPs/W), it looks like that the Radxa ROCK 5C is the winner at $14.58/(GFLOPs/W).

If you want the most cost efficient 16 GB system, that you're looking at the Turing Machines RK1 at $51.14/(GFLOPs/W).

As my RAM requirements goes up, the $/(GFLOPs/W) gets higher and higher.

eXelcoVert, 2025-10-20:
Hey Jeff appreciate your review on this SBC. I was wondering if you did any test's with this SBC and the Compute Blade.