geerlingguy/sbc-reviews #41: Radxa Rock 5C

geerlingguy, 2024-04-04:
[image]

## Basic information

  - Board URL (official): https://radxa.com/products/rock5/5c/
  - Board purchased from: Arace
  - Board purchase date: April 4, 2024
  - Board specs (as tested): 4GB RAM
  - Board price (as tested): $59.90

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

ThomasKaiser, 2024-04-12:
Radxa put a Rock-5-ITX and a Rock 5C Lite (RK3582 with some SoC cores disabled) in the mail. Interestingly idle consumption with 5C Lite is slightly higher than on my 5B (with same RPi USB-C power brick on same Netio powermeter):

    root@rock-5c:/home/radxa# Netio=192.168.83.72/4 sbc-bench.sh -m
    Power monitoring on socket 4 of powerbox-1 (Netio 4KF, FW v3.2.0, XML API v2.4, 235.23V @ 50.02Hz)
    
    Rockchip RK3582 / 35 82 12 fe 21 41  32 47 41 31 00 00 00 00, Kernel: aarch64, Userland: arm64
    
    CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                     cpufreq   min    max
     CPU    cluster  policy   speed  speed   core type
      0        0        0      408    1800   Cortex-A55 / r2p0
      1        0        0      408    1800   Cortex-A55 / r2p0
      2        0        0      408    1800   Cortex-A55 / r2p0
      3        0        0      408    1800   Cortex-A55 / r2p0
      4        1        4      408    2400   Cortex-A76 / r4p0
      5        1        4      408    2400   Cortex-A76 / r4p0
    
    Thermal source: /sys/devices/virtual/thermal/thermal_zone0/ (soc-thermal)
    
    Time       big.LITTLE   load %cpu %sys %usr %nice %io %irq   Temp      mW
    20:28:01:  408/ 600MHz  0.06   1%   1%   0%   0%   0%   0%  34.2°C     1640
    20:28:06:  408/1800MHz  0.06   2%   1%   0%   0%   0%   0%  35.2°C     1640
    20:28:11:  408/ 600MHz  0.05   0%   0%   0%   0%   0%   0%  35.2°C     1630
    20:28:16:  408/ 600MHz  0.05   1%   1%   0%   0%   0%   0%  35.2°C     1630
    20:28:21:  408/ 600MHz  0.04   1%   1%   0%   0%   0%   0%  34.2°C     1640
    20:28:26:  600/ 600MHz  0.12   2%   1%   0%   0%   0%   0%  35.2°C     1660
    20:28:31:  408/ 600MHz  0.11   1%   0%   0%   0%   0%   0%  35.2°C     1670
    20:28:37:  408/ 600MHz  0.10   1%   1%   0%   0%   0%   0%  35.2°C     1660
    20:28:42:  408/ 600MHz  0.09   2%   1%   0%   0%   0%   0%  35.2°C     1680
    20:28:47:  408/1800MHz  0.08   1%   0%   0%   0%   0%   0%  35.2°C     1690
    20:28:52:  408/1800MHz  0.08   1%   0%   0%   0%   0%   0%  35.2°C     1700
    20:28:57:  408/ 600MHz  0.07   1%   1%   0%   0%   0%   0%  35.2°C     1710
    20:29:02:  408/ 408MHz  0.23   1%   1%   0%   0%   0%   0%  35.2°C     1720
    20:29:07:  408/1800MHz  0.21   1%   1%   0%   0%   0%   0%  35.2°C     1720
    20:29:13:  408/ 600MHz  0.19   1%   1%   0%   0%   0%   0%  35.2°C     1720

And since you seem to have the RK3588S2 variant can you please post NVMEM contents from your board once you start testing?

For my RK3582 it looks like this:

    root@rock-5c:/home/radxa# RK_NVMEM_FILE="$(find /sys/bus/nvmem/devices/rockchip*/* -name nvmem 2>/dev/null | head -n1)"
    root@rock-5c:/home/radxa# hexdump -C <"${RK_NVMEM_FILE}"
    00000000  52 4b 35 82 12 fe 21 41  32 47 41 31 00 00 00 00  |RK5...!A2GA1....|
    00000010  00 00 00 00 18 15 04 07  07 09 20 0d 08 00 04 00  |.......... .....|
    00000020  00 00 00 00 00 00 00 00  07 0a 00 00 00 c1 37 9b  |..............7.|
    00000030  4e e6 07 00 00 a2 06 b3  06 00 00 91 03 00 00 00  |N...............|
    00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

geerlingguy, 2024-05-16:
...and it finally arrived today. I have RS131-D4R26, Radxa ROCK 5C 4GB, and I went to the Getting Started link on the box (https://rock.sh/5c), but can't find any place to download an image for the board. The [downloads](https://radxa.com/products/rock5/5c/#downloads) section of the page just has:

[image]

I've asked over on the [Radxa forums](https://forum.radxa.com/t/announcing-the-rock-5c-power-performance-and-versatility-for-just-30/20429/110?u=geerlingguy) where I can find an image to boot the board and test it. I also don't see anything Armbian-wise for the Rock 5C yet, not sure where to go to get something to boot!

geerlingguy, 2024-05-16:
It's also interesting the packaging says "Radxa Compute Module", and "ROCKPI 5C" whereas the board itself says "Radxa ROCK 5C V1.1" and it seems the official name is Rock 5C. Maybe an older packaging design got through to production, but it did make me do a double take, like I had ordered a weird model or something.

ThomasKaiser, 2024-05-16:
https://github.com/radxa-build/rock-5c/releases/ and https://github.com/armbian/community/releases/tag/24.5.0-trunk.559

geerlingguy, 2024-05-16:
@ThomasKaiser - Thanks! It'd be nice if they linked to those anywhere on the Radxa site / Wiki :)

ThomasKaiser, 2024-05-16:
Well, to arrive at Radxa's correct download locations it's quite a journey through various Github/wiki pages that contain deprecation notices, further links and so on. Also a bit worrying that the device is shipped to customers now and still only images labeled 'for internal testing' are available :)

BTW: I would either choose the t3 build from https://github.com/radxa-build/rock-5c-6_1/releases or the Armbian flavour of your choice. But no idea whether the 6.1 builds work correctly asides headless use (that's all I tried to far with my 5C Lite)

geerlingguy, 2024-05-16:
> Also a bit worrying that the device is shipped to customers now and still only images labeled 'for internal testing' are available.

Radxa seem to still be in the 'ship hardware, then customers help us get software side running' stage. I like their hardware, but it's impossible to recommend them over something like Pi until they get out of that mindset. At minium, a full tested build should be ready before the first board is shipped (and it should be linked from the product page, downloads, wiki, etc.), or they should very prominently mark the board as 'devkit' or 'beta/alpha' (kinda like what Lichee has done with the RISC-V boards).

geerlingguy, 2024-05-16:
@ThomasKaiser - That image download did work, I'm running kernel 6.1.x and things work okay; it seems like there's no GPU acceleration though; things are a little choppy, UI-wise.

HDMI output works fine, though, unlike on CM5 with the CM4 IO Board.

geerlingguy, 2024-05-16:
Just a quick test of the UI, it's a little stuttery (enough to be a little jarring), like when resizing windows or dragging windows around (I'd rather that decoration be turned off if it's not buttery smooth). YouTube plays back okay in 1080p, with a little stuttering at 4K.

When I shut down the board, it actually powers off to < 1W (near 0W) poweroff state.

Pressing the power button, it boots right back up, and consumes around 2W idle (with or without HDMI plugged in).

geerlingguy, 2024-05-16:
Geekbench 6 power draw:

[image]

ginkage, 2024-05-16:
My experience is a bit different. I have full HW acceleration, and I'm successfully using an NVMe hat created for Pi 5...
[image]
But, I'm used to building my own system images and kernels, so I'm really non-representative. I'm the kind of person this board was made for. Happy to try and assist you in achieving the best possible experience with the board (although I'm not affiliated with Radxa in any way, I'm afraid, so I'm just doing some work for free, as you've rightly pointed out).

geerlingguy, 2024-05-16:
@ginkage - What I don't get is how, if you can get that going as an individual in the community, Radxa can't get that going on their own OS image? Where is it even documented? Not sure if maybe Armbian has a better default image for the Rock 5C or if you have a custom concoction of kernel patches?

ginkage, 2024-05-16:
I'm using a custom kernel, yes. And a custom image. And then I put some more magic on top, so I have this board running perfectly just a couple hours after having it delivered...
But, I'm a wizard, I've been preparing in advance for a few weeks, ordering the right accessories and figuring out the kernel quirks, and I had my own custom build scripts for rk3588 family for a while.
That's what I'm saying: I'm non-representative. Radxa works mostly with vanilla Rockchip kernel (which is a bad baseline from the start), and I'm using a slightly more recent Rockchip kernel, with Armbian patches on top, plus some of my bleeding edge findings to boot. I will do my best to try and figure out the minimal set of changes to make these things work out of the box with the next Armbian release (that's roughly two weeks from now), but all in all, it's indeed really tricky, as we all depend on the upstream work from Rockchip, and they're themselves lagging behind heavily.
If it weren't for Rockchip, we'd probably have Vulkan support on rk3588 in Linux a year ago.

geerlingguy, 2024-05-16:
> If it weren't for Rockchip, we'd probably have Vulkan support on rk3588 in Linux a year ago.

😭 why can't we have nice things! :D

Thanks for your work though—is there anywhere I could follow along, like do you keep build logs or a repo with patches or anything like that? It seems like it would be handy if they're not centralized anywhere else.

ginkage, 2024-05-16:
I've only received my 5C a few hours ago, so no real repo yet. :)
The kernel, I keep it at: https://github.com/ginkage/mirrors/
And I build Armbian images from trunk with a bunch of customizations we've gathered up here: https://github.com/StonedEdge/Retro-Lite-CM5/tree/main/armbian — it's for a DIY portable console build, but we're using desktop Armbian as a base, plus just a few custom scripts and blobs to have GPU acceleration working out of the box (much like "amazingfated" builds).
So, with all those custom patches, there's just one extra thing you need to build the image with my kernel (should you want to go with that one; that would make two of us):
```
diff --git a/config/sources/families/rockchip-rk3588.conf b/config/sources/families/rockchip-rk3588.conf
index 25c669c74..cbd3059b3 100644
--- a/config/sources/families/rockchip-rk3588.conf
+++ b/config/sources/families/rockchip-rk3588.conf
@@ -32,8 +32,8 @@ case $BRANCH in
                BOOTDIR='u-boot-rockchip64'
                declare -g KERNEL_MAJOR_MINOR="6.1"    # Major and minor versions of this kernel.
                declare -g -i KERNEL_GIT_CACHE_TTL=120 # 2 minutes; this is a high-traffic repo
-               KERNELSOURCE='https://github.com/armbian/linux-rockchip.git'
-               KERNELBRANCH='branch:rk-6.1-rkr1'
+               KERNELSOURCE='https://github.com/ginkage/mirrors.git'
+               KERNELBRANCH='branch:kernel-6.1'
                KERNELPATCHDIR='rk35xx-vendor-6.1'
                LINUXFAMILY=rk35xx
                ;;
```
One thing I don't have is a readily built and uploaded image you can flash though, sorry about that. And, I'm having difficulties with the wireless module. As in, I made it work, but I had to compile some extra stuff from https://github.com/radxa-pkg/aic8800
There is probably a better way to do all of this, but again, I'm only a few hours in, give it a few days or a couple weeks (look out for Armbian 24.05 release!), and it's going to be way different.

geerlingguy, 2024-05-17:
Using the Raspberry Pi M.2 HAT+ with the Pinedrive 2242 256GB NVMe SSD:

```
0004:41:00.0 Non-Volatile memory controller: Phison Electronics Corporation PS5013 E13 NVMe Controller (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Phison Electronics Corporation PS5013-E13 PCIe3 NVMe Controller (DRAM-less)
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 102
	Region 0: Memory at f4200000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [80] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [d0] MSI-X: Enable+ Count=9 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [e0] MSI: Enable- Count=1/8 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [110 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=220us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Kernel driver in use: nvme
```

It's running at PCIe Gen 2 x1 speed. Benchmarking now and I'll update the post above.

ThomasKaiser, 2024-05-17:
@geerlingguy could you please provide output from `cat /etc/radxa_image_fingerprint` of the Radxa image?

geerlingguy, 2024-05-17:
```
radxa@rock-5c:~$ cat /etc/radxa_image_fingerprint
FINGERPRINT_VERSION='2'
RSDK_BUILD_DATE='Thu, 16 May 2024 10:21:34 +0000'
RSDK_REVISION=''
RSDK_CONFIG='/etc/rsdk/config.yaml'
```

ThomasKaiser, 2024-05-17:
Thank you. So they changed their metadata format. After three weeks in rural France with only MacBooks as ARM thingies around I'm about to get back to the SBC zoo soon so I'll investigate later on my own :)

geerlingguy, 2024-05-24:
The Rock 5c gets a couple mentions in [today's video on the LattePanda Mu](https://www.youtube.com/watch?v=GKGtRrElu30).

cweickhmann, 2024-05-28:
A small question on the side: Did you have a look at using the eMMC on your Rock 5Cs? I've obtained the 5C Lite and I'm sort of lost because I cannot find an SPI loader for it.

geerlingguy, 2024-05-28:
@cweickhmann - I didn't, sorry!

ginkage, 2024-05-28:
Do you really need a special SPI loader? I've simply flashed an Armbian image to my eMMC on the 5C, and it just worked.

cweickhmann, 2024-05-29:
No worries. The wiki says you have to and for the 5A and B there are instructions and images. And so far I've only had a blank screen when starting it up without a working µSD card (same on the Zero 3E I got in parallel, btw).

I'll check if your idea just works, @ginkage . Have you had success with this on other 5s? That would be great and I'll let you know what I got. But I have the hunch that changing the SPI loader is necessary to tell it to boot from eMMC instead of µSD.

cweickhmann, 2024-05-29:
If that is the case, it's unclear to me why the docs describe the process like that. `rkdeveloptool` does more than just flashing data to the eMMC, afaik.

I think this is not a Linux on ARM issue. It's a unclear docs issue. In particular, for whatever reason there are two things, it seems: U-Boot files labelled "... SPI ..." and one labelled "...spl..." (yeah, lower case and actually not i but L).
I'll go through the process and give you an update.

ThomasKaiser, 2024-05-29:
> for whatever reason there are two things, it seems: U-Boot files labelled "... SPI ..." and one labelled "...spl..."

Yeah, that's two different things: [Serial Peripheral Interface (SPI)](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface) (totally irrelevant here since SPI NOR flash is missing on the Rock 5C) and [secondary program loader (SPL)](https://github.com/ARM-software/u-boot/blob/master/doc/README.SPL). On some devices the latter can be accessed via the former (protocol). But not on Rock 5C unless you buy [an adapter and then can't use eMMC any more](https://forum.radxa.com/t/announcing-the-rock-5c-power-performance-and-versatility-for-just-30/20429/122?u=tkaiser).

And of course this is a 'Linux on ARM' issue since documentation over here sucks a lot (or to be more precise this affects the 'Linux on Android e-waste' world where all the SoCs originate from we have on these cheap ARM thingies. Devices relying on those ARM SoCs from vendors who take Linux seriously (NXP, TI, Renesas and a few others) come with proper documentation and are at least threet times as expensive.

cweickhmann, 2024-05-29:
All right. Let's not get into this debate here. Maybe elsewhere ;-)

So, here's my feedback so far:

* My configuration
  * Rock 5C Lite 16GB
  * additional 32GB eMMC module
  * intended: Use the Penta Hat to make it a NAS

Radxa's Wiki is a bit weird* as when you get to [Installing The Operating System](https://docs.radxa.com/en/rock5/rock5c/getting-started/install-os) and select eMMC it guides you through a process where flashing the eMMC is done using `rkdeveloptool` through the OTG USB port (top-right, when facing ports and LAN port is on the right). Without explaining, it asks the user to use rkdeveloptool to flash the secondary program loader and a system image. Turns out, at least for me this doesn't work. rkdeveloptool quits with the famous and likewise not further explained error message "Creating Comm Object failed!".

So, I've done what many others have: Flash a small system on a µSD, put the image file on there, boot into the system and `dd` the desired image onto the eMMC.

Now, it does in fact boot. At least with the Radxa-provided Debian Bullseye.

*) Others may find this choice of words amusing. I know.

cweickhmann, 2024-05-29:
> rkdevtool is useful when you don't have an eMMC to USB adapter, otherwise you can simply flash eMMC directly.

That's what I wanted to do, and in fact: after doing it there were two partitions on the eMMC, but apparently incomplete or faulty.
In any case, thanks for the feedback!

Altirix, 2024-07-13:
Btw rockchip are not fusing off the disabled cores. if you are lucky youll only have some of the things disabled faulty, they can be reenabled with a uboot patch.

Armbian rolling release currently has the patch https://github.com/armbian/build/blob/main/patch/u-boot/legacy/u-boot-radxa-rk35xx/board_rock-5c/reopen_disabled_nodes.patch

I have two Rock 5c Lite 4gb. one only had a bad GPU and the other only a bad encoder core.

you can also check what ends up being reenabled with this script provided by a user on radax forums https://forum.radxa.com/t/rk3582-soc-broken-ip-node-check/21562

its output is in chinese but its:
CPU
GPU
Enc0
Enc1
Dec0
Dec1

@cweickhmann 
@ThomasKaiser 
as you both mentioned having lites and might want to try this.

ThomasKaiser, 2024-07-13:
> as you both mentioned having lites and might want to try this.

My Lite runs with all CPU cores available since months thanks to Jianfeng's `u-boot` patch.

cweickhmann, 2024-11-09:
Hey, little update on the Rock 5C Lite plus Penta SATA HAT using Armbian in case your SATA devices just vanish after a kernel update:

Armbian 2.8.4's `linux-image-vendor-rk35xx` does not detect the SATA controller (JMicron Technology Corp. JMB58x). I had to revert back to 24.8.2.
Both are 6.1.75-vendor-rk35xx Linux kernels. I suspect this is an overlay problem.

If anyone ran into (or will run into) a similar problem, please let me know how you fixed it. For now, I'm just staying on the Armbian 24.8.2 kernel.

LockonS, 2024-11-16:
> Hey, little update on the Rock 5C Lite plus Penta SATA HAT using Armbian in case your SATA devices just vanish after a kernel update:
>
> Armbian 2.8.4's `linux-image-vendor-rk35xx` does not detect the [...]

Just got into possible simliar issues booting system from EMMC with Rock 5C board.
I was running the official debian system downloaded from the site `[ROCK 5C Desktop System Image: Debian12 KDE b1](https://github.com/radxa-build/rock-5c/releases/download/rsdk-b1/rock-5c_bookworm_kde_b1.output.img.xz)`

The board seems not able to recognize the nvme ssd connected by pcie slot. 
The hardware have no problem since when I boot the system from SD card(emmc removed), the board could recognize the nvme ssd on pcie slot.

LockonS, 2024-11-16:
> Hey, little update on the Rock 5C Lite plus Penta SATA HAT using Armbian in case your SATA devices just vanish after a kernel update:
>
> Armbian 2.8.4's `linux-image-vendor-rk35xx` does not detect the [...]

Some updates in the tests.
After I switched from debian based Radxa OS to the ubuntu-rockchip (https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v2.4.0), the issue seems to be solved. 
Both 22.04 and 24.04 worked, the corresponding kernel versions are `5.10.0-1012-rockchip` and `6.1.0-1025-rockchip`.

ddeath, 2024-11-17:
@cweickhmann looks like there is a fix for that in armbian already: https://github.com/armbian/linux-rockchip/commit/74b836e737d3fa07169657c7c64769b9afd9771a

cweickhmann, 2024-12-02:
Thanks! I hope I get around to testing it this weekend.

cweickhmann, 2024-12-07:
Hey, I skipped 24.11.1 (that your link was referring to @ddeath ) and upgraded to 24.11.2, which: Works. :-)

thet0ast3r, 2025-01-02:
hi, is there any data on how much onboard hardware (enc, decoder, gpu, cores) is usually not working on the rock5c's; i.e. if i buy 5 pieces, how many will still have gpu; decoders... etc?

Did they deliberately 'undersell' to be able to move as many boards as possible? 
I'd imagine the chance is pretty high that only 1 thing is broken per board, either 1 cpu core, or the gp, or one encoder or one decoder core. it should never be nothing, i suppose.

p-web-git, 2025-01-14:
I bought a rock 5c lite and depending on the image I run I have different hardware available. I am currently running `Linux rock-5c 6.1.75-vendor-rk35xx` and every works, from the 8 cores CPU, to the GPU to the 2 encoders/decoders. Unfortunately there is some error related with pcie 

When running the edge variant from armbian only the * cores of the CPU show as working.

VoxelTek, 2025-03-18:
If I understand correctly, it looks like the Rock 5C has slower PCIe than the Pi 5. Am I understanding correctly?

Then again, different drives were tested between the Pi 5 and Rock 5C, so that *might* be affecting speeds.

c0m4r, 2025-03-30:
@VoxelTek Yes, Rock 5C can only work with Gen.2 x1 speeds, which means around ~400MB/s tops. For me, a pendrive connected via USB 3.0 actually works faster and also more stable. [Here's my setup](https://gist.github.com/c0m4r/b3fea6342bcf5a1b25b608fc36100d68#PCIe). I would appreciate it if you could do a speed test of downloading a file from the network to an NVMe drive and compare it with downloading to a pendrive or SD card. I wonder if your speed also drops significantly as in my case.

ryderhutchings, 2025-07-08:
Can someone please explain why the ROCK 5C has an eMMC Connector? Is that something people in the SBC-er community use? And was the micro-SD slot not intended to be the main storage source? 🤣

[image]

cweickhmann, 2025-07-08:
Yes, you cab even buy fitting eMMC modules from Radxa. The memory chips used there tend to work way better for recuuring write operations (unlike those on SD cards). You're not forced to use it.
But if you plan to have yout SBC running like a mini server, I strongly recommend not using SDs as the system drive (same goes for USB drives).