geerlingguy/sbc-reviews #42: LattePanda Mu

geerlingguy, 2024-05-03:
[image]

## Basic information

  - Board URL (official): https://www.lattepanda.com/lattepanda-mu
  - Board purchased from: (Provided for review by LattePanda)
  - Board purchase date: 2024-05-02
  - Board specs (as tested): 8G RAM / 64G eMMC
  - Board price (as tested): $139

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | TODO MB/s |
| iozone 1M random read | TODO MB/s |
| iozone 1M random write | TODO MB/s |
| iozone 4K random read | TODO MB/s |
| iozone 4K random write | TODO MB/s |

`curl https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh | sudo bash`

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading. Download the script with `curl -o disk-benchmark.sh [URL_HERE]` and run `sudo DEVICE_UNDER_TEST=/dev/sda DEVICE_MOUNT_PATH=/mnt/sda1 ./disk-benchmark.sh` (assuming the device is `sda`).

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-05-03:
A few notes unboxing and assembling this SBC with it's [Lite Carrier](https://www.dfrobot.com/product-2822.html) ($40) and their official [Aluminum Active Cooler](https://www.dfrobot.com/product-2823.html) ($12):

  - The lite board includes an open-ended PCIe Gen 3 x4 slot, which apparently is only powered when using DC barrel jack power (12V DC), but not using 15V USB-C PD input.
  - The Lite board includes 2x USB 3.0, 2x USB 2.0, Ethernet (TODO — 1 Gbps? 2.5?), HDMI, and 12V barrel plug or 15V USB-C PD power input.
  - The active cooler fits over the rather large (compared to typical SBCs) N100 chip using three spring-loaded screws. I would not crank the screws down 100% tight, as there is ever-so-slight board flex if you attempt that.
  - DFRobot intends to make a higher-RAM variant at some point in the future ([16GB was mentioned](https://www.dfrobot.com/product-2820.html#comment-6434241728)).
  - The official web page for the product lists a few potential future expansion boards: NAS carrier with 10 GbE (4x?) and 8 NVMe M.2 2280 slots, Router Carrier with 4x Ethernet jacks with PoE power output (maybe?), Graphics Carrier with MXM, SXM, OAM, or general PCIe cards, and a Cluster Carrier for multiple Mu boards. The illustrations show extremely basic designs lacking networking, power circuitry, etc. meaning those are likely quick illustrations to show potential future products based on what's stuck for things like the Pi Compute Module 4.

As MKBHD says (paraphrased), "don't buy a product based on future promises" — but is the Mu good enough to stand on its own merits, and could an ecosystem develop around this form factor like it has around the Compute Modules?

geerlingguy, 2024-05-03:
Some other coverage of the board:

  - [ETA Prime video](https://www.youtube.com/watch?v=z3XHWwfqXJg)
  - [CNX Software](https://www.cnx-software.com/2024/04/11/lattepanda-mu-x86-compute-module-intel-processor-n100-cpu/)

geerlingguy, 2024-05-03:
First boot:

  - I booted it up from the eMMC the first time; it rebooted three times and in the middle boot mentioned there was a CMOS checksum error (no doubt because there was no CMOS battery, it came with one I installed after setting up the board).
  - It booted into Windows 11 Home (no license attached)
  - In Windows: playing back 4K video on YouTube, the Intel UHD Graphics iGPU goes up to 30-35% utilization, and CPU maxes out at 100% utilization, but only a couple frames are dropped overall, no real issues there.
  - In Windows: 8 GB creates a lot of memory pressure. Just one browser tab opened to YouTube pumped up utilization to 5 GB. Two more browser tabs (CNN and Microsoft), and we hit 6 GB!
  - I flashed Ubuntu 22.04 to a USB drive, and installed a Kioxia BG4 2230-size M.2 NVMe SSD in the M.2 M-key slot, and will attempt installing Ubuntu on the LattePanda Mu

geerlingguy, 2024-05-03:
Ubuntu installation / notes:

  - With the USB drive plugged in, a boot option selection menu popped up on the next power on, and I could select the UEFI-based installer, then went into Ubuntu's installer.
  - Ubuntu's installer didn't see the Kioxia BG4, so I popped out of the Installer and launched Terminal.
  - I pulled the BG4 and installed a WD SN520 instead.

geerlingguy, 2024-05-03:
Ubuntu installation:

  - I decided to just overwrite the Windows installation on the 64 GB built-in eMMC.
  - Installation was simple and straightforward once I went with the eMMC.
  - Power usage fluctuated between 9-22W (with some peaks hitting 25W), in both Windows and Ubuntu—the BIOS seemed to have all turbos enabled, so no chip limits in place.

[image]

geerlingguy, 2024-05-03:
Power consumption during Geekbench 6 run:

[image]

geerlingguy, 2024-05-05:
Pibenchmarks.com test results:

```
     Category                  Test                      Result      
HDParm                    Disk Read                 291.80 MB/sec            
HDParm                    Cached Disk Read          281.63 MB/sec            
DD                        Disk Write                103 MB/s                 
FIO                       4k random read            26323 IOPS (105295 KB/s) 
FIO                       4k random write           21222 IOPS (84891 KB/s)  
IOZone                    4k read                   54314 KB/s               
IOZone                    4k write                  44254 KB/s               
IOZone                    4k random read            31812 KB/s               
IOZone                    4k random write           44036 KB/s               

                          Score: 10891                                       
```

I don't see them yet on https://pibenchmarks.com/user/geerlingguy/ though.

ThomasKaiser, 2024-05-05:
Interestingly the device shares the 'product name' `ADL-N` with this [Weibu Mini PC](https://www.cnx-software.com/2023/03/18/intel-core-i3-n305-mini-pc-review-windows-11-ubuntu-22-04/).

geerlingguy, 2024-05-05:
Power consumption while off (after I did a `sudo shutdown now` and waited a few hours):

[image]

UnKnoWn-Consortium, 2024-05-07:
Based on what I read from their schematics and Discord plus my tinkering.

> * The Lite board includes 2x USB 3.0, 2x USB 2.0, Ethernet (TODO — 1 Gbps? 2.5?), HDMI, and 12V barrel plug or 15V USB-C PD power input.

The Ethernet on the lite carrier is provided by a good old gigabit Realtek RTL8111H. 

> * DFRobot intends to make a higher-RAM variant at some point in the future (16GB was mentioned).

From the "manual" included, DFRobot also has a 24GB variant listed. 

> * I pulled the BG4 and installed a WD SN520 instead.
...and BIOS didn't see that either.

If you are using the lite carrier, that M.2 M-key slot only has SATA signal wired.

geerlingguy, 2024-05-07:
> If you are using the lite carrier, that M.2 M-key slot only has SATA signal wired.

Oh! That's a bit crazy, I don't think I've even seen a 2230-sized SATA M.2 drive. But I guess they must exist...

UnKnoWn-Consortium, 2024-05-08:
@geerlingguy Sorry my fault I stand corrected. I dig out their lite carrier schematic and confirm again because 2230 SATA SSD doesn't make a lot of sense... (I have seen 2242 "NGFF" SATA SSD in the late 2010s but never 2230)

The M-key slot is indeed connected to a **PCIe x1 signal**. 

[image]
[image]

geerlingguy, 2024-05-24:
Video covering the Mu: https://www.youtube.com/watch?v=GKGtRrElu30 (and [blog post](https://www.jeffgeerling.com/blog/2024/lattepanda-mu-crams-x86-pc-som-form-factor)).

kesavvvvv, 2024-06-07:
Were you able to get the SSD working finally? Don't think you went over that in the video. The M-key is rated for PCI-E x1. 

The standards for m.2 are 

M-key is for PCIe x4 / NVMe x4 or SATA.

B-key is for PCIe x2 / NVMe x2 or SATA.

E-key is for wifi like you said.

So if i understand this right, an m.2 SATA or nvme SSD should'nt work in that slot.

widom8ker, 2024-08-23:
I have the same issue but with an older latte panda

preetam1409, 2025-05-16:
> I received an email back from LattePanda saying:
> 
> > The issue likely stems from some unknown factors causing confusion in the BIOS settings area. Numerous options are in an uncertain state, leading to abnormal operation. Specifically, for PCIe interfaces, this results in a complete halt of the PCIe REFCLK, preventing PCIe devices from functioning properly.
> 
> The solution they sent across was:
> 
> 1. Download the [correct version of the BIOS branch](https://github.com/LattePandaTeam/LattePanda-Mu/tree/main/Softwares/BIOS) and reflash it. Set the [AFUWIN v5.16 software](https://www.ami.com/bios-uefi-utilities/) as shown in the image below.
> 2. After the reflash is complete, shut down the computer, unplug the power cord, and remove the RTC battery. Wait 30 seconds to ensure the RTC is completely powered down and the BIOS settings are reset to default.
> 3. Reinstall the RTC battery, plug in the power, and start the computer normally.
> 
> [image]

how can I get the software afuwingui software ?

RobbyRed98, 2026-02-08:
Hey @geerlingguy, I think there's a slight mistake in the [results.yaml](https://github.com/geerlingguy/sbc-reviews/blob/d0dfab32ebe25fd5c21f627b564fdb655d9b8fc8/web/data/results.yml#L832) reagarding the LattePanda Mu. Since it's powered by an Intel N100, shouldn't the architecture be listed as x86_64 instead of aarch64?

geerlingguy, 2026-02-09:
@RobbyRed98 - Yes indeed, oops! Would you like to submit a PR for that fix or I can fix it otherwise.

RobbyRed98, 2026-02-09:
@geerlingguy – I'd be happy to! I opened the PR here: https://github.com/geerlingguy/sbc-reviews/pull/101