geerlingguy/sbc-reviews #43: Raspberry Pi Zero 2W

geerlingguy, 2024-05-16:
[image]

## Basic information

  - Board URL (official): https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/
  - Board purchased from: Micro Center
  - Board purchase date: October 2021
  - Board specs (as tested): 512MB RAM
  - Board price (as tested): $15

## Linux/system information

```
# output of `neofetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 --reverse -c $SERVER_IP`: TODO Mbps
  - `iperf3 --bidir -c $SERVER_IP`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-05-16:
I realize I set up this repository after the Zero 2W came out, and I don't have full numbers for it.

geerlingguy, 2024-05-17:
Heh, the lack of RAM really kills a few applications, like the video transcode seems to have just locked up after a few minutes:

```
x264 2022-02-22:
    pts/x264-2.7.0 [Video Input: Bosphorus 4K]
    Test 1 of 2
    Estimated Trial Run Count:    3                      
    Estimated Test Run-Time:      10 Minutes             
    Estimated Time To Completion: 20 Minutes [23:25 UTC] 
        Started Run 1 @ 23:06:03
```

I've let it run a few hours now and I can't SSH in from another terminal either.

geerlingguy, 2024-05-17:
After running Geekbench 6 for a couple hours, it eventually hung, seemingly out of memory:

[image]

bobqianic, 2024-05-30:
BTW, how did you manage to get the Pi Zero 2W connected through USB Ethernet? Any guidance here? I saw many tutorials online, but they are very old and don't work anymore.

geerlingguy, 2024-06-05:
I have a rather ancient micro USB to Ethernet 10/100 Mbps adapter that I just plugged in, and it worked. Didn't have to do anything special. I can't find where I even bought it from back in the day!