geerlingguy/sbc-reviews #45: Figure out replacement for neofetch

geerlingguy, 2024-05-24:
Currently I use neofetch to grab a quick friendly-looking overview of the hardware for each system:

https://github.com/geerlingguy/sbc-reviews/blob/master/.github/ISSUE_TEMPLATE/sbc-board.md#linuxsystem-information

Neofetch is not maintained, however, so I should pick an alternative before it starts breaking on newer systems.

I like the idea of fastfetch, but it's not just an `apt install` away. I like a tool I can pop on quickly, even over SSH, and copying and pasting `wget` commands and changing version numbers for `.deb` releases is annoying.

Is there another tool that's in the Debian repos (and ideally somewhat universal, also easy to install on Ubuntu, Fedora, etc.) that replaces `neofetch`'s functionality?

Atropa-Solanaceae, 2024-05-24:
I've since shifted to employing `screenfetch` to my systems:
https://github.com/KittyKatt/screenFetch

It has an apt repo for debian, dnf for fedora, aur for arch and so on and so forth, all listed under the install tab on their wiki:
https://github.com/KittyKatt/screenFetch/wiki/Installation

It looks pretty snazzy too! Very similar to neofetch and very customizable -- and still maintained.

geerlingguy, 2024-05-24:
Indeed, not hard to install and use at all!

[image]

geerlingguy, 2024-05-24:
There we go!

closed 2024-05-24 by geerlingguy