geerlingguy/sbc-reviews #46: Milk-V Mars

geerlingguy, 2024-07-01:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://milkv.io/mars
  - Board purchased from: Provided by Milk-V, available at [Arace](https://arace.tech/products/milk-v-mars?variant=42315817025716)
  - Board purchase date: June 26, 2024
  - Board specs (as tested): MV050-D8 (8GB RAM)
  - Board price (as tested): $68.99

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-07-01:
One quick note: There is an [official Debian image](https://milkv.io/docs/mars/getting-started/images#official-image), but also Ubuntu 24.04 Server supports the Mars, but you have to update the bootloader via SPI first, if you want to use that.

geerlingguy, 2024-07-02:
I noticed the default system partition `/dev/mmcblk1p4` was only 4GB, and it ran out of space trying to install `git` to clone the `top500-benchmark`. So I ran through the [expanding the boot partition guide](https://milkv.io/docs/mars/getting-started/extendpartition) helpfully published on the Milk-V docs site. It would be nice if that were run automatically on first boot like on the Pi and many other SBCs, but it's at least nice they include a guide :)

geerlingguy, 2024-07-02:
Oops... trying to run some of the pts benchmarks, I run out of free space :(

```
    pts/phpbench-1.1.6:
        Test Installation 1 of 1
            There is not enough space at /var/lib/phoronix-test-suite/installed-tests/pts/phpbench-1.1.6/ for the test files.

Lock creation failed due to lack of disk space.
```

geerlingguy, 2024-07-02:
And after finding [this tweet from @mediocreDevops](https://x.com/mediocreDevops/status/1808144692753686817), I ran through the gauntlet of getting Ansible installed finally. I've documented it here: [Installing Ansible on a RISC-V computer](https://www.jeffgeerling.com/blog/2024/installing-ansible-on-risc-v-computer).

geerlingguy, 2024-07-02:
I couldn't get the disk benchmarks to finish on the microSD card I had installed. Going to re-test with a larger potentially faster card...

geerlingguy, 2024-07-02:
Another oddity: I can shut down the Mars using `sudo shutdown now`, but then to turn it back on, the button I *assume* is a power button doesn't do anything. Tried pressing it, holding for 3s, 5s, and 10s, but the only way to power back on is to unplug and re-plug the power supply.

geerlingguy, 2024-07-03:
`lspci` isn't included with the default packages:

```
user@milkv:~$ lspci
-bash: lspci: command not found
```

Fixed with `sudo apt install -y pciutils`, and now I see this with an Inland 256 GB M.2 SSD plugged in using an [A+E Key to M-key adapter](https://amzn.to/4bCxSek):

```
user@milkv:~$ lspci
0000:00:00.0 PCI bridge: PLDA XpressRich-AXI Ref Design (rev 02)
0000:01:00.0 USB controller: VIA Technologies, Inc. VL805/806 xHCI USB 3.0 Controller (rev 01)
0001:00:00.0 PCI bridge: PLDA XpressRich-AXI Ref Design (rev 02)
0001:01:00.0 Non-Volatile memory controller: Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller (rev 03)

user@milkv:~$ sudo lspci -vvv
...
0001:01:00.0 Non-Volatile memory controller: Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller (rev 03) (prog-if 02 [NVM Express])
	Subsystem: Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 56
	Region 0: Memory at 38000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <8us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable+ Count=16 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00002100
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [158 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [178 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [180 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: nvme
```

geerlingguy, 2024-07-03:
For some reason with an NVMe SSD, I can't get random write speeds to register with iozone. Very weird.

I've also notice in regular use, I'll randomly see:

```
user@milkv:~$ 
Broadcast message from Debian-gdm@milkv on tty1 (Wed 2024-07-03 15:14:31 UTC):

The system will suspend now!
```

This has happened during benchmarks, and sometimes when I'm doing nothing at all, like right now.

geerlingguy, 2024-07-03:
Ah, that looks like it's related to the image's hibernation settings for the desktop OS, see https://forums.debian.net/viewtopic.php?t=156005

eshattow, 2024-07-06:
If you can tolerate not having proprietary drivers (graphics monitor output) then please see https://github.com/geerlingguy/sbc-reviews/issues/22#issuecomment-1908676618 for a guide that applies to VisionFive2/Star64/Mars/MarsCM with upstream U-Boot firmware and Linux kernel with an example of official Debian GNU/Linux installation.

eshattow, 2025-12-27:
> [@geerlingguy](https://github.com/geerlingguy) - I had previous posted on [issue#2](https://github.com/geerlingguy/sbc-reviews/issues/2#issuecomment-2662307270) regarding Milk-V Mars GPU benchmarks. I was curious if an external GPU can be added to this board.

If you can make/find an adapter, then any graphics card with RX580 (similar also working are RX550 or RX590) should be plug and play no problems on Milk-V Mars with updated U-Boot and authentic Debian 13 Trixie distro: https://wiki.debian.org/InstallingDebianOn/StarFive/VisionFiveV2

For the time and cost of a PCIe format adapter and so on you could probably source a Pine64 Star64 which is based off the same reference design as the Milk-V Mars is based from as well, and also features a PCIe connector.

There is some work being reviewed upstream for the integrated Imagination graphics however it is initially targeting the T-Head 1520 SoC: [PATCH v4 Verisilicon DC8200 driver and adaption to TH1520](https://lore.kernel.org/lkml/20251224161205.1132149-1-zhengxingda@iscas.ac.cn/)

Anyways the desktop Linux performance on in-order StarFive JH-7110 will not be fast enough for modern web browser or other typical use. It is possible to get some development work done. See the Pine64 PineTab-V for comparable usage with having the same StarFive JH-7110 SoC and being dependent on integrated graphics.

Also note that (as of year end 2025) there's not enough upstream support yet for any of the Eswin 77-series SoC with this similar Imagination graphics IP core to be as usable as the Milk-V Mars. I comment that the SpacemiT K3 is probably the most exciting SoC expected this next year that may be available to consumers as a lot of it is already being reviewed for upstream before there's even any hardware being sold (we have DeepComputing's Yuning to thank for encouraging these efforts).

So, yeah the situation with StarFive JH-7110 is not changed much, it remains the best choice for a mostly-upstream RISC-V application SoC. There's even a new product with the leftover silicon at a lower clock rating, the StarFive VisionFive 2 Lite board. Milk-V apparently discontinued the Mars CM and Mars product lines even though there was some suggestion that there would be another batch of Mars CM...  anyhow if you are looking for examples of adapting full-size PCIe devices to Milk-V Mars it has been done before with the VisionFive 2 and I think more recently with the VisionFive 2 Lite.