geerlingguy/sbc-reviews #47: Milk-V Jupiter

geerlingguy, 2024-07-18:
[image]

## Basic information

  - Board URL (official): https://milkv.io/jupiter
  - Board purchased from: (Provided by Milk-V)
  - Board purchase date: July 17, 2024
  - Board specs (as tested): M1 @ 1.8GHz, 16GB RAM
  - Board price (as tested): $115 (pre-order pricing)

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-07-18:
After speaking with Milk-V about some of their RISC-V boards, they wanted to send over a Jupiter for me to test. I plan on running it through my usual tests, and also seeing how the board behaves as a regular ol' PC ITX board, testing front panel connectors, USB 3.0 and other IO. It also comes with a heatsink/fan (a pretty small one, mind you), and an IO shield. I haven't unboxed it fully but will hopefully do that tomorrow or Monday!

geerlingguy, 2024-07-19:
First test round will be with https://github.com/milkv-jupiter/jupiter-ubuntu-build/releases - Ubuntu 23.04 desktop release 1.0.8. I will flash it to a microSD card first.

geerlingguy, 2024-07-19:
Apparently there's some amount of standardization in the world... got this from Milk-V:

[image]

geerlingguy, 2024-07-19:
I can boot and log in via serial console; it also has an Ethernet connection, at 1 Gbps, and I can access it via UART or SSH, but so far I don't have an HDMI output working. `xrandr` gives me 'Couldn't open display'. I will try another HDMI monitor and see if that works. For some reason a lot of RISC-V boards don't seem to like my HP EliteDisplay E231.

geerlingguy, 2024-07-19:
The docs for the Jupiter describe using `gparted` to expand the SD partition, but I elected to use [the RVSpace guide Extend Partition on SD Card or eMMC](https://doc-en.rvspace.org/VisionFive2/Quick_Start_Guide/VisionFive2_QSG/extend_partition.html).

geerlingguy, 2024-07-19:
I was originally wondering if the fan or fan connector was dead, but no—just the heatsink seems to be good enough for 99% of the time—the fan will only spin up deep into a multicore test, and briefly at that. It's noticeable but not annoying.

geerlingguy, 2024-07-19:
I'm able to get the HDMI output to feed into OBS through my CamLink 4K, so it's probably an incompatibility with that HP monitor:

[image]

geerlingguy, 2024-07-22:
I just noticed there's a newer release of the 23.10 desktop: https://github.com/milkv-jupiter/jupiter-ubuntu-build/releases/tag/v1.0.9

(As well as [Bianbu](https://github.com/milkv-jupiter/jupiter-bianbu-build/releases/tag/v1.0.9)). Fedora also has a RISC-V task force building images for the Jupiter—see the downloadable images here: https://mirror.iscas.ac.cn/fedora-riscv/dl/Milk-V/Jupiter/images/latest/

geerlingguy, 2024-07-22:
Notes on Bianbu OS minimal install:

  - It truly is minimal, doesn't include utilities like `nmcli`, `openssh-server`, `lsusb`, or `lspci` (have to `apt install network-manager rfkill openssh-server usbutils pciutils` etc.
  - The premade image on GitHub is set to Chinese locale by default, so run `sudo dpkg-reconfigure locales` and select your language.

As with the Ubuntu image, I'm not seeing the WiFi device, just the two Ethernet ports and a `sit0` interface (something to do with IPv6 I think):

```
root@milkv-jupiter:~# nmcli d
DEVICE  TYPE      STATE      CONNECTION 
end0    ethernet  unmanaged  --         
end1    ethernet  unmanaged  --         
sit0    iptunnel  unmanaged  --         
lo      loopback  unmanaged  --   
```

geerlingguy, 2024-07-22:
I'm also re-running Geekbench 6 under the Bianbu minimal install to see if there's any difference between it and the Ubuntu release.

geerlingguy, 2024-07-22:
I've also asked on the Community forum about the [`SATAPWR1` and `2` headers](https://community.milkv.io/t/satapwr1-and-satapwr2-headers/2504), which look like 4-pin ATX fan connectors, but I'm scared to plug in a fan to one of them in case the pinout is incorrect :)

It would be nice to provide a couple fan headers so I can run a case fan off the ITX board, but I'm not sure if those are wired up to any PWM controls.

geerlingguy, 2024-07-22:
Fedora 40 with Gnome runs stable, if slow. No GPU drivers present in that build, it seems—everything goes to the CPU and runs a bit slowish. YouTube can kinda play back at 240p, whereas on Bianbu and Ubuntu I can hit 720p passably.

[image]

geerlingguy, 2024-07-22:
I don't seem to be getting any GPU acceleration under Ubuntu or Bianbu, at least not by default... it seems like they should both be able to render a lot better using the built-in GPU (see [Banana Pi BPI-F3 here](https://www.youtube.com/watch?v=cT9ZNAIPgBY)).

geerlingguy, 2024-07-22:
Kioxia XG8 over PCIe:

```
0001:01:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller XG8 (rev 01) (prog-if 02 [NVM Express])
	Subsystem: KIOXIA Corporation Device 0001
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 56
	Region 0: Memory at 90200000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI: Enable- Count=1/32 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [b0] MSI-X: Enable+ Count=65 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [c0] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd- ExtTag+ PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 16GT/s, Width x4, ASPM L1, Exit Latency L1 <32us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
		LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [150 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [260 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [360 v1] Data Link Feature <?>
	Capabilities: [380 v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [400 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
			  PortCommonModeRestoreTime=60us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [470 v1] Lane Margining at the Receiver <?>
	Kernel driver in use: nvme
```

geerlingguy, 2024-07-22:
Testing my 32GB eMMC module that I used before with Rock 5 B:

```
root@milkv-jupiter:~# dmesg | grep mmc2
[    4.279458] mmc2: new HS400 Enhanced strobe MMC card at address 0001
[    4.286636] mmcblk2: mmc2:0001 DA4032 29.1 GiB 
[    4.295504] mmcblk2boot0: mmc2:0001 DA4032 4.00 MiB 
[    4.301771] mmcblk2boot1: mmc2:0001 DA4032 4.00 MiB 
[    4.307801] mmcblk2rpmb: mmc2:0001 DA4032 16.0 MiB, chardev (242:0)
[    4.314220] mmc2: SDHCI controller on d4281000.sdh [d4281000.sdh] using ADMA
```

Adding the results for that and NVMe up in the OP.

geerlingguy, 2024-07-22:
I also tested Docker on this board, since that's something called out in the Jupiter's documentation. Installing was easy (install `docker.io` and that's it!). The `hello-world` image runs successfully, and there are a few hundred risc-v supporting images on Docker Hub already. I was able to run Debian and Alpine easy enough, though `httpd` (Apache) doesn't have a riscv64 build yet.

For many images, like [Redis](https://hub.docker.com/_/redis), they say `riscv64` is in the supported architectures, but that's not on the main image. You have to instead use [riscv64/redis](https://hub.docker.com/r/riscv64/redis/), and there's no `latest` tag on that repo, so you have to dig around and find a tag you like, for example:

```
root@milkv-jupiter:~# docker run --name some-redis -d redis
Unable to find image 'redis:latest' locally
latest: Pulling from library/redis
docker: no matching manifest for linux/riscv64 in the manifest list entries.
See 'docker run --help'.
root@milkv-jupiter:~# docker run --name some-redis -d riscv64/redis
Unable to find image 'riscv64/redis:latest' locally
docker: Error response from daemon: manifest for riscv64/redis:latest not found: manifest unknown: manifest unknown.
See 'docker run --help'.
root@milkv-jupiter:~# docker run --name some-redis -d riscv64/redis:7.4-rc-alpine3.20
Unable to find image 'riscv64/redis:7.4-rc-alpine3.20' locally
7.4-rc-alpine3.20: Pulling from riscv64/redis
d4714cc4c8bb: Pull complete 
5923aa7cbf3a: Pull complete 
5431027fcd2f: Pull complete 
e01b77e4c90b: Pull complete 
1789fa2a0e8f: Pull complete 
2d89db90b7dd: Pull complete 
4f4fb700ef54: Pull complete 
f90f4ee3aef7: Pull complete 
Digest: sha256:03d38d93104eb9f10aaa01b63cf1a7f2f18f7389103cf097b4d51f3e036dd5fd
Status: Downloaded newer image for riscv64/redis:7.4-rc-alpine3.20
502e8abbd2f57af0081e0441108741468588dc9906acb995fc5a3bd702582da5
root@milkv-jupiter:~# docker ps
CONTAINER ID   IMAGE                                              COMMAND                  CREATED         STATUS         PORTS      NAMES
502e8abbd2f5   riscv64/redis:7.4-rc-alpine3.20                    "docker-entrypoint.s…"   5 seconds ago   Up 3 seconds   6379/tcp   some-redis
```

It feels like the `armv7`/`arm64` situation 5-8 years ago currently; if you do a lot with containers, expect to have to build your own on top of the primitives for a while, yet. And some software compiles, some doesn't (without some tweaking), so it's not for the 'driveby sysadmin' yet, you have to actually know the basics to be productive.

geerlingguy, 2024-07-22:
Testing an old AMD R5 230 2GB GPU, I plugged it into the board while powered off, and was surprised to see the fan spin up—thus I opened [PCIe slot powered even when Jupiter is powered off](https://community.milkv.io/t/pcie-slot-powered-even-when-jupiter-is-powered-off/2505).

The system boots with the card installed, and `dmesg` shows:

```

```

Then there's `lspci`:

```
TODO
```

I am going to try this [k1-gpu](https://gitee.com/icenowy/linux-6.1/commits/k1-gpu) branch to see if I can get the AMD GPU to work in Bianbu Linux. That's a fork of [Bianbu Linux 6.1](https://gitee.com/bianbu-linux/linux-6.1), and the patch seems to perform a few of the write-combining hacks that we had to do for `radeon/amdgpu` for Arm as well.

carbonfix, 2024-07-23:
For GPU testing, the command provided by SpaceMIT is as follows:
```
sudo apt install glmark2-es2-wayland
glmark2-es2-wayland --run-forever
```

hokamilkv, 2024-07-23:
> I was originally wondering if the fan or fan connector was dead, but no—just the heatsink seems to be good enough for 99% of the time—the fan will only spin up deep into a multicore test, and briefly at that. It's noticeable but not annoying the few times it does spin up.

Our default logic is to turn on the fan when the core temperature reaches above 60 degrees. We will update to PWM speed regulation in later versions.

hokamilkv, 2024-07-23:
> For GPU testing, the command provided by SpaceMIT is as follows:
> 
> ```
> sudo apt install glmark2-es2-wayland
> glmark2-es2-wayland --run-forever
> ```

You can also watch the gpu status:
sudo watch cat /sys/kernel/debug/pvr/status

geerlingguy, 2024-07-23:
@carbonfix - Thanks! I was able to get a much more reasonable score of 459 with that!

ThomasKaiser, 2024-07-24:
Memory performance (especially latency) is not that great (well, bad). Are the Milk-V guys aware of this and might provide 'better' DRAM initialization code?

geerlingguy, 2024-07-24:
@ThomasKaiser - They are watching this thread, would be good to hear their thoughts—not sure if this is the first board they're using the Spacemit SoC with, but I wonder if it's an issue with Spacemit's controller or with something more under their control.

geerlingguy, 2024-07-25:
> Just why? By looking at https://browser.geekbench.com/v6/cpu/compare/6979805?baseline=5769184 it's 100% obvious that Geekbench on RISC-V is not capable to generate any meaningful numbers that would represent anything 'real-world performance'. It's just nonsense.

Because people will ask regardless, and I like having the data, regardless of the fitness of Geekbench 6 as a universal benchmarking tool :)

The reason I run a few dozen other tests too is because:

  1. Having one simple benchmark is fun for giggles, but having an array of more real-world examples like real code compilation, glmark, memory test, MP3/MP4 transcoding, etc. is better.
  2. Different tests show differences in architecture differently :D

To that point, I still can't run a few common benchmarks (like PTS tests) because they aren't set up for RISC-V. Whether that reflects real-world performance ('0' vs some metric that is above '0') is immaterial—instead, it reflects reality: if you buy one of these boards today, the performance might as well be 0 if you can't compile or run software the same as you can on x86 and Arm currently. At least, if you're not willing to roll up your sleeves and patch makefiles, compile newer versions than are available in repositories, build your own containers, etc.

geerlingguy, 2024-07-30:
I have a new board today, and here are a few notes:

First, idle power draw as tested with a USB-C power adapter is slightly higher from the wall (maybe slightly less efficient supply), measuring 5W idle, 10W under stress-ng all cores:

[image]

Second, WiFi works (yay!), so now I can run my network performance tests with WiFi instead of wired.

geerlingguy, 2024-08-05:
YouTube video: [The PC industry is changing: RISC-V goes mainstream](https://www.youtube.com/watch?v=YxtFctEsHy0).

apcameron, 2024-08-05:
Try this benchmark on your systems. 
https://github.com/PlummersSoftwareLLC/Primes/tree/drag-race/PrimeCPP/solution_2
Just tried it on my Risc-v VisionFive 2 
It scores better than the PI 4 in this test. See Below

-------------------------------------------------------------------------
Computing primes to 10000000 on 4 threads for 5 seconds.
Passes: 64, Threads: 4, Time: 5.07541, Average: 0.0793033, Limit: 10000000, Counts: 664579/664579, Valid : Pass

davepl_par;64;5.07541;4;algorithm=base,faithful=yes,bits=1

-------------------------------------------------------------------------
It will be interesting to see the results

I got the idea to run it on my system after watching https://www.youtube.com/watch?v=7lYX0Eo4cbE&t=1s

swamytk, 2024-10-16:
> Kioxia XG8 over PCIe:
> 
> ```
> 0001:01:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller XG8 (rev 01) (prog-if 02 [NVM Express])
> 	Subsystem: KIOXIA Corporation Device 0001
>
>
> 		LnkCap:	Port #0, Speed 16GT/s, Width x4, ASPM L1, Exit Latency L1 <32us
> 			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
> 		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
> 			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
> 		**LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)**
> 			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
>
>
> 	Kernel driver in use: nvme
> ```

Hi  @geerlingguy , thanks a lot for the detailed work on this board. I am yet to receive my board which is on the way. 
The spec says PCIe Gen2x2 (2 lanes), but lspci shows link status as "Width x1" as shown above. Did you get a chance to analyze this? This is the reason you get only 346MB/s max on NVMe SSD ([benchmark test](https://github.com/geerlingguy/sbc-reviews/issues/47#issue-2417496122))? With 2 lanes, we should get around 800MB/s performance.

mliezun, 2024-10-18:
Hi @geerlingguy!

I know this is a little bit off-topic, but I've been building a site for hosting CI/CD runners powered by RISC-V architecture. I was in the need of it and noticed there wasn't one out there that was easy to setup.

Here's the site:
https://www.riscvrunners.com/

For now it only supports Gitlab Runners, but I plan to add support for other platforms too!

If you want to try it without paying, check out this Gitlab Project: https://gitlab.com/riscvrunners/runner-test

That project is open to the public, so you can push code and run CI pipelines to see if it works as you expect.

Haui1112, 2024-12-07:
> > Ubuntu 23.10 desktop
> 
> Dumb question: can this be upgraded in place to a later Ubuntu release?

There is no reason for it not to be able to be upgraded to a newer version. The spacemit k1 is already being used with ubuntu 24.04 on other boards (DC ROMA PAD II for example)

frankenstein91, 2025-03-13:
Did you get it to boot with an USB-C power supply... I'm stuck in a loop...
UART:
 ```
try sd...
bm:3
sys: 0x200
```