geerlingguy/sbc-reviews #48: Radxa X4

geerlingguy, 2024-07-20:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://radxa.com/products/x/x4
  - Board purchased from: Arace
  - Board purchase date: July 19, 2024
  - Board specs (as tested): 8GB
  - Board price (as tested): $79.90 (+ $14.90 heatsink+fan)

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-07-20:
Some coverage: [CNX](https://www.cnx-software.com/2024/07/19/radxa-x4-low-cost-credit-card-sized-intel-n100-sbc-raspberry-pi-5-alternative/).

It's the first time an Intel-based SBC has challenged Raspberry Pi / RK3588 Arm boards in pricing. Running windows with 4/8 GB of RAM would be a bit of a pain, but Linux on it could be interesting.

Just because it's x86 doesn't mean it will be perfect, though (as I experienced with the [LattePanda Mu](https://github.com/geerlingguy/sbc-reviews/issues/42). It is a very intriguing board, Intel must have a massive stock of N100 SoCs, with how many devices are integrating the things!

geerlingguy, 2024-07-30:
A few notes from my assembly:

  - The heatsink case requires removal of the 'bottom feet' to get at the screws on the X4; if they had cutouts in the metal for a screwdriver, or if the cut was different, they could avoid users having to remove and replace 6 extra screws to get at the board.
  - The included thermal pad was definitely crusty. [Bret.dk also mentioned this](https://bret.dk/intel-n100-radxa-x4-first-thoughts/), it seems like at least whatever batch we were shipped, the thermal pads may be expired, or were stored in a very dry place that rotted them out. I'll attach a picture in a bit illustrating the problem—I think the crusty nature also resulted in incomplete contact, as the smaller die was only contacting the thermal pad about 50%.
  - I replaced the included thermal pad with my own, cut from a 1mm thick [set](https://amzn.to/3WphLew).
  - The fan cable from the fan on the included heatsink is not very long—I had to re-remove the X4 to attach the little 2-pin JST connector, then re-attach the X4, stretching the fan cable to its limit. There's no way to route the fan cable differently—it could definitely use an extra cm or so of length.
  - The antennas and CMOS battery came pre-attached—and kind of just stick out from the case like antennae. I used a tiny piece of VHB tape to stick the CMOS battery onto the 'leg' of the heatsink case, and stuck the WiFi/BT antennas to the other side. There's not really a place to route these little wires, so they just stick out the open sides.
  - There was no M.2 mounting screw included, so I had to fish one out of my little M.2 screw baggie.
  - There's no thermal interface between the memory and the heatsink, so it doesn't get the benefit of the giant heatsink hovering just a couple mm or so above it.

geerlingguy, 2024-07-30:
Things I would like to check on with GPIO:

  - [ ] How is the RP2040 attached to the system? (I didn't see it using `lsusb`)
  - [ ] Is there any way to get HATs to work?
  - [ ] Do any Pi libraries work with the X4, or will they need rewriting to work through the RP2040?

geerlingguy, 2024-07-30:
Power levels between 9-22W, while running Geekbench:

[image]

I'm guessing I should use a higher power USB-C adapter (currently using Argon's 27W PWR GaN...). They recommend their PD 30W PSU, or anything that "supports 12V and the current >= 2.5A under 12V".

geerlingguy, 2024-07-30:
I'm seeing some _very_ high temps:

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +95.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +95.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +95.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +95.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +95.0°C  (high = +105.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C

nvme-pci-0300
Adapter: PCI adapter
Composite:    +72.8°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +85.8°C)
Sensor 1:     +72.8°C  (low  = -273.1°C, high = +65261.8°C)
```

NVMe stays up around that temperature at all times. The CPU spikes to 95°C when I start hammering it, then it seems to throttle down to settle in at 74°C or so. It idles at 45°C.

geerlingguy, 2024-07-30:
Initial conclusion—assuming there isn't something massively wrong with my heatsink case: you need a _lot_ more cooling / better thermal interface than a Raspberry Pi / Arm chip to cool the N100 adequately. Its's odd because the similar-size heatsink on the [LattePanda Mu](https://github.com/geerlingguy/sbc-reviews/issues/42) seems to do an adequate job of keeping it cool, at least for benchmarking runs.

geerlingguy, 2024-07-30:
PiBenchmarks.com result:

```
Alias (leave blank for Anonymous): geerlingguy
Result submitted successfully and will appear live on https://pibenchmarks.com within a couple of minutes.

     Category                  Test                      Result      
HDParm                    Disk Read                 2837.27 MB/sec           
HDParm                    Cached Disk Read          1644.80 MB/sec           
DD                        Disk Write                732 MB/s                 
FIO                       4k random read            187889 IOPS (751559 KB/s)
FIO                       4k random write           84979 IOPS (339917 KB/s) 
IOZone                    4k read                   103342 KB/s              
IOZone                    4k write                  140635 KB/s              
IOZone                    4k random read            61937 KB/s               
IOZone                    4k random write           235508 KB/s              

                          Score: 51977                                       
```

geerlingguy, 2024-07-31:
For GPIO use, see [Les Pounder's video on using GPIO](https://www.youtube.com/watch?v=rUkpIG_3D9k) — you press the button on the board (near the RADXA logo, unmarked), and it resets the RP2040 and holds down BOOTSEL at the same time, so it appears as a USB device:

```
Bus 003 Device 005: ID 2e8a:0003 Raspberry Pi RP2 Boot
```

Under Ubuntu Desktop, it automatically mounts the volume:

```
sda           8:0    1   128M  0 disk 
└─sda1        8:1    1   128M  0 part /media/jgeerling/RPI-RP2
```

There, you can drag a UF2 file on it (like MicroPython), and use Thonny to interact with the RP2040, upload code to it, etc.

I have yet to find a way to virtually 'press the button'; some people mention the access is the [same as on the X2L](https://docs.radxa.com/en/x/x2l/software-development/flash), but I tried using their code to manually reset the RP2040 with `gpiod` installed and got:

```
jgeerling@radxa-x4:~/Downloads$ source usb.sh
gpioset: error setting the GPIO line values: No such file or directory
gpioset: error setting the GPIO line values: No such file or directory
gpioset: error setting the GPIO line values: No such file or directory
gpioset: error setting the GPIO line values: No such file or directory
```

geerlingguy, 2024-07-31:
I've squished in some Noctua thermal paste in place of the thermal pads, to see if it conducts the heat away any better.

[image]

Re-running Geekbench 6, I'm getting: https://browser.geekbench.com/v6/cpu/7138811 ([very _slightly_ faster than before](https://browser.geekbench.com/v6/cpu/compare/7138811?baseline=7127802)).

camerahacks, 2024-08-03:
Hey Jeff, this is André from dphacks/rpilocator.

I just got mine today! I was able to reroute the fan cable by removing the 3 screws on the fan and repositioning it so the wires are closer to the connector.

geerlingguy, 2024-08-03:
@camerahacks ah, that would definitely help! Maybe a QC issue and they can fix it on the production line for that heatsink fan.

rcarmo, 2024-08-04:
The ideal thing would be for the fan wire to go down to the board via a notch (I have that in my “improvements” section). That way I could 3D print dust covers for the front and back that didn’t leave a little turn of wire on the outside…

alansill, 2024-08-04:
Has anyone tried the POE+ HAT with this? Our interest as usual is in building SBC clusters for outreach and training, and while we have been able to cobble together POE for the 12V needed for the X2L boards by combining POE-to-12V barrel adapters with automotive 12V-to-USB-C adapters that support 12V output by soldering barrel connector leads onto them, we'd prefer an integrated solution.

camerahacks, 2024-08-04:
@alansill , Radxa is offering a PoE+ HAT specific to the X4. The HAT has the power pins that line up with the X4's power pins.

alansill, 2024-08-04:
> @alansill , Radxa is offering a PoE+ HAT specific to the X4. The HAT has the power pins that line up with the X4's power pins.

HI, thanks! I'm definitely aware of this and asking if anyone has any experience with using it yet.

geerlingguy, 2024-08-05:
@alansill IIRC @bretmlw has one? He has his first thoughts here: https://bret.dk/intel-n100-radxa-x4-first-thoughts/

alansill, 2024-08-05:
Thanks. Now it's down to comparing good cheap PoE+ 2.5 GbE switches, I guess. 

The only reason I want this feature is for remote hard power cycling control. Wish I could find a nice way to do this directly. The other option we're exploring is a simple option-isolator for each node conventional power controlled by the GPIO of the head node.

willgryan, 2024-08-05:
Is there any mention of In-band ECC in the BIOS? I am wondering, because both the LattePanda Mu & ODROID H-4 use (LP)DDR5 and offer IBECC.

webash, 2024-08-29:
Anyone had a chance to benchmark or even just stat-dump the eMMC? Otherwise anyone have a view about whether it'll be suitable for the OS install of a Docker Swarm Cluster; likely fairly stateless as I'll be externalising most logs to a Loki/Prom instance, so there should be limited continuous writes. Can't find any information about the eMMC on Radxa's website.

rcarmo, 2024-08-29:
I don't recall anyone having mentioned receiving a board with the eMMC soldered on. Mine came with the BGA location blank, and I don't think they're currently selling SKUs with it.

willgryan, 2024-08-30:
I have an X4 8GB model w/ 64GB eMMC. Below are fio results from running [YetAnotherBenchScript](https://github.com/masonr/yet-another-bench-script) on Debian 12.5. Would you like to see anything else?

```
fio Disk Speed Tests (Mixed R/W 50/50) (Partition /dev/mmcblk0p2):
---------------------------------
Block Size | 4k            (IOPS) | 64k           (IOPS)
  ------   | ---            ----  | ----           ---- 
Read       | 22.38 MB/s    (5.5k) | 77.02 MB/s    (1.2k)
Write      | 22.39 MB/s    (5.5k) | 77.43 MB/s    (1.2k)
Total      | 44.77 MB/s   (11.1k) | 154.46 MB/s   (2.4k)
           |                      |                     
Block Size | 512k          (IOPS) | 1m            (IOPS)
  ------   | ---            ----  | ----           ---- 
Read       | 119.52 MB/s    (233) | 117.75 MB/s    (114)
Write      | 125.87 MB/s    (245) | 125.59 MB/s    (122)
Total      | 245.39 MB/s    (478) | 243.34 MB/s    (236)
```

schtschenok, 2024-09-03:
Hey there! I just got my X4 (8GB), and my temps seem to be way better. At roughly 55c under load (with all cores running constantly at 2900MHz) & 25c idle. I tweaked something in BIOS so they don't ever lower their frequency under load from 2900, something that has to do with PL1/PL2. If the load is asymmetrical - some cores can go up to 3.4GHz at roughly 65c for a moment, but never 90 or anything like that. These thermals were the same for me both during synthetic stress tests (s-tui) and during the actual compilation of FFMPEG that I did, so it's probably not a testing methodology issue.

While assembling it, I noticed that the supplied thermal pad was just way too thin (and crumbly) and didn't make the full contact between the die and the radiator. They put two pieces of the same pad in the box, so I, without much hesitation, just slapped one on top of another, and it worked for me!

So my GeekBench results with two crumbly thermal pads slapped onto each other (and not even covering the whole chip's PCB area - only the shiny dies and some space around them) are even better than @geerlingguy's thermal paste results. Which is kind of weird, honestly.
https://browser.geekbench.com/v6/cpu/7615536

zarir94, 2025-03-03:
Hello everyone! Can any of you install and try the LLM called `Qwen 2.5 3b` on Radxa X4? I just want to know how it performs with it's 4 cores of N100 cpu. Revealing LLM performance (eval rate, prompt rate etc) on it might help many of us like me. Just a random geek asking for an overview! ;)

jace, 2025-05-07:
`lspci` is showing a Thunderbolt 4 controller:

```
00:0d.0 USB controller: Intel Corporation Alder Lake-N Thunderbolt 4 USB Controller
00:14.0 USB controller: Intel Corporation Alder Lake-N PCH USB 3.2 xHCI Host Controller
```

`lsusb` also shows a 20 Gbps interface:

```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 20000M/x2
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 007: Dev 002, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 007: Dev 002, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
```

I can't find this port though. It's not any of the four Type A ports, and the Type C port is only accepting power, not recognising a hub plugged into it.

Has anyone confirmed if the Type C port is indeed only for power, with no USB capabilities? Thunderbolt capability exists but is not exposed in the hardware?