geerlingguy/sbc-reviews #49: Orange Pi 5 Max

geerlingguy, 2024-08-17:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-Max.html
  - Board purchased from: Amazon
  - Board purchase date: August 17, 2024
  - Board specs (as tested): 8GB RAM
  - Board price (as tested): $113

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-08-17:
Thanks to [@Sknashville@theatl.social on Mastodon](https://theatl.social/@Sknashville/112978340015287071) for making me aware of the Max's existence. Will be interesting to see if this board is more of the goldilocks that Radxa was trying to thread with the X4...

martinbone, 2024-09-07:
@geerlingguy   Very interesting - I've been looking for an OP5 Max benchmark like this. The main thing that stands out to me is the speed of NVMe compared with the first OP5 (which isn't surprising considering the PCIe / lanes on the Max). This is what I got on my OP5:

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 50.70 MB/s |
| iozone 4K random write     | 123.35 MB/s |
| iozone 1M random read      | 363.86 MB/s |
| iozone 1M random write     | 366.33 MB/s |
| iozone 1M sequential read  | 368.94 MB/s |
| iozone 1M sequential write | 372.99 MB/s |

Your glmark2-es2-wayland result looks quite low ... but I imagine this is due to the OS and drivers installed. On my OP5 I've got a score of 4800 (Joshua Riek's Ubuntu 22.04, running CPU and GPU at Performance and PAN_MESA_DEBUG=gofaster).

Many thanks for the data on the NVMe speed!!

jimmyhon, 2024-09-13:
I also experienced the slow RX bandwidth on RTL8125 by default.

I was able to get full speed (2.35 Gbps) by moving the irq handler to the big A76 cores

`
ls /proc/irq/125/enP3p49s0
echo f0 > /proc/irq/125/smp_affinitiy
`

So it's not a limitation of the hardware.

butterdori, 2024-12-14:
Hi Jeff! Thanks for the very great and detailed reviews, both here and on your YouTube channel. I've been looking to build a portable NAS to complement my main home NAS, and looks like beefed up Pis are a great solution. I came across this but also noticed that Radxa released a 5B+, which has dual m.2 slots (albeit at half (x2) the speeds) and built-in Wifi with a cheaper cost than the original 5B (around $100 for 8GB ram and 64GB eMMC on board). For my needs, it seems that Radxa has more capable hardware, but I've also been hearing Orange Pi has better software support. Given the finicky nature of rockchips on Linux anyway, the difference between the two may not be great. After all, if I can get Debian running, it won't take much to run a NAS server (OMV etc), so my software support needs won't be extensive, especially if I'm not looking to do some extreme stuff like PCIe GPUs. So, I'm torn. Should I get the Orange Pi 5 Max or the Radxa Rock 5B+? Am I too worried about the software support of Radxa? I would very much appreciate your advice. Thanks in advance.

dtuantran, 2025-05-09:
Thanks for your benchmark. Just for your information, the rockchip has a NPU 6 TOPS. Maybe it could be interesting to do the benchmark with NPU.