geerlingguy/sbc-reviews #50: Sipeed Lichee Pi 3A (Spacemit K1)

geerlingguy, 2024-09-04:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://en.wiki.sipeed.com/hardware/en/lichee/K1/lpi3a/1_intro.html
  - Board purchased from: (provided by Sipeed)
  - Board purchase date: September 4, 2024
  - Board specs (as tested): Spacemit K1, 8GB RAM, 32GB eMMC
  - Board price (as tested): $99

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-09-04:
Other coverage:

  - [CNX Software](https://www.cnx-software.com/2024/09/05/licheepi-3a-a-spacemit-k1-risc-v-development-board-with-som-and-carrier-board/)

geerlingguy, 2024-09-05:
PCIe layout:

```
jgeerling@k1:~$ lspci
0001:00:00.0 PCI bridge: Device 201f:0001 (rev 01)
0001:01:00.0 Non-Volatile memory controller: SK hynix Gold P31/BC711/PC711 NVMe Solid State Drive
0002:00:00.0 PCI bridge: Device 201f:0001 (rev 01)

jgeerling@k1:~$ sudo lspci -vvv
[sudo] password for jgeerling: 
0001:00:00.0 PCI bridge: Device 201f:0001 (rev 01) (prog-if 00 [Normal decode])
	Device tree node: /sys/firmware/devicetree/base/soc/pcie@ca400000/interrupt-controller@0
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 58
	Region 0: Memory at 90000000 (32-bit, non-prefetchable) [size=1M]
	Region 1: Memory at 90100000 (32-bit, non-prefetchable) [size=1M]
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: f000-0fff [disabled] [16-bit]
	Memory behind bridge: 90200000-902fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2- AuxCurrent=375mA PME(D0+,D1+,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/32 Maskable- 64bit+
		Address: 0000000060600000  Data: 0000
	Capabilities: [70] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0
			ExtTag+ RBE+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x2, ASPM L0s L1, Exit Latency L0s <4us, L1 <16us
			ClockPM- Surprise+ LLActRep+ BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x2
			TrErr- Train- SlotClk+ DLActive+ BWMgmt+ ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP+ LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd+
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn+ NFERptEn+ FERptEn+
[... 124 lines cut ...]
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn+ NFERptEn+ FERptEn+
		RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
			 FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
		ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0000
	Capabilities: [148 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [160 v1] Vendor Specific Information: ID=0006 Rev=0 Len=018 <?>
	Kernel driver in use: pcieport
```

geerlingguy, 2024-09-05:
I had to expand the filesystem space like I did the Milk-V Jupiter (https://github.com/geerlingguy/sbc-reviews/issues/47#issuecomment-2239867481), but now I can't get the system to reboot. D'oh!

geerlingguy, 2024-09-06:
Strange behavior when using it with NanoKVM (though I can't verify yet if it's a problem on the NanoKVM side, or an issue with power via USB-A ports on the Lichee Pi): https://github.com/sipeed/NanoKVM/issues/40

geerlingguy, 2024-09-10:
For some reason, I kept having issues compiling mpich via Ansible, so I had to log in and run `configure` manually, as well as `make`. I think it was having trouble finding the paths to dependencies required for the build.

I also still can't get the disk partition expanded without borking the boot, but apparently that could be related to the Lichee Pi 3A firmware revision I'm running. I'll try upgrading that soon.

geerlingguy, 2024-09-10:
Reading more into the 'firmware', it seems like that term references the OS maybe according to https://en.wiki.sipeed.com/hardware/en/lichee/K1/lpi3a/3_burn_image.html ? I'm going to ask Sipeed for clarification.

geerlingguy, 2024-09-11:
There's a WiFi chipset on the carrier board, but at least using the default Bianbu install, I can't seem to use it for WiFi network access.

ThomasKaiser, 2024-09-29:
Jeff, if time permits could you please run on the Lichee Pi 3A (or _any other_ SpacemiT K1/M1 thingy you own) the following?

    ls -laR /sys/devices/soc0

([background](https://github.com/Dr-Noob/cpufetch/issues/295))

faried, 2024-10-08:
```
root@k1:/home/fn# grep . /sys/devices/soc0/* 2>/dev/null
/sys/devices/soc0/family:spacemit socs
/sys/devices/soc0/machine:SiPEED LPi3A Board
/sys/devices/soc0/revision:C
/sys/devices/soc0/serial_number:0059D968E583D118
/sys/devices/soc0/soc_id:K1-6370
root@k1:/home/fn# 
```
and
```
root@k1:/home/fn# grep -r . /sys/devices/soc0/* 2>/dev/null
/sys/devices/soc0/family:spacemit socs
/sys/devices/soc0/machine:SiPEED LPi3A Board
/sys/devices/soc0/power/runtime_active_time:0
/sys/devices/soc0/power/runtime_status:unsupported
/sys/devices/soc0/power/runtime_suspended_time:0
/sys/devices/soc0/power/control:auto
/sys/devices/soc0/revision:C
/sys/devices/soc0/serial_number:0059D968E583D118
/sys/devices/soc0/soc_id:K1-6370
/sys/devices/soc0/subsystem/drivers_autoprobe:1
root@k1:/home/fn# uname -a
Linux k1 6.6.36 #2.0~rc7.2 SMP PREEMPT Mon Sep 30 10:36:26 UTC 2024 riscv64 riscv64 riscv64 GNU/Linux
root@k1:/home/fn# 
```
This is with Bianbu 2.0rc2.

brucehoult, 2024-12-14:
> I've now tried extending the partition two times, and both times, after a reboot, the system won't finish booting—it just has a blinking cursor on the HDMI output.

Yeah, that happened to me (Bianbu 2.0.4) if I used `fdisk`.

Doing `sudo apt update; sudo apt install gparted; sudo gparted` worked fine. On startup gparted warned that the GPT table wasn't using the full disk size, and offered to fix it. I think fdisk might have also warned on that, but idk how to fix it there.