geerlingguy/sbc-reviews #51: Qualcomm Snapdragon X Dev Kit for Windows

geerlingguy, 2024-09-25:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.arrow.com/en/products/c8380-12c-mp-32g/thundercomm
  - Board purchased from: Arrow
  - Board purchase date: 2024-07-16 (Arrived 2024-09-25)
  - Board specs (as tested): X Elite - 12 core, 32GB RAM, 512 GB NVMe SSD
  - Board price (as tested): $855.42

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-09-25:
A few notes as I'm unboxing:

  - Box includes a power adapter and USB-C to HDMI adapter (I'm guessing the latter was added when they decided to drop the internal HDMI port).
  - The bottom of the unit says "Evaluation only; not FCC approved for resale"
  - According to devs on Discord, the unit comes with Windows 11 Home — and so far besides on-device partitions, there is no official way to download a recovery image or download other editions of Windows 11 built for Arm that work on it.
  - The warranty states it is void if any parts are changed—and two of the phillips head screws in the base are covered by little 🚫 stickers
  - Comparing the size, it's close to, but a little wider and less deep than a typical 1L 'square' mini PC. It's much larger (but shorter) than my GMKTek box (NUC-sized), and it's nearly the same width (just a hair wider) and a little deeper than the Windows Dev Kit (the one from 2023 / Project Volterra)

MagicAndre1981, 2024-09-25:
>  there is no official way to download a recovery image or download other editions of Windows 11 built for Arm that work on it.

upgrading to a higher Windows SKU should work via [entering a product key in settings app](https://support.microsoft.com/en-us/windows/upgrade-windows-home-to-windows-pro-ef34d520-e73f-3198-c525-d1a218cc2818).

geerlingguy, 2024-09-25:
Some hardware details:

  - Ethernet port is on a separate daughtercard marked "Running HDMI Board" with an unpopulated HDMI port, and a display connector that is not connected to anything.
  - Ethernet provided by RTL8125BG (2.5 Gbps Realtek NIC) on mPCIe card
  - WiFi is a "T99H432.10" module, presumably Qualcomm though I can't find documentation on it. I have found _one_ laptop, the [MSI Alpha 17 C7V](https://www.msi.com/Laptop/Alpha-17-C7VX/Specification), which has the same model number (but .03 instead of .10), and it is listed as "WiFi 7". The WiFi module is E-key, and has a little heatsink mounted to it (with thermal pads on top and bottom).
  - There's an unpopulated PCIe slot (no slot is installed, but the pads are on the board. I would really like that to be populated. Looks like full size x16, not sure if electrically connected as x16, x8, x4, x1, or not at all...
  - There's a 2280 size NVMe SSD, a Foresee XP2200F512G, made by Longsys. It is mounted with a full-length heatsink (with thermal pads on top and bottom).
  - There's a tiny daughtercard called "Running ECB" - does it have something to do with cryptography? Platform security? Not sure, it's mounted above a NanoSIM and microSD slot. The plug it is plugged into is labeled "EC_I/F"
  - That's right, there's a hidden NanoSIM slot (with nothing in it), next to the microSD card slot that is exposed through the front of the case.
  - The cooler is a massive copper heatsink with heat pipes and a massive cooling stack for the SoC itself. There are thermal pads on all the little power circuits, and a large amount of thermal compound on all four RAM chips and the SoC itself.
  - The fan is APEXE model B82DBHA2467, with a 4-pin standard fan header on the motherboard (easy to remove, separate from the copper heatsink). It is connected to the heatsink with screws, and there is a taped on duct to port the exhaust out the side of the Dev Kit.
  - There are a couple CSI ports, many unpopulated headers, an unpopulated eMMC spot near the RAM, and there's a micro USB debug port labeled `DBG_UART`

geerlingguy, 2024-09-26:
[image]

Blog post with many photos and more detailed notes after zooming way way in: [Qualcomm Snapdragon Dev Kit for Windows Teardown (2024)](https://www.jeffgeerling.com/blog/2024/qualcomm-snapdragon-dev-kit-windows-teardown-2024).

utp216, 2024-09-26:
> * Box includes a power adapter and USB-C to HDMI adapter (I'm guessing the latter was added when they decided to drop the internal HDMI port).

I wonder if you solder the appropriate HDMI port to the PCB if it would function. Maybe they didn't want to pay for HDMI licensing?

apache405, 2024-09-26:
The ECB is an embedded controller board.  ITE is one of the more popular EC chips out there.

jglathe, 2024-09-26:
Arrow has just canceled my order for this one - I ordered on the same day :raised_eyebrow:. Thought they would ship none, then, but... whatever. The hardware seems nice enough, though. Do you want to run Linux on the thing? That was my intent. You also can run the [Windows Dev Kit 2023 with Linux](https://github.com/jglathe/linux_ms_dev_kit).

SpieringsAE, 2024-09-26:
> > * Box includes a power adapter and USB-C to HDMI adapter (I'm guessing the latter was added when they decided to drop the internal HDMI port).
> 
> I wonder if you solder the appropriate HDMI port to the PCB if it would function. Maybe they didn't want to pay for HDMI licensing?

doesn't the DP to HDMI converter get around that? as in the licensing cost is already part of the converter chip?

But yeah I wondered the same thing, maybe even make a different daughterboard that just exposes DP directly.

Same with the unpopulated PCIE slot, wonder how far populating that gets, on windows it might be locked out of the ACPI tables, but on Linux you might be able to just add it to the devicetree blob

talynone, 2024-09-26:
Richard Campbell with hardware design experience speculated HDMI FCC approval was the issue, more info:

https://www.youtube.com/watch?v=NlEV38cO8Gs

talynone, 2024-09-26:
Getting the Windows ARM ISO is doable but requires a bit more effort right now:

https://www.makeuseof.com/download-install-windows-11-arm-iso/

anonymix007, 2024-09-26:
> T99H432.10

I believe the actual model is NCM865A. NCM865 is used in some WiFi 7 products and AFAIK the only difference between them is the Bluetooth connection interface (USB vs UART).

SpieringsAE, 2024-09-26:
> Richard Campbell with hardware design experience speculated HDMI FCC approval was the issue, more info:
> 
> https://www.youtube.com/watch?v=NlEV38cO8Gs

tbh, why even hdmi at all, DP is right there, on the board, why put extra chips on a board to make the display interface worse (arguably).

theholypumpkin, 2024-09-26:
I'm very excited about the Linux benchmarks!!!

I'm waiting for Tuxedo Computers with their Snapdragon X Model to be released maybe at the end of 2024 and Wondering what performance I can expect in comparison to my Late 2013 MacBook Pro.

Now with kernel 6.11 out, the Snapdragon X chips should at work, but looking at the release notes looks like it's still not fully baked. Probably by 6.12.

lukasMega, 2024-09-26:
what is the power consumption in common use cases (idle in headless mode, idle with desktop, load, ...) Thank you!

talynone, 2024-09-26:
There's not many disk imaging/cloning apps that work on Windows ARM right now, this procedure seemed to have worked for others (read the comments for gotchyas):

https://old.reddit.com/r/Surface/comments/1dniufo/ssd_clonemigrate_on_arm/

talynone, 2024-09-26:
Can't seem to find an Windows ARM native build of iperf3, but x86/x64 version should work fully in emulation.

In general the following is a great resource for finding native Windows ARM apps (well maintained/updated):

https://armrepo.ver.lt/

driver1998, 2024-09-26:
> There's not many disk imaging/cloning apps that work on Windows ARM right now

WinPE on ARM64 supports x64 emulation, and some disk partitioning/imaging software works fine under emulation. I tried DiskGenius and it works fine. dd for Windows might work as well.

jglathe, 2024-09-26:
MiniTool ShadowMaker works on arm for an image. restore... better an x86 box, but works reliably.

geerlingguy, 2024-09-26:
Regarding the HDMI port, [Richard Campbell speculates](https://www.youtube.com/watch?v=NlEV38cO8Gs) that port could be at least a significant reason for the shipping delay, if they weren't able to get it cleared for compliance (maybe also the reason the FCC clearance didn't happen).

geerlingguy, 2024-09-26:
Regarding Windows 11 image creation—[I have created aarch64 images using UUPdump.net in the past](https://www.jeffgeerling.com/blog/2023/ampere-altra-max-windows-gpus-and-gaming), and that works fine, it's just... not official in any way, shape, or form.

What I would _really_ like to see is Microsoft providing an arm64 ISO like they do for x86 on their [Windows 11 Download page](https://www.microsoft.com/en-us/software-download/windows11/).

talynone, 2024-09-26:
Little off topic, but a bigger reason some smaller open source projects don't have WIndows ARM64 builds even more than the availability of the Dev Kit is because GitHub (which is owned by Microsoft) Windows ARM64 runners are currently only available for enterprise customers. GitHub has said it should open it up to everyone soon but if Microsoft wants faster adoption they really should light a fire on that.

geerlingguy, 2024-09-27:
Notes from my first boot:

  - The Ethernet LEDs don't seem to light. Don't see an issue with the cabling, it's like the NIC EEPROM doesn't have the LEDs lighting (connected in Windows fine at 2.5 Gbps).
  - Windows 11 Home is silly but at least the [account setup bypass](https://pureinfotech.com/bypass-internet-connection-install-windows-11/) still works.
  - After setting up an account, I think there were around 6 or 7 total reboots for Windows Updates, plus a firmware update that made it so my DVI-input monitor no longer worked with the Dev Kit (had to switch to a newer HDMI monitor).
  - The fan is a *lot* louder than I expected. It ramps down to almost silent at idle (thankfully), but the fan ramps up aggressively any time I do anything (even opening an app, or for background activity). The lowest speed is still louder than my Mac Studio maxed out.

geerlingguy, 2024-09-27:
Single core gets up to around 30W, multi-core Cinebench 2024 gets up to 100+ W, and the fans ramp up to about 58 dBa measured 1' away. Definitely a loud little machine.

Want to get Linux going so I can do some real testing ;)

SpieringsAE, 2024-09-27:
Do you have a link to the geekbench test results? Would love to see results of the independent tests

geerlingguy, 2024-09-27:
I'm updating all the final results in the OP at the top of the issue, and just added a link to the Geekbench result there.

talynone, 2024-09-27:
Heaven has no native ARM64 tests, so that may not give the best results. 

Solar Bay, Wild Life: and Wild Life: Extreme in 3DMark have Windows ARM64 native versions.

geerlingguy, 2024-09-27:
@talynone - I understand that — just wanting to get a picture of how well x86 stuff is emulated and runs on this platform ;)

geerlingguy, 2024-09-27:
Was able to install WSL with Ubuntu 24.04.

  1. Run PowerShell as Administrator
  2. Run `wsl --install`
  3. After it completes, reboot
  4. Run `wsl --list --online` to show available Distros, select one to install
  5. (I ran `wsl.exe --install -d Ubuntu-24.04`)
  6. WSL dumps you into the Ubuntu environment once it's done

To quit, type `exit`. To get back in, type `wsl`.

jglathe, 2024-09-27:
Could you dump the acpi tables of this thing and pr them to https://github.com/aarch64-laptops/build ? That would help building a device tree for it - to boot Linux directly.

jglathe, 2024-09-27:
fun find: open terminal, type ubuntu, and it comes up in the same window :raised_eyebrow:

talynone, 2024-09-27:
Yep WSL works great, so does Windows Sandbox (needs to be added on PRO edition via add Windows components), even graphics partitioning with Hyper-V works (I did so just an experiment, not much practical use), 

Windows Subsystem for Android doesn't work on Elite computers, because Elite processors only support ARM64 binaries, and Windows Subsystem for Android has 32bit ARM binaries.

Someone did create a hack for it with ARM64 binaries injected into a custom image:

https://www.xda-developers.com/windows-subsystem-for-android-snapdragon-how-to/

Latest version can be found here:
https://github.com/snickler/WSA-Patched/releases

geerlingguy, 2024-09-27:
> Could you dump the acpi tables of this thing

Haven't done that before, what's the best way; is there a tool that runs under Windows and would work on aarch64?

SpieringsAE, 2024-09-27:
> Could you dump the acpi tables of this thing and pr them to https://github.com/aarch64-laptops/build ? That would help building a device tree for it - to boot Linux directly.

I wonder how close it is to the CRD

jglathe, 2024-09-27:
[I just used these](https://www.intel.com/content/www/us/en/download/774881/acpi-component-architecture-downloads-windows-binary-tools.html) for the WDK2023 and the HP Omnibook X14. and a little [phind](https://www.phind.com/)

jglathe, 2024-09-27:
CRD/QCP could be quite close, as it is for may of these

Mis012, 2024-09-27:
to my understanding it's literally CRD with some components not soldered on (and with secureboot on because F U), I assume upstream Linux has a devicetree for the CRD but idk how complete it is

geerlingguy, 2024-09-30:
Was told by someone from Linaro on Mastodon to use [this guide](https://dortania.github.io/Getting-Started-With-ACPI/Manual/dump.html#from-windows), so will try that now.

Using [SSDTTime](https://github.com/corpnewt/SSDTTime), I dumped the following acpi table data:

[acpi.zip](https://github.com/user-attachments/files/17192636/acpi.zip)

I posted it in a PR to the aarch64-laptops/build repo here: https://github.com/aarch64-laptops/build/pull/110

sanderhollaar, 2024-10-01:
I tried to boot kernel `6.12-rc1` on a `Thinkpad T14s Gen 6`.

The kernel loads and seems to start initrd but almost immediately the screen turns blank and the laptop freezes hard.

The output captured with high-speed video and `earlyprintk=vga` and `ignore_loglevel` does not give a hint. It does not seem to detect USB.

jglathe, 2024-10-01:
Hmm a little off topic, but: Try using the USB type-c connectors for booting. Upstream dtb for T14s doesn't have working USB type-a yet.

Mis012, 2024-10-01:
`earlyprintk=vga` is really not going to work here... you want `earlycon=efifb`

sanderhollaar, 2024-10-01:
I thought my experience booting Linux on a Snapdragon X Elite could be of use to Jeff but please do tell me if this is considered too off topic.

I've duplicated the usb-a drive to usb-c but this did not affect the boot. I don't think it boots far enough.

I've exchanged `earlyprintk=vga` with `earlycon=efifb` which gives better readable output until it says:
```
printk: legacy console [tty0] enabled
printk: legacy bootconsole [efifb0] disabled
```
I'll try `keep_bootcon` later.

Last lines before the display turns black:
```
vreg_bob1: Setting 3008000-3960000uV
vreg_l1d_0pr Setting 880000-880000uV
vreg_l1i_1p0 Setting 1800000-1800000uV
vreg_l2i_1p2 Setting 1200000-1200000uV
vreg_l3i_0o8 Setting 880000-880000uV
vreg_l2d_0p9 Setting 912000-912000uV
input: gpio-keys as /devices/platform/gpio-keys/input/input0
vreg_bob2: Setting 2504000-3000000uV
```
Could contain typo's, output is hard to read.

jglathe, 2024-10-01:
Hmm wrong discussion. You could try the [multiboot Ubuntu image](https://github.com/jglathe/linux_ms_dev_kit/wiki/Bootable-Image-for-multiple-Snapdragon-(SC8280XP)-and-Snapdragon-X-Elite-(X1E80100)-laptops) (with type-c stick) to see if you get further.

talynone, 2024-10-01:
@geerlingguy Looks like someone was listening to your wishes

https://x.com/_sumitdh/status/1841124915392430098

geerlingguy, 2024-10-01:
WOW! After a few years of everyone wondering, it sounds like it might finally happen. Hope that line doesn't become a meme in a few months though. They need to get that done quick.

fobrs, 2024-10-02:
> Can't seem to find an Windows ARM native build of iperf3, but x86/x64 version should work fully without issue in emulation.

>In general the following is a great resource for finding native Windows ARM apps (well maintained/updated):

>https://armrepo.ver.lt/

I've just successfully build iperf 2.2.0 for ARM64 with msys2 clangarm64. 

If anyone wants to test/use it, please drop me a note...

fobrs, 2024-10-08:
Est. Ship: 9 Jan 2025Est. Delivery: 13 Jan 2025
C8380-12C-MP-32G
Thundercomm Technology
😭

jglathe, 2024-10-08:
Yeah same here. @geerlingguy got the only one they had, I'd wager

geerlingguy, 2024-10-09:
@jglathe 😱 

Maybe my first tweet about this back in June was true... https://twitter.com/geerlingguy/status/1812980507308601549

SpieringsAE, 2024-10-10:
@geerlingguy there does seem to be a preliminary dtb for the devkit in the works by qualcomm:  
https://lore.kernel.org/all/20241005182250.788272-1-quic_sibis@quicinc.com/

fobrs, 2024-10-15:
It's shipping for me!
[image]

jglathe, 2024-10-15:
Also got an advance shipment notice.

fobrs, 2024-10-16:
Got it, it's 2 times faster compiling my app than a dell core i9 (9th gen) precision 7540 laptop. But making more noise too, and the dell is noisy...

jglathe, 2024-10-16:
I've seen that x1e is fast, builds ~3 times faster than the Windows Dev Kit 2023, and faster than the equivalent of a Ryzen 9-3900 when cross compiling. Tested on a HP Omnibook X14 with x1e-78-100. Which is not that noisy, but clearly audible when its on full (compile) load.

CampGareth, 2024-10-17:
Um, Qualcomm and Arrow just killed this product. No further units, no future support, but also full refunds for all purchasers and you can keep the hardware.

SpieringsAE, 2024-10-17:
> Um, Qualcomm and Arrow just killed this product. No further units, no future support, but also full refunds for all purchasers and you can keep the hardware. 

For real? :(
You got a link or something?

fobrs, 2024-10-17:
[here](https://app.connected.arrow.com/e/es?s=600830862&e=1064569&elqTrackId=ba8a0b8bb1f34ec3bf6f5a005cfd0be3&elq=4032e435a10a45f890baf8778e659555&elqaid=64924&elqat=1&elqak=8AF5CC44ECD6C631F5115896DD31BC969AE9FD2AC9A5360C65BA21BDD33914360847)
I have a collectors item!

2company, 2024-10-17:
One going spare. Unit arrived today. Now surplus, using a ARM64 laptop instead.

geerlingguy, 2024-10-18:
Heh, also blogged about it today here: [Qualcomm cancels Snapdragon Dev Kit, refunds all orders](https://www.jeffgeerling.com/blog/2024/qualcomm-cancels-snapdragon-dev-kit-refunds-all-orders).

What an odd situation! Still want to get Linux on it.

jglathe, 2024-10-18:
Mine just arrived, thankfully. There is a basic dt already, will try to bring it up with Linux in the next days. No dp or HDMI might complicate matters a bit, we'll see.

Mis012, 2024-10-18:
well, the (e)DP connector is there :P

[107 more replies]