geerlingguy/sbc-reviews #52: AmpereOne A192-32X (SuperMicro)

geerlingguy, 2024-10-16:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.supermicro.com/en/products/system/megadc/2u/ars-211me-fnr
  - Board purchased from: (Provided by Ampere/Supermicro)
  - Board purchase date: Oct 14, 2024
  - Board specs (as tested): A192-32X, 512GB DDR5 (5200 MT/s)
  - Board price (as tested): (If you have to ask...)

## Linux/system information

```
# output of `screenfetch`
ubuntu@ubuntu:~$ screenfetch 
                          ./+o+-       ubuntu@ubuntu
                  yyyyy- -yyyyyy+      OS: Ubuntu 24.04 noble
               ://+//////-yyyyyyo      Kernel: aarch64 Linux 6.8.0-39-generic-64k
           .++ .:/++++++/-.+sss/`      Uptime: 23m
         .:++o:  /++++++++/:--:/-      Packages: 810
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 5.2.21
       .:+o:+o/.          `+sssoo+/    Disk: 19G / 101G (20%)
  .++/+:+oo+o:`             /sssooo.   CPU: Ampere Ampere-1a @ 192x 3.2GHz
 /+++//+:`oo+o               /::--:.   GPU: ASPEED Technology, Inc. ASPEED Graphics Family (rev 52)
 \+/+o+++`o++o               ++////.   RAM: 31390MiB / 522867MiB
  .++.o+++oo+:`             /dddhhh.  
       .+.o+oo:.          `oddhhhh+   
        \+.++o+o``-````.:ohdhhhhh+    
         `:o+++ `ohhhhhhhhyo++os:     
           .o:`.syhhhhhhh/.oo++o`     
               /osyyyyyyo++ooo+++/    
                   ````` +oo+++o\:    
                          `oo++.     

# output of `uname -a`
Linux ubuntu 6.8.0-39-generic-64k #39-Ubuntu SMP PREEMPT_DYNAMIC Sat Jul  6 11:08:16 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-10-16:
Getting full 25 Gbps Ethernet on the 2nd interface:

```
ubuntu@ubuntu:~$ ethtool eno2np1
Settings for eno2np1:
	Supported ports: [ FIBRE ]
	Supported link modes:   25000baseCR/Full
	                        1000baseX/Full
	                        10000baseCR/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: RS	 BASER
	Advertised link modes:  25000baseCR/Full
	                        1000baseX/Full
	                        10000baseCR/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 25000Mb/s
	Lanes: 1
	Duplex: Full
	Auto-negotiation: on
	Port: Direct Attach Copper
	PHYAD: 1
	Transceiver: internal
netlink error: Operation not permitted
        Current message level: 0x00002081 (8321)
                               drv tx_err hw
	Link detected: yes
```

If I try running Geekbench 6 I get a core dump, lol:

```
ubuntu@ubuntu:~/Geekbench-6.3.0-LinuxARMPreview$ ./geekbench6
<jemalloc>: Unsupported system page size
<jemalloc>: Unsupported system page size
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

I opened up a support issue for that: [Can't run Geekbench 6 Arm Preview on AmpereOne 192-core system](http://support.primatelabs.com/discussions/geekbench/86728-cant-run-geekbench-6-arm-preview-on-ampereone-192-core-system)

geerlingguy, 2024-10-16:
And yes, I know this system is not really an SBC. I still want to test it against Arm SBCs, though ;)

geerlingguy, 2024-10-25:
To get `btop` to show the CPU SoC temps instead of `apm_xgene/IO Power`, I went into options `o`, tabbed to the CPU tab, and under 'Cpu sensor' changed it to `apm_xgene/SoC Temperature`.

[image]

ThomasKaiser, 2024-10-26:
Jeff, if time permits could you please check this:

    grep CONFIG_ARM64_MTE /boot/config-6.8.0*

Background: the CPU cores should be capable of MTE but [your machine doesn't expose the feature via `/proc/cpuinfo`](https://community.amperecomputing.com/t/weekend-read-memory-tagging-in-ampere-one/963/5).

hrw, 2024-10-26:
No GPU in it but can you check it with some AMD/NVIDIA graphic cards?

geerlingguy, 2024-10-26:
@hrw - I'd love to find a way to get a test sample of one of AMD or Nvidia's enterprise server cards—right now the best fit I have is an older Quadro RTX card, but it won't fit in this chassis.

@ThomasKaiser I'll try to run that next time I have the server booted (remind me if I forget next week); I shut it down over the weekend and a boot cycle takes 5-10 minutes, so I'm too lazy to sit and wait today for one command!

hrw, 2024-10-27:
@geerlingguy "add pcie x16 riser cable to your shopping list" was my first idea but then I realized that server case would lack power cables for gpu as well.

geerlingguy, 2024-10-28:
@hrw - The server actually includes 2x8 pin PCIe power connections, it's designed for up to 1 fanless GPU (needs high CFM to keep cool).

geerlingguy, 2024-10-28:
@ThomasKaiser:

```
ubuntu@ubuntu:$ grep CONFIG_ARM64_MTE /boot/config-6.8.0*
/boot/config-6.8.0-39-generic-64k:CONFIG_ARM64_MTE=y
/boot/config-6.8.0-47-generic:CONFIG_ARM64_MTE=y
```

geerlingguy, 2024-10-29:
Attempting qemu-coremark, during setup I'm getting an error: [meson setup fails with 'Dependency "glib-2.0" not found'](https://github.com/AmpereComputing/qemu-coremark/issues/3)

bexcran, 2024-10-29:
> Attempting qemu-coremark, during setup I'm getting an error: [meson setup fails with 'Dependency "glib-2.0" not found'](https://github.com/AmpereComputing/qemu-coremark/issues/3)

@geerlingguy  What distro are you using? If Ubuntu, it should install the dependencies including glib2.0-dev automatically.

geerlingguy, 2024-10-29:
Had to install `libglib2.0-dev` manually, then add myself to the `kvm` group, but now the benchmark runs.

geerlingguy, 2024-10-30:
I noticed when I run `sudo shutdown now`, I get logged out of `ubuntu` and SSH goes away, but then the server won't actually _power off_ (and go into BMC-only mode) for many minutes.

Watching the SOL Console today, I saw tons of errors like:

```
[ 5261.993963] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x3ee0} len:0
[ 5270.120534] {1788}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 2
[ 5270.129045] {1788}[Hardware Error]: It has been corrected by h/w and requires no further action
[ 5270.137729] {1788}[Hardware Error]: event severity: corrected
[ 5270.143461] {1788}[Hardware Error]:  Error 0, type: corrected
[ 5270.149193] {1788}[Hardware Error]:   section_type: memory error
[ 5270.155186] {1788}[Hardware Error]:    error_status: Storage error in DRAM memory (0x0000000000000400)
[ 5270.164478] {1788}[Hardware Error]:   node:0 card:5 module:16 device:7 
[ 5270.171078] {1788}[Hardware Error]:   error_type: 13, scrub corrected error
[ 5270.178026] EDAC MC0: 1 CE scrub corrected error on unknown memory (node:0 card:5 module:16 device:7 page:0x0 offset:0x0 grain:1 syndrome:0x0 - APEI location: node:0 card:5 module:16 device:7 status(0x0000000000000400): Storage error in DRAM memory)
[ 5271.187341] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x3ee4} len:0
[ 5280.388425] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x3ef0} len:0
```

So it looks like that DIMM is throwing a bunch of errors, maybe causing the Ethernet driver to throw other errors?

```
[ 5372.462135] bnxt_en 0003:02:00.1 eno2np1: Resp cmpl intr err msg: 0x51
[ 5372.468653] bnxt_en 0003:02:00.1 eno2np1: hwrm_ring_free type 2 failed. rc:fffffff0 err:0
[ 5381.671651] bnxt_en 0003:02:00.1 eno2np1: Resp cmpl intr err msg: 0x51
[ 5381.678169] bnxt_en 0003:02:00.1 eno2np1: hwrm_ring_free type 2 failed. rc:fffffff0 err:0
```

bexcran, 2024-10-30:
I saw the shutdown of an AmpereOne machine I was testing take a really long time too due to the Broadcom Ethernet driver. But I didn’t see any of the DRAM or APEI issues, so I’m not sure they’re related.

geerlingguy, 2024-10-30:
> I saw the shutdown of an AmpereOne machine I was testing take a really long time too due to the Broadcom Ethernet driver. 

Hmm, maybe that's it then — those messages kept popping in amidst all the DIMM messages. Might be nice to figure out how to fix the `bnxt_en` driver!

geerlingguy, 2024-10-30:
Testing a RAID 0 array of all the NVMe drives [following my guide](https://www.jeffgeerling.com/blog/2021/htgwa-create-raid-array-linux-mdadm):

```
ubuntu@ubuntu:~$ sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=6 /dev/nvme0n1p1 /dev/nvme1n1p1 /dev/nvme2n1p1 /dev/nvme3n1p1 /dev/nvme5n1p1 /dev/nvme6n1p1

ubuntu@ubuntu:~$ sudo mdadm --detail /dev/md0
/dev/md0:
           Version : 1.2
     Creation Time : Wed Oct 30 16:37:22 2024
        Raid Level : raid0
        Array Size : 11251445760 (10.48 TiB 11.52 TB)
      Raid Devices : 6
     Total Devices : 6
       Persistence : Superblock is persistent

       Update Time : Wed Oct 30 16:37:22 2024
             State : clean 
    Active Devices : 6
   Working Devices : 6
    Failed Devices : 0
     Spare Devices : 0

            Layout : original
        Chunk Size : 512K

Consistency Policy : none

              Name : ubuntu:0  (local to host ubuntu)
              UUID : 6dd22af6:0fd54fa0:9463f73f:636afb4e
            Events : 0

    Number   Major   Minor   RaidDevice State
       0     259       11        0      active sync   /dev/nvme0n1p1
       1     259       13        1      active sync   /dev/nvme1n1p1
       2     259       12        2      active sync   /dev/nvme2n1p1
       3     259       14        3      active sync   /dev/nvme3n1p1
       4     259       15        4      active sync   /dev/nvme5n1p1
       5     259       16        5      active sync   /dev/nvme6n1p1

ubuntu@ubuntu:~$ sudo mkfs.ext4 -m 0 -E lazy_itable_init=0,lazy_journal_init=0 /dev/md0
ubuntu@ubuntu:~$ sudo mkdir /mnt/raid0
ubuntu@ubuntu:~$ sudo mount /dev/md0 /mnt/raid0
```

Running my disk benchmark on the array...

TODO

geerlingguy, 2024-10-31:
Ampere sent over a replacement DIMM, and it seems to have corrected all the memory issues.

_However_, shutdown is still excruciating — about 10 minutes, and I just see tons of Ethernet NIC errors, maybe a bug in the `bnxt_en` driver on arm64?

```
[  224.516490] infiniband bnxt_re0: Couldn't change QP1 state to INIT: -110
[  224.523180] infiniband bnxt_re0: Couldn't start port
[  224.528173] bnxt_en 0003:02:00.0 bnxt_re0: Failed to destroy HW QP
[  224.534384] ------------[ cut here ]------------
[  224.538988] WARNING: CPU: 97 PID: 2721 at drivers/infiniband/core/cq.c:322 ib_free_cq+0x13c/0x1d8 [ib_core]
[  224.548759] Modules linked in: tls xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat bridge stp llc nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo xt_addrtype nft_compat nf_tables qrtr overlay nls_iso8859_1 bnxt_re(+) ampere_cspmu cfg80211 dax_hmem acpi_ipmi ib_uverbs cxl_acpi ast cxl_core ipmi_ssif arm_cspmu_module arm_spe_pmu i2c_algo_bit ib_core onboard_usb_hub acpi_tad arm_cmn ipmi_msghandler xgene_hwmon cppc_cpufreq sch_fq_codel binfmt_misc dm_multipath nvme_fabrics efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 rndis_host cdc_ether usbnet btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq libcrc32c raid1 raid0 crct10dif_ce polyval_ce polyval_generic ghash_ce sm4_ce_gcm sm4_ce_ccm sm4_ce sm4_ce_cipher sm4 sm3_ce sm3 sha3_ce sha2_ce nvme sha256_arm64 sha1_ce nvme_core bnxt_en xhci_pci xhci_pci_renesas nvme_auth aes_neon_bs aes_neon_blk aes_ce_blk aes_ce_cipher [last unloaded: ipmi_devintf]
[  224.637726] CPU: 97 PID: 2721 Comm: (udev-worker) Not tainted 6.8.0-39-generic-64k #39-Ubuntu
[  224.646237] Hardware name: Supermicro Super Server/R13SPD, BIOS T20241001152934 10/01/2024
[  224.654487] pstate: 63401009 (nZCv daif +PAN -UAO +TCO +DIT +SSBS BTYPE=--)
[  224.661437] pc : ib_free_cq+0x13c/0x1d8 [ib_core]
[  224.666152] lr : ib_mad_port_open+0x220/0x450 [ib_core]
[  224.671388] sp : ffff80010920f520
[  224.674690] x29: ffff80010920f520 x28: 0000000000000000 x27: ffffb4c746059120
[  224.681813] x26: 0000000000000000 x25: ffff0002527e8870 x24: ffff0002527e88f8
[  224.688936] x23: ffffb4c7465f3e90 x22: 00000000ffffff92 x21: ffffb4c7465fc550
[  224.696060] x20: ffff000246000000 x19: ffff00015794bc00 x18: ffff8000e8d400f0
[  224.703182] x17: 0000000000000000 x16: 0000000000000000 x15: 6c6c6174735f7766
[  224.710305] x14: 0000000000000000 x13: 505120574820796f x12: 7274736564206f74
[  224.717429] x11: 2064656c69614620 x10: 0000000000000000 x9 : ffffb4c7465c58b0
[  224.724552] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[  224.731675] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[  224.738798] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000002
[  224.745921] Call trace:
[  224.748355]  ib_free_cq+0x13c/0x1d8 [ib_core]
[  224.752723]  ib_mad_port_open+0x220/0x450 [ib_core]
[  224.757609]  ib_mad_init_device+0x78/0x228 [ib_core]
[  224.762582]  add_client_context+0xfc/0x208 [ib_core]
[  224.767556]  enable_device_and_get+0xe0/0x1e0 [ib_core]
[  224.772790]  ib_register_device.part.0+0x130/0x218 [ib_core]
[  224.778459]  ib_register_device+0x38/0x68 [ib_core]
[  224.783345]  bnxt_re_ib_init+0x120/0x238 [bnxt_re]
[  224.788135]  bnxt_re_probe+0x14c/0x268 [bnxt_re]
[  224.792746]  auxiliary_bus_probe+0x50/0x108
[  224.796920]  really_probe+0x1c0/0x420
[  224.800575]  __driver_probe_device+0x94/0x1d8
[  224.804920]  driver_probe_device+0x48/0x188
[  224.809091]  __driver_attach+0x14c/0x2c8
[  224.813002]  bus_for_each_dev+0x88/0x110
[  224.816913]  driver_attach+0x30/0x60
[  224.820476]  bus_add_driver+0x17c/0x2d0
[  224.824300]  driver_register+0x68/0x178
[  224.828125]  __auxiliary_driver_register+0x78/0x148
[  224.832990]  bnxt_re_mod_init+0x54/0xfff8 [bnxt_re]
[  224.837861]  do_one_initcall+0x64/0x3b8
[  224.841687]  do_init_module+0xa0/0x280
[  224.845425]  load_module+0x7b8/0x8f0
[  224.848988]  init_module_from_file+0x98/0x118
[  224.853332]  idempotent_init_module+0x1a4/0x2c8
[  224.857850]  __arm64_sys_finit_module+0x70/0xf8
[  224.862368]  invoke_syscall.constprop.0+0x84/0x100
[  224.867147]  do_el0_svc+0xe4/0x100
[  224.870536]  el0_svc+0x48/0x1c8
[  224.873673]  el0t_64_sync_handler+0x148/0x158
[  224.878019]  el0t_64_sync+0x1b0/0x1b8
[  224.881670] ---[ end trace 0000000000000000 ]---
[  224.886282] bnxt_en 0003:02:00.0 bnxt_re0: Free MW failed: 0xffffff92
[  224.892720] infiniband bnxt_re0: Couldn't open port 1
[  257.266860] INFO: task (udev-worker):2732 blocked for more than 122 seconds.
[  257.273911]       Tainted: G        W          6.8.0-39-generic-64k #39-Ubuntu
[  257.281123] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  284.760123] systemd-shutdown[1]: Waiting for process: 2721 ((udev-worker)), 2732 ((udev-worker))
[  326.899586] bnxt_en 0003:02:00.1: QPLIB: bnxt_re_is_fw_stalled: FW STALL Detected. cmdq[0xe]=0x3 waited (101989 > 100000) msec active 1 
[  326.911840] bnxt_en 0003:02:00.1 bnxt_re1: Failed to modify HW QP
[  326.917924] infiniband bnxt_re1: Couldn't change QP1 state to INIT: -110
[  326.924614] infiniband bnxt_re1: Couldn't start port
[  326.929635] bnxt_en 0003:02:00.1 bnxt_re1: Failed to destroy HW QP
[  326.935847] bnxt_en 0003:02:00.1 bnxt_re1: Free MW failed: 0xffffff92
[  326.942289] infiniband bnxt_re1: Couldn't open port 1
[... cut ...]
[  381.507812] bnxt_en 0003:02:00.0 eno1np0: Error (timeout: 5000015) msg {0xb4 0x5fa} len:0
[  387.048016] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x6b9} len:0
[  391.749581] bnxt_en 0003:02:00.0 eno1np0: Error (timeout: 5000015) msg {0xb4 0x5ff} len:0
[  396.814118] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x6c6} len:0
[  403.018680] bnxt_en 0003:02:00.0 eno1np0: Error (timeout: 5000015) msg {0x23 0x606} len:0
[  406.578705] bnxt_en 0003:02:00.1 eno2np1: Error (timeout: 5000015) msg {0x41 0x6cf} len:0
...
```

bexcran, 2024-10-31:
Could you try blacklisting the `bnxt_re` module? To do so, edit `/etc/modprobe.d/blacklist.conf` and add:
```
blacklist bnxt_re
```

From https://utcc.utoronto.ca/~cks/space/blog/linux/BroadcomNetworkDriverAndRDMA?showcomments :
>The driver stalls during boot and spits out kernel messages like:
```
    bnxt_en 0000:ab:00.0: QPLIB: bnxt_re_is_fw_stalled: FW STALL Detected. cmdq[0xf]=0x3 waited (102721 > 100000) msec active 1
    bnxt_en 0000:ab:00.0 bnxt_re0: Failed to modify HW QP
    infiniband bnxt_re0: Couldn't change QP1 state to INIT: -110
    infiniband bnxt_re0: Couldn't start port
    bnxt_en 0000:ab:00.0 bnxt_re0: Failed to destroy HW QP
    [... more fun ensues ...]
```
> This causes systemd-udev-settle.service to fail:

>    udevadm[1212]: Timed out for waiting the udev queue being empty.
>    systemd[1]: systemd-udev-settle.service: Main process exited, code=exited, status=1/FAILURE

>This then causes Ubuntu 24.04's ZFS services to fail to completely start, which is a bad thing on hardware that we want to >use for [our ZFS fileservers](https://utcc.utoronto.ca/~cks/space/blog/linux/ZFSFileserverSetupIII).

>We aren't the only people with this problem, so I was able to find various threads on the Internet, [for example](https://>forum.proxmox.com/threads/broadcom-nics-down-after-pve-8-2-kernel-6-8.146185/). These gave me the solution, which >is to blacklist the bnxt_re kernel module, but at the time left me with the mystery of how and why the bnxt_re module >was even being loaded in the first place.

geerlingguy, 2024-10-31:
@bexcran - Will try that, after waiting 15 minutes I just pushed an immediate shutdown so I could finally power cycle.

bexcran, 2024-10-31:
@geerlingguy That's what I'd do too! If you want to be a _bit_ nicer to your system/filesystem and you have "Magic SysRq Keys" enabled you can do:

ALT+PrintScreen+s,u,b

That is, press and hold ALT and SysRq (will probably be labeled PrintScr on your keyboard instead of SysRq) while pressing 's', then 'u' then 'b' without letting go of ALT and SysRq.

That'll sync data to disk, attempt to unmount filesystems and the reboot.

https://docs.kernel.org/admin-guide/sysrq.html

geerlingguy, 2024-10-31:
I added `/etc/modprobe.d/blacklist-bnxt.conf` with `blacklist bnxt_re` inside, and rebooted.

Now, it reaches `poweroff.target` within 3 seconds, and `Power down` state after about 12. SOOOOO much nicer lol.

I guess if I ever need Infiniband over Ethernet, I can figure out that `bnxt_re` module, otherwise, not sure why it would load by default!

geerlingguy, 2024-10-31:
@bexcran - Is there any simple way of switching the kernel I'm booting on here? I would like to try the 4K kernel just to see if Geekbench will complete a run, but the default kernel that it's running right now (for performance reasons) is 64K.

bexcran, 2024-10-31:
@geerlingguy  Sorry, I don't know.

geerlingguy, 2024-10-31:
I may do a reinstall of the OS on a separate drive just to do that test then.

geerlingguy, 2024-11-01:
Also, now that I have my Ampere Altra 32-core NAS server upgraded to 25 Gbps Ethernet: https://github.com/geerlingguy/arm-nas/issues/16

I can finally run the `iperf3` test between these two machines!

geerlingguy, 2024-11-01:
Testing a copy over SMB from one of the NVMe on this system to the NVMe on the HL15:

```
$ sudo apt install cifs-utils
$ sudo mkdir /mnt/mercury
$ sudo mount -t cifs -o user=jgeerling,uid=$(id -u),gid=$(id -g) //nas01.mmoffice.net/mercury /mnt/mercury

# Inside /mnt/nvme/test, create a large file
$ fallocate -l 100G largefile

# Benchmark file copy over SMB
ubuntu@ubuntu:/mnt/nvme/test$ rsync --info=progress2 -a largefile /mnt/mercury/test/largefile
107,374,182,400 100%    1.14GB/s    0:01:27 (xfr#1, to-chk=0/1)
```

Not quite as fast as I was hoping, but this is dealing with SMB + Ethernet overhead, and I saw it going between 8-15 Gbps on the NAS.

geerlingguy, 2024-11-01:
To switch kernels on Ubuntu, I did the following:

```
$ sudoedit /etc/default/grub

# Comment out the `GRUB_TIMEOUT_STYLE=hidden` line so it looks like:
#GRUB_TIMEOUT_STYLE=hidden
GRUB_TIMEOUT=0

# After saving the file, run
$ sudo update-grub
```

Technically I could hit `Esc` (I think? Maybe `Shift`?) during boot, but the timing for that is pretty narrow, so it's nicer to just have the menu appear during boot.

geerlingguy, 2024-11-01:
Now that I have the kernel back to 4k page size, I am running Geekbench 6 test. I noticed someone else ran one on the same motherboard/CPU in May: https://browser.geekbench.com/v6/cpu/6131970

geerlingguy, 2024-11-21:
From Wendell / L1Techs, running on Sapphire Rapids (60 core / 120 thread / 1.90 GHz):

```
59 instances of pts/coremark running in parallel in arm64 VMs!
Round 1 - Total CoreMark Score is: 491008
w@prod-baremetal:~/qemu-coremark$ cat /proc/cpuinfo |grep Xeon |head -n 1
model name      : Intel(R) Xeon(R) Platinum 8490H
```

ThomasKaiser, 2024-11-21:
> I wanted to get more numbers for the qemu-coremark test Ampere suggested

Why exactly? What is the purpose of running the arm64 version of coremark emulated on x86? 

Why is your 192-core system executing 47 parallel instances, the 120-thread Platinum 8490H executing 59 and the 512-thread Xeon 6980P 127? The Ampere and the 6980P seem to fire up `($count-of-threads / 4 ) - 1` instances while the 8490H fires up `($count-of-threads / 2 ) - 1`... just why? Why not `$count-of-threads` instead if it's all about running fully parallel?

geerlingguy, 2024-11-21:
@ThomasKaiser those are questions best directed at Ampere. I know coremark is one of the few benchmarks where their CPU really shines in general (compared to AMD/Intel), and running emulated code further puts their machine ahead of the others in this specific instance (Hyperthreading always muddies things too).

I think marketing-wise, Ampere would like to take some W's somewhere, and this is how they want to do it. It's a very niche case—where you have arm64 code you want to run, and you're running it on x86 servers, but I do get it. Some people run Ampere machines for CI for their arm stuff (targeting embedded), so this is the one case where it could give an indication that doing that is better native on Arm than emulated on your existing x86 servers.

Is it contrived? Yes. Is it useful? Well, maybe a little. I'm not going to give it much weight in my ultimate review of this hardware. (In fact, for full benchmarking, I am pointing people towards Phoronix and ServeTheHome, who have more extensive testing already).

joespeed, 2024-11-21:
@ThomasKaiser the purpose of qemu-coremark is simply to educate automakers and other developers of arm software of something that is pretty obvious to you and Jeff. If you are developing and testing arm software then it is better done on arm. Automakers do a ton of arm CI testing in arm64 emulators on x86. But emulation is slow and imperfect. You get better quality software faster when doing arm64 software development and testing on .... arm64. [Here is a talk](https://drive.google.com/file/d/18aX8VAZOavTdjfOePmTMxf3qekkvR6Hm/view?pli=1) that @bexcran and I gave to the [SOAFEE TSC](https://www.soafee.io) about this.

So what qemu-coremark does is run Phoronix Test Suite CoreMark in as many 4-core arm64 SoCs as the host will support, emulated on x86 and virtualized on arm64. Automakers, Tier 1s and automotive ISVs do CI testing in large (no larger) numbers of emulated or virtualized arm64 SoCs, actually automotive ECUs, so that they can run their 10,000s to million CI tests in parallel. Vehicles have 100s of millions of lines of code these days so must be able to massively parallelize the CI testing or it'll take a week. [Apex.ai](https://www.apex.ai) takes CI testing that consumes days on a physical automotive ECU and cranks through it in hours using arm64 runners with virtual automotive ECUs with QNX RTOS on racked Ampere arm64 servers. And because it is virtualized, not emulated, the quality or correctness of the test environment is much better.

geerlingguy, 2024-11-21:
@joespeed - Thanks for the clarification!

ThomasKaiser, 2024-11-22:
> arm64 CI testing

Thanks, that explains this niche use case and is important for Ampere's target audience (not so much Jeff's ;) )

> These high core count servers probably deserve some new benchmarking setups though

I personally use 7-zip's internal benchmark as a rough representation of these 'server workloads' (Integer / memory latency) and so far within the last decade when comparing with 'real workloads' between different machines the scores were a pretty good representation of tasks that were able to run on different hardware.

The Ampere machine here with its 192 cores shows a 1:156 ratio between single-threaded und multi score: 4783 vs. 745720 7-ZIP MIPS... which is simply excellent. Would be interesting how that compares to the Drupal setup :)

geerlingguy, 2024-11-22:
@ThomasKaiser - Indeed! At some point I'll get time to work on it again. Right now in crunch mode for some... other projects too.

hrw, 2024-11-24:
@geerlingguy can you run [my ArmCpuInfo](https://github.com/hrw/edk2-armcpuinfo/) in EFI shell and share results?

I have some idea and need such data from some Arm servers.