geerlingguy/sbc-reviews #53: System76 Thelio Astra Workstation

geerlingguy, 2024-10-22:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://system76.com/desktops/thelio-astra
  - Board purchased from: (Provided by System76 for review)
  - Board purchase date: October 22, 2024
  - Board specs (as tested): Ampere Altra Max M128-30, 512 GB ECC DDR4-3200, Nvidia A402
  - Board price (as tested): $3,299 (base), TBD (as tested)

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-10-22:
Phoronix has a review up with some preliminary benchmarks: [System76 Thelio Astra Reviewed: High-End ARM64 Developer Desktop](https://www.phoronix.com/review/system76-thelio-astra).

geerlingguy, 2024-10-22:
Note: I was originally going to have most of the benchmarking done already... but I had to be silly and try out a bunch of other GPUs. I then ran into the fun parade of Nvidia proprietary vs Ubuntu included vs Nouveau drivers, and totally borked my Ubuntu install...

Couple that with Arm64 needing specific card hardware support for video output pre-OS boot, and I got it nice and mangled. Need to reinstall Ubuntu and start fresh again with the A402 they included ;)

joespeed, 2024-10-23:
https://github.com/AmpereComputing/qemu-coremark

geerlingguy, 2024-10-23:
Ubuntu reinstalled. I downloaded [Ubuntu 24.04.1 Server for arm64](https://ubuntu.com/download/server/arm), installed it through OpenBMC's remote KVM (could also use SOL Serial-Over-LAN console, it works surprisingly well for text/console-based install), and am running through Ampere's guide for setting up [Nvidia graphics accelerated Linux Desktop environment](https://github.com/AmpereComputing/NVIDIA-GPU-Accelerated-Linux-Desktop-on-Ampere/tree/main):

```
# 1 - Disable PCIe ASPM:
sudo nano /etc/default/grub

# Add pcie_aspm=off to kernel parameters
GRUB_CMDLINE_LINUX_DEFAULT="pcie_aspm=off"

sudo update-grub

# 2 - Install Ubuntu's desktop environment
sudo apt install -y ubuntu-desktop
sudo reboot

# 3 - Install Nvidia graphics drivers
TODO - Should I install the Ubuntu-managed proprietary driver, or Nvidia's proprietary driver package?
```

geerlingguy, 2024-10-23:
It seems like it could be a nouveau issue, after looking in dmesg logs:

```
[   21.727151] nouveau 0004:01:00.0: Adding to iommu group 28
[   22.774703] i2c_designware APMC0D0F:00: controller timed out
... [message repeats] ...
[   54.294665] watchdog: BUG: soft lockup - CPU#122 stuck for 26s! [jbd2/dm-0-8:1631]
[   54.302232] Modules linked in: nouveau(+) snd_hda_intel(+) snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event drm_gpuvm snd_rawmidi drm_exec gpu_sched nls_iso8859_1 drm_ttm_helper snd_seq acpi_ipmi ttm ipmi_ssif(+) snd_seq_device irdma drm_display_helper snd_timer i40e cec snd rc_core soundcore ib_uverbs ast ib_core ipmi_devintf ipmi_msghandler arm_dmc620_pmu xgene_hwmon arm_cmn acpiphp_ampere_altra cppc_cpufreq arm_dsu_pmu acpi_tad joydev input_leds apple_mfi_fastcharge dm_multipath efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq libcrc32c raid1 raid0 cdc_ether usbnet hid_apple hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid uas usb_storage crct10dif_ce polyval_ce polyval_generic ghash_ce sm4 sha2_ce sha256_arm64 ice ixgbe nvme sha1_ce igb nvme_core xhci_pci xfrm_algo i2c_algo_bit mdio xhci_pci_renesas nvme_auth gnss aes_neon_bs
[   54.302392]  aes_neon_blk aes_ce_blk aes_ce_cipher
[   54.302402] CPU: 122 PID: 1631 Comm: jbd2/dm-0-8 Not tainted 6.8.0-47-generic #47-Ubuntu
[   54.302408] Hardware name: System76 Thelio Astra/Thelio Astra, BIOS 3.02 08/20/2024
[   54.302412] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   54.302418] pc : queued_spin_lock_slowpath+0x90/0x4f0
[   54.302430] lr : _raw_spin_lock+0x84/0xb8
[   54.302438] sp : ffff80008e98b8b0
[   54.302440] x29: ffff80008e98b8b0 x28: 0000000000000000 x27: ffff80008e98bcd8
[   54.302449] x26: ffff07ff9d751740 x25: ffffc7ec57f40000 x24: ffffc7ec57f40000
[   54.302457] x23: ffffc7ec5d78cc84 x22: 0000000000000000 x21: ffff07ff9d751748
[   54.302464] x20: 0000000000000000 x19: ffff07ff9d751748 x18: ffff80008e871080
[   54.302472] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[   54.302479] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[   54.302486] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffc7ec5e59047c
[   54.302493] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[   54.302499] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[   54.302506] x2 : 0000000000000001 x1 : 0000000000000001 x0 : 0000000000000001
[   54.302513] Call trace:
[   54.302516]  queued_spin_lock_slowpath+0x90/0x4f0
[   54.302524]  _raw_spin_lock+0x84/0xb8
[   54.302530]  nvme_queue_rqs+0xe4/0x2f8 [nvme]
[   54.302552]  blk_mq_flush_plug_list.part.0+0x16c/0x1c0
[   54.302561]  blk_add_rq_to_plug+0x1ac/0x2a0
[   54.302567]  blk_mq_submit_bio+0x530/0x6e0
[   54.302573]  __submit_bio+0x100/0x220
[   54.302580]  __submit_bio_noacct+0x68/0x1f0
[   54.302586]  submit_bio_noacct_nocheck+0x1e8/0x230
[   54.302591]  submit_bio_noacct+0x134/0x658
[   54.302597]  submit_bio+0xc0/0x190
[   54.302602]  submit_bh_wbc+0x150/0x220
[   54.302609]  submit_bh+0x20/0x50
[   54.302615]  jbd2_journal_commit_transaction+0x4d0/0x1708
[   54.302622]  kjournald2+0xc8/0x298
[   54.302627]  kthread+0xf8/0x110
[   54.302632]  ret_from_fork+0x10/0x20
[   54.326691] i2c_designware APMC0D0F:00: controller timed out
... [message repeats] ...
... [lots more of the spinlock messages too] ...
[  379.890828] nouveau 0004:01:00.0: gr: DATA_ERROR 0000009c [] ch 1 [00ffb40000 gnome-shell[4279]] subc 0 class c797 mthd 0d78 data 00000004
[  379.893910] ast 0003:02:00.0: swiotlb buffer is full (sz: 8388608 bytes), total 32768 (slots), used 0 (slots)
[  379.903289] nouveau 0004:01:00.0: gr: DATA_ERROR 0000009c [] ch 1 [00ffb40000 gnome-shell[4279]] subc 0 class c797 mthd 0d78 data 00000004
[  379.915735] nouveau 0004:01:00.0: gr: DATA_ERROR 0000009c [] ch 1 [00ffb40000 gnome-shell[4279]] subc 0 class c797 mthd 17e0 data 00000018
```

geerlingguy, 2024-10-23:
After blacklisting the `nouveau` driver and rebooting (the ubuntu 550 nvidia driver was preinstalled), I am now getting display output.

However, it seems to not be using any GPU acceleration... `glmark2` shows LLVM rendering.

geerlingguy, 2024-10-23:
Going to pause my testing on the workstation for the time being—it looks like there are two main issues I'm hitting:

  1. The fans remain at idle and don't spin up when needed—at least the CPU coolers—and this leads to lockups under high load over long periods. See https://github.com/geerlingguy/top500-benchmark/issues/44#issuecomment-2432830384
  2. I can't get external display output to work through the Nvidia card with either Nvidia's driver install or the Ubuntu driver install. The system doesn't get past 'Checkpoint 92'.

geerlingguy, 2024-10-23:
Just a note, with this configuration (a lot of sticks of ECC RAM), it idles around 100W, and at system shutdown, uses about 7W of power to run the BMC/IPMI:

[image]

bexcran, 2024-10-23:
Some more detailed information about the boot process:

00:00: Press power button
 - 00:10: SCP completes booting. Powers on the host ARMv8-A CPU.
 - 00:25: TF-A finishes DRAM initialization and training.
 - 00:35: TF-A finishes copying UEFI from SPI-NOR into DRAM.
01:00: Motherboard BIOS splash screen (ASRock Rack logo and boot commands displayed on screen)
...

Also, if you ssh to BMC ports 2200, 2201, 2202 (using the same login as the BMC) you can see the SOL consoles for the host, SCP (PMPro and SMPro) and the secure TF-A console.

geerlingguy, 2024-12-08:
I ran into overheating / thermal shutdown issues (RAM and SoC were both throwing temperature errors after 10-20 minutes of a heavy workload).

The fix was to install System76's 'driver' package, which includes a `system76-power` daemon that runs in the background and adjusts fan speeds on the system:

The [System76 Driver (Install)](https://support.system76.com/articles/system76-driver/) documentation directed me to install the driver:

```
sudo apt-add-repository -y ppa:system76-dev/stable
sudo apt-get update
sudo apt install system76-driver
```

That succeeded, and now:

```
system76@thelio-astra:~$ ps ax | grep system76-power
  12737 ?        Ss     0:00 /usr/bin/system76-power daemon
  18538 pts/0    S+     0:00 grep --color=auto system76-power
```

I'd like to know if there's a way to set a default fan curve or something that doesn't require the daemon to be running. Also see if there's a way to get sensor data into OpenBMC.

geerlingguy, 2024-12-10:
@bexcran - FYI I am not able to run Geekbench 6 (not that it gives great data for this platform... but still)—it bails out each time during the multi core 'Background Blur' test, which I believe uses NEON and is the main benchmark that seems to give a lot of Arm CPUs a hard time (my overclocked Pi 5's always barfed at that point!).

I was wondering if you might be able to give it a run and see what happens, if it might be able to run on another system?

An interesting aside, it seems like Primate Labs support site has been practically silent for a year or so now. I have a few questions out about these things, but nothing's gotten a response :(

geerlingguy, 2024-12-10:
I also tried running GravityMark (with Vulkan) and `glmark2-es2-wayland`, and neither completed. I would eventually get a `(core dumped)`. I didn't see anything relevant in `dmesg`, just a bunch of message every few minutes about apparmor denying snap from doing stuff:

```
[ 1442.033754] audit: type=1400 audit(1733860961.332:179): apparmor="DENIED" operation="capable" class="cap" profile="/usr/lib/snapd/snap-confine" pid=10601 comm="snap-confine" capability=12  capname="net_admin"
```

I am trying GravityMark with OpenGL instead of Vulkan now...

geerlingguy, 2024-12-10:
Ollama seems to install and detect the Nvidia GPU successfully:

```
system76@thelio-astra:~/Downloads$ curl -fsSL https://ollama.com/install.sh | sh
>>> Installing ollama to /usr/local
>>> Downloading Linux arm64 bundle
######################################################################## 100.0%
>>> Creating ollama user...
>>> Adding ollama user to render group...
>>> Adding ollama user to video group...
>>> Adding current user to ollama group...
>>> Creating ollama systemd service...
>>> Enabling and starting ollama service...
Created symlink /etc/systemd/system/default.target.wants/ollama.service → /etc/systemd/system/ollama.service.
>>> NVIDIA GPU installed.
```

I downloaded `llama3.2:3b`, ran it, and it was definitely GPU-accelerated, giving me good speed—will bench it and put those results above.

geerlingguy, 2024-12-11:
In my testing of Ollama on the Thelio Astra, it works out of the box with any Nvidia GPU via CUDA, but AMD GPUs don't work due to ROCm not yet supporting arm64, and Ollama not supporting Vulkan; see https://github.com/geerlingguy/ollama-benchmark/issues/5

But I could successfully run llama.cpp on my AMD Pro W7700, as well as an Nvidia RTX A4000 that was a drop-in upgrade.

geerlingguy, 2024-12-11:
I also tried setting up Box86/Box64/Steam using [Ampere's Steam guide](https://github.com/AmpereComputing/Steam-on-Ampere), but ran into https://github.com/ptitSeb/box86/issues/1015 — I might need to revamp their guide entirely. It would be nice if Pi-Apps would work out of the box on Ampere machines... maybe worth pursuing.

geerlingguy, 2024-12-11:
Windows 11 install is going off without a hitch—based on https://github.com/AmpereComputing/Windows-11-On-Ampere/issues/6#issuecomment-2476779227 (the UUPdump download). And when it booted, I still had my display plugged into the A4000's DisplayPort connection—and Windows 11 seems to have identified that as an output. There's no acceleration, and the resolution is like 480p stretched, but I'm surprised it's giving output through the Nvidia GPU, and not only through VGA!

geerlingguy, 2024-12-15:
This is an odd one... the A4000 I inserted is showing up at PCIe Gen 1 x16 (2.5 GT/sec) - confirmed in `lspci`:

[image]

It's inserted into slot 1. I'll have to reseat it, maybe messed something up when I was swapping NVMe drives.

geerlingguy, 2024-12-16:
Well, fun mystery: The Nvidia cards are showing up as `Speed 2.5GT/s (downgraded), Width x16` on the ADLINK Dev Workstation as well! At least in the top slot. Going to try another slot and see what happens.

geerlingguy, 2024-12-16:
To add on to the strangeness:

I tried reinstalling Ubuntu 24.04.1 desktop (using a daily live ISO download: https://cdimage.ubuntu.com/daily-live/20240421/), and in the installer, I'm seeing `Speed 16GT/s, Width x16` now...

However, I can't get a clean Ubuntu install on either of my Ampere systems—they bail out after copying over files, during some mount/umount commands :/

So now I have a machine in a half-installed state lol (Windows installs fine, but I can't find an ISO download of Ubuntu Desktop that will install cleanly. 24.04.1 Server (from https://ubuntu.com/download/server/arm) also errors out in a similar way. Only 20.04 seems to install happily, so I wonder if I'm just missing something? cc @bexcran @joespeed

bexcran, 2024-12-16:
I'd guess it's getting confused about existing partitions.
Can you boot from a rescue disk and delete any partitions you don't need (e.g. the Ubuntu partitions you're trying to recreate)? Or maybe use the custom disk partitioning mode?

geerlingguy, 2024-12-16:
@bexcran - I was erasing / formatting the entire drive on each install, at least using the GUI installer's option to do that. But I'll go in and manually delete all partitions before starting the install this time, as maybe there's something weird with the way the installer does it's thing!

geerlingguy, 2024-12-16:
Tried again by running Terminal in the Ubuntu Live CD ISO, and running:

```
sudo wipefs /dev/nvme0n1
sudo parted /dev/nvme0n1 print  # to confirm the partition table is wiped
```

That... seems to have worked! I guess I shouldn't trust the "Erase and install" option in the Ubuntu installer—that only erases the existing partition, I guess, but doesn't reformat the whole drive with the right partition layout! Or maybe a bug in the installer, not sure.

In any case, it's finally rebooted, and checking in Ubuntu, now I see the A4000 as:

```
LinkSta: Speed 16GT/s, Width x16
```

Very odd! Of course, I still have an issue with first boot—I'm wondering if it's because of missing ASPEED or Nvidia drivers (I'm plugged into VGA right now), but the desktop environment/GUI don't appear. I just see a mouse cursor on a blank grey background, and I tried plugging into VGA, then DisplayPort on the A4000, and even both at the same time.

I had to press Ctrl-Alt-3 to bring up the console, then I could log in there.

Then I enabled SSH:

```
sudo apt update
sudo apt install -y openssh-server
sudo ufw allow ssh
sudo systemctl start ssh
```

Then via SSH I installed the Nvidia drivers for Ubuntu as listed in the [Ubuntu Nvidia driver installation guide](https://ubuntu.com/server/docs/nvidia-drivers-installation):

```
sudo ubuntu-drivers install
```

I had elected not to install the drivers as part of the install process, but am now realizing that may have made the GUI work out of the box, heh. As it is, after doing that install (it installed version `550`) and rebooting... the DisplayPort connections were outputting nothing, but at least VGA had a working GUI :D

But I noticed, the card was back down to `2.5 GT/s (downgraded)`, so maybe a driver issue?!

```
jgeerling@thelio-astra:~$ sudo lspci -vvvv -s 0004:01:00.0
[sudo] password for jgeerling: 
0004:01:00.0 VGA compatible controller: NVIDIA Corporation GA104GL [RTX A4000] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: NVIDIA Corporation GA104GL [RTX A4000]
	Physical Slot: 1-3
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 323
	NUMA node: 0
	IOMMU group: 26
	Region 0: Memory at 20000000 (32-bit, non-prefetchable) [size=16M]
	Region 1: Memory at 280000000000 (64-bit, prefetchable) [size=32G]
	Region 3: Memory at 280800000000 (64-bit, prefetchable) [size=32M]
	Expansion ROM at 21000000 [virtual] [disabled] [size=512K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fffaf040  Data: 0000
	Capabilities: [78] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 16GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <16us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
[... 42 lines cut ...]
	Capabilities: [600 v1] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Capabilities: [900 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [bb0 v1] Physical Resizable BAR
		BAR 0: current size: 16MB, supported: 16MB
		BAR 1: current size: 32GB, supported: 64MB 128MB 256MB 512MB 1GB 2GB 4GB 8GB 16GB 32GB
		BAR 3: current size: 32MB, supported: 32MB
	Capabilities: [c1c v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [d00 v1] Lane Margining at the Receiver <?>
	Capabilities: [e00 v1] Data Link Feature <?>
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia
```

And I also noticed `nouveau` was in the mix, so to avoid weird issues, I blacklisted it:

```
sudo nano /etc/modprobe.d/blacklist-nouveau.conf

# Put this inside the file and save it:
blacklist nouveau
```

After a reboot, still no output through the card :/

geerlingguy, 2024-12-16:
I've uninstalled all nvidia-related packages with `sudo apt --purge remove '*nvidia*'`, and then rebooted. Then I installed Nvidia's [latest aarch64 driver](https://www.nvidia.com/en-us/drivers/unix/linux-aarch64-archive/) with:

```
wget https://us.download.nvidia.com/XFree86/aarch64/565.77/NVIDIA-Linux-aarch64-565.77.run
chmod +x NVIDIA-Linux-aarch64-565.77.run
sudo ./NVIDIA-Linux-aarch64-565.77.run
```

TODO.

bexcran, 2024-12-16:
> But I noticed, the card was back down to 2.5 GT/s (downgraded), so maybe a driver issue?!

I was confused by that a while ago too. I think it's that the driver drops the speed down to save power.
If you find something that puts the card under load, it should just up to 16 GT/s I think.

geerlingguy, 2024-12-16:
@bexcran - Interesting, I've never seen a driver dynamically change the link speed on a PCIe device! (I've done it manually before, but didn't know that was a thing any driver would ever do...).

geerlingguy, 2024-12-16:
Steam installs without a hitch using Pi-Apps now, though, lol.

bexcran, 2024-12-16:
@geerlingguy Which NVIDIA driver version did you install? Version 565.77 released a week or two ago on the New Feature Branch has quite a few fixes you probably want.

bexcran, 2024-12-16:
This has more information on the link being downgraded: https://forums.developer.nvidia.com/t/prevent-pcie-from-reducing-gen/72537

> The GPU’s own power management reduces core clock, reduces memory clock, reduces core voltage, and downgrades the PCIe link.

geerlingguy, 2024-12-16:
> Which NVIDIA driver version did you install? Version 565.77 released a week or two ago on the New Feature Branch has quite a few fixes you probably want.

I installed `NVIDIA-Linux-aarch64-565.77.run`

It seems to be handling Steam and games quite well, no issues to report right now other than GLmark not being happy, and Firefox not launching.

bexcran, 2024-12-16:
You might get more information about why Firefox is unhappy by launching it from a Terminal window.

geerlingguy, 2024-12-16:
Firefox seems to die on some GTK errors:

```
jgeerling@thelio-astra:~$ DISPLAY=:1 firefox
[58904, Main Thread] WARNING: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:187

(firefox:58904): Gtk-WARNING **: 17:48:22.839: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 17:48:22.839: Failed to load module "canberra-gtk-module"
[58904, Main Thread] WARNING: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:187

(firefox:58904): Gtk-WARNING **: 17:48:22.841: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 17:48:22.841: Failed to load module "canberra-gtk-module"
ExceptionHandler::GenerateDump attempting to generate:/home/jgeerling/snap/firefox/common/.mozilla/firefox/y3sk7gtw.default/minidumps/2788d54a-6437-4d63-72dc-fe09062bc862.dmp
ExceptionHandler::GenerateDump cloned child 59114
ExceptionHandler::SendContinueSignalToChild sent continue signal to child
ExceptionHandler::WaitForContinueSignal waiting for continue signal...
ExceptionHandler::GenerateDump minidump generation succeeded

(crashreporter:59128): Gtk-WARNING **: 17:48:24.149: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 17:48:24.149: Failed to load module "canberra-gtk-module"

(crashreporter:59128): Gtk-WARNING **: 17:48:24.150: GTK+ module /snap/firefox/4792/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 17:48:24.150: Failed to load module "canberra-gtk-module"
Exiting due to channel error.
Bus error (core dumped)
```

bexcran, 2024-12-17:
I wonder if your problems with Firefox and/or glmark are related to x11 vs wayland?
Could you check which you're running? For example, I'm using x11:
```
$ loginctl
SESSION  UID USER  SEAT  LEADER  CLASS   TTY    IDLE SINCE
      2 1000 bcran seat0 5420    user    tty2   no   -    
      3 1000 bcran -     5440    manager -      no   -    
      9 1000 bcran -     1532584 user    pts/10 no   -
$ loginctl show-session 2 -p Type
Type=x11
```

I built glmark from source hoping to debug it, but it runs fine on my system with a NVIDIA T400 card.

geerlingguy, 2024-12-17:
Same thing, even after I run `sudo apt install -y libcanberra-gtk-module`:

<details>
<summary>Click to show full output</summary>

```
jgeerling@thelio-astra:~$ DISPLAY=:1 firefox
[46210, Main Thread] WARNING: GTK+ module /snap/firefox/5434/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:201

(firefox:46210): Gtk-WARNING **: 09:26:02.449: GTK+ module /snap/firefox/5434/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 09:26:02.449: Failed to load module "canberra-gtk-module"
[46210, Main Thread] WARNING: GTK+ module /snap/firefox/5434/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:201

(firefox:46210): Gtk-WARNING **: 09:26:02.450: GTK+ module /snap/firefox/5434/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 09:26:02.450: Failed to load module "canberra-gtk-module"
ExceptionHandler::GenerateDump attempting to generate:/home/jgeerling/snap/firefox/common/.mozilla/firefox/y3sk7gtw.default/minidumps/73db0e22-dfff-58e3-de03-ce26c2833b07.dmp
ExceptionHandler::GenerateDump cloned child 46430
ExceptionHandler::WaitForContinueSignal waiting for continue signal...
ExceptionHandler::SendContinueSignalToChild sent continue signal to child
ExceptionHandler::GenerateDump minidump generation succeeded
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /usr/share/fonts/truetype/ubuntu/UbuntuSans[wdth,wght].ttf: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidia0 for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/browser/features/webcompat@mozilla.org.xpi: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/browser/features/screenshots@mozilla.org.xpi: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidia0 for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /usr/share/fonts/truetype/ubuntu/UbuntuSans[wdth,wght].ttf: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /usr/share/fonts/truetype/ubuntu/UbuntuSans[wdth,wght].ttf: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/browser/features/formautofill@mozilla.org.xpi: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /var/snap/firefox/common/fontconfig/2300eef321c393bfd76478a5c0e95b23-le64.cache-7: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /var/snap/firefox/common/fontconfig/0bd3dc0958fa2205aaaa8ebb13e2872b-le64.cache-7: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to map /dev/nvidiactl for debug info: Invalid argument (os error 22)
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/gnome-platform/usr/share/mime/mime.cache: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /home/jgeerling/snap/firefox/common/.mozilla/firefox/y3sk7gtw.default/extensions/langpack-en-US@firefox.mozilla.org.xpi: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/browser/features/pictureinpicture@mozilla.org.xpi: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/browser/omni.ja: Unknown file magic
[ERROR minidump_unwind::symbols::debuginfo] failed to parse file kind for /snap/firefox/5434/usr/lib/firefox/omni.ja: Unknown file magic
[... cut ...]
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 09:26:03.855: Failed to load module "canberra-gtk-module"

(crashreporter:46444): Gtk-WARNING **: 09:26:03.856: GTK+ module /snap/firefox/5434/gnome-platform/usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
Gtk-Message: 09:26:03.856: Failed to load module "canberra-gtk-module"
```
</details>

geerlingguy, 2024-12-17:
lol, Ubuntu's "App Store" also won't launch, for some reason (get a ton of `apparmor="DENIED"` there too).

I tried disabling apparmor with `sudo systemctl stop apparmor && sudo systemctl disable apparmor` but at least without a reboot, it seems like it's still doing stuff.

To try to get a good install of Firefox, I ran:

```
sudo snap remove firefox
sudo apt update
sudo apt install firefox
```

However, it says "firefox is already installed" :P

So I went and tried installing Firefox using Mozilla's instructions: [Install Firefox on Linux](https://support.mozilla.org/en-US/kb/install-firefox-linux#w_install-firefox-deb-package-for-debian-based-distributions-recommended)... however, it still says `firefox is already the newest version (1:1snap1-0ubuntu5)`. I'm going to try rebooting and reinstalling.

I hate snap.

geerlingguy, 2024-12-17:
Ah, found [this answer](https://askubuntu.com/a/1514036/88829) which says I have to run:

```
sudo apt purge firefox
```

...BEFORE installing the new apt version direct from Mozilla.

I did that, but it's _still_ installing the snap version, grr! `Installing the firefox snap`.

geerlingguy, 2024-12-17:
I also wanted to test Wayland instead of x11... but editing `/etc/gdm3/custom.conf` and rebooting doesn't seem to make a difference, it seems like wayland might not even be installed?

bexcran, 2024-12-17:
@geerlingguy https://manpages.ubuntu.com/manpages/xenial/man1/snap.1.html says

> remove
>       Remove a snap from the system
>
>       The remove command removes the named snap from the system.
>
>       The snap's data is currently not removed; use purge for that. This behaviour  will  change
>       before 16.04 is final.

Despite the "before 16.04 is final", I think you still need to pass "--purge" when running "snap remove".

geerlingguy, 2024-12-17:
I've also tried running `SuperTuxKart`, which I believe uses OpenGL... it also core dumps!

```
[info   ] IrrDriver: OpenGL version: 3.2
[info   ] IrrDriver: OpenGL vendor: NVIDIA Corporation
[info   ] IrrDriver: OpenGL renderer: NVIDIA RTX A4000/PCIe
[info   ] IrrDriver: OpenGL version string: OpenGL ES 3.2 NVIDIA 565.77
[info   ] IrrDriver: OpenGL total memory: 16376
[info   ] GLDriver: Explicit Attrib Location Present
[info   ] GLDriver: ARB Uniform Buffer Object Present
[info   ] GLDriver: EXT texture format BGRA8888 Present
[info   ] GLDriver: EXT Color Buffer Float Present
[info   ] GLDriver: EXT Texture Compression S3TC Present
[info   ] GLDriver: EXT Texture Compression S3TC sRGB Present
Bus error (core dumped)
```

bexcran, 2024-12-17:
I'd guess you have bits of different NVIDIA drivers on your system causing problems.
It might be best to reinstall if you can, and only use the GPL/MIT driver version 565.77 from nvidia.com.

geerlingguy, 2024-12-17:
@bexcran - Yeah. Just did that, heh.

I chose all the default options and let the installer install the Nvidia driver during install (hoping that would fare better than doing it later)... but the Nvidia DisplayPort outputs show nothing with that driver, but at least VGA works.

And FireFox launches now, yay!

`glmark2-es2` runs via llvmpipe, and doesn't die when it hits `[effect2d] kernel`.

Now I'll try uninstalling the Ubuntu drivers and installing Nvidia's again...

geerlingguy, 2024-12-17:
After a reinstall _without_ installing the drivers, the DisplayPort output does work, but the VGA output just outputs a blank screen with just a cursor (same thing as seen in the KVM in OpenBMC, though with a different cursor through the browser viewport):

[image]

To be productive, I had to switch to the DisplayPort output, right-click on the desktop, choose Display Settings, and then switch from 'Join' to 'Mirror', and then the task bar / etc was visible on the DisplayPort output (but things are pretty choppy).

Now I'm going to attempt installing Nvidia's latest driver (this time using MIT/GPL version):

```
sudo apt install -y build-essential
wget https://us.download.nvidia.com/XFree86/aarch64/565.77/NVIDIA-Linux-aarch64-565.77.run
chmod +x NVIDIA-Linux-aarch64-565.77.run
sudo ./NVIDIA-Linux-aarch64-565.77.run
```

geerlingguy, 2024-12-17:
The Nvidia installer warned about nouveau being loaded:

>  One or more modprobe configuration files to disable Nouveau have been written.  You will need to reboot your system   
  and possibly rebuild the initramfs before these changes can take effect.  Note if you later wish to reenable          
  Nouveau, you will need to delete these files: /usr/lib/modprobe.d/nvidia-installer-disable-nouveau.conf,              
  /etc/modprobe.d/nvidia-installer-disable-nouveau.conf 

So I've let it write those files and update initram. The machine won't reboot, it just kinda gets stuck (graphically), with a softlock in the SOL console. I had to hard-reset via BMC (or front panel power button).

Trying again...

TODO.

geerlingguy, 2024-12-17:
So...

  - Firefox still won't launch with the Nvidia driver installed
  - `glmark2-es2` still locks up after the `[bump] bump-render=height`, with `SIGBUS in __memcpy_generic()`
  - `supertuxkart` still won't launch

I'd love to know the magical incantation to get all these things working with a fresh Ubuntu 24.04 install! :(

bexcran, 2024-12-18:
Didn't you install the Ubuntu NVIDIA drivers first, _then_ the version downloaded from nvidia.com?
I think that might be where you're going wrong.

geerlingguy, 2024-12-18:
@bexcran - No, this last time I installed Ubuntu, I made absolutely sure _not_ to check the box for 'Install 3rd party drivers', and indeed, once the install was complete, nouveau was the only thing installed (part of the kernel), which is also why the behavior was so flaky until I got the Nvidia driver installed (which blacklists the nouveau module).

So I've now installed Ubuntu 24.04 Desktop fresh 3 times:

  - Install no 3rd party drivers (during initial install), then install `nvidia-driver-550` using ubuntu-drivers, then uninstall that, then install Nvidia proprietary driver 565.77
  - Install 3rd party drivers, then uninstall `nvidia-driver-550`, then install Nvidia proprietary driver 565.77
  - Install no 3rd party drivers (during initial install), then install Nvidia proprietary driver 565.77

So far at no point in any of these processes have I been able to get Firefox to launch, nor GL-based applications to work, under Ubuntu on this machine :(

With AMD and the default / built-in `amdgpu` driver, it all works fine, barring some text rendering glitches. (That's assuming I unplug the Nvidia card and plug in the AMD—if I have both plugged in, the Nvidia driver, whether Ubuntu's or from Nvidia directly, seems to screw something up).

geerlingguy, 2024-12-18:
It'd be helpful to know _exactly_ what steps System76 takes to build or install a complete Ubuntu 24.04 Desktop environment, from scratch, because the install that came from them seemed to be _more_ functional (and worked with the Ubuntu 550 driver, at least), it just had some weird issues with package management, which is why I wanted to go to a clean install in the first place.

ahoneybun, 2024-12-18:
I don't know if these steps are fully completed:

https://github.com/system76/tech-docs/blob/astra/src/models/thelio-astra-a1-n1/ubuntu-24.04.md

https://github.com/system76/tech-docs/blob/astra/src/models/thelio-astra-a1-n1/ubuntu-20.04.md

geerlingguy, 2024-12-18:
@bexcran - https://cdimage.ubuntu.com/noble/daily-live/current/ — that's the only source I've found for an arm64 desktop build

ahoneybun, 2024-12-19:
@geerlingguy yep glad to help!

geerlingguy, 2024-12-19:
@ahoneybun - for one bit:

> Ensure that DMI information is properly set at this point. If not, set it and reboot.

What does that mean? Or is that for internal setup processes?

geerlingguy, 2024-12-19:
It looks like one of the last steps loads in `nvidia-560.35.03` (when I do a `dist-upgrade` after adding `ppa:system76-dev/stable` and installing the system76 driver...

ahoneybun, 2024-12-19:
> @ahoneybun - for one bit:
> 
> > Ensure that DMI information is properly set at this point. If not, set it and reboot.
> 
> What does that mean? Or is that for internal setup processes?
> 
> Also, there was a step saying to open `nm-connection-editor`, but when I typed it in the CLI, I got:
> 
> ```
> jgeerling@thelio-astra:~$ nm-connnection-editor
> Command 'nm-connnection-editor' not found, did you mean:
>   command 'nm-connection-editor' from deb network-manager-gnome (1.34.0-1ubuntu1)
> Try: sudo apt install <deb name>
> ```
> 
> Is that something I need to do through the GUI?

I'm not sure about DMI, I think is set in firmware. For network manager I don't think it is installed on a  server install and instead netplan is.

ahoneybun, 2024-12-19:
> I'm not sure about DMI, I think is set in firmware. For network manager I don't think it is installed on a server install and instead netplan is.

To double reply to this question, the DMI is checked by the system76-driver software to note that it is a System76 product and apply any quirks that the hardware needs. I don't think that applies to you and other end users in this case.

geerlingguy, 2024-12-19:
@ahoneybun - After following all those directions, I wound up not getting a display output through the graphics card (A4000 at least), here's the dmesg logs:

```
[   19.623218] nvidia: module license 'NVIDIA' taints kernel.
[   19.623228] nvidia: module license taints kernel.
[   19.752203] nvidia-nvlink: Nvlink Core is being initialized, major device number 508
[   19.759505] nvidia 0004:01:00.0: Adding to iommu group 26
[   19.762455] nvidia 0004:01:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[   19.829971] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  560.35.03  Fri Aug 16 21:13:32 UTC 2024
[   19.838555] [drm] [nvidia-drm] [GPU ID 0x00040100] Loading driver
[   21.484369] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0004:01:00.0 on minor 2
[   21.509226] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
[   21.522424] nvidia-uvm: Loaded the UVM driver, major device number 506.
```

VGA is working, and `nvidia-smi` works:

```
jgeerling@thelio-astra:~$ nvidia-smi
Thu Dec 19 23:26:11 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.35.03              Driver Version: 560.35.03      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA RTX A4000               Off |   00000004:01:00.0 Off |                  Off |
| 41%   33C    P8              6W /  140W |      15MiB /  16376MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      4194      G   /usr/lib/xorg/Xorg                              4MiB |
+-----------------------------------------------------------------------------------------+
```

I'm just getting no display output through DisplayPort... this was the same behavior I found installing Ubuntu Desktop and using it's built-in 550 version of the driver.

The System76 software package update installed `560.35.03`, which is slightly newer, but not as new as the `565.77` driver that I had to install previously to get display output working (but which seemed to wreck Firefox and OpenGL...).

geerlingguy, 2024-12-20:
There is a new simpler instruction for setting everything up — I will test this out today!

https://github.com/system76/tech-docs/blob/astra/src/models/thelio-astra-a1-n1/ubuntu-24.04.md

Basically:

  1. Download Ubuntu Desktop (daily): https://cdimage.ubuntu.com/noble/daily-live/20240827/noble-desktop-arm64.iso
  2. Install Ubuntu Desktop
  3. Run the following commands:

```
sudo apt-get update && sudo apt-get dist-upgrade
sudo add-apt-repository ppa:system76-dev/stable
sudo apt-get install system76-driver-nvidia
reboot
```

ahoneybun, 2024-12-20:
@geerlingguy just to confirm when you added the PPA and driver did it also upgrade the kernel to the system76 6.9.3 or are you still on the 6.8.0 Ubuntu kernel?

geerlingguy, 2024-12-20:
@ahoneybun - Ah, looks like I'm still on:

```
jgeerling@thelio-astra:~$ uname -a
Linux thelio-astra 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:32:09 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

How should I change that?

ahoneybun, 2024-12-20:
No that is what I have on this lab unit with a RTX 4090. Can you share this output?

apt policy system76-driver-nvidia
apt policy nvidia-driver-560

geerlingguy, 2024-12-20:
@ahoneybun - Oh wait!

I ran one more cycle of `full-upgrade` after the reboot (just to see if any more packages would need updating...), and I got:

```
jgeerling@thelio-astra:~$ sudo apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  chrony tzdata-legacy
The following upgrades have been deferred due to phasing:
  python3-distupgrade ubuntu-release-upgrader-core ubuntu-release-upgrader-gtk
The following packages will be upgraded:
  libnss-systemd libpam-systemd libsystemd-shared libsystemd0 libudev1 linux-tools-common systemd systemd-dev
  systemd-oomd systemd-resolved systemd-sysv systemd-timesyncd udev
13 upgraded, 2 newly installed, 0 to remove and 3 not upgraded.
Need to get 11.9 MB of archives.
After this operation, 876 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
```

After a reboot, the display output is working! Firefox is working... running GLmark2 now.

ahoneybun, 2024-12-20:
@geerlingguy I also see the issue with `glmark2-es2` so that might be something with mesa or something related.

[37 more replies]