geerlingguy/sbc-reviews #56: ArmSoM Sige7 / Banana Pi BPI-M7

geerlingguy, 2024-11-08:
[image]

## Basic information

  - Board URL (official): https://www.armsom.org/sige7 (or https://banana-pi.org/en/banana-pi-sbcs/169.html)
  - Board purchased from: Provided by ArmSoM/BPI
  - Board purchase date: November 8, 2024
  - Board specs (as tested): 32G RAM / 128G eMMC
  - Board price (as tested): $244.40 ('Pro Max')

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-11-08:
Regarding OSes, ArmSoM provides a download link to a Google Drive folder with a bunch of distros to choose from: https://drive.google.com/drive/folders/1ijH2PoVUtHwe7fyiIpiQ7bsDZV3nx-MB

I am going to try out Debian, specifically `armsom-sige7-debian-bullseye-xfce4-arm64-20241029`

geerlingguy, 2024-11-08:
Hmm... the device seems not to boot up with that image flashed to a microSD card.

When I plug in power to the 'Power' USB-C port, I see the red LED light up, but nothing else happens, and there's no HDMI output either. Also no blinking on the Ethernet port (I'm plugged into the 'LAN' port).

geerlingguy, 2024-11-08:
Trying again with `ubuntu-22.04-preinstalled-desktop-arm64-armsom-sige7` this time...

When I flashed the Debian image, Etcher gave me a warning about the partitions, it didn't do it with this Ubuntu image, so maybe that's a good sign...

Well... halfway through the image writing process, I get this warning:

[image]

I shall now attempt with `ubuntu-22.04.3-preinstalled-server-arm64-armsom-sige7`!

geerlingguy, 2024-11-08:
The Ubuntu server image does seem to boot—I get both the 'R' and 'G' LEDs lit up when I plug in power, and both remain lit constantly. I got HDMI output after 5-10 seconds, and that was all well and good... but then a few seconds later, it would shut back down / reset. Going to capture the screen output to see where in boot it's failing.

The Ethernet jack even lights up for about .5 seconds towards the end of the false-boot-loop.

geerlingguy, 2024-11-08:
Ah, just found [the Sige7 FAQ](https://docs.armsom.org/ru/armsom-faqs#sige7-faq):

> Q1: My ArmSoM-Sige7 cannot boot / is stuck in an infinite boot loop.
> 
> The ArmSoM-Sige7 supports USB PD power negotiation for higher voltage requirements, such as 9V, 12V, 15V, and 20V, to meet the total system power load requirements (around 30W, including WiFi, SSD, USB peripherals, etc.). Currently, PD negotiation is implemented in the kernel driver, and we need to boot to the kernel to start the negotiation. However, some power supplies may not be patient while waiting for the ArmSoM-Sige7 to negotiate and will cut off power when a timeout occurs, leading to an infinite boot loop.

geerlingguy, 2024-11-08:
Also to boot off NVMe, they just posted this in the forum: [[Sige7 installation system to NVMe startup](https://forum.armsom.org/t/sige7-installation-system-to-nvme-startup/234)](https://forum.armsom.org/t/sige7-installation-system-to-nvme-startup/234)

quintuple-lained, 2024-11-08:
now this might be a stupid question but wouldnt chrooting into the boot medium from a different machine and running passwd solve the login issue?

geerlingguy, 2024-11-09:
That could, however I don't have my Linux workstation booted up right now, and also don't have my SD card reader plugged into it, it'd be a lot of friction to hack at the default password that way. I also want to make sure I can set this thing up in a way I could recommend to viewers of a potential video in the future, I don't want people to have to have a Linux machine they can use to hack in their own password on the base install :)

EyalRo, 2024-11-09:
If you have access to the filesystem, there's always the trick where you edit /etc/shadow and overwrite the row for root with an empty password `root::0:0:root:/root:/bin/bash` then login with root and no password

itlhd, 2024-11-11:
> Regarding OSes, ArmSoM provides a download link to a Google Drive folder with a bunch of distros to choose from: https://drive.google.com/drive/folders/1ijH2PoVUtHwe7fyiIpiQ7bsDZV3nx-MB
> 
> I am going to try out Debian, specifically `armsom-sige7-debian-bullseye-xfce4-arm64-20241029`

This system requires SDDiskTool to run from sdcard, You can refer to the official documentation for [3. Burn Debian system](https://docs.armsom.org/general-tutorial/flash-img)

itlhd, 2024-11-11:
> Now that I have this thing booting... I don't know how to log in. I've tried `ubuntu`/`ubuntu`, `armsom`/`armsom`, and `root`/`root` so far...
> 
> After seeing the issue [Sige7 username and password](https://forum.armsom.org/t/sige7-username-and-password/176) and trying `armsom` a few more times... I tried registering an account on the ArmSoM forum, but after waiting 10 minutes I still haven't received the activation email.
> 
> So I can't ask there for help, maybe this image is not the best image to use, but the Debian image did not seem to be able to flash at all... 1 hour in and I haven't been able to boot this board to the point I can access it yet :(

When third-party images are first launched, custom account passwords are required. The default account password for official images is armsom/armsom
The [forum](https://forum.armsom.org/) should be able to register now

geerlingguy, 2024-11-11:
@itlhd - Is there any way to prepare an SD card without having to use a Windows-only tool like SDDiskTool? I normally work on a Mac, and have Linux always booted next to me as well, it's inconvenient to have to move my SD reader over to my Windows PC, which is normally not powered on.

Otherwise, I may also try with Armbian; it seems like they provide a well-supported image: https://www.armbian.com/bananapi-m7/

geerlingguy, 2024-11-11:
I flashed the latest `Armbian 24.8.1 Noble Gnome` image to a microSD card using Etcher, but it doesn't seem to boot—the green 'G' LED doesn't light like it does with the Ubuntu image, and I get no output over HDMI either.

I'd like to have a way to get an image onto this device without employing a Windows PC. I may do it just to get it going, but I don't want to recommend a device to someone that _requires_ Windows to get it booted :/

itlhd, 2024-11-12:
> armsom-sige7-debian-bullseye-xfce4-arm64-20241029

This post may be helpful to you：[Upgrading Sige7 Image in Linux](https://forum.armsom.org/t/upgrading-sige7-image-in-linux/237)

as-jackson, 2024-11-12:
Hello Jeff, we applied for a new sige7 (32+128) today after seeing the problem.
We used balenaetcher to burn [armsom-sige7](https://dl.armbian.com/armsom-sige7/Noble_current_gnome-kisak)
After completion, we inserted the sd into armsom-sige7 and everything seemed to be normal. When you start it for the first time, you need to plug in the mouse, keyboard and HDMI, and you need to set your account and password.
[image]
After I set the password, I entered the desktop system normally.
[image]

We understand that it is not a pleasant thing when customers or people who watch the video cannot start Linux after getting the product. We will try our best to find out what problems you encounter and solve them.

Can you see the log after burning the armbian system to the sd card and inserting the sige7 to start it?

as-jackson, 2024-11-13:
Ubuntu system is ok [ubuntu sige7](https://joshua-riek.github.io/ubuntu-rockchip-download/boards/armsom-sige7.html)
[image]

as-jackson, 2024-11-13:
If you want to use armsom-sige7-debian-bullseye-xfce4-arm64-20240802.img, this is the chip manufacturer's sdk firmware, which does not support balenaEtcher. You need to use the original tool [SDDiskTool_v1.78](https://drive.google.com/drive/folders/12ucrWgg0JfyyxGiSbTCliuQg-14nEhPe)
[image]
[image]