geerlingguy/sbc-reviews #57: M4 Pro Mac Mini

geerlingguy, 2024-11-11:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.apple.com/mac-mini/
  - Board purchased from: Apple (direct)
  - Board purchase date: October 29, 2024 (arrived Nov 11, 2024)
  - Board specs (as tested): M4 Pro 10/10/16-core, 32GB RAM, 1TB SSD, 10 GbE
  - Board price (as tested): 1499.00

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2`
3. Replace this block of text with the results.
```

Note: This benchmark requires an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2024-11-12:
I think my setup at my home desk leaves something to be desired. I noticed the M4 Pro CPU spikes to 100°C during multi-core tests like Cinebench. Geekbench 6 doesn't hit the cores as hard for as long, but it may have brief throttling as well, affecting the overall score.

[image]

I'm going to re-run all the tests in a better test environment at the studio tomorrow.

ThomasKaiser, 2024-11-12:
> why Apple still doesn't just go to M.2 is crazy

M.2 as in 'standard NVMe SSD with own controller'? Well, two reasons against:

  * lower profits
  * power efficiency ruined. At least with any consumer SSD so far I experienced **way** higher consumption figures than without (SBC use case: eMMC vs. OS on NVMe) and I guess a lot of what Apple is doing with the I/O subsystem is optimizing for both performance and battery life (or low consumption in the Mini's case). Unfortunately I haven't seen any tests so far emphasizing on the 'race to idle' concept also including I/O but by doing any fine-grained measurements with any Apple Silicon HW it's obvious that the I/O subsystem is most of the times in 'deep sleep' state or something lower (not that familiar with terminology in this area)

geerlingguy, 2024-11-12:
At the office, where the Mac mini is in open air (ambient temp 23°C / 73°F), Cinebench is still pushing the SoC to 100°F pretty quickly. Maybe Apple's fan curve isn't aggressive enough?

I can hear the fan, but it's certainly very quiet. Only barely audible in the studio.

geerlingguy, 2024-11-12:
After installing Macs Fan Control and setting the fan to 4900 rpm (max), the fan is audible (sounds about like the Qualcomm Dev Kit, lol), and temps are now down to 80°C after a minute:

[image]

Going to re-run Cinebench multi at max fan speed to see how it fares, and compare the two runs:

| Test | Default fan settings (1900 rpm) | High fan speed (4000 rpm) |
| --- | --- | --- |
| Cinebench 2024 Single | TODO | TODO |
| Cinebench 2024 Multi | TODO | TODO |
| Geekbench 6.3.0 Single | TODO | TODO |
| Geekbench 6.3.0 Multi | TODO | TODO |

geerlingguy, 2024-11-12:
Cinebench results with Apple's default fan curve:

[image]

Cinebench results with 3000 rpm fan speed:

TOOD

geerlingguy, 2024-11-12:
Heh... I thought I had ordered an M4 Pro. Little did I know, I ordered an M4. Oopsie! It's still way faster than my M2 Air I'm replacing, and it still has 32 GB of RAM (double my Air), 1 TB HDD (double my Air), and 10 GbE (which my Air had to use a TB3 dongle for), so I'm happy with it, but disappointed I didn't look as closely when I ordered it.

Though looking at the prices now, I'm okay with not paying the like $500 premium for the Pro + RAM upgrade.

geerlingguy, 2024-11-12:
Here's the power graph through the Cinebench runs:

[image]

geerlingguy, 2024-11-12:
@ThomasKaiser - Since the `sec-bench.sh` script doesn't run on macOS/Darwin, would you like any tests in particular since I have my machine up and ready for some benchmarks? (Outside of what I've already run.)

geerlingguy, 2024-11-12:
I ran my https://github.com/geerlingguy/ollama-benchmark a couple times, even tried loading up the llama3.1:70b model but that tried consuming almost 40 GB of RAM—which meant 12 GB of swap, and slaughtered the performance. Need more RAM to run larger models.

andrewginns, 2024-11-13:
Would be great to run this standardised benchmark of llama models!

https://github.com/ggerganov/llama.cpp/discussions/4167

film42, 2024-11-13:
> I can hear the fan, but it's certainly very quiet. Only barely audible in the studio. Fan is at 1920 rpm

That's good news, but ouch! That seems like a really slow fan. Curious to know how well it handles high-res video transcoding. I'm sure I'm not the only one who thinks this screams "my next media server" for only $550 (on Amazon).

nreilly, 2024-11-13:
> At the office, where the Mac mini is in open air (ambient temp 23°C / 73°F), Cinebench is still pushing the SoC to 100°F pretty quickly. Maybe Apple's fan curve isn't aggressive enough?
> 
> I can hear the fan, but it's certainly very quiet. Only barely audible in the studio. Fan is at 1920 rpm

Please consider testing in [High Power Mode](https://support.apple.com/en-us/101613).

geerlingguy, 2024-11-13:
@nreilly - Low Power mode has been switched off in all my testing:

[image]

nreilly, 2024-11-13:
> @nreilly - Low Power mode has been switched off in all my testing

The Mac mini (2024) should have 3 options for the energy level. Low, Automatic and High. The [Apple article](https://support.apple.com/en-us/101613) indicates it's available for the Mac mini (2024), and doesn't clarify it's only for the M4 Pro versions, so I would expect it to be an option for you, but your screenshot obviously says no.

[image]

imadcat, 2024-11-13:
Great thx for sharing! Is there also a computer vision (non LLM) AI benchmark result?

JayBrown, 2024-11-13:
If only APFS and Disk Utility offered RAID5 functionality… I'd use this as a home server in a heartbeat, with two external M.2 thunderbolt enclosures, each sporting an M.2 gen3 x2 to 6xSATA adapter… using three SATA SSDs on each adapter for a combined storage pool. Afaik the only way to implement RAID5 is by using OWC's SoftRAID, and that's a subscription model, and I don't know if we should trust their software for an application as fundamental as storage. OpenZFS seems to be buggy as hell on macOS, so RAIDz1 is probably out of the question. Come on, Apple, give us more RAID levels already. 🙏

mibosshard, 2024-11-13:
the M4 (non-Pro) only has the Low Power Mode. only M4 Pro Mac mini has High Power Mode. in Low Power Mode, the M4 SoC consumes less than 7 W under full load (HandBrake H.265 encoding on the CPUs).
https://browser.geekbench.com/v6/cpu/compare/8799758?baseline=8802915

geerlingguy, 2024-11-20:
I unplugged 10 GbE and just used WiFi 6, and idle power consumption goes from 6W to 4.1W.

The HPL efficiency score increased from 6.74 Gflop/W to **7.57 Gflops/W** (much wow).

film42, 2024-11-20:
Wow! Does apple power down the radio completely when wifi and Bluetooth are disabled? Can’t help but wonder if radios are using power at idle when 10 GbE is connected.

ThomasKaiser, 2024-11-20:
> I unplugged 10 GbE and just used WiFi 6

Have you also tried to 'downgrade' the Ethernet link by connecting to a GbE or 2.5GbE switch port and measured idle consumption?

geerlingguy, 2024-11-20:
@ThomasKaiser - I haven't tried that yet. May do so soon, left my power cord at home so I can't test right now :(

geerlingguy, 2024-11-20:
For some comparisons:

[image]

[image]

akarabach, 2024-11-21:
Looks like a good candidate to run as a home server. With 2 thunderbolt 5 ports its not a problem anymore to connect even 990 pro ssds. Would love to see such kind of video on the channel!

ThomasKaiser, 2024-11-21:
> Since the sbc-bench.sh script doesn't run on macOS/Darwin, would you like any tests in particular

@geerlingguy in case the Mini is still in 'testing territory' with Xcode installed it should take just a few minutes of your time to get 7-zip 16.02 scores (16.02 to be comparable with `sbc-bench` scores – versions from 17.04 on up may perform different):

  git clone https://github.com/ThomasKaiser/p7zip
  cd p7zip
  make -j$(sysctl -n hw.ncpu) INSTALL=install CC=gcc CXX=g++ OPTFLAGS='-O3'
  for i in 1 2 3 4 5 ; do bin/7za b -mmt=1 | grep "Tot:" ; done | sort -n | tail -n1 ; for i in 1 2 3 4 5 ; do bin/7za b -mmt=6 | grep "Tot:" ; done | sort -n | tail -n1

This will download 16.02 sources (with all known vulnerabilities patched), build from source and run the 7-zip benchmark single-threaded and on all 6 P-cores in parallel (5 continous runs each, then displaying best score each).

Numbers to be compared with (counting/measuring only P-cores on the Macs):

  * 192-core AmpereOne A192-32X (3200 MHz): single: 4783, multi: 745720 (155.91 ratio)
  * 8-core Apple M1 Pro (3228 MHz): single: 5540, multi: 46540 (8.39 ratio)
  * 8-core Apple M1 Max (3228 MHz): single: 5845, multi: 52170 (8.93 ratio)
  * 8-core Apple M2 Pro (3504 MHz): single: 6830, multi: 57120 (8.36 ratio)
  * 4-core Apple M3 (4054 MHz): single: 7570, multi: 31380 (4.15 ratio)
  * 6-core Apple M4 (? MHz): single: ?, multi: ? (? ratio)

(the ratio for all the Apple results is obviously strange since should be lower than count of threads)

I checked parallel with `sudo powermetrics | grep frequency` in a following run whether the cpufreqs were consistent when running single-threaded vs. multi and found on the systems tested no difference (unlike M1 and M2 in the past that were clocked lower when all P-cores were active compared to only 1 or 2).

DaveHarwoodNZ, 2024-12-06:
> > why Apple still doesn't just go to M.2 is crazy
> 
> M.2 as in 'standard NVMe SSD with own controller'? Well, two reasons against:
> 
> * lower profits
> * power efficiency ruined. 

The integrated controller appears to offer low-overhead encryption which shows almost no performance degradation. Not that many people will compare devices with encryption enabled.

ThomasKaiser, 2025-04-07:
> @geerlingguy in case the Mini is still in 'testing territory' with Xcode installed it should take just a few minutes of your time to get 7-zip 16.02 scores (16.02 to be comparable with sbc-bench scores – versions from 17.04 on up may perform different):

Since an M4 Air arrived today, numbers below:

  * 192-core AmpereOne A192-32X (3200 MHz): single: 4783, multi: 745720 (155.91 ratio)
  * 8-core Apple M1 Pro (3228 MHz): single: 5540, multi: 46540 (8.39 ratio)
  * 8-core Apple M1 Max (3228 MHz): single: 5845, multi: 52170 (8.93 ratio)
  * 8-core Apple M2 Pro (3504 MHz): single: 6830, multi: 57120 (8.36 ratio)
  * 4-core Apple M3 (4054 MHz): single: 7570, multi: 31380 (4.15 ratio)
  * 4-core Apple M4 (4400 MHz): single: 8477, multi: 35356 (4.17 ratio)

Or in other words: memory performance hasn't increased the same way as CPU clockspeeds comparing M4 to M3.