geerlingguy/sbc-reviews #58: Raspberry Pi Compute Module 5

geerlingguy, 2024-11-27:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.raspberrypi.com/products/compute-module-5/
  - Board purchased from: Provided by Raspberry Pi
  - Board purchase date: November, 2024
  - Board specs (as tested): 4GB RAM, 32GB eMMC (also 8GB RAM, 8GB eMMC as noted)
  - Board price (as tested): TODO

All tests were run on the 4GB board, except as noted. Some tests scale with RAM.

## Linux/system information

```
# output of `screenfetch`
         _,met$$$$$gg.           pi@cm5
      ,g$$$$$$$$$$$$$$$P.        OS: Debian 12 bookworm
    ,g$$P""       """Y$$.".      Kernel: aarch64 Linux 6.6.51+rpt-rpi-2712
   ,$$P'              `$$$.      Uptime: 12m
  ',$$P       ,ggs.     `$$b:    Packages: 1630
  `d$$'     ,$P"'   .    $$$     Shell: bash 5.2.15
   $$P      d$'     ,    $$P     Disk: 9.4G / 237G (5%)
   $$:      $$.   -    ,d$$'     CPU: ARM Cortex-A76 @ 4x 2.4GHz
   $$\;      Y$b._   _,d$P'      GPU: 
   Y$$.    `.`"Y$$$$P"'          RAM: 677MiB / 4045MiB
   `$$b      "-.__              
    `Y$$                        
     `Y$$.                      
       `$$b.                    
         `Y$$b.                 
            `"Y$b._             
                `""""      

# output of `uname -a`
Linux cm5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

## Benchmark results

### CPU

  - Geekbench 6: (804 single / 1651 multi - https://browser.geekbench.com/v6/cpu/9095882)
  - 26.15 Gflops (2.79 Gflops/W) ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) [HPL result](TODO)) - 31.48 Gflops for 8GB (3.147 Gflops/W)
  - TODO: Re-test with microSD card or eMMC CM5.

### Power

  - Idle power draw (at wall): 2.4 W (2.3 W for 'Lite' models)
  - Maximum simulated power draw (`stress-ng --matrix 0`): 8.5 W
  - During Geekbench multicore benchmark: 8.2 W
  - During `top500` HPL benchmark: 9.4 W (10 W on 8GB)

### Disk

#### Pinedrive 256GB 2242 NVMe SSD at PCIe Gen 3

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 63.01 MB/s |
| iozone 4K random write     | 298.06 MB/s |
| iozone 1M random read      | 820.16 MB/s |
| iozone 1M random write     | 759.24 MB/s |
| iozone 1M sequential read  | 823.04 MB/s |
| iozone 1M sequential write | 758.51 MB/s |

#### Built-in eMMC (32GB)

| Benchmark                  | Result |
| -------------------------- | ------ |
[... 251 lines cut ...]
   4194304 :   51.5 ns          /    77.4 ns 
   8388608 :   79.8 ns          /   108.0 ns 
  16777216 :   94.9 ns          /   119.5 ns 
  33554432 :  104.4 ns          /   126.1 ns 
  67108864 :  110.0 ns          /   130.3 ns
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here: TODO

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: 11.739 sec
  - pts/x264 4K: 4.32 fps
  - pts/x264 1080p: 18.06 fps
  - pts/phpbench: 435778
  - pts/build-linux-kernel (defconfig): 2222.151 sec

## Other benchmarks

  - Boot time (Pi OS 64-bit Desktop): 22.92s to SSH login, 23.90s to GUI with menu bar

schoolpost, 2024-11-27:
Thanks for all the thorough performance numbers, can you confirm CM5 uses the D0 variant of the BCM2712?

geerlingguy, 2024-11-27:
@schoolpost - All the ones I've seen are D0, yes.

ThomasKaiser, 2024-11-27:
Do you have any information what the term ECC in 'LPDDR4-4267 SDRAM with ECC' should exactly mean?

geerlingguy, 2024-11-27:
[Home Assistant Yellow is also a drop-in upgrade](https://www.jeffgeerling.com/blog/2024/home-assistant-yellow-instant-2x-iot-speedup-cm5), in addition to all the Compute Module carrier boards I mentioned in my video: https://www.youtube.com/watch?v=X4blR5Ua3S0

melroy89, 2024-12-01:
Are you sure the Broadcom BCM2712 has H264 hardware decode/encode bock? I read a lot online that HEVC decoder in the BCM2712 is not provided by Broadcom, but designed by the Raspberry Foundation themselves. And with this BCM2712 chip they removed H264 decoder totally:

https://www.phoronix.com/forums/forum/hardware/processors-memory/1412037-raspberry-pi-5-benchmarks-significantly-better-performance-improved-i-o?p=1412202#post1412202

geerlingguy, 2024-12-01:
@melroy89 - It has H.265 decode (up to 4K 60 fps), but not hardware H.264 encode or decode (nor any other encode/decode, at least none that is usable by Raspberry Pi... there could be dark silicon on there that Broadcom used for other customers).

CM5 is fully capable of 1080p encode and decode, and can even hit 4K depending on bitrate and what you're after, outside of H.265. See: [Can the Raspberry Pi 5 handle 4K?](https://www.jeffgeerling.com/blog/2024/can-raspberry-pi-5-handle-4k).

I haven't yet tested the CM5 on IP KVM cards—in my video I was clear that the Compute Module fits, but that and many other CM4 boards have not been tested with CM5 yet. For tracking on _that_ testing, please follow this issue: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/686

geerlingguy, 2024-12-09:
Added some results up above, but here's new SBC-bench results: https://0x0.st/Xh2H.txt

Some interesting diffs:

```diff
- libc memcpy copy                                 :   5705.0 MB/s (3, 0.2%)
- libc memchr scan                                 :  13722.6 MB/s (2)
- libc memset fill                                 :  12567.9 MB/s (3, 0.9%)
+ libc memcpy copy                                 :   5892.4 MB/s (2)
+ libc memchr scan                                 :  14188.5 MB/s (2)
+ libc memset fill                                 :   9675.1 MB/s (3, 1.4%)

-  * memcpy: 5705.0 MB/s, memchr: 13722.6 MB/s, memset: 12567.9 MB/s
-  * 16M latency: 119.2 118.6 118.8 117.8 120.8 134.6 130.2 139.9 
-  * 128M latency: 136.3 135.2 147.2 135.1 136.4 134.9 135.8 137.1 
-  * 7-zip MIPS (3 consecutive runs): 11250, 11265, 11217 (11240 avg), single-threaded: 3164
-  * `aes-256-cbc     540307.67k  1003613.67k  1256029.53k  1332866.39k  1365516.29k  1367834.62k`
-  * `aes-256-cbc     540608.23k  1003560.11k  1255918.42k  1332845.23k  1365215.91k  1368135.00k`
+  * memcpy: 5892.4 MB/s, memchr: 14188.5 MB/s, memset: 9675.1 MB/s
+  * 16M latency: 100.1 101.0 103.3 100.9 99.88 114.8 130.4 146.9 
+  * 128M latency: 116.7 115.2 116.6 115.2 119.5 115.8 116.6 118.4 
+  * 7-zip MIPS (3 consecutive runs): 11819, 11858, 11809 (11830 avg), single-threaded: 3306
+  * `aes-256-cbc     540521.66k  1003777.26k  1256054.36k  1332929.88k  1365549.06k  1368053.08k`
+  * `aes-256-cbc     540683.01k  1003568.53k  1256005.03k  1332878.68k  1365235.03k  1368211.46k`
```

BlackS52, 2025-05-22:
Hi all,
I'm newbie in that topic and try to deep dive to it. 
I'm trying to benchmark the same board and get the same results.
I have a trouble. I can't find tinymembench v0.4.**10**  Only tinymembench v0.4.09 but it looks like that it didn't optimize perferctly
Can you share where I can get tinymembench v0.4.**10** ?

geerlingguy, 2025-05-22:
I build from https://github.com/rojaster/tinymembench from the main source repository, and that is set to .10 (which is not yet technically released).

BlackS52, 2025-05-22:
> I build from https://github.com/rojaster/tinymembench from the main source repository, and that is set to .10 (which is not yet technically released).

Ohh Hadn't even thought of that. Thank you.
And maybe you can share your flags and compiler version? 
Just to repeat and validate my board. Just to check that my board have the same perf like your

BlackS52, 2025-05-22:
> [@BlackS52](https://github.com/BlackS52) all the exact commands I run are done through automation so they're consistent board to board — here's the Pyinfra script, you can pull the commands out of the server.shell tasks: https://github.com/geerlingguy/sbc-reviews/blob/master/benchmark/tasks/tinymembench.py

Thanks a lot