geerlingguy/sbc-reviews #60: Raspberry Pi 500

geerlingguy, 2024-12-09:
[image]

## Basic information

  - Board URL (official): https://www.raspberrypi.com/products/raspberry-pi-500/
  - Board purchased from: Provided by Raspberry Pi
  - Board purchase date: Nov 27, 2024
  - Board specs (as tested): 8GB RAM
  - Board price (as tested): $90 ($120 if purchased in Kit)

## Linux/system information

```
# output of `screenfetch`
         _,met$$$$$gg.           pi@pi500
      ,g$$$$$$$$$$$$$$$P.        OS: Debian 12 bookworm
    ,g$$P""       """Y$$.".      Kernel: aarch64 Linux 6.6.51+rpt-rpi-2712
   ,$$P'              `$$$.      Uptime: 8m
  ',$$P       ,ggs.     `$$b:    Packages: 1579
  `d$$'     ,$P"'   .    $$$     Shell: bash 5.2.15
   $$P      d$'     ,    $$P     Disk: 4.9G / 33G (16%)
   $$:      $$.   -    ,d$$'     CPU: ARM Cortex-A76 @ 4x 2.4GHz
   $$\;      Y$b._   _,d$P'      GPU: 
   Y$$.    `.`"Y$$$$P"'          RAM: 861MiB / 8048MiB
   `$$b      "-.__              
    `Y$$                        
     `Y$$.                      
       `$$b.                    
         `Y$$b.                 
            `"Y$b._             
                `""""

# output of `uname -a`
Linux pi500 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - 29.812 Gflops / **2.59 Gflops/W** ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) [HPL result](TODO))

### Power

  - Idle power draw (at wall): 3.1 W
  - Maximum simulated power draw (`stress-ng --matrix 0`): 9 W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: 11.5 W

### Disk

#### Raspberry Pi A2 microSD Card (32GB)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 9.58 MB/s |
| iozone 4K random write     | 2.65 MB/s |
| iozone 1M random read      | 88.08 MB/s |
| iozone 1M random write     | 21.99 MB/s |
| iozone 1M sequential read  | 88.93 MB/s |
| iozone 1M sequential write | 22.71 MB/s |

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

#### Built-in Ethernet

`iperf3` results:
[... 218 lines cut ...]
   8388608 :   83.8 ns          /   115.1 ns 
  16777216 :   99.2 ns          /   126.9 ns 
  33554432 :  108.9 ns          /   133.4 ns 
  67108864 :  114.5 ns          /   137.2 ns 
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: 11.775 sec
  - pts/x264 4K: 4.12 fps
  - pts/x264 1080p: 17.41 fps
  - pts/phpbench: 433941
  - pts/build-linux-kernel (defconfig): 2486.594 sec

srcshelton, 2024-12-10:
Any clues yet as to what the populated 2-pin header just below the GPIO connector might be?

I was perhaps hoping for a way to connect an RTC battery/super-capacitor, but a UART port for the Debug Probe would also be useful!

geerlingguy, 2024-12-11:
@srcshelton 99% certain that's UART for like the Pi Debug Probe. And there's an RTC battery header space to the right of the Keyboard connector.