geerlingguy/sbc-reviews #62: https://radxa.com/products/orion/o6

geerlingguy, 2024-12-20:
[image]

## Basic information

  - Board URL (official): https://radxa.com/products/orion/o6
  - Board purchased from: ARACE Tech
  - Board purchase date: December 18, 2024 (pre-order, 'shipping after the Spring Festival 2025', so early Feb)
  - Board specs (as tested): Radxa Orion O6 32GB
  - Board price (as tested): $299.00

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

### glmark2

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2` (with `DISPLAY=:0` prepended if running over SSH)
3. Replace this block of text with the results.
```

### GravityMark

GravityMark results:

```
1. Download the latest version of GravityMark: https://gravitymark.tellusim.com
2. Run `chmod [downloaded_file.run]`
3. Run `sudo ./[downloaded_file.run]` and press `y` to accept the terms.
4. Open the link it prints, and run the Benchmark defaults, changing to 720p resolution and 50,000 asteroids.
```

Note: These benchmarks require an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

### Ollama

`ollama` LLM model inference results:

```
# Install ollama
curl -fsSL https://ollama.com/install.sh | sh

# Download some models
ollama pull llama3.2:3b && ollama pull llama3.1:8b

# Run benchmark
git clone https://github.com/geerlingguy/ollama-benchmark.git
cd ollama-benchmark
./obench.sh
```

Note that Ollama will run on the CPU if no valid GPU / drivers are present. Be sure to note whether Ollama runs on the CPU, GPU, or a dedicated NPU.

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

ThomasKaiser, 2024-12-21:
Slightly/highly off-topic... but I thought I ask anyway since the question might interest some of your audience too (if you disagree simply mark my comment as off-topic).

To test 5GbE capabilties a TrendNet TEG-S762 just arrived (cheapest unmanaged passively-cooled switch with at least 2 5GbE ports I could find two days ago) and an RTL8126 based M.2 NIC is on its way (10 bucks taxes and shipping included, [currently listed for a lot more again](https://aliexpress.com/item/1005007013394229.html))

Since you recommended WisdPi equipment to me may I ask where you ordered and if from their store whether you got something like a commercial invoice? I'm thinking about ordering two RTL8157 based WP-UT5 you already reviewed to test the Onion O6 against Mac clients but need a 'real' invoice :)

geerlingguy, 2024-12-22:
@ThomasKaiser - Their site uses Shopify, so it spits out the normal Shopify-style invoice (lists billing/customer address, items, shipping, tax—I presume it would also list VAT information in EU (not sure on that).

For my most recent order, it took 3 days from departing China to clear customs in the US, then four days to hit the regional mail facility here.

MagicAndre1981, 2024-12-22:
> I'm thinking about ordering two RTL8157 based WP-UT5 you already reviewed to test the Onion O6 against Mac clients but need a 'real' invoice :)

I bought a RTL8157 based adapter from [Wavlink](https://www.wavlink.com/en_us/product/WL_NWU340G.html) on German Amazon (your name may indicate you are German, too). On USB 3.0x Gen1 (5Gbit) you get 3.5GBit/s so around 400MB/s and full 5 GBit/s on USB 3.x Gen2 (10GBit) Port. But I use Windows devices, so I have no idea about Mac. You may need newer drivers.

ThomasKaiser, 2024-12-22:
> Shopify-style invoice (lists billing/customer address, items, shipping, tax—I presume it would also list VAT information in EU (not sure on that).

Since they listed 'included taxes' at checkout I simply gave it a try and edit this comment later when items and invoice arrived.

> no idea about Mac. You may need newer drivers.

RTL8157 has pretty good support by stock macOS (confirmed to work out of the box with at least macOS 13-15). I checked the macOS driver package and it only contains two directories named '10.8' and '10.9-10.15' so this is only for ancient OS versions :)

spx86, 2025-01-15:
Can you help test if this development board supports Arm CCA?
I also want to buy this model development board because I want to do some development work in Realm, so I need to know if orion o6 supports Arm CCA. i can hardly find any information about it!

geerlingguy, 2025-01-27:
There's an [Orion O6 Debug Party thread](https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054) over on the Radxa community forum. I just got a special box in the mail today...

ThomasKaiser, 2025-01-27:
'Mine' arrived as well:

[image]

Let's hope the aforementioned thread will bundle information needed at this point (and it doesn't happen in some Discord channels)

hrw, 2025-01-27:
Do tests, share data etc. 

Planning to order, but winter vacations first :D

hrw, 2025-01-28:
Can you add [BSA ACS](https://github.com/ARM-software/bsa-acs/blob/main/prebuilt_images/BSA_ACPI/v24.11_REL1.0.9/Bsa.efi) to list of tests?

You run it from EFI Shell: `bsa.efi` for normal run + `bsa.efi -v 1` for verbose one (and share both).

This checks for BSA compliance.

And my [ArmCpuInfo](https://github.com/hrw/edk2-armcpuinfo/releases) as it shows better which cpu features are supported than running Linux and checking /proc/cpuinfo ;D

mkesper, 2025-01-28:
If this thing gets mainline support it will be rocking!

ThomasKaiser, 2025-01-30:
@geerlingguy @cnxsoft: latest `sbc-bench` v0.9.70 should now contain all adoptions/hacks needed for dealing with CD8180.

Especially the weird cluster setup is addressed and the mystery why `cpu11` is not only faster than its `cpu9` and `cpu10` siblings from same cpufreq cluster but also faster than `cpu0` with which the core shares the real CPU clockspeeds but not access to DRAM.

See commit comment for further details: https://github.com/ThomasKaiser/sbc-bench/commit/6b0cd05fb78fb75b2161fa307111a05b0e642356#commitcomment-151945077

geerlingguy, 2025-01-30:
@ThomasKaiser - Quite strange! I wonder if there could be any chip to chip variance here? Seems very weird to have one core behave with a measurable difference (especially wrt memory access). If that's as designed, what was the goal??

ThomasKaiser, 2025-01-30:
> If that's as designed, what was the goal??

Good question. I'm hoping @wtarreau as announced will soon start [his detective work about the inner workings](https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054/94?u=tkaiser) :)

(Though `$dayjob` might be in the way just as in my case right now)

wtarreau, 2025-01-30:
Yeah I really want to, but indeed I didn't plan on receiving one so I haven't reserved time yet. But probably this week-end. I already downloaded the image to flash on it, and already have an SSD available.

geerlingguy, 2025-01-31:
I was trying to get the OS loaded on an NVMe drive, but when I flashed the B3 version of the Debian desktop image, there's no .img file included in the Home folder as per the instructions: https://docs.radxa.com/en/orion/o6/getting-started/quick-start#3-write-the-image

I was considering cloning the USB drive, but I have a 1TB USB drive and 256 GB NVMe installed, so that won't work either.

I guess I can just try downloading the B3 img.gz file to the Live install USB drive, then flash from there... but I had assumed it would be included. Maybe a change from revision B1 to B3?

wtarreau, 2025-01-31:
I had the image on mine, it was the B3. I downloaded the "usb-preinstalled-something-b3" image, uncompressed it to a USB hard drive, the image was 20 GB. Then once booted, the NVME image was in the radxa user account. I just copied it over the /dev/nvme0n1 device, rebooted and it worked first time.

Also, apparently the USB image and the NVME ones are not exactly the same.  For example I noticed lscpu missing from the USB one while it was on the NVME one. Thus there's *at least* one difference.

BTW be careful if you ever test openssl. There is an outdated version 3.0.1 in /usr/share/cix/something that is not correct, and unfortunately that path is before /usr/bin for the radxa user. Same for their libs, so you can quickly end up with tests reporting garbage numbers.

And you can change the max CPU freq in the BIOS from 2.5 to 2.6, as indicated by Thomas (I did it, since better test it closer to the expected final specs).

geerlingguy, 2025-01-31:
That's what I downloaded (https://dl.radxa.com/orion/o6/images/debian/orion-o6-debian12-preinstalled-desktop-b3.img.gz), and I flashed that download directly with Etcher, then booted off the USB stick.

The home directory had no extra files in it (confirmed in Terminal and in the file browser, not just looking in Etcher)... so I just downloaded that entire file again on the Live booted image.

Flashed it with Etcher, and it is running now.

However, the fan was not going full blast in the Live CD boot from USB, nor during BIOS operations... but once I booted from the internal NVMe, the fan is now going at 100% continuously (and is quite loud!). Any way to force the fan to use PWM? CPU temp is reporting as 30°C so it's certainly not necessary to be going all out :D

wtarreau, 2025-01-31:
Yes, I got the same as well. I explained on the forum how I calmed it down, by echoing value 17 to pwm1 (at 16 it eventually stopped spinning). Found it, it's here: https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054/126?u=willy
```
$ echo 17 > /sys/class/hwmon/hwmon1/pwm1
```
Do not hesitate to read some contents there, as there's already quite a lot of data shared (CPU cores arrangement in clusters, RAM latency/BW etc).

geerlingguy, 2025-01-31:
@wtarreau Thanks, and no worries, right after commenting, I finished reading through the debug party thread and found your comment on the PWM there too.

I'll carry on with my testing here :)

geerlingguy, 2025-01-31:
Power consumption during a top500 HPL run:

[image]

geerlingguy, 2025-02-03:
I posted an [initial pre-release blog post](https://www.jeffgeerling.com/blog/2025/oryon-o6-itx-arm-v9-board-temper-your-expectations) about the board, as I think it's important for Radxa to tone down some of the marketing hyperbole around the board, if they don't want the first wave of testers to lambast this board.

It's good hardware, and I'd hate for it to be let down by early software issues. (And for the hundredth time, I wish Radxa would give more time between these 'debug parties' where tons of issues are found, and the public launch, when people who pre-ordered get boards that are _pretty_ good but lacking in some essential features, or they have to do things like pin old software packages since updating would remove functionality they expect to have...).

cnxsoft, 2025-02-04:
I usually take my time or do a preview (as opposed to a review) for new boards from Radxa.

geerlingguy, 2025-02-04:
@cnxsoft - The tricky bit for that (in general) is when to actually do a full-on _review_, though. Do you do it when they start shipping boards to people who pre-ordered, or wait some time after that?

From prior experience, I don't have a lot of confidence we'll be in a much different place in a few weeks, when these boards start shipping :(

I want to be proven wrong, because I really like this board hardware, but as I say in the blog post, I think I need to temper my expectations!

cnxsoft, 2025-02-06:
My method is quite basic. I review a few other boards first, and once I've done I switch to it. I try not to delay more than one month though.

geerlingguy, 2025-02-23:
A couple updates over the past few weeks:

  - It looks like [Cix is beginning work on Linux upstreaming](https://patchwork.kernel.org/project/linux-arm-kernel/cover/20250220084020.628704-1-peter.chen@cixtech.com/), though the first patch set will take some time
  - [OpenBSD includes the O6 in its arm64 support page](https://www.openbsd.org/arm64.html); see [mailing list discussion](https://marc.info/?l=openbsd-arm&m=173823317816570&w=2)
  - I've seen users report success with AMD and Nvidia graphics cards (seemingly without driver modification?), so I am eager to test some of my cards out...

Also been monitoring the O6 Radxa forum, looks like [one user has an RX 7600 running Skyrim at 4K](https://forum.radxa.com/t/arm-workstation-build/25922?u=geerlingguy), and [another has Windows 11 up and running](https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054/135?u=geerlingguy).

ThomasKaiser, 2025-02-23:
> [another has Windows 11 up and running](https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054/135?u=geerlingguy).

The another is Mario known for [Windows on R](https://github.com/worproject). And thanks for the link to the OpenBSD mailing list, great to see Jared McNeill in action (again, [he did a lot of the ground work to boot any arm64 based distro on recent Rockchip platforms](https://github.com/jaredmcneill/quartz64_uefi)).

hrw, 2025-02-23:
@geerlingguy thanks for link to *BSD discussion.

"The more I read, the more I postpone the order" :(

Civil, 2025-02-25:
@geerlingguy they also promised to release source code for the kernel (source: https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054/199 ) and edkII next week and they are very active on their discord.

But note - they apparantly have a bug in EC firmware right now: https://forum.radxa.com/t/warning-nearly-cooked-my-amd-pro-while-the-orion-was-turned-off/25948 - Radxa promised to fix that ASAP.

> I've seen users report success with AMD and Nvidia graphics cards (seemingly without driver modification?), so I am eager to test some of my cards out...

People are basically running mainline kernel - you lose mali, vpu and npu, but get recent enough drivers. It have some problems, as you can't just install Linux there because EFI have some bugs in ACPI mode, but it is doable even right now.

ThomasKaiser, 2025-02-25:
> they are very active on their discord.

So same shit show as with Rock 5B back then? High activity behind closed curtains?

Civil, 2025-02-25:
> So same shit show as with Rock 5B back then? High activity behind closed curtains?

Not really. It is not closed curtains, and they monitor their forums as well. But I'm seriously upset that kernel source code is not available from day 1.

wtarreau, 2025-02-26:
> It is not closed curtains

Ah ? Then can you explain why I can't find these conversations on Google when searching for anything related to Orion O6, while the Radxa forums, Jeff's & CNX pop up first ? That's precisely the problem of these shitty closed discussion places, they're not indexed, not searchable, and any possibly useful info that's exchanged between participants is lost forever and cannot benefit others looking for the same. The only consolation is that the level of the people going there generally doesn't warrant the production of useful info...

Regarding the forum, several of us have shared a lot of feedback, tests, measurements, suggestions for improvements, and this was left with little to no feedback. I just thought they were extremely busy (which is quite possible), but maybe they're again just not watching the feedback that comes there since they're disturbed with those asking "what do I do now?" after being dictated basic commands to type on their keyboards.

hrw, 2025-02-26:
@wtarreau in SBC (Seriously Bad Computers) land there are many people who buy things because marketing slogans were nice. Without looking at state, support etc. And then complain or abandon with "this SBC sucks, vendor sucks".

I have decided to ignore Orion C6 forum/discord for a few weeks to give Radxa time to release source for firmware/BSP and start fixing errors. And then will reconsider ordering 32GB version of board.

Arm SBC land is terrible. Full of vendors knowing only how to sell but not how to provide support. Radxa team did some work on their previous products so I am curious how well they will handle this one.

https://society.oftrolls.com/@hrw/114068723579492658

wtarreau, 2025-02-26:
Well, I'm having good hopes on this one. First, Radxa makes efforts. They're a small team yet they're trying to upstream patches, and they're having a lot of consideration for customers feedback. Not that many vendors are listening to customers. It's true that the Arm SBC land is bad. But it's improving. ext2linux support by default on u-boot allowed to make a great step forward. UEFI (as much as I hate it) even more. On this board we have working UEFI+ACPI, it's only the 3rd Arm board that I can boot with a mainline kernel without any device tree. The previous ones were SolidRun's LX2 and Adlink's Altra dev kit (whose bios was full of bugs). All of this significantly reduces the need for mainlining. People are not asking for mainlining code when they buy an x86 SBC, because it's expected to work out of the box. We're getting closer to this here.

I too am waiting for EDK2 sources. I'm sure those who know this area will be able to improve it. And it will also allow to re-enable the initially promised CPU frequencies for example.

But overall I tend to think that this board will likely be Radxa's most successful board by being the first one which will boot mainstream distro images out of the box, and I understand why they're taking their time (even though they've advertised its upcoming availability a bit too fast IMHO). There's nothing worse than having a million of users claim everywhere that your device is broken, you just can't fix such claims afterwards. They'd rather make sure to avoid most possibly negative reactions now. And yes, I too am interested in a buying 32GB one soon for LLM experiments (unless I find a better solution in the mean time).

Civil, 2025-02-26:
The upstream kernel is not that hard to get up & running, btw.

I got 6.13.0 (`git clone https://github.com/torvalds/linux.git && cd linux && git checkout v6.13` but downloading kernel works) working with some caveats.

I used defconfig where I then went through it and enabled (as compiled-in) stuff that is required to boot, here is the final config: https://gist.github.com/Civil/8bf3259974136b05df9642139b7a60ae

My rx6400 was hanging at boot for some reason, but rx550 works ok. I've also needed to install dracut (they somehow messed up initramfs-tools scripts, and I've decided not to invest time into them)

```
sudo mkdir -p /boot/efi
sudo mount /dev/nvme0n0p1 /boot/efi
sudo apt install dracut build-essential bison flex libssl-dev
make -j12
sudo make modules_install
sudo cp arch/arm64/boot/Image.gz /boot/vmlinuz-6.13.0
sudo dracut --force
sudo cp arch/arm64/boot/Image.gz /boot/efi/vmlinuz-6.13.0
sudo cp /boot/initrd.img-6.13.0 /boot/efi/
echo -e "\n\nmenuentry '31 Mainline 6.13.0' {\n    linux /vmlinuz-6.13.0 root=/dev/nvme0n1p2 rootwait rw\n    initrd /initrd.img-6.13.0\n}\n"  | sudo tee -a /boot/efi/GRUB/GRUB.CFG
sudo sed -i 's/default=.*/default="31"/;s/timeout=.*/timeout=10/' /boot/efi/GRUB/GRUB.CFG
```

Civil, 2025-02-26:
Oh, and before that, in EFI, you need to switch it into ACPI mode. That doesn't prevent booting stock kernel,  btw.

Civil, 2025-02-26:
Also, for newer GPUs, more recent firmware might be required, so:
```
git clone --depth 1 https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
cd linux-firmware
sudo make install
```

wtarreau, 2025-02-26:
You should definitely follow the thread on radxa's forums, as these points regarding boot, acpi, mainline etc have already been covered by several participants: https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054

Civil, 2025-02-26:
> You should definitely follow the thread on radxa's forums, as these points regarding boot, acpi, mainline etc have already been covered by several participants: https://forum.radxa.com/t/orion-o6-debug-party-invitation/25054

I know about the thread, I haven't seen all the info in a single place, though, plus an exact list of commands can be used as a sort of script (I've put better version in a thread, that is just copy-pastable).

wtarreau, 2025-03-01:
@runaway97 I found the CPU OPP (frequency and voltages for each half-cluster) here:

   https://gitlab.com/cix-linux/cix_opensource/release/edk2-non-osi/-/blob/874c4/bbdf2/cix_beta2_radxa_dev/Platform/CIX/Sky1/PackageTool/pm_config/opp_config_custom.h?ref_type=heads

Browsing in gitlab is a real mess and I have not found any way to look for a file name ("find") nor contents ("grep"). I found these files out of luck yesterday and couldn't find them again anymore this morning without cloning. The total size is 800MB so if you have enough space and bandwidth you'll save time cloning these trees and exploring them locally.

I don't know these files are included in the build process, but the surrounding files are interesting as well. That's where you discover that the chip supports DDR5 up to 6400 MT/s hence the claim of ">100 GB/s", but with 5500 MT/s DRAM it's obviously not achievable.

wtarreau, 2025-03-01:
FWIW I could rebuild the whole image, but I can confirm that changing the definitions there indeed has exactly zero effect :-/

Civil, 2025-03-03:
> P.S. Also in case you wanted to try - Intel Arc doesn't work, "Unhandled page fault" on driver load. Though my RX 5700 works.

Update on that - I misread the stacktrace becuase it got mixed up with the efi's unahdled pagefault, so I thought that was caused by Xe driver. So with my old patch from https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/1824 it works if you build mesa with Xe support (stock one doesn't have it).

Also performance probably would be horrendous, because:
```
[    2.908081] xe 0000:c3:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC). Consider enabling 'Resizable BAR' support in your BIOS
```

geerlingguy, 2025-03-03:
@Civil - The BAR size issue could be related to [PCI: Release unused bridge resources during resize](https://patchwork.kernel.org/project/linux-pci/patch/20240507213125.804474-1-alex.williamson@redhat.com/). See related discussion getting an A750 and B580 running on the Pi here: [Add Intel A750](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/510#issuecomment-2637227605).

Civil, 2025-03-03:
> The BAR size issue could be related to [PCI: Release unused bridge resources during resize](https://patchwork.kernel.org/project/linux-pci/patch/20240507213125.804474-1-alex.williamson@redhat.com/)

I doubt it is in this case. Even with the patch from the mailing list it fails to resize BAR. And I haven't seen such problems on Altra for example.

geerlingguy, 2025-03-19:
@q4a - Honestly I had been waiting for a more finalized firmware to be ready, that hopefully solves some of the many little issues people have been having (not the least of which is idle power draw—but that looks like it *could* be just the way the board was designed?).

I am still waiting a bit for whatever else may come from the debug party, I don't want to leave a negative impression based on beta firmware, since I really hope this board can go further than the Rock 5 ITX in terms of being a better alternative for small, efficient builds versus something like Ampere.

wtarreau, 2025-03-19:
I personally think the hardware is great and promising. Yes there are issues with idle power consumption, CPU ordering, CPU frequency, DRAM frequency, BIOS compatibility with operating systems, power-off apparently still draining power etc. All of these look like software issues so I'm not much worried. I don't actually *need* that board but I'm confident that once these issues will be fixed, it will be the best Arm development board around for several years, which is why I've ordered a new one, as in its current state it's already a great product (but still way below that advertised).

I think the Radxa team is a small team and issues are not progressing as fast as we'd like. Maybe they're busy on other issues we've not noticed (and likely with other products as well). It's clear they could communicate better, at least acknowledging issues (maybe even use an issue tracker instead of the forum to try to focus efforts on certain topics). The state of the BIOS that seems to ignore manual tweaking in the sources, probably due to using too many blobs, is annoying as we can hardly help there as well. What they're risking now is a massive marketing success with many orders and non-technical customers all opening issues in parallel before the issues above had a chance to be fixed. We'll see...

hrw, 2025-03-19:
Hardware feels like easy part as SoC vendors often have some reference designs.

I wonder will Radxa is capable of managing software/firmware side of this board. Cixtech did basic BSP so SBC can be used but their plans look like "be patient, we plan to plan upstreaming".

ACPI mode needs firmware improvements, 23MB of blobs scare potential contributors away.

Do not wait to 2026 with review. Fw/sw side may get better at some point but is it granted to happen "soon"? No. 

Cix plans to work on edk2 upstreaming in Q3. Will it mean opening blobs? No idea.

Kernel side did not even started as Cix only sent basic DT which is not merged yet.

hannescam, 2025-03-19:
I feel like we got dough for a cake with this board.

Civil, 2025-03-19:
> has been acknowleged but still not fixed correctly from what i have seen on discord.

Yeah... though they are sort of trying.

> that person left and they have nobody to deal with the issues anymore.

Could be, but then I understand even less why they are so clinging to the source code.

> Here you go:

Yeah, I saw the answer to my remark ;)

geerlingguy, 2025-03-19:
> I feel like we got dough for a cake with this board.

@hannescam - Indeed, that is what it feels like at this point.

If Raspberry Pi sent me an _alpha_ version of the Pi 5 or any other product that was in the state of the O6 now a month _post launch_, I would lambast them if they didn't completely turn that ship around.

I grade Radxa on a curve, and know that they especially can't polish the software / firmware side the same as Pi, who has more than one full-time engineer on both Linux and firmware dev. But then I know I'll get lambasted (as usual) in the comments for being a Pi shill, and ignoring all other vendors, no matter what I say on the matter :P

Civil, 2025-03-19:
> I grade Radxa on a curve, and know that they especially can't polish the software / firmware side the same as Pi, who has more than one full-time engineer on both Linux and firmware dev.

But they need to learn to depend on community more and to be more honest with people. If nobody would tell the true story - they will just continue to ship products in the same shape as Orion now.

And it doesn't matter that they are better than average arm sbc on a lunch date. What matters is that they are still worse than x86 20 years ago

wtarreau, 2025-03-20:
I'm seeing it a bit differently.

SBC vendors are always entirely tied to what SoC vendors deliver. 9 years ago, HardKernel was the very first victim of Amlogic lying on the S905's frequency. Similarly those selling RockChip-based boards have to live with advertised features that depend on totally outdated BSP kernels, and cannot diverge too much from the reference design. The reason is simple: SBC vendors are first and foremost electronic device designers and makers. They identify a set of components which, based on specs, could make a nice solution once combined together. And it's natural to announce the shiny new features that your board will come with. But since they're essentially electronics designers and not developers, they solely rely on the shitty software coming from the SoC vendor to deliver the advertised features. With some of them this is particularly visible as you realize that they indicate that their board aims to run Linux, yet all the docs explaining how to connect to the board or flash it are all windows-based, just because that's what they're using all day to run their EDA software, they don't need to know Linux at all to make an SBC because you don't need to touch software. The farthest they go consists in modifying a default DTS file to properly enumerate onboard devices, and sometimes you even find CRLF at the end of lines in the DTS, indicating a lot about how they were manipulated!

I think that with CIX announcing the "opensource SoC", Radxa saw a huge opportunity for switching from the previous model where they just have to swallow whatever crap the SoC vendor pushes down their throat, to a model where they could rely on the community's claimed knowledge to adjust the details that are not handled by the SoC vendor. That's reasonable, with every chip vendor we've been complaining "where's the datasheet?" to fix bogus drivers, so this time the opportunity to put an end to this situation very likely looked super appealing.

But for whatever reason CIX has "opensource soon" plans, not "opensource now". Most of the bogus stuff is done in the EDK2 part, and that's the one that's full of blobs. So I think that for now Radxa is in the same situation with CIX as they've been with other SoC vendors: they're just between customers complaining about false advertising and the maker saying "be patient it will come". Except that with Rockchip nobody expected anything to work any time soon, but with CIX's claimed opensource chip, everyone hoped it would be possible to start hacking on it.

For me the problem is essentially the timing. The device was advertised and sold too early, or CIX is late. But if you look on the board, there are 3 logos there: arm china, CIX, Radxa. I suspect that at some point the 3 together saw an opportunity for making a great product that breaks the long-established BSP model, and counted on arm to do the marketing ("very first opensource armv9"), CIX to provide the chips that enable everything, and Radxa's ability to quickly design boards that assemble everything and sell them through their well-established resellers channel. It does make a lot of sense. When working on products, small opportunistic windows like this are super important and must absolutely not be missed. But now everyone is waiting on CIX to respond to queries about idle consumption, CPU frequency settings, CPU ordering, DRAM frequency and whatnot.

It's possible that CIX is also made of mostly chip designers and overestimated their ability to write boot loader code and drivers, I don't know. It's possible as well that they absolutely want to provide themselves a good working base without the help of community to tell other chip makers "see how this should be done?". And even if late this could be a well deserved slam in the face of rockchip, amlogic and co.

Thus my feeling is that we're observing an experimental project that tries to change the 2-decades long situation that we've all been complaining about, that the selected hardware has really promising specs and that it will eventually work. If it's a first time at such an experiment for these companies, it's understandable that it doesn't work that well and takes longer than imagined, as they probably were a bit optimistic. But I still think that as long as the 2 companies work together on the project, nothing is lost, and what needs to be fixed is essentially software. Due to this I accepted to take the risk to pre-order another board now, counting on fixes to eventually come.

I mean, we've all waited 2 years for rock5b to work more or less with mainline, and that was not at all thanks to Rockchip's efforts nor public documentation. Here it's been 3 months and it's CIX's first attempt. Maybe we can be a bit less impatient. However we should continue to pressure them to release the necessary software and documentation so that we can help them by experimenting each on our side and report with our findings. If they postpone too much, they risk everyone complaining loudly about the early defects and delivering the image of a broken product at release date, which would be contrary to the intended message. If instead they'd been releasing the software part before people complained, they could have claimed it worked out of the box, because, honestly, this board works way better out of the box on mainline than any rockchip-based device we've seen to date. Thus for me their experiment was a success and they should surf on this wave instead of trying to deliver by themselves something perfect.

Civil, 2025-03-26:
>  It's only a little faster than the various RISC-V chips I've tested, which I constantly harp on for their (so-far) lack of efficiency.

I'd disagree. Performance-wise it is way better than any of the RISC-V chips out there. Single-core performance is like a 10x of fastest RISC-V out there (P550 or TH1520, depends on how you count).

Multi-core - you can argue that 64 of C920 in SG2042 are faster, but not by much.

And that doesn't take into account that GPU in this one is far better than any of the imagination ax/bx that are popular amongst RISC-V ones.

Efficiency wise it is indeed bad. Not the worst out there, but somewhere at the bottom of the list. But remind you - it is the cheapest ARM board that have more than 4 lanes of PCIe available to users (and those are actually x8 + x4 m.2 and x2 m.2) - that is also something positive.

But overall its state is arguably worse (if you take into account promises) compared to RK3588 when it was released.

derTim, 2025-03-26:
> it _still_ hasn't shipped for anyone who ordered the AI kit

I am happy about that, allowed me to cancel my 32GB + AI Kit order last week  😄 

If SW support improves over the next months, I will maybe order it again, hopefully with an EU based shipping option, so I don't have to pay ~60€ for shipping and an additional 19%  in taxes + handling fees.

geerlingguy, 2025-03-26:
> I'd disagree. Performance-wise it is way better than any of the RISC-V chips out there. Single-core performance is like a 10x of fastest RISC-V out there (P550 or TH1520, depends on how you count).

Sorry, I meant to say "only a little bit more efficient", totally mis-typed that!

geerlingguy, 2025-03-27:
@wtarreau 

> But we're still far from this openness, and the overall silence make the situation look embarrassing for the vendors, which is not reassuring for the customers who can think that it's far worse than what they're observing.

Definitely. Though I do notice a pattern with Radxa sometimes, where they will be responsive a while, then I'm guessing the people who I am in contact go heads-down into some projects... then it will take a while to get a response again. If the Cix culture is similar, it is sometimes frustrating to deal with—even as someone who has more 'sway' in terms of getting a response in the first place!

In a related vein, I noticed Radxa launched a [CM3J](https://www.cnx-software.com/2025/03/27/radxa-cm3j-industrial-grade-rockchip-rk3568j-som-is-compatible-with-raspberry-pi-compute-module-4/) as well... they continue pushing out new product after new product, it would be nice if they could slow down and support the things they have already launched better.

cnxsoft, 2025-03-27:
@geerlingguy  I also think they have too many products, so it makes it hard to focus on software. It also depends on the size of the company.  If their software engineers have to consistently jump from one target to the other depending on the emergency of the day, it's not good for productivity...

wtarreau, 2025-03-27:
If there are more hardware people than software people, it makes sense to issue many products. The principle is then to consider that there will be a certain success ratio and that some products will just fail early and be abandoned while others will be successful, without knowing upfront which ones will be in each category. I think it's the approach that FrendlyArm has been taking for a long time, for example. Remember the S5P6818-based devices ? They were promising but zero effort was made on the software, which killed them, and it didn't even seem to cause them the slightest problem. I personally don't support this approach at all but it can be understood why some companies would work like this.

hrw, 2025-03-28:
Unsubscribing.

https://society.oftrolls.com/@hrw/114239243576231795

lyuchuny3, 2025-04-02:
hi, guys, you can see this document to learn how to use the NPU on Radxa O6.

https://zhouyi-npu-tutorial.readthedocs.io/en/latest/index.html

Shivansps, 2025-04-12:
Today radxa launched bios 9.0.0 (???? up from 0.2.1 something?).

With that bios, ACPI is now default, the small cores has been disabled by default (you can re-enable), TPM 2.0 was added. And... i cant belive this... you can install Windows 11 in a complete normal way! 
Just download the iso from Microsoft site, burn it to a pendrive with Rufus, boot from it and install. First time EVER i see this working.

Also in Windows the cpu is now up to 2.6ghz, seems to boost, the fan goes up and down depending on load (man the bar was reallly low). You will need an USB ethernet adapter that Windows has drivers for it to get internet, but other than a thousand of missing drivers its OK.

Even my RTX3060 was showing up in device manager (cant start, code 10, something about not being the same device that was brought up during post) but i was completely unable to get any form of video out from it no matter what i did.

And thats another thing, some pcie 4.0 gpus were not being detected before, now they are, my RTX3060 now shows up in Windows and in lspci in linux, but i havent had luck with normal linux installers yet.

[63 more replies]