geerlingguy/sbc-reviews #65: HiFive Premier P550

geerlingguy, 2025-01-24:
[comment]: # (If desired, delete this line and add an image of the board here)

## Basic information

  - Board URL (official): https://www.sifive.com/boards/hifive-premier-p550
  - Board purchased from: Provided for review
  - Board purchase date: January 20, 2025
  - Board specs (as tested): TODO
  - Board price (as tested): TODO

## Linux/system information

```
# output of `screenfetch`
PASTE_HERE

# output of `uname -a`
PASTE_HERE
```

## Benchmark results

### CPU

  - Geekbench 6: (TODO single / TODO multi - PASTE_URL)
  - TODO Gflops ([geerlingguy/top500-benchmark](https://github.com/geerlingguy/top500-benchmark) HPL result)

### Power

  - Idle power draw (at wall): TODO W
  - Maximum simulated power draw (`stress-ng --matrix 0`): TODO W
  - During Geekbench multicore benchmark: TODO W
  - During `top500` HPL benchmark: TODO W

### Disk

#### MANUFACTURER_AND_MODEL_OF_DISK_HERE

[comment]: # (Run `lsblk -o NAME,FSTYPE,LABEL,MOUNTPOINT,SIZE,MODEL` to get model)

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | TODO MB/s |
| iozone 4K random write     | TODO MB/s |
| iozone 1M random read      | TODO MB/s |
| iozone 1M random write     | TODO MB/s |
| iozone 1M sequential read  | TODO MB/s |
| iozone 1M sequential write | TODO MB/s |

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/ TEST_SIZE=1g ./disk-benchmark.sh
```

Run benchmark on any attached storage device (e.g. eMMC, microSD, NVMe, SATA) and add results under an additional heading.

Also consider running [PiBenchmarks.com script](https://www.jeffgeerling.com/blog/2023/using-pibenchmarkscom-sbc-disk-performance-testing).

### Network

`iperf3` results:

  - `iperf3 -c $SERVER_IP`: TODO Mbps
  - `iperf3 -c $SERVER_IP --reverse`: TODO Mbps
  - `iperf3 -c $SERVER_IP --bidir`: TODO Mbps up, TODO Mbps down

(Be sure to test all interfaces, noting any that are non-functional.)

## GPU

### glmark2

`glmark2-es2` / `glmark2-es2-wayland` results:

```
1. Install glmark2-es2 with `sudo apt install -y glmark2-es2`
2. Run `glmark2-es2` (with `DISPLAY=:0` prepended if running over SSH)
3. Replace this block of text with the results.
```

### GravityMark

GravityMark results:

```
1. Download the latest version of GravityMark: https://gravitymark.tellusim.com
2. Run `chmod [downloaded_file.run]`
3. Run `sudo ./[downloaded_file.run]` and press `y` to accept the terms.
4. Open the link it prints, and run the Benchmark defaults, changing to 720p resolution and 50,000 asteroids.
```

Note: These benchmarks require an active display on the device. Not all devices may be able to run `glmark2-es2`, so in that case, make a note and move on!

### Ollama

`ollama` LLM model inference results:

```
# Install ollama
curl -fsSL https://ollama.com/install.sh | sh

# Download some models
ollama pull llama3.2:3b && ollama pull llama3.1:8b

# Run benchmark
git clone https://github.com/geerlingguy/ollama-benchmark.git
cd ollama-benchmark
./obench.sh
```

Note that Ollama will run on the CPU if no valid GPU / drivers are present. Be sure to note whether Ollama runs on the CPU, GPU, or a dedicated NPU.

TODO: See [this issue](https://github.com/geerlingguy/sbc-reviews/issues/2) for discussion about a full suite of standardized GPU benchmarks.

## Memory

`tinymembench` results:

<details>
<summary>Click to expand memory benchmark result</summary>

```
# Run the two commands below, then replace this code block with the full result.
git clone https://github.com/rojaster/tinymembench.git && cd tinymembench && make
./tinymembench
```
</details>

## `sbc-bench` results

Run sbc-bench and paste a link to the results here:

```
wget https://raw.githubusercontent.com/ThomasKaiser/sbc-bench/master/sbc-bench.sh
sudo /bin/bash ./sbc-bench.sh -r
```

## Phoronix Test Suite

Results from [pi-general-benchmark.sh](https://gist.github.com/geerlingguy/570e13f4f81a40a5395688667b1f79af):

  - pts/encode-mp3: TODO sec
  - pts/x264 4K: TODO fps
  - pts/x264 1080p: TODO fps
  - pts/phpbench: TODO
  - pts/build-linux-kernel (defconfig): TODO sec

geerlingguy, 2025-02-05:
Bringup notes: I just plugged everything into the SiFive-provided enclosure, plugging in the front panel header connectors to standard pin locations... but it seems like it may not use the standard pin locations? The power button does nothing, and I don't see anything on the power supply to indicate it is not active (there's no separate power switch). So I'll check the documentation and see if I'm missing something obvious!

geerlingguy, 2025-02-05:
Apparently "SW2" needs to be switched towards the edge of the board, instead of towards the middle of the board. There's no particular "ON" indication so that was a little confusing, but now I have it booted and running Ubuntu 24.04 (it was pre-installed).

Going to do a little exploration before jumping into any benchmarking.

Note on the official case—the tiny PSU (not sure if it's Flex ATX?) has a fairly loud fan. It ramps down immediately after startup, but it has a bit of a whine to it that makes it less pleasant. Fine for under a desk or in another room but as a desktop unit its pretty noisy, even at idle.

geerlingguy, 2025-02-05:
Ollama doesn't install, I get:

```
    Starting nested operation: tasks/ollama-benchmark.py | Run Ollama Install Script 
    [10.0.2.234] nested ERROR: Unsupported architecture: riscv64
```

But it looks like I could possibly compile it separately: https://www.bit-brick.com/2024/10/02/在pi-one-上安装ollama/

Also, I filed an issue on the ollama project: https://github.com/ollama/ollama/issues/8857 — we'll see where that goes!

geerlingguy, 2025-02-05:
I was expecting to have to manually compile `cryptography` for Ansible, but it installed out of the box...

```
--> Starting operation: Install Ansible dependencies 
    [10.0.2.234] Success

--> Starting operation: Remove Python 3.11 EXTERNALLY-MANAGED file 
    [10.0.2.234] No changes

--> Starting operation: Remove Python 3.12 EXTERNALLY-MANAGED file 
    [10.0.2.234] No changes

--> Starting operation: Install Ansible 
    [10.0.2.234] Success
```

I've updated my [Installing Ansible on a RISC-V computer](https://www.jeffgeerling.com/blog/2024/installing-ansible-on-risc-v-computer) blog post with this note. Might not be required at least on Ubuntu anymore!

That said, I'm going to let the HPL benchmark run overnight... taking quite a while :)

geerlingguy, 2025-02-06:
For all my benchmarking... the idle to full power range is pretty tight:

[image]

geerlingguy, 2025-02-06:
Some Ethernet info for end1 (one of the two 1 Gbps ports);

```
ubuntu@ubuntu:~$ ethtool end1
Settings for end1:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: master
	Port: MII
	PHYAD: 0
	Transceiver: external
netlink error: Operation not permitted
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
	Link detected: yes

ubuntu@ubuntu:~$ ethtool -T end1
Time stamping parameters for end1:
Capabilities:
	software-transmit
	software-receive
	software-system-clock
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```

geerlingguy, 2025-02-06:
Looking at some of the other features of this board:

  - GPIO access is currently not exposed in software
  - Imagination AXM-8-256 GPU not yet supported (so no encode/decode support)?
  - Integrated 20 TOPS NPU not yet supported
  - WiFi slot is it actually A+E-key?
  - The PCIe slot should work okay, need to test that with a few different devices
  - I haven't tested the remote Ethernet MCU management port either

geerlingguy, 2025-02-06:
Vulkan info (`vulkaninfo` output):

<details>
<summary>Click to expand</summary>

```
$ vulkaninfo
'DISPLAY' environment variable not set... skipping surface info
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.275

Instance Extensions: count = 21
===============================
	VK_EXT_acquire_drm_display             : extension revision 1
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 10
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_portability_enumeration         : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_wayland_surface                 : extension revision 6
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6
	VK_LUNARG_direct_driver_loading        : extension revision 1

Layers: count = 2
=================
VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (PowerVR A-Series AXM-8-256)
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (PowerVR A-Series AXM-8-256)
		Layer-Device Extensions: count = 0

Device Properties and Extensions:
=================================
GPU0:
VkPhysicalDeviceProperties:
---------------------------
	apiVersion        = 1.3.225 (4206817)
	driverVersion     = 1.516.125 (6307965)
	vendorID          = 0x1010
	deviceID          = 0x30010101
	deviceType        = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName        = PowerVR A-Series AXM-8-256
	pipelineCacheUUID = 7d406000-1e03-0098-0065-00eed95e6600

VkPhysicalDeviceLimits:
-----------------------
	maxImageDimension1D                             = 16384
	maxImageDimension2D                             = 16384
	maxImageDimension3D                             = 2048
[... 770 lines cut ...]
	vulkanMemoryModelDeviceScope                       = true
	vulkanMemoryModelAvailabilityVisibilityChains      = true
	shaderOutputViewportIndex                          = false
	shaderOutputLayer                                  = false
	subgroupBroadcastDynamicId                         = true

VkPhysicalDeviceVulkan13Features:
---------------------------------
	robustImageAccess                                  = true
	inlineUniformBlock                                 = true
	descriptorBindingInlineUniformBlockUpdateAfterBind = true
	pipelineCreationCacheControl                       = true
	privateData                                        = true
	shaderDemoteToHelperInvocation                     = true
	shaderTerminateInvocation                          = true
	subgroupSizeControl                                = true
	computeFullSubgroups                               = true
	synchronization2                                   = true
	textureCompressionASTC_HDR                         = false
	shaderZeroInitializeWorkgroupMemory                = true
	dynamicRendering                                   = true
	shaderIntegerDotProduct                            = true
	maintenance4                                       = true
```
</details>

geerlingguy, 2025-02-07:
After upgrading to the latest versions of everything (I was avoiding that initially, as it has caused issues on RISC-V dev boards in the past...), GLMark2 actually runs now! Though most other 3D/Mesa utilities keep showing various failures.

geerlingguy, 2025-02-07:
`lspci -vvvv` output with AMD Radeon HD R5 230 Graphics Card installed:

```
ubuntu@ubuntu:~$ sudo lspci -vvvv
00:00.0 PCI bridge: Beijing ESWIN Computing Technology Co., Ltd. EIC7700 Root Complex (rev 01) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR+ <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 43
	IOMMU group: 5
	Region 0: Memory at 48000000 (32-bit, non-prefetchable) [size=128M]
	Region 1: Memory at 41000000 (32-bit, non-prefetchable) [size=1M]
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 1000-1fff [size=4K] [16-bit]
	Memory behind bridge: 41100000-411fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 8000000000-800fffffff [size=256M] [32-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR+ <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/32 Maskable+ 64bit+
		Address: 00000000fffff000  Data: 0000
		Masking: fffffffe  Pending: 004440e2
	Capabilities: [70] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag+ RBE+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L0s L1, Exit Latency L0s <4us, L1 <16us
			ClockPM- Surprise- LLActRep+ BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x4
			TrErr- Train- SlotClk+ DLActive+ BWMgmt+ ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP+ LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn+ NFERptEn+ FERptEn+
		RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
			 FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
		ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0100
	Capabilities: [148 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
[... 92 lines cut ...]
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fffff000  Data: 0001
	Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 14, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 40000001 0000000f fffff000 00000000
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
```

I get display output, but all the OpenGL/Vulkan stuff still uses the internal PowerVR GPU.

geerlingguy, 2025-02-07:
SuperTuxKart also gives errors, as it is trying to use OpenGL:

```
[info   ] ServerConfig: Could not read server config file '/home/ubuntu/.config/supertuxkart/config-0.10/server_config.xml'. A new file will be created.
..:: Antarctica Rendering Engine 2.0 ::..
[info   ] [IrrDriver Logger]: ..:: Antarctica Rendering Engine 2.0 ::..
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  113
  Current serial number in output stream:  114
```

geerlingguy, 2025-02-07:
Testing an RX 550 with a monstrosity of a riser... and a CPU 8-pin to PCIe 6-pin power adapter :D

geerlingguy, 2025-02-07:
And booting with the AMD RX 550 GPU installed:

```
ubuntu@ubuntu:~$ sudo dmesg | grep amdgpu
[   30.396777] [drm] amdgpu kernel modesetting enabled.
[   30.396898] amdgpu 0000:01:00.0: Failed to enable PASID
[   30.405968] amdgpu 0000:01:00.0: Adding to iommu group 5
[   30.406261] amdgpu 0000:01:00.0: enabling device (0000 -> 0003)
[   30.620928] amdgpu 0000:01:00.0: amdgpu: Fetched VBIOS from ROM BAR
[   30.620948] amdgpu: ATOM BIOS: 113-BAFFIN_PRO_160513_D5_4GB_MIC_0_W82
[   30.621114] amdgpu 0000:01:00.0: vgaarb: deactivate vga console
[   30.621125] amdgpu 0000:01:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[   30.621133] amdgpu 0000:01:00.0: amdgpu: PCIE atomic ops is not supported
[   31.092725] amdgpu 0000:01:00.0: BAR 2: releasing [mem 0x8010000000-0x80101fffff 64bit pref]
[   31.092739] amdgpu 0000:01:00.0: BAR 0: releasing [mem 0x8000000000-0x800fffffff 64bit pref]
[   31.092834] amdgpu 0000:01:00.0: BAR 0: assigned [mem 0x8000000000-0x80ffffffff 64bit pref]
[   31.092860] amdgpu 0000:01:00.0: BAR 2: assigned [mem 0x8100000000-0x81001fffff 64bit pref]
[   31.092921] amdgpu 0000:01:00.0: amdgpu: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
[   31.092929] amdgpu 0000:01:00.0: amdgpu: GART: 256M 0x000000FF00000000 - 0x000000FF0FFFFFFF
[   31.093169] [drm] amdgpu: 4096M of VRAM memory ready
[   31.093177] [drm] amdgpu: 7999M of GTT memory ready.
[   31.123006] amdgpu: hwmgr_sw_init smu backed is polaris10_smu
[   31.502814] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
[   31.646982] amdgpu 0000:01:00.0: amdgpu: SE 2, SH per SE 1, CU per SH 8, active_cu_number 14
[   31.652147] amdgpu 0000:01:00.0: amdgpu: Using BACO for runtime pm
[   31.659567] [drm] Initialized amdgpu 3.54.0 20150101 for 0000:01:00.0 on minor 2
[   31.666522] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
[   41.352126] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
[  117.653629] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
[  367.197695] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
[  435.621750] amdgpu 0000:01:00.0: [drm] Cannot find any crtc or sizes
```

And lspci:

```
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X] (rev cf) (prog-if 00 [VGA controller])
	Subsystem: XFX Pine Group Inc. Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X]
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 91
	IOMMU group: 5
	Region 0: Memory at 8000000000 (64-bit, prefetchable) [size=4G]
	Region 2: Memory at 8100000000 (64-bit, prefetchable) [size=2M]
	Region 4: I/O ports at 1000 [size=256]
	Region 5: Memory at 41100000 (32-bit, non-prefetchable) [size=256K]
	Expansion ROM at 41140000 [disabled] [size=128K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1+,D2+,D3hot+,D3cold+)
		Status: D3 NoSoftRst+ PME-Enable+ DSel=0 DScale=0 PME-
	Capabilities: [58] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr+ FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #4, Speed 8GT/s, Width x8, ASPM L1, Exit Latency L1 <1us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s (downgraded), Width x4 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix+, MaxEETLPPrefixes 1
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit+ 64bit+ 128bitCAS-
[... 79 lines cut ...]
			 FRS-
			 AtomicOpsCap: 32bit+ 64bit+ 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fffff000  Data: 0001
	Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 14, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 40000001 0000000f fffff000 00000000
	Capabilities: [328 v1] Alternative Routing-ID Interpretation (ARI)
		ARICap:	MFVC- ACS-, Next Function: 0
		ARICtl:	MFVC- ACS-, Function Group: 0
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
```

geerlingguy, 2025-02-07:
Testing a Kioxia XG8 NVMe PCIe SSD in a x1 adapter (my only x4 adapter is installed somewhere, not sure where it ran off to lol):

```
01:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller XG8 (rev 01) (prog-if 02 [NVM Express])
	Subsystem: KIOXIA Corporation NVMe SSD Controller XG8
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 12
	IOMMU group: 5
	Region 0: Memory at 41100000 (64-bit, non-prefetchable) [size=16K]
...
		LnkCap:	Port #0, Speed 16GT/s, Width x4, ASPM L1, Exit Latency L1 <32us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
```

Run my disk benchmark script:

```
wget https://raw.githubusercontent.com/geerlingguy/pi-cluster/refs/heads/master/benchmarks/disk-benchmark.sh
chmod +x disk-benchmark.sh
sudo MOUNT_PATH=/mnt/nvme TEST_SIZE=1g ./disk-benchmark.sh
```

Results:

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 45.00 MB/s |
| iozone 4K random write     | 82.66 MB/s |
| iozone 1M random read      | 585.13 MB/s |
| iozone 1M random write     | 614.41 MB/s |
| iozone 1M sequential read  | 591.46 MB/s |
| iozone 1M sequential write | 616.79 MB/s |

**IMPORTANT NOTE**: This drive should give a lot better performance if I run it with x4, so please don't draw any conclusions from the above results until after I re-run with the full bandwidth available ;)

geerlingguy, 2025-02-07:
Over in the SiFive forum post, I got a reply directing me to the [Release 2025.02.00 notes]().

Now:

```
$ export MESA_LOADER_DRIVER_OVERRIDE=zink
$ supertuxkart  # WORKS!

$ glxinfo -B
name of display: :1
WARNING: Some incorrect rendering might occur because the selected Vulkan device (PowerVR A-Series AXM-8-256) doesn't support base Zink requirements: feats.features.fillModeNonSolid 
display: :1  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Collabora Ltd (0x1010)
    Device: zink (PowerVR A-Series AXM-8-256) (0x30010101)
    Version: 805372161.1.8
    Accelerated: yes
    Video memory: 15999MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 15999 MB, largest block: 15999 MB
    VBO free aux. memory - total: 0 MB, largest block: 0 MB
    Texture free memory - total: 15999 MB, largest block: 15999 MB
    Texture free aux. memory - total: 0 MB, largest block: 0 MB
    Renderbuffer free memory - total: 15999 MB, largest block: 15999 MB
    Renderbuffer free aux. memory - total: 0 MB, largest block: 0 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 15999 MB
    Total available memory: 15999 MB
    Currently available dedicated video memory: 15999 MB
OpenGL vendor string: Collabora Ltd
OpenGL renderer string: zink (PowerVR A-Series AXM-8-256)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

geerlingguy, 2025-02-07:
I ran:

```
sudo mv /usr/share/X11/xorg.conf.d/10-pvr.conf /usr/share/X11/xorg.conf.d/10-pvr.conf.hidden
```

Then shut down, plugged the RX 550 back in, and booted back up.

Now:

```
$ glxinfo -B
name of display: :1
display: :1  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.7, DRM 3.54, 6.6.21-10-premier) (0x67ef)
    Version: 26607.1.8
    Accelerated: yes
    Video memory: 4096MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 3870 MB, largest block: 3870 MB
    VBO free aux. memory - total: 7967 MB, largest block: 7967 MB
    Texture free memory - total: 3870 MB, largest block: 3870 MB
    Texture free aux. memory - total: 7967 MB, largest block: 7967 MB
    Renderbuffer free memory - total: 3870 MB, largest block: 3870 MB
    Renderbuffer free aux. memory - total: 7967 MB, largest block: 7967 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 4096 MB
    Total available memory: 12095 MB
    Currently available dedicated video memory: 3870 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.7, DRM 3.54, 6.6.21-10-premier)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 1:22.3.5-0ubuntu0~ppa10
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

geerlingguy, 2025-02-07:
GLMark2 on the RX 460:

```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      AMD
    GL_RENDERER:    AMD Radeon RX 460 Graphics (polaris11, LLVM 15.0.7, DRM 3.54, 6.6.21-10-premier)
    GL_VERSION:     4.6 (Compatibility Profile) Mesa 1:22.3.5-0ubuntu0~ppa10
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 912 FrameTime: 1.097 ms
[build] use-vbo=true: FPS: 3097 FrameTime: 0.323 ms
[texture] texture-filter=nearest: FPS: 3238 FrameTime: 0.309 ms
[texture] texture-filter=linear: FPS: 3196 FrameTime: 0.313 ms
[texture] texture-filter=mipmap: FPS: 3230 FrameTime: 0.310 ms
[shading] shading=gouraud: FPS: 3231 FrameTime: 0.310 ms
[shading] shading=blinn-phong-inf: FPS: 3204 FrameTime: 0.312 ms
[shading] shading=phong: FPS: 3203 FrameTime: 0.312 ms
[shading] shading=cel: FPS: 3238 FrameTime: 0.309 ms
[bump] bump-render=high-poly: FPS: 3235 FrameTime: 0.309 ms
[bump] bump-render=normals: FPS: 3105 FrameTime: 0.322 ms
[bump] bump-render=height: FPS: 3094 FrameTime: 0.323 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 3085 FrameTime: 0.324 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 3154 FrameTime: 0.317 ms
[pulsar] light=false:quads=5:texture=false: FPS: 2985 FrameTime: 0.335 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 1832 FrameTime: 0.546 ms
[desktop] effect=shadow:windows=4: FPS: 1788 FrameTime: 0.560 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 220 FrameTime: 4.564 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 219 FrameTime: 4.582 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 398 FrameTime: 2.518 ms
[ideas] speed=duration: FPS: 1325 FrameTime: 0.755 ms
[jellyfish] <default>: FPS: 2845 FrameTime: 0.352 ms
[terrain] <default>: FPS: 973 FrameTime: 1.029 ms
[shadow] <default>: FPS: 2687 FrameTime: 0.372 ms
[refract] <default>: FPS: 1508 FrameTime: 0.663 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 3104 FrameTime: 0.322 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 3096 FrameTime: 0.323 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 3072 FrameTime: 0.326 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 3080 FrameTime: 0.325 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 2916 FrameTime: 0.343 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 3475 FrameTime: 0.288 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 3476 FrameTime: 0.288 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 3487 FrameTime: 0.287 ms
=======================================================
                                  glmark2 Score: 2565 
=======================================================
```

smzahraee, 2025-02-09:
The memory test results between the P550 and RK3588 show significantly lower performance than expected. Could you confirm the accuracy of these tests or provide additional context on the testing methodology?

geerlingguy, 2025-02-09:
@smzahraee - See: https://github.com/geerlingguy/sbc-reviews/blob/master/benchmark/tasks/tinymembench.py

I run tinymembench, and also sbc-bench.sh (from @tkaiser), and the memory bandwidth does seem to be hampered by the SoC in this case, even though it's LPDDR5 at I believe something like 6000 MT/s? It certainly jives with the rest of the benchmark results.

It could be something firmware/OS updates could help with, but I think the memory controller on the P550 just isn't as high performance as we hoped judging from the RAM they use here.

ThomasKaiser, 2025-02-10:
> the idle to full power range is pretty tight

I wouldn't call this not 'tight' but 'an indication that's something seriously wrong' ... most probably at the firmware layer.

1.4W difference between 'idle' and fully loaded with HPL would make this thing an efficiency monster which obviously is not the case and 'idle consumption' is in reality 'firmware going berserk' or something like that :)

As for the memory results not only bandwidth suffers but also latency (right with the top rows), measured with @wtarreau's `ramlat`, comparing this board with Rock 5B (RK3588) and Orion O6 (CD8180):

    Executing ramlat on cpu0 (eswin,eic770x), results in ns:
    
           size:  1x32  2x32  1x64  2x64 1xPTR 2xPTR 4xPTR 8xPTR
             4k: 3.697 3.670 3.581 3.579 2.148 2.147 2.862 5.726 
             8k: 3.667 3.669 3.583 3.579 2.148 2.148 2.863 5.725 
            16k: 3.668 3.670 3.579 3.579 2.147 2.147 2.864 5.726 
            32k: 3.673 3.670 3.581 3.582 2.149 2.150 2.865 5.729 
            64k: 10.12 11.54 10.02 11.55 9.303 10.76 13.17 25.60 
           128k: 10.12 11.55 10.03 11.61 9.316 10.89 13.36 25.80 
           256k: 22.39 21.36 22.26 21.28 21.72 20.76 22.21 29.21 
           512k: 30.77 29.80 30.68 29.68 30.59 29.42 37.03 55.31 
          1024k: 34.51 34.29 34.39 33.89 34.01 33.85 51.98 85.22 
          2048k: 41.25 39.43 44.31 38.73 40.84 38.70 67.21 101.7 
          4096k: 96.49 90.53 90.34 89.18 90.91 87.24 117.7 136.6 
          8192k: 175.7 175.5 175.7 174.7 175.3 168.8 196.3 204.2 
         16384k: 196.9 198.6 195.6 197.9 196.0 195.6 245.3 253.8 
         32768k: 207.4 210.5 207.1 210.0 206.5 208.7 275.7 293.3 
         65536k: 209.5 213.3 209.3 212.7 208.6 211.1 282.0 300.7 
        131072k: 216.0 218.5 214.7 218.3 214.3 216.5 290.4 307.5 
    
    Executing ramlat on cpu6 (Cortex-A76), results in ns:
    
           size:  1x32  2x32  1x64  2x64 1xPTR 2xPTR 4xPTR 8xPTR
             4k: 1.721 1.721 1.720 1.721 1.720 1.720 1.721 3.263 
             8k: 1.720 1.721 1.720 1.721 1.720 1.720 1.721 3.353 
            16k: 1.720 1.721 1.720 1.720 1.720 1.720 1.721 3.353 
            32k: 1.720 1.720 1.720 1.721 1.720 1.720 1.721 3.356 
            64k: 1.721 1.721 1.721 1.721 1.721 1.721 1.722 3.356 
           128k: 5.214 5.217 5.214 5.216 5.215 5.920 7.426 13.04 
           256k: 6.129 6.073 6.104 6.098 6.119 6.079 7.574 13.04 
           512k: 11.60 9.961 11.28 9.966 11.65 10.62 12.43 19.14 
          1024k: 17.95 18.47 17.88 18.47 18.09 18.13 19.84 30.12 
          2048k: 19.32 19.45 19.30 19.45 19.25 19.78 21.89 31.66 
          4096k: 55.24 42.18 49.56 42.21 49.36 43.65 46.08 59.47 
          8192k: 101.8 85.42 98.06 86.08 101.1 84.98 84.20 91.70 
         16384k: 128.6 114.1 119.0 110.4 118.3 106.1 104.0 107.7 
         32768k: 127.5 128.2 127.3 128.3 127.7 125.0 120.6 116.0 
         65536k: 134.3 134.5 134.0 134.5 133.1 127.4 124.8 130.4 
        131072k: 137.9 138.2 138.0 138.3 137.2 132.3 130.5 134.8 
    
    Executing ramlat on cpu11 (Cortex-A720), results in ns:
    
           size:  1x32  2x32  1x64  2x64 1xPTR 2xPTR 4xPTR 8xPTR
             4k: 1.539 1.540 1.539 1.539 1.539 1.539 1.539 2.843 
             8k: 1.540 1.539 1.539 1.540 1.539 1.539 1.539 2.903 
            16k: 1.539 1.539 1.540 1.539 1.540 1.539 1.539 2.906 
            32k: 1.539 1.539 1.543 1.540 1.540 1.539 1.540 2.908 
            64k: 1.540 1.540 1.540 1.540 1.540 1.541 1.540 2.910 
           128k: 3.724 3.759 3.725 3.751 3.727 4.542 6.398 16.94 
           256k: 3.524 3.544 3.524 3.540 3.524 4.563 8.657 17.29 
           512k: 3.479 3.478 3.477 3.478 3.476 4.562 8.672 17.31 
          1024k: 11.55 13.33 11.68 13.34 11.80 15.60 22.79 33.30 
          2048k: 11.99 13.90 12.16 13.91 12.16 18.34 20.53 36.12 
          4096k: 19.63 16.78 19.22 16.85 19.52 21.69 21.17 33.59 
          8192k: 26.60 21.68 26.59 21.78 26.46 24.44 25.74 35.03 
         16384k: 33.12 25.20 32.83 25.18 33.04 27.55 30.31 41.83 
         32768k: 41.80 43.55 41.85 43.14 41.89 50.44 70.08 113.9 
         65536k: 50.24 61.97 50.23 60.87 50.30 59.56 87.23 133.0 
        131072k: 49.58 71.82 49.01 70.11 48.89 63.16 94.08 136.9 

Checking [272 results from various systems 251 have better timings than this board](https://github.com/ThomasKaiser/sbc-bench/tree/master/results). Those systems being slower almost all being in a different league (original RPi, Allwinner A10 or D1, i.MX6, Kendryte K510) and of course the other [Eswin EIC7700 submission from last summer](https://github.com/ThomasKaiser/sbc-bench/issues/99).

@geerlingguy are you in touch with SiFive guys? Would be an idea to ask them what they've to say about both memory timings (new firmware release lying around somewhere?) and 'idle' consumption :)

geerlingguy, 2025-02-10:
@ThomasKaiser - I've opened a forum topic: [EIC7700X Memory Bandwidth / RAM Timings](https://forums.sifive.com/t/eic7700x-memory-bandwidth-ram-timings/7002).

ThomasKaiser, 2025-02-10:
> I also reviewed the source code of tinymembench and found that it does not include any RISC-V-specific assembly code.

Well, we're using a tool that is unmaintained for 8 full years by Siarhei (where you just opened an issue). Jeff uses [this fork](https://github.com/rojaster/tinymembench) (most probably since supporting execution in macOS?) and my sbc-bench since exactly two years [that](https://github.com/nuumio/tinymembench) for reasons I can't remember :(

Maybe not the ideal tool any more...

wtarreau, 2025-02-10:
No that's not a problem. What's important is that these tools (like ramlat) use the memory just like any application would do, and report the measured impact on the application's performance. I don't know for tinymembench, but ramlat just walks over pre-filled areas using a sufficiently unpredictable sequence and counts the number of operations done in a certain amount of time. This is *exactly* what an application experiences when walking down a list, a hash table or a tree. That's also why gcc build times are usually proportional to ramlat's output, since compilers manipulate a lot of objects in memory. By the way, by default (without -n) ramlat reports the number of ops per period, which really is the performance indicator (but not easy to place in a context).

In fact I'd say that tools that optimize for a specific platform try to get into the deepest details. That can be fine to try to get a specific metric without the platform-specific noise, but they just forget to mention the impact on applications in the end. If the goal is to figure a chip's specs in the end, reading a reference and a datasheet gives it. What matters is the integration of that chip and the final results. Seeing that my visionfive with LPDDR4-3200 only delivers 6.5 GB/s with ~200ns access time says a lot about it. It's probably a DRAM controller setup issue, I don't know, but the point is that the result makes a pathetic product.

geerlingguy, 2025-02-10:
Using the same Kioxia XG8 with a [GLOTRENDS PA09-HS M.2 NVMe to PCIe 4.0 X4 Adapter](https://amzn.to/4hsfVCI), I'm now seeing PCIe Gen 3 x4:

```
01:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller XG8 (rev 01) (prog-if 02 [NVM Express])
	Subsystem: KIOXIA Corporation NVMe SSD Controller XG8
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 12
	IOMMU group: 5
	Region 0: Memory at 41100000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI: Enable- Count=1/32 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [b0] MSI-X: Enable+ Count=65 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [c0] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd- ExtTag+ PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr+ FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 16GT/s, Width x4, ASPM L1, Exit Latency L1 <32us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (downgraded), Width x4
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
		LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq+ ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 14, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 40000001 0000000f fffff000 00000000
	Capabilities: [150 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [260 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [360 v1] Data Link Feature <?>
	Capabilities: [380 v1] Physical Layer 16.0 GT/s <?>
	Capabilities: [400 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
			  PortCommonModeRestoreTime=60us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=216064ns
		L1SubCtl2: T_PwrOn=150us
	Capabilities: [470 v1] Lane Margining at the Receiver <?>
	Kernel driver in use: nvme
	Kernel modules: nvme
```

Adding the disk-benchmark.sh results to the OP above.

geerlingguy, 2025-02-10:
Next up: Testing Box64 on the board, with both the iGPU and the AMD GPU.

  - [Support issue for Box86 on RISC-V](https://github.com/ptitSeb/box86/issues/678)
    - Box86 won't support RISC-V directly, so box64 + box32 is the best solution
  - I can use `ccmake ..` for interactive box64 build options (install with `sudo apt install cmake-curses-gui`)
  - [Compile instructions for box64 on RISC-V](https://github.com/ptitSeb/box64/blob/main/docs/COMPILE.md#for-risc-v)

geerlingguy, 2025-02-10:
Aside: I noticed a fan on hwmon0, but I couldn't change the fan speeds by applying different values...

```
root@ubuntu:/home/ubuntu# ls /sys/class/hwmon/hwmon0
device      fan1_label    fan_pwm_free    name     power  subsystem
fan1_input  fan_pwm_duty  fan_pwm_period  of_node  pwm1   uevent

# does nothing different, but does update `pwm1` value (default is 50)
echo 10 > /sys/class/hwmon/hwmon0/pwm1
```

geerlingguy, 2025-02-10:
```
root@ubuntu:/home/ubuntu# cat /sys/class/hwmon/hwmon0/name
eswin_fan_control
```

Setting any other value for `/sys/class/hwmon/hwmon0/fan_pwm_duty` (default is `5`) also does nothing.

zyz1dmail, 2025-02-11:
@geerlingguy   openGL based application need to have zink enabled. So you can run any openGL app like this:
MESA_LOADER_DRIVER_OVERRIDE=zink glxinfo

zyz1dmail, 2025-02-11:
@geerlingguy  glmark2-es2 score on Ubuntu is around 1300 with the internal GPU, so your  score of 1240 is expected. Since Ubuntu's desktop system (mutter) consumes more graphics resource than Debian's desktop system(xfwm4). You may try to run glmark2-es2 on Debian. And expected score is around 1800.  Maybe you can also install xfwm4 as your ubuntu window manager and the glmark2-es2 score should be higher.
Please also keep in mind all openGL application have to run with zink enabled.  like below:
MESA_LOADER_DRIVER_OVERRIDE=zink  myGfxApp

lianghujun258, 2025-02-11:
Hi，I am a developer at ESWIN, and I have seen the the inference results of LLM. I would like to provide some additional information on the large model inference results that we support.

lianghujun258, 2025-02-11:
Hi，
I am a developer at ESWIN, and I have seen the the inference results of LLM. I would like to provide some additional information on the large model inference results that we support.
| System                                      | CPU/NPU | Model                      | Eval Rate   |
|---------------------------------------------|---------|----------------------------|-------------|
| HiFive Premier P550(4-core RISC-V)          | NPU     | Qwen2 0.5B                 | 35.7 Tokens/s |
| HiFive Premier P550(4-core RISC-V)          | NPU     | Qwen2 7B                   | 3.08 Tokens/s |
| HiFive Premier P550(4-core RISC-V)          | NPU     | deepseek-distill-qwen-7b 7B| 3.08 Tokens/s |

ksco, 2025-02-11:
To enable box32 mode, use `ccmake ..` and enable `BOX32`, `BOX32_BINFMT`, and then recompile and reinstall box64. After that, box64 will automatically switch to box32 mode for 32bit programs.

ksco, 2025-02-11:
> In terms of something I could demo for a video, do you have any guide for games or downloads I could get onto here? I usually just use Steam since that makes it easy :D

Sure, if you're not in a hurry, I can give you a list of games we've tested on the P550 tomorrow.

geerlingguy, 2025-02-11:
@ksco That'd be great! I can wait, still have some other things I'm testing as well.

geerlingguy, 2025-02-11:
@ksco - I bought The Witcher 3 from Gog.com and downloaded all 16 parts... but when I try running the setup.exe, I get:

```
ubuntu@ubuntu:/run/user/1000/gvfs/smb-share:server=nas01.mmoffice.net,share=ingest/The Witcher 3$ ls
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-1.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-10.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-11.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-12.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-13.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-14.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-15.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-2.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-3.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-4.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-5.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-6.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-7.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-8.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)-9.bin'
'setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883).exe'
ubuntu@ubuntu:/run/user/1000/gvfs/smb-share:server=nas01.mmoffice.net,share=ingest/The Witcher 3$ box64 setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_\(73883\)
[BOX64] Warning: DynaRec is available on this host architecture, an interpreter-only build is probably not intended.
[BOX64] Running on unknown riscv64 cpu with 4 cores, pagesize: 4096
[BOX64] Will use hardware counter measured at 1.0 MHz emulating 2.0 GHz
[BOX64] Box64 v0.3.3 a50d34e4 built on Feb 10 2025 22:37:30
[BOX64] Detected 48bits at least of address space
[BOX64] Counted 25 Env var
[BOX64] BOX64 LIB PATH: 
[BOX64] BOX64 BIN PATH: ./:bin/:/home/ubuntu/.local/bin/:/usr/local/sbin/:/usr/local/bin/:/usr/sbin/:/usr/bin/:/sbin/:/bin/:/usr/games/:/usr/local/games/:/snap/bin/
[BOX64] Looking for setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883)
[BOX64] BOX64ENV: Variables overridden via env and/or RC file:
	BOX64_DYNAREC_BIGBLOCK=0
	BOX64_DYNAREC_CALLRET=0
	BOX64_DYNAREC_NATIVEFLAGS=0
	BOX64_DYNAREC_SAFEFLAGS=2
[BOX64] Error: File is not found. (setup_the_witcher_3_wild_hunt_4.04a_redkit_update_2_(73883))
```

geerlingguy, 2025-02-11:
@ksco - Ah thanks. These are things I never have done manually, so didn't explore all the fun complexity behind the scenes with Pi-Apps and other scripts I've used.

Civil, 2025-02-12:
I wonder, if Premier P550 can be overclocked to 1.8 Ghz (or maybe even beyond that)? Some other boards on the same chip (Milk-V Megrez) can run at 1.8 GHz, but that would require at least modifying device tree, copying some definitions from milk-v:
https://github.com/rockos-riscv/rockos-kernel/blob/rockos-v6.6.y/arch/riscv/boot/dts/eswin/eic7700-milkv-megrez.dts#L898-L919

Or even compiling kernel from rockos.

Min1123, 2025-02-12:
The iGPU driver appears to not support Wayland. I can get weston to operate, but gnome-shell or SDDM in Wayland mode change the gamma (I think, the screen brightens) and have a working cursor over top the VT you attempt spawning from, however the environment never starts.

Also, using the zink rendering on blender is insufficient to run blender (4.0.2, the version that is with Ubuntu Noble (24.04.2)), it will eventually crash out, but it gives an unpleasant blank gray window while attempting to work. The driver also gives a warning "WARNING: Some incorrect rendering might occur because the selected Vulkan device (PowerVR A-Series AXM-8-256) doesn't support base Zink requirements: feats.features.fillModeNonSolid".

Min1123, 2025-02-12:
Also, OpenCL works (on the iGPU), at least from clinfo and clpeak, at version 3.0.

Kernel modules are not there for nouveau or for xe, so, without a more generic kernel (and newer in the case of xe), it looks like embedded or amdgpu.

I really appreciate the the board has HDMI support and will show its u-boot and allow interruptions to get into the u-boot without having an external serial connection and using that from another computer.

geerlingguy, 2025-02-12:
I have Witcher 3 running... slowly:

[image]

[image]

Going to also test World of Goo just to see if a lightweight game can get playable FPS. It took me over 1 hr to get through the opening cutscene and out of the first room in Witcher!

geerlingguy, 2025-02-12:
Running `llama.cpp` compiled with Vulkan (see [instructions](https://www.jeffgeerling.com/blog/2024/llms-accelerated-egpu-on-raspberry-pi-5)), I got the following performance:

## llama3.2:3b on RX 480 4GB

```
llama_perf_sampler_print:    sampling time =     232.37 ms /   450 runs   (    0.52 ms per token,  1936.58 tokens per second)
llama_perf_context_print:        load time =   12751.58 ms
llama_perf_context_print: prompt eval time =    1210.39 ms /    13 tokens (   93.11 ms per token,    10.74 tokens per second)
llama_perf_context_print:        eval time =   18819.80 ms /   449 runs   (   41.91 ms per token,    23.86 tokens per second)
llama_perf_context_print:       total time =   23924.14 ms /   462 tokens
```

## llama3.2:3b on RX 580 8GB

```
llama_perf_sampler_print:    sampling time =     397.17 ms /   769 runs   (    0.52 ms per token,  1936.19 tokens per second)
llama_perf_context_print:        load time =    7999.67 ms
llama_perf_context_print: prompt eval time =    4900.60 ms /    35 tokens (  140.02 ms per token,     7.14 tokens per second)
llama_perf_context_print:        eval time =   62029.51 ms /  1831 runs   (   33.88 ms per token,    29.52 tokens per second)
llama_perf_context_print:       total time =  115494.99 ms /  1866 tokens
```

geerlingguy, 2025-02-12:
I also tried an RX 6700 XT, but it seems that card is too new, there's a kernel oops when I try booting with the card installed:

```
[   27.475116] [drm] Not enough PCI address space for a large BAR.
[   27.475122] amdgpu 0000:03:00.0: BAR 0: assigned [mem 0x8000000000-0x800fffffff 64bit pref]
[   27.475149] amdgpu 0000:03:00.0: BAR 2: assigned [mem 0x8010000000-0x80101fffff 64bit pref]
[   27.475180] amdgpu 0000:03:00.0: amdgpu: VRAM: 12272M 0x0000008000000000 - 0x00000082FEFFFFFF (12272M used)
[   27.475189] amdgpu 0000:03:00.0: amdgpu: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
[   27.475195] amdgpu 0000:03:00.0: amdgpu: AGP: 267894784M 0x0000008400000000 - 0x0000FFFFFFFFFFFF
[   27.475207] [drm] Detected VRAM RAM=12272M, BAR=256M
[   27.475212] [drm] RAM width 192bits GDDR6
[   27.475512] [drm] amdgpu: 12272M of VRAM memory ready
[   27.475520] [drm] amdgpu: 7999M of GTT memory ready.
[   27.475633] [drm] GART: num cpu pages 131072, num gpu pages 131072
[   27.475816] [drm] PCIE GART of 512M enabled (table at 0x0000008000000000).
[   29.689824] amdgpu 0000:03:00.0: amdgpu: STB initialized to 2048 entries
[   29.690322] [drm] Loading DMUB firmware via PSP: version=0x02020020
[   29.698008] [drm] use_doorbell being set to: [true]
[   29.698115] [drm] use_doorbell being set to: [true]
[   29.698190] [drm] Found VCN firmware Version ENC: 1.30 DEC: 3 VEP: 0 Revision: 4
[   29.698220] amdgpu 0000:03:00.0: amdgpu: Will use PSP to load VCN firmware
[   29.784851] [drm] reserve 0xa00000 from 0x82fd000000 for PSP TMR
[   29.886410] amdgpu 0000:03:00.0: amdgpu: RAS: optional ras ta ucode is not available
[   29.904700] amdgpu 0000:03:00.0: amdgpu: SECUREDISPLAY: securedisplay ta ucode is not available
[   29.904733] amdgpu 0000:03:00.0: amdgpu: smu driver if version = 0x0000000e, smu fw if version = 0x00000012, smu fw program = 0, version = 0x00413e00 (65.62.0)
[   29.904743] amdgpu 0000:03:00.0: amdgpu: SMU driver if version not matched
[   29.904780] amdgpu 0000:03:00.0: amdgpu: use vbios provided pptable
[   29.962893] amdgpu 0000:03:00.0: amdgpu: SMU is initialized successfully!
[   29.962968] [drm] Display Core failed to initialize with v3.2.247!
[   29.962981] Unable to handle kernel access to user memory without uaccess routines at virtual address 0000000000000008
[   29.974497] Oops [#1]
[   29.978151] Modules linked in: amdgpu(+) snd_hda_codec_hdmi pvrsrvkm binfmt_misc amdxcp drm_exec gpu_sched drm_buddy snd_soc_i2s drm_suballoc_helper snd_seq_midi snd_seq_midi_event snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_rawmidi snd_hwdep nls_iso8859_1 drm_ttm_helper snd_hda_core ttm snd_soc_audio_graph_card esw_es8328_codec snd_soc_hdmi_codec snd_soc_simple_card_utils backlight snd_soc_core at24 ina2xx pac193x snd_pcm_dmaengine snd_seq snd_pcm regmap_i2c snd_seq_device snd_timer snd input_leds dw_axi_dmac_platform dw_wdt joydev soundcore virt_dma sch_fq_codel dm_multipath efi_pstore nfnetlink ip_tables x_tables btrfs blake2b_generic xor raid6_pq libcrc32c hid_logitech_hidpp hid_logitech_dj hid_generic dwmac_eic7700 usbhid hid stmmac_platform stmmac fusb303b gpio_dwapb typec pcs_xpcs spi_eswin_bootspi phylink gpio_generic uio_pdrv_genirq uio
[   30.055552] CPU: 1 PID: 435 Comm: (udev-worker) Not tainted 6.6.21-10-premier #7
[   30.064375] Hardware name: SiFive HiFive Premier P550 (DT)
[   30.071305] epc : drm_atomic_private_obj_fini+0x14/0x58
[   30.077972]  ra : amdgpu_dm_fini+0x3e/0x1cc [amdgpu]
[   30.092815] epc : ffffffff807200fe ra : ffffffff03055be6 sp : ffff8f800279b5b0
[   30.100114]  gp : ffffffff81d90148 tp : ffffaf80e2c1b900 t0 : 213734322e322e33
[   30.100120]  t1 : 0000000000000000 t2 : 2e33762068746977 s0 : ffff8f800279b5d0
[   30.100124]  s1 : ffffaf80e383f720 a0 : ffffaf80e383f6e8 a1 : 0000000000000000
[   30.121994]  a2 : 0000000000000000 a3 : 0000000000000000 a4 : 0000000000000000
[   30.130686]  a5 : 0000000000000000 a6 : 0000000000000000 a7 : 0000000000000000
[   30.139368]  s2 : ffffaf80e383f000 s3 : ffffaf80e3800000 s4 : ffffaf80e3800010
[   30.148051]  s5 : ffff8f800279b6a0 s6 : 0000000000000001 s7 : ffffaf80e3841000
[   30.156734]  s8 : 0000000000000004 s9 : 0000000000000002 s10: 0000000000000001
[   30.165417]  s11: 0000000000000000 t3 : 0000000000000000 t4 : 0000000000000000
[   30.174100]  t5 : 0000000000000000 t6 : ffff8f800279b3f8
[   30.180503] status: 0000000200000120 badaddr: 0000000000000008 cause: 000000000000000f
[   30.188501] [<ffffffff807200fe>] drm_atomic_private_obj_fini+0x14/0x58
[   30.195109] [<ffffffff03055be6>] amdgpu_dm_fini+0x3e/0x1cc [amdgpu]
[   30.209878] [<ffffffff03062148>] amdgpu_dm_init.isra.0+0x140/0x92e [amdgpu]
[   30.222870] [<ffffffff0306294a>] dm_hw_init+0x14/0x2a [amdgpu]
[   30.235486] [<ffffffff0310001c>] amdgpu_device_ip_init+0x2f4/0x518 [amdgpu]
[   30.248192] [<ffffffff02e09720>] amdgpu_device_init+0xa22/0xe48 [amdgpu]
[   30.262935] [<ffffffff02e0b7c8>] amdgpu_driver_load_kms+0x1e/0x154 [amdgpu]
[   30.275760] [<ffffffff02e0386e>] amdgpu_pci_probe+0xde/0x360 [amdgpu]
[   30.289858] [<ffffffff80646414>] pci_device_probe+0x8a/0x132
[   30.296957] [<ffffffff807a22bc>] call_driver_probe+0x1a/0x13a
[   30.303800] [<ffffffff807a2d82>] really_probe+0x82/0x26a
[   30.309167] [<ffffffff807a2fd4>] __driver_probe_device+0x6a/0x122
[   30.315328] [<ffffffff807a30b8>] driver_probe_device+0x2c/0xac
[   30.315336] [<ffffffff807a32ea>] __driver_attach+0xda/0x1ce
[   30.315342] [<ffffffff807a0afa>] bus_for_each_dev+0x62/0xbc
[... cut ...]
[   34.896926] audit: type=1400 audit(1739394604.103:144): apparmor="DENIED" operation="capable" class="cap" profile="/usr/sbin/cupsd" pid=1096 comm="cupsd" capability=12  capname="net_admin"
[   36.032376] loop2: detected capacity change from 0 to 8
[   38.784926] eic7700-eth-dwmac 50410000.ethernet end1: Link is Up - 1Gbps/Full - flow control rx/tx
[   43.997589] pipewire[1256]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   56.316063] codec read reg[1]:0x0
[   57.281634] codec read reg[1]:0xc0
[   57.417774] codec read reg[1]:0xc0
[   57.965144] codec read reg[1]:0xc0
```

geerlingguy, 2025-02-12:
I tried launching World of Goo, but I got:

```
[BOX64] Using native(wrapped) libasound.so.2
0100:fixme:avrt:AvSetMmThreadCharacteristicsW (L"Pro Audio",00000000819AFE6C): stub
[BOX64] Warning: Mmap64 for (@0x7fff05000000 0x32f110) for elf "/home/ubuntu/Downloads/wine/lib/wine/x86_64-unix/opengl32.so" returned 0x3f5bc00000(0x3f5bba8000/0x200000) instead
[BOX64] Using emulated /home/ubuntu/Downloads/wine/lib/wine/x86_64-unix/opengl32.so
0024:err:winediag:nodrv_CreateWindow Application tried to create a window, but no driver could be loaded.
0024:err:winediag:nodrv_CreateWindow L"The explorer process failed to start."
0024:fixme:ver:GetCurrentPackageId (00000000800EFB00 0000000000000000): stub
```

And I was dumped back to the command line :(

ksco, 2025-02-13:
You can also try the Linux version, which generally performs better than Windows ones.

geerlingguy, 2025-02-13:
@ksco - oh, didn't even realize there was a Linux download available!

geerlingguy, 2025-02-13:
Just wanted to note—in addition to a Debian image from ESWIN (which seems to be the bleeding edge where they're doing the most enablement work), there's a Fedora image for it as well. See:

  - https://fedoraproject.org/wiki/Architectures/RISC-V/SiFive/HiFivePremierP550

(Thanks to @NeilHanlon for pointing this out!)

RevySR, 2025-02-13:
> I also tried an RX 6700 XT, but it seems that card is too new, there's a kernel oops when I try booting with the card installed:
>
> ```
> [   27.475116] [drm] Not enough PCI address space for a large [...]

https://rockos-riscv.github.io/rockos-docs/en/docs/faq/#my-amd-rdna-or-newer-gpu-isnt-working-with-kernel-oops-on-boot

geerlingguy, 2025-02-13:
@RevySR - Interesting, have you successfully tested a newer AMD card on here? Also, is RockOS the name for the official Debian flavored Linux build for the ESWIN? I hadn't heard of it before your comment, and had associated it with the 'Rock' branding from Radxa until I read up on it more...

KevinMX, 2025-02-14:
> Interesting, have you successfully tested a newer AMD card on here? 

Check that AMD Radeon 7900XTX + Milk-V Megrez Video. I guess you've seen that already.

We're currently building up the website for docs and stuff. dGPU related docs will be added soon™.

> Also, is RockOS the name for the official Debian flavored Linux build for the ESWIN?

Yes.

> I hadn't heard of it before your comment, and had associated it with the 'Rock' branding from Radxa until I read up on it more...

No, RockOS is not affiliated with Radxa and has nothing to do with their products. I guess you mean Radxa's ROCK series (all caps).

drmpeg, 2025-03-21:
ESWIN has released some patches for the P550 to run at 1.8 GHz. https://github.com/sifive/riscv-linux/tree/dev/clk-1.8

Here's the .deb files for the new kernel.

https://www.w6rz.net/linux-image-6.6.21_6.6.21-g6e64cce95cec-2_riscv64.deb

https://www.w6rz.net/linux-headers-6.6.21_6.6.21-g6e64cce95cec-2_riscv64.deb

Just use ```sudo dpkg -i linux-image-6.6.21_6.6.21-g6e64cce95cec-2_riscv64.deb``` and ```sudo dpkg -i linux-headers-6.6.21_6.6.21-g6e64cce95cec-2_riscv64.deb``` to install. Everything will be installed automatically, grub will be updated and initramfs will be run.

Reboot and select the new kernel in grub with "Advanced options for Ubuntu".

There's a new device tree, so the fan control moves from ```hwmon0``` to ```hwmon2```.

geerlingguy, 2025-03-22:
@drmpeg - Thanks for the updates! I'll have to try firing this up again soon...

BlackS52, 2025-05-22:
> ESWIN has released some patches for the P550 to run at 1.8 GHz. https://github.com/sifive/riscv-linux/tree/dev/clk-1.8
>
> Here's the .deb files for the new kernel. [...]

Great. It loioks like it works well. 
However maybe you know why by default Premier use 1.4GHz instead of 1.8GHz ?