geerlingguy/sbc-reviews #66: Comparison page

mkesper, 2025-01-28:
Hi Heff, thanks for your fantastic work!

Would you think it makes sense to collect the infos in a comparison page? A little bit like the top500 page but also for comparing other stats.

geerlingguy, 2025-01-31:
I've considered making this more like the Pi PCIe Database site, where there's a page for each benchmarked SBC, with an index page. Then at some point another page or two could be added with tabular data (ideally pulled in from each issue, otherwise a manually curated page).

That page could have a table that is filterable/searchable with JS...

Even cooler would be having a few graphs, too, like for top500 scores ranked, things like that.

ValdikSS, 2025-02-04:
Take a look at https://hackerboards.com/
Would it be easier to ask the creators to add some new performance-related columns and use this website?

geerlingguy, 2025-02-04:
@ValdikSS - I think that would be an uphill battle—as with any PC benchmark comparison site, adding benchmark numbers to a page immediately provides a lot of fodder for discussion and sometimes accusations of bias, even with pretty standard / widely accepted benchmarks and techniques.

That site is a great, simple, database of all the base boards and specs, and doesn't have much in the way of potentially-subjective data.

It's also nearly impossible to validate data coming from multiple parties, because of variations in temperature, power, environment, choices in how a board is set up, OSes, and even the time it is tested...

ValdikSS, 2025-02-04:
Just found this
https://dietpi.com/survey/#benchmark

geerlingguy, 2025-02-25:
_Ideally_, I'd like my Hugo setup (see above) to be able to pull in the first comment on every issue (maybe with a specific label) and drop that into page under the `web/content/reviews` directory as a markdown file, with some standard metadata like the title of the issue as the title of the page, and some way to sort by date.

geerlingguy, 2025-02-25:
It may be possible to get data out in an Action with https://stackoverflow.com/a/70244956/100134 — however, to get *all* issues and iterate through them, I might need to use GitHub's API...

geerlingguy, 2025-02-25:
Closing this out to move the rest of this work in to #69 and other issues.

closed 2025-02-25 by geerlingguy