geerlingguy/sbc-reviews #67: pyinfra giving '

geerlingguy, 2025-02-04:
With my `main.py` task file using a callback to loop through all the task files with a pause between... I am now getting the following error when attempting to include another task file with a recursive loop inside:

```
    Starting nested operation: tasks/ollama-benchmark.py | Execute Ollama loop 
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 900, in gevent._gevent_cgreenlet.Greenlet.run
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/command.py", line 231, in execute_function
    self.function(*self.args, **self.kwargs)
  File "/Users/jgeerling/Development/sbc-reviews/benchmark/tasks/ollama-benchmark.py", line 48, in callback
    server.shell(
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/operation.py", line 216, in decorated_func
    if host.in_op:
       ^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 975 more times]
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 49, in __getattr__
    if self._get_module() is None:
       ^^^^^^^^^^^^^^^^^^
RecursionError: maximum recursion depth exceeded
2025-02-04T00:38:42Z <Greenlet at 0x10725b060: execute_function> failed with RecursionError

    Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/operations.py", line 94, in _run_host_op
    status = command.execute(state, host, connector_arguments)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/command.py", line 234, in execute
    return greenlet.get()
           ^^^^^^^^^^^^^^
  File "src/gevent/greenlet.py", line 823, in gevent._gevent_cgreenlet.Greenlet.get
  File "src/gevent/greenlet.py", line 373, in gevent._gevent_cgreenlet.Greenlet._raise_exception
  File "/opt/homebrew/lib/python3.12/site-packages/gevent/_compat.py", line 51, in reraise
    raise value
  File "src/gevent/greenlet.py", line 900, in gevent._gevent_cgreenlet.Greenlet.run
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/command.py", line 231, in execute_function
    self.function(*self.args, **self.kwargs)
  File "/Users/jgeerling/Development/sbc-reviews/benchmark/tasks/ollama-benchmark.py", line 48, in callback
    server.shell(
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/api/operation.py", line 216, in decorated_func
    if host.in_op:
       ^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 975 more times]
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 49, in __getattr__
    if self._get_module() is None:
       ^^^^^^^^^^^^^^^^^^
RecursionError: maximum recursion depth exceeded

    [10.0.2.247] nested Unexpected error in Python callback: RecursionError('maximum recursion depth exceeded',)
    [10.0.2.247] nested Error: executed 0 commands
```

It continues on past that task include to the next one, but that is not ideal, as I'd like to be able to run through all the tasks via one main file, without a bunch of duplicated code to manage sleeps between task includes...

From https://stackoverflow.com/a/3323013/100134, it looks like I can check and raise the recursion limit in Python:

```python
# Check recursion limit
import sys
print(sys.getrecursionlimit())

# Change recursion limit
sys.setrecursionlimit(1500)
```

geerlingguy, 2025-02-04:
Testing a recursion limit of `2000` (was `1000`).

geerlingguy, 2025-02-04:
Still getting:

```
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 51, in __getattr__
    return getattr(self._get_module(), key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 1975 more times]
  File "/opt/homebrew/lib/python3.12/site-packages/pyinfra/context.py", line 49, in __getattr__
    if self._get_module() is None:
       ^^^^^^^^^^^^^^^^^^
RecursionError: maximum recursion depth exceeded

    [10.0.2.247] nested Unexpected error in Python callback: RecursionError('maximum recursion depth exceeded',)
    [10.0.2.247] nested Error: executed 0 commands
```

The first time it was:

```
[Previous line repeated 975 more times]
```

And this time it is:

```
[Previous line repeated 1975 more times]
```

So it does seem like I've hit some kind of recursive loop here...

Fizzadar, 2025-02-04:
Good 'ol recursion. Smaller repro here: https://github.com/pyinfra-dev/pyinfra/issues/1283, PR attached to that.

geerlingguy, 2025-02-05:
@Fizzadar fixed this upstream, so I don't need to rely on a patched local install now, hooray!

closed 2025-02-05 by geerlingguy

geerlingguy, 2025-04-15:
Just as a note, when installing on a new system, until a release newer than 3.2 is tagged, I need to:

```
pip3 install git+https://github.com/pyinfra-dev/pyinfra.git
```

geerlingguy, 2025-04-28:
@Fizzadar woohoo! Thanks!